morrownr/USB-WiFi #101: Add Realtek 8812BU/8822BU to in-kernel list

tejasraman, 2022-08-17:
Kernel 5.19 (mainline) works perfectly with my 8812BU (Cudy WU600). Maybe it would be good to consider adding the aforementioned Realtek chipset to the in kernel supported list? I'll do testing if you would like.

morrownr, 2022-08-17:
Hi @tejasraman 

I had watching the initial patches come through linux-wireless but did not know that the driver had been mainlined. Yes, please test and report. I will be glad to move it to in-kernel if you beat it up good and see good results. Can you test managed, monitor and master modes?

Can I get you to post the results of:

$ iw list

>  because my (ralink/mediatek) WiFi adapter broke)

That is not good but it is an excuse to get a new mt7921au WiFi 6e adapter... <grin>

tejasraman, 2022-08-17:
I will do this tomorrow. Thanks @morrownr!

morrownr, 2022-08-17:
Hi again @tejasraman

My dev box runs kernel 5.19.1 currently so, of course, I had to grab and plug a rtl8812bu based adapter in to see what would happen. Nothing. My distro is Mint 21 with the Ubuntu 5.19.1 kernel from here:

https://kernel.ubuntu.com/~kernel-ppa/mainline/

This kernel obviously does not contain the usb patch series so please specify which distro you are using and if the maintainers are adding the patches to their kernel.

While the test information will be valuable to many folks, I need to hold off on moving the 8812bu/8822bu chipsets to the in-kernel list until such time as the usb patch series is approved and included in the mainline kernel. I hope you understand. That last thing I need to do is cause confusion.

Nick

morrownr, 2022-08-17:
Copy all. So that there is no confusion for others that may read this thread, I am not suggesting anyone try the Ubuntu mainline kernels unless they use Ubuntu or one of the Ubuntu based distros such as Linux Mint. If one is not technically inclined, you can and probably will break your system so only do this if you know what you are doing and you have any important data backed up.

tejasraman, 2022-08-17:
I have a spare PC with Neon Dev Edition on it. I will try it on there later. (I use Fedora for my main PC so.....)

tejasraman, 2022-08-17:
> While the test information will be valuable to many folks, I need to hold off on moving the 8812bu/8822bu chipsets to the in-kernel list until such time as the usb patch series is approved and included in the mainline kernel. I hope you understand. That last thing I need to do is cause confusion.

@morrownr I understand that it might cause confusion. My testing with 5.19 was biased due to a patch set being used. (No, I do not have your drivers on my main PC. This was used on a Win11 PC in my home for some time (despite that Windows is not my favorite OS by any stretch of the word, and I have been using Linux for years))

tejasraman, 2022-08-17:
[image]
Hey @morrownr
Because the Ubuntu PPA only has 6.0-rc1 for arm64 (with this option) I need to compile a kernel myself for x86. In doing so I found some promising looking entries in the kernel config interface.

Tejas

morrownr, 2022-08-17:
> I found some promising looking entries in the kernel config interface.

That is not what you are looking for. The 8812bu support is in rtw88. I think the option for that is `Realtek 802.11 ac...` but it is already selected.

I think the only way to do this is to manually grab and apply the patches.

tejasraman, 2022-08-17:
Well that's not the point. If I have to do manual work, the whole point of testing is gone. I'll test once the compile finishes and report back with an (obviously mainline) kernel.

morrownr, 2022-08-17:
> Oh, and sorry if I'm wasting your time.

You are not wasting my time. I watch linux-wireless and have not seen anything on this topic lately so I needed to see where we are and we did that. I've been very careful about my wording on this topic around here because this is not an effort backed by Realtek programmers. It may or may not ever happen. I will get excited when Realtek actually gets involved with in-kernel support for their usb wifi adapters.

My opinion is that the big thing to get excited about right now is the new mt7921au chipset (WiFi 6e). The driver is in the mainline kernel, mt7921u, and there are a couple of adapters on the market that we can buy with more expected.

tejasraman, 2022-08-17:
Sadly [this](https://www.phoronix.com/news/Realtek-RTW88-WiFi-Linux-Driver) is the last time Realtek seemed to care about the concept of _in-kernel_ drivers. (Even my main PC came with a built in Realtek 8821CE which is now mainline since ~5.8 (could be earlier))

morrownr, 2022-08-17:
When thinking of Realtek and wireless, you need to think of chipsets for cards and chipsets for usb wifi adapters. Realtek supports the card chipsets with in-kernel drivers. No comment on quality but the support from real, no-kidding Realtek employees is there for the cards. The problem is with usb chipsets. No in-kernel support at all. I closely watch the happenings in the linux-wireless mailing list. There is a stark difference in support. Mediatek is doing it right. Realtek ?!?

I just can't wait until ALFA releases a high power mt7921au based adapter!

My forecast is that within the next 2-3 years, Mediatek will own the Linux USB WiFi adapter market as the word spreads.

tejasraman, 2022-08-17:
My main PC has an 8821CE PCIe card in it and it has worked amazingly, no drops for 2 years now. Realtek PCIe drivers are 👍. 

You're right. If only Mediatek would make cheaper chipsets. (Not that I'm opposed to spending for Linux compatibility - I don't like Windows at all and support for Realtek on Mac is way worse that Linux (I know...) 

But look at the ~30 chipsets when I paid $13 for my Cudy WU600 (8812BU) and $10 for my 8821CU which are perfectly reliable except for the in-kernel drivers not existing?

If Realtek ever bothers with Linux support, Mediatek is in trouble (suffice to say) but I still think that some Mediatek chipsets are higher quality and have better Windows and Linux drivers than Realtek. (Seriously, 100MBPs with MS's built in Realtek driver on Windows and ~20MBPs with Realtek. 
> Realtek

tejasraman, 2022-08-18:
I don't trust AliExpress honestly..... but if it was a reliable site, all the Realtek adapters would be gone here. (Yes, I use Linux for a lot of things)

morrownr, 2022-08-18:
> I don't trust AliExpress honestly...

Me either. I'm looking for an alternate source and/or other adapters that use the mt7921au chipset. I sure wish ALFA would hurry up.

tejasraman, 2022-08-18:
True…..Hopefully, I can stop using painful Realtek adapters for Linux. $23 for a WIFI 6E Tri-Band? My Google WIFI (Gen 1) only supports 802.11 AC

morrownr, 2022-08-19:
@tejasraman 

A US based dealer:

https://www.cpusolutions.com/store/pc/COMFAST-CF-951AX-Wifi-6-USB-Adapter-802-11AX-Wi-fi-Dongle-1800Mbps-Dual-Band-Wireless-Adapter-p6657.htm

morrownr, 2022-08-19:
@tejasraman 

A phone call later and...

CPU Solutions agreed to ship via USPS ground. The cost was $5 so total cost is $24.99.

Their web site is only set up for UPS and their rates are ridiculously high right now. However, if you call and ask for shipping USPS ground, which is still fast, then shipping is reasonable.

tejasraman, 2022-08-19:
Oh my god – thanks so much @morrownr

morrownr, 2022-08-19:
Hope it helps. I ordered a 951 so hopefully by early next week I will have something to test.

The phone number for CPU Solutions is 800-474-4278.

FWIW: I told CPU Solutions that they should probably carry the CF-953AX also.

tejasraman, 2022-08-19:
I’m done with Realtek ;)

morrownr, 2022-08-19:
Hey, if you get one, start a new thread and let's have a contest to see who can write the best review.

Oh, I have a question: I think you mentioned having a RasPi4B. If so, how do we upgrade RasPiOS to the latest mainline kernel so we can we can test? It is well documented for Ubuntu (and all flavors) and Debian and also for other major distros but I cannot find a thing about RasPiOS.

Cheers

tejasraman, 2022-08-19:
1. I have a 3B+, not a 4B
2. I vaguely remember cross-compiling once using something called _gcc-Linaro_. I have a spare RasPI so I'll try to document the procedure for you.

Tejas

tejasraman, 2022-08-19:
I just went to the Github website and looked at some email comments. Sorry that Outlook left a lot of formatting stuff at the top. I was using my MacBook to email (I use Mailspring on Linux) so....

morrownr, 2022-08-19:
> Sorry that Outlook left a lot of formatting stuff at the top.

No problem. I cleaned it up.

> I have a spare RasPI so I'll try to document the procedure for you.

I hope there us a procedure that does not involved brain surgery.

tejasraman, 2022-08-19:
I can image a stock raspiOS with kernel 6.0rc1 for rasPi4.

morrownr, 2022-08-19:
That would be great. It does not have to be a super stable thing as this is just for testing my new mt7921au based adapter when it gets here.

It seems that the RasPiOS folks are only upgrading kernels when a new LTS comes along. They are on 5.15 right now. The next LTS will probably be 6.0 and then you add a couple of months for the new release and it could be a while before the support is widely available on the RasPiOS. I want to see how things are working and maybe put a bug report or two in like for them to update the firmware to the latest version... they are notorious for messing up firmware.

Thanks.

tejasraman, 2022-08-19:
Will do. Thanks @morrownr

tejasraman, 2022-08-19:
Hey @morrownr I am compiling 6.0rc1 right now. Just got back so.... Anyway, I enabled USB support + all the Mediatek support for MediaTek USB (RPI Foundation and most distros ship this default) so...
I'll send you an imaged copy of Raspi-OS in a sec.

tejasraman, 2022-08-19:
Sorry, I am using mainline-6.0rc1 as RPI Foundation is stuck at 5.15 right now.

tejasraman, 2022-08-20:
@morrownr 
Here is a status update: The kernel compiled and I am almost done an image - just installing kernel modules now - but I cannot test it because I used the blob for rPi4 and not rPi3.

tejasraman, 2022-08-20:
Hey @morrownr 
Status update number 2: IT'S READY!
Here is a link to an XZ-compressed img (should work with balenaEtcher/RPI Imager)
https://1drv.ms/u/s!AkBuD6XsET8x-FVrCHyXJXgsgd9y - yes this is a onedrive link.

Tejas

morrownr, 2022-08-20:
Hey @tejasraman 

Downloaded. Thanks. I'm busy this weekend but will burn an sd and give it a try early next week. Will let you know.

ViRb3, 2022-08-21:
> I don't trust AliExpress honestly..... but if it was a reliable site, all the Realtek adapters would be gone here. (Yes, I use Linux for a lot of things)

Just curious, don't trust it due to quality/delivery or other concerns? FWIW, I ordered the adapter from that very link and it arrived 7 days later to the UK, brand new sealed and fully functional. Now I just need to decide if I should go through the trouble of compiling Linux 5.19 for my Ubuntu RPi4 server, or wait patiently for the next official release 😄

tejasraman, 2022-08-21:
@ViRb3 IME there's a ton of scammy stuff on Aliexpress. My ISO should work for RPi4 if you want mainline - I compiled it with most of the official, stable MediaTek support.

ViRb3, 2022-08-21:
I see. I noticed your ISO, but that's Raspberry Pi OS, right? Unfortunately I am hardly dependent on Ubuntu...

tejasraman, 2022-08-21:
I can send you the raw kernel files if you want to put them on your server manually. I still have all the blobs and other stuff that are needed.

tejasraman, 2022-09-05:
Hi @morrownr did the iso work or is the kernel broken?