morrownr/USB-WiFi #103: 6ghz Access Point Creation with MT7921

desultory, 2022-08-28:
I'm not able to get this card to create an access point.  I've upgraded the firmware to the latest using this wiki as a guide, and had to compile hostapd myself for SAE support, but it seems to be having issues with ACS.  I'm sure my hostapd config is wrong, but I can't find any good resources on making wifi 6 APs on linux,

Here is my hostapd compile config:
`# Example hostapd build time configuration
#
# This file lists the configuration options that are used when building the
# hostapd binary. All lines starting with # are ignored. Configuration option
# lines must be commented out complete, if they are not to be included, i.e.,
# just setting VARIABLE=n is not disabling that variable.
#
# This file is included in Makefile, so variables like CFLAGS and LIBS can also
# be modified from here. In most cass, these lines should use += in order not
# to override previous values of the variables.

# Driver interface for Host AP driver
CONFIG_DRIVER_HOSTAP=y

# Driver interface for wired authenticator
#CONFIG_DRIVER_WIRED=y

# Driver interface for drivers using the nl80211 kernel interface
CONFIG_DRIVER_NL80211=y

# QCA vendor extensions to nl80211
#CONFIG_DRIVER_NL80211_QCA=y

# driver_nl80211.c requires libnl. If you are compiling it yourself
# you may need to point hostapd to your version of libnl.
#
#CFLAGS += -I$<path to libnl include files>
#LIBS += -L$<path to libnl library files>

# Use libnl v2.0 (or 3.0) libraries.
#CONFIG_LIBNL20=y

# Use libnl 3.2 libraries (if this is selected, CONFIG_LIBNL20 is ignored)
CONFIG_LIBNL32=y

# Driver interface for FreeBSD net80211 layer (e.g., Atheros driver)
#CONFIG_DRIVER_BSD=y
#CFLAGS += -I/usr/local/include
#LIBS += -L/usr/local/lib
#LIBS_p += -L/usr/local/lib
#LIBS_c += -L/usr/local/lib

# Driver interface for no driver (e.g., RADIUS server only)
#CONFIG_DRIVER_NONE=y

# WPA2/IEEE 802.11i RSN pre-authentication
CONFIG_RSN_PREAUTH=y

# Support Operating Channel Validation
#CONFIG_OCV=y

# Integrated EAP server
CONFIG_EAP=y

# EAP Re-authentication Protocol (ERP) in integrated EAP server
CONFIG_ERP=y

# EAP-MD5 for the integrated EAP server
CONFIG_EAP_MD5=y

# EAP-TLS for the integrated EAP server
CONFIG_EAP_TLS=y

# EAP-MSCHAPv2 for the integrated EAP server
#CONFIG_EAP_MSCHAPV2=y

# EAP-PEAP for the integrated EAP server
[... 379 lines cut ...]
#wmm_ac_be_cwmin=4
#wmm_ac_be_cwmax=10
#wmm_ac_be_txop_limit=0
#wmm_ac_be_acm=0
#wmm_ac_vi_aifs=2
#wmm_ac_vi_cwmin=3
#wmm_ac_vi_cwmax=4
#wmm_ac_vi_txop_limit=94
#wmm_ac_vi_acm=0
#wmm_ac_vo_aifs=2
#wmm_ac_vo_cwmin=2
#wmm_ac_vo_cwmax=3
#wmm_ac_vo_txop_limit=47
#wmm_ac_vo_acm=0
broadcast_deauth=1
eapol_key_index_workaround=0
eap_server=0
own_ip_addr=127.0.0.1
wpa=2
wpa_passphrase=password
wpa_key_mgmt=SAE WPA-PSK
rsn_pairwise=CCMP-256 CCMP
ieee80211w=1
sae_password=password6
sae_require_mfp=1`

desultory, 2022-08-28:
Quartz64, armbian latest `Linux quartz64a 5.19.4-media #22.08.1 SMP PREEMPT Fri Aug 26 13:31:57 UTC 2022 aarch64 aarch64 aarch64 GNU/Linux`

The adapter is an ebay special mt7921 that has a big heatsink but I'm assuming is just a m.2 to pcie adapter board.

desultory, 2022-08-28:
An unfortunate thing about this board is that the usb3 is kinda broken on it, I may do a hardware mod that can fix that though

morrownr, 2022-08-28:
The first thing I would do is change the channel:

> channel=100

channel=36

Your country code is US and that makes 100 a DFS channel. I'm not sure the driver support DFS. It may but I have not tested yet. You could also go with 149.

What does your output of `iw dev` look like after you start hostapd?

desultory, 2022-08-28:
```
phy#0
        Interface wlan0
                ifindex 3
                wdev 0x1
                addr 10:6f:d9:29:19:d7
                ssid test6
                type AP
                channel 36 (5180 MHz), width: 20 MHz, center1: 5180 MHz
                txpower 3.00 dBm
                multicast TXQ:
                        qsz-byt qsz-pkt flows   drops   marks   overlmt hashcol tx-bytes        tx-packets
                        0       0       0       0       0       0       0       0               0
```
I'm not seeing it with the ax210 in my laptop.

desultory, 2022-08-28:
```
phy#0
        Interface wlan0
                ifindex 3
                wdev 0x1
                addr 10:6f:d9:29:19:d7
                ssid test6
                type AP
                channel 149 (5745 MHz), width: 20 MHz, center1: 5745 MHz
                txpower 3.00 dBm
                multicast TXQ:
                        qsz-byt qsz-pkt flows   drops   marks   overlmt hashcol tx-bytes        tx-packets
                        0       0       0       0       0       0       0       0               0
```
on channel 149, still can't see it

morrownr, 2022-08-28:
Hmmm...

> type AP

Generally if hostapd is able to take the interface into AP mode successfully then there is not much wrong. That is a good sign.

Let me suggest you expend your channel settings so that you can do 80 MHz channel width:

# Channel
channel=36
# Channel width (0 = 40 MHz. 1 = 80 Mhz)
vht_oper_chwidth=1
# VHT center channel (chan + 6)
vht_oper_centr_freq_seg0_idx=42

Now give me a chance to go through the rest of the posting above in detail.

morrownr, 2022-08-28:
In looking at the above postings, I'm wondering if it would to keep things simple and work toward for more complicated.

A WPA2, 5 GHz band setup working well and then we try to push the buttons for AX and then after that is working well, get WPA3 going. 

If you are interested, I have an AP guide up that gives pretty hostapd.conf examples for 2.4 GHz and 5 GHz. It isn't updated yet for WiFi 6 or 6 GHz channels but it could help:

https://github.com/morrownr/USB-WiFi/blob/main/home/AP_Mode/Bridged_Wireless_Access_Point.md

desultory, 2022-08-29:
I had a config that was working on a pi4 with an 80211ac usb3 adapter, i copied that config and it gave me an error about 80+80 channel width.

There is my iw list:

```
Wiphy phy0
        wiphy index: 0
        max # scan SSIDs: 4
        max scan IEs length: 482 bytes
        max # sched scan SSIDs: 10
        max # match sets: 16
        Retry short limit: 7
        Retry long limit: 4
        Coverage class: 0 (up to 0m)
        Device supports AP-side u-APSD.
        Device supports T-DLS.
        Supported Ciphers:
                * WEP40 (00-0f-ac:1)
                * WEP104 (00-0f-ac:5)
                * TKIP (00-0f-ac:2)
                * CCMP-128 (00-0f-ac:4)
                * CCMP-256 (00-0f-ac:10)
                * GCMP-128 (00-0f-ac:8)
                * GCMP-256 (00-0f-ac:9)
                * CMAC (00-0f-ac:6)
                * CMAC-256 (00-0f-ac:13)
                * GMAC-128 (00-0f-ac:11)
                * GMAC-256 (00-0f-ac:12)
        Available Antennas: TX 0x3 RX 0x3
        Configured Antennas: TX 0x3 RX 0x3
        Supported interface modes:
                 * managed
                 * AP
                 * AP/VLAN
                 * monitor
        Band 1:
                Capabilities: 0x9ff
                        RX LDPC
                        HT20/HT40
                        SM Power Save disabled
                        RX Greenfield
                        RX HT20 SGI
                        RX HT40 SGI
                        TX STBC
                        RX STBC 1-stream
                        Max AMSDU length: 7935 bytes
                        No DSSS/CCK HT40
                Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
                Minimum RX AMPDU time spacing: No restriction (0x00)
                HT TX/RX MCS rate indexes supported: 0-15
                HE Iftypes: managed
                        HE MAC Capabilities (0x08011a000040):
                                +HTC HE Supported
                                Trigger Frame MAC Padding Duration: 2
                                OM Control
                                Maximum A-MPDU Length Exponent: 3
                                A-MSDU in A-MPDU
                        HE PHY Capabilities: (0x2270ce120dc0b306423f00):
                                HE40/2.4GHz
                                242 tone RUs/2.4GHz
                                Device Class: 1
                                LDPC Coding in Payload
                                HE SU PPDU with 1x HE-LTF and 0.8us GI
                                NDP with 4x HE-LTF and 3.2us GI
                                STBC Tx <= 80MHz
                                STBC Rx <= 80MHz
                                Full Bandwidth UL MU-MIMO
                                Partial Bandwidth UL MU-MIMO
                                DCM Max Constellation: 2
                                DCM Max Constellation Rx: 2
[... 585 lines cut ...]
ignore_broadcast_ssid=0
macaddr_acl=0
auth_algs=1
wpa=2
wpa_key_mgmt=WPA-PSK
#wpa_pairwise=TKIP
rsn_pairwise=CCMP

```
password and ssid left out

```
wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
80/80+80 MHz: no second channel offset
Could not set channel for kernel driver
Interface initialization failed
wlan0: interface state COUNTRY_UPDATE->DISABLED
wlan0: AP-DISABLED
wlan0: Unable to setup interface.
wlan0: interface state DISABLED->DISABLED
wlan0: AP-DISABLED
wlan0: CTRL-EVENT-TERMINATING
hostapd_free_hapd_data: Interface wlan0 wasn't started
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
```

desultory, 2022-08-29:
That config was working on a usb mt7612 I think

morrownr, 2022-08-29:
You may need to add the following to hostapd.conf:

vht_capab=[MAX-MPDU-11454][RXLDPC][SHORT-GI-80][TX-STBC-2BY1][RX-STBC-1][SU-BEAMFORMEE][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN]

Well, actually, after looking at your hostapd log above:

> Interface initialization failed
> hostapd_free_hapd_data: Interface wlan0 wasn't started

Hmmm... interface is not coming. What is causing that?

morrownr, 2022-08-29:
> 5180 MHz [36] (20.0 dBm) (no IR)
> 5200 MHz [40] (20.0 dBm)
> 5220 MHz [44] (20.0 dBm)
> 5240 MHz [48] (20.0 dBm) (no IR)

Okay, what is going on?

"The NO-IR flag then means you cannot use modes of operation that require you to initiate radiation first"

What is the result of:

$ iw reg get

morrownr, 2022-08-29:
FYI: I just happened to reread the title of this issue:

> 6ghz Access Point Creation with MT7921

The mt7921 chipset that you can do WiFi 6 as in AX. It cannot do 6 GHz channels. WiFi 6e can do 6 GHz channels. You need a mt7921k (for pcie) or mt7921au (for usb) to do 6 GHz channels. Take a look in your `iw list` output under band 4.

desultory, 2022-08-29:
> > 5180 MHz [36] (20.0 dBm) (no IR)
> > 5200 MHz [40] (20.0 dBm)
> > 5220 MHz [44] (20.0 dBm)
> > 5240 MHz [48] (20.0 dBm) (no IR)
> 
> Okay, what is going on?
> 
> "The NO-IR flag then means you cannot use modes of operation that require you to initiate radiation first"
> 
> What is the result of:
> 
> $ iw reg get

```
root@quartz64a:/etc/wireguard# iw reg get
global
country US: DFS-FCC
        (902 - 904 @ 2), (N/A, 30), (N/A)
        (904 - 920 @ 16), (N/A, 30), (N/A)
        (920 - 928 @ 8), (N/A, 30), (N/A)
        (2400 - 2472 @ 40), (N/A, 30), (N/A)
        (5150 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
        (5250 - 5350 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW
        (5470 - 5730 @ 160), (N/A, 24), (0 ms), DFS
        (5730 - 5850 @ 80), (N/A, 30), (N/A), AUTO-BW
        (5850 - 5895 @ 40), (N/A, 27), (N/A), NO-OUTDOOR, AUTO-BW, PASSIVE-SCAN
        (5925 - 7125 @ 320), (N/A, 12), (N/A), NO-OUTDOOR, PASSIVE-SCAN
        (57240 - 71000 @ 2160), (N/A, 40), (N/A)
```

desultory, 2022-08-29:
> You may need to add the following to hostapd.conf:
>
> vht_capab=[MAX-MPDU-11454][RXLDPC][SHORT-GI-80][TX-STBC-2BY1][RX-STBC-1][SU-BEAMFORMEE][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN]
>
> Well, actually, [...]

added this and am getting the same error

morrownr, 2022-09-03:
I'm going to be busy until sometime next week but will try to help if you are still having problems when I am available.

Nick

desultory, 2022-09-03:
Thanks, I'm still having issues but I don't know how many are board related, this board seems to have issues. I'm considering finding another sbc with pcie support or trying with usb, but I couldn't find any good looking wifi6 usb cards a month ago.

bjlockie, 2022-09-12:
> FYI: I just happened to reread the title of this issue:
>
>
> The mt7921 chipset that you can do WiFi 6 as in AX. It cannot do 6 GHz channels. WiFi 6e can do 6 GHz channels. You need a mt7921k (for [...]

I wanted to add that I bought a dual band mt7921 from Amazon that said it was triband so I sent it back.
I then bought another adapter from a different seller on Amazon that advertised triband and specified an mt7921k chipset.
It is dual band too.
I decided to keep it instead of hoping.
A lot of stuff on Amazon are imported from China and are the same things on ebay.
I don't have a 6GHz router.

desultory, 2022-09-12:
> I wanted to add that I bought a dual band mt7921 from Amazon that said it was triband so I sent it back. I then bought another adapter from a different seller on Amazon that advertised triband and [...]

I had similar issues, amazon support and the seller both called me a liar and got my reviews removed saying the cards were dual band.  Weird that my reviews on the amazon renewed hard drive were also removed when I showed it had it's SMART stats wiped and failed the initial smart test I ran on it.  The card I'm currently using was from ebay and is also dual band but was cheaper and had a heatsink unlike the trash that was being sold on amazon.  It's extremely frustrating that amazon is now full of bad cards, the card I got on amazon was also flashed to be a ryzen card or whatever which just made loading firmware and drivers more of a pain but still possible.  

I've tried a few cards and none of them seem to actually be tri band, I think some people have had success using modded firmware on intel ax210 cards though.  I'm about at my limits trying to order stuff on amazon, I don't really know if ebay or aliexpress are that much more reliable though.

bjlockie, 2022-09-12:
My card has a heatsink so you can't see the chip but it is a 7921.
I figure those chips are in plentiful supply.
I don't know why mediatek even made it, I think it was AFTER Intel already had a 6GHz chip.

morrownr, 2022-09-12:
The good news for usb is that it appears no dual band chipset will be made since they are starting off with the tri band mt7921au (which appears to be a mt7921k with support for usb.)

A lot of sellers could give a s**t about the details that matter to us. I give first preference to Rokland as they have folks behind the scenes that work hard to understand Linux users. You might give them a call and request the type of cards you are looking for with the mt7921k chipset:

https://store.rokland.com/

They service the US and Canada for sure but you might ask if you are elsewhere in the world.

bjlockie, 2022-09-12:
> A lot of sellers could give a s**t about the details that matter to us. I give first preference to Rokland as they have folks behind the scenes that work hard to understand Linux users. You might give them a call and request the type of cards you are looking for with the mt7921k chipset:

I'll keep the non-6ghz.
It's just a bit disappointing that a lot of sellers are doing that.
The good thing is that it is mediatek. :-)
I am in Canada and if get a 6GHz router I'll get a new card.
It might not be until wifi 8. :-)

morrownr, 2022-09-12:
My wifi 5 router officially turned into a Wifi 6 router last night. It is a ZyXEL NBG6817. The cf-951ax and an elbow extender turned it into a WiFi 6 router. OpenWRT 22.03 has the driver for the mt7921au. It works well and when Alfa releases a good adapter with long range, I'll replace the Comfast and will be able to test 6 GHz.

bjlockie, 2022-09-12:
> I had similar issues, amazon support and the seller both called me a liar and got my reviews removed saying the cards were dual band.

I tried the adapter on Windows.
The drive makes it LOOK like it is 6e but there is no 6GHZ band.
Linux tells even more. :-)

[image]

morrownr, 2022-09-12:
If `iw list` is not showing band 4, then it is likely a mt7921 chipset and not the mt7921k. I did not think AMD licensed the mt7921, only the mt7921k and later. Beats me what is going on.

bjlockie, 2022-09-12:
I don't think it is an AMD thing, I think it is the driver for the AMD that Ubit (the seller) pointed me at that happens to work but doesn't report the actual chip.
I couldn't find drivers hosted on the Mediatek website which is annoying.
They don't seem to be very end-user friendly.

bjlockie, 2022-09-13:
This what Ubit (the manufacturer) said:
>Does your router support 6E? If your router doesn't have 6E function, then the NIC can't receive 6G signal.

Sigh.

ayyyuki1, 2022-09-13:
> My wifi 5 router officially turned into a Wifi 6 router last night. It is a ZyXEL NBG6817. The cf-951ax and an elbow extender turned it into a WiFi 6 router. OpenWRT 22.03 has the driver for the [...]

can you do 6GHz iperf tests now?

desultory-zz, 2022-10-25:
I'm on the same exact hardware, just on the latest armbian, this is what I'm seeing from `iw list` now:

```
Wiphy phy0
        wiphy index: 0
        max # scan SSIDs: 4
        max scan IEs length: 482 bytes
        max # sched scan SSIDs: 10
        max # match sets: 16
        Retry short limit: 7
        Retry long limit: 4
        Coverage class: 0 (up to 0m)
        Device supports AP-side u-APSD.
        Device supports T-DLS.
        Supported Ciphers:
                * WEP40 (00-0f-ac:1)
                * WEP104 (00-0f-ac:5)
                * TKIP (00-0f-ac:2)
                * CCMP-128 (00-0f-ac:4)
                * CCMP-256 (00-0f-ac:10)
                * GCMP-128 (00-0f-ac:8)
                * GCMP-256 (00-0f-ac:9)
                * CMAC (00-0f-ac:6)
                * CMAC-256 (00-0f-ac:13)
                * GMAC-128 (00-0f-ac:11)
                * GMAC-256 (00-0f-ac:12)
        Available Antennas: TX 0x3 RX 0x3
        Configured Antennas: TX 0x3 RX 0x3
        Supported interface modes:
                 * managed
                 * AP
                 * AP/VLAN
                 * monitor
        Band 1:
                Capabilities: 0x9ff
                        RX LDPC
                        HT20/HT40
                        SM Power Save disabled
                        RX Greenfield
                        RX HT20 SGI
                        RX HT40 SGI
                        TX STBC
                        RX STBC 1-stream
                        Max AMSDU length: 7935 bytes
                        No DSSS/CCK HT40
                Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
                Minimum RX AMPDU time spacing: No restriction (0x00)
                HT TX/RX MCS rate indexes supported: 0-15
                HE Iftypes: managed
                        HE MAC Capabilities (0x08011a000040):
                                +HTC HE Supported
                                Trigger Frame MAC Padding Duration: 2
                                OM Control
                                Maximum A-MPDU Length Exponent: 3
                                A-MSDU in A-MPDU
                        HE PHY Capabilities: (0x2270ce120dc0b306423f00):
                                HE40/2.4GHz
                                242 tone RUs/2.4GHz
                                Device Class: 1
                                LDPC Coding in Payload
                                HE SU PPDU with 1x HE-LTF and 0.8us GI
                                NDP with 4x HE-LTF and 3.2us GI
                                STBC Tx <= 80MHz
                                STBC Rx <= 80MHz
                                Full Bandwidth UL MU-MIMO
                                Partial Bandwidth UL MU-MIMO
                                DCM Max Constellation: 2
                                DCM Max Constellation Rx: 2
                                SU Beamformee
                                Beamformee STS <= 80Mhz: 3
[... 538 lines cut ...]
                 * managed: 0x40 0xb0 0xd0
                 * AP: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
                 * AP/VLAN: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
                 * mesh point: 0xb0 0xc0 0xd0
                 * P2P-client: 0x40 0xd0
                 * P2P-GO: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
                 * P2P-device: 0x40 0xd0
        Supported extended features:
                * [ RRM ]: RRM
                * [ SET_SCAN_DWELL ]: scan dwell setting
                * [ BEACON_RATE_LEGACY ]: legacy beacon rate setting
                * [ BEACON_RATE_HT ]: HT beacon rate setting
                * [ BEACON_RATE_VHT ]: VHT beacon rate setting
                * [ FILS_STA ]: STA FILS (Fast Initial Link Setup)
                * [ CQM_RSSI_LIST ]: multiple CQM_RSSI_THOLD records
                * [ CONTROL_PORT_OVER_NL80211 ]: control port over nl80211
                * [ TXQS ]: FQ-CoDel-enabled intermediate TXQs
                * [ AIRTIME_FAIRNESS ]: airtime fairness scheduling
                * [ AQL ]: Airtime Queue Limits (AQL)
                * [ CONTROL_PORT_NO_PREAUTH ]: disable pre-auth over nl80211 control port support
                * [ SCAN_FREQ_KHZ ]: scan on kHz frequency support
                * [ CONTROL_PORT_OVER_NL80211_TX_STATUS ]: tx status for nl80211 control port support
              
```
`Linux quartz64a 6.0.3-media #trunk.0080 SMP PREEMPT Sun Oct 23 18:40:39 UTC 2022 aarch64 aarch64 aarch64 GNU/Linux`

desultory-zz, 2022-10-25:
My old 802.11ac conf just werks now, testing ax/wpa3

desultory-zz, 2022-10-25:
I was able to get it to work with this conf, but the link speeds are showing up as 86mbps on my phone; my laptop, with an ax210, can't even see the network. 

```
interface=wlan0
logger_syslog=-1
logger_syslog_level=2
logger_stdout=-1
logger_stdout_level=2
ctrl_interface=/var/run/hostapd
ctrl_interface_group=0

ssid=test6
country_code=US
ieee80211d=1
ieee80211h=1
hw_mode=a
#channel=100

channel=36
ieee80211ac=1
ieee80211ax=1

vht_oper_chwidth=1
channel=36
vht_oper_centr_freq_seg0_idx=42

ht_capab=[HT40+][HT40-][GF][SHORT-GI-20][SHORT-GI-40]
vht_capab=[RXLDPC][SHORT-GI-80][TX-STBC-2BY1][RX-STBC-1][MAX-A-MPDU-LEN-EXP3][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN]

beacon_int=100
dtim_period=2
max_num_sta=2007
rts_threshold=-1
fragm_threshold=-1
macaddr_acl=0
auth_algs=3
ignore_broadcast_ssid=0
#wmm_enabled=1
#wmm_ac_bk_cwmin=4
#wmm_ac_bk_cwmax=10
#wmm_ac_bk_aifs=7
#wmm_ac_bk_txop_limit=0
#wmm_ac_bk_acm=0
#wmm_ac_be_aifs=3
#wmm_ac_be_cwmin=4
#wmm_ac_be_cwmax=10
#wmm_ac_be_txop_limit=0
#wmm_ac_be_acm=0
#wmm_ac_vi_aifs=2
#wmm_ac_vi_cwmin=3
#wmm_ac_vi_cwmax=4
#wmm_ac_vi_txop_limit=94
#wmm_ac_vi_acm=0
#wmm_ac_vo_aifs=2
#wmm_ac_vo_cwmin=2
#wmm_ac_vo_cwmax=3
#wmm_ac_vo_txop_limit=47
#wmm_ac_vo_acm=0
broadcast_deauth=1
eapol_key_index_workaround=0
eap_server=0
own_ip_addr=127.0.0.1
wpa=2
wpa_passphrase=password
wpa_key_mgmt=SAE WPA-PSK
rsn_pairwise=CCMP-256 CCMP
ieee80211w=1
sae_password=password6
sae_require_mfp=1
```

Here is the `iw dev` output:
```
root@quartz64a:/etc/hostapd# iw dev
phy#0
        Interface wlan0
                ifindex 3
                wdev 0x1
                addr 10:6f:d9:29:19:d7
                ssid test6
                type AP
                channel 36 (5180 MHz), width: 40 MHz, center1: 5190 MHz
                txpower 3.00 dBm
                multicast TXQ:
                        qsz-byt qsz-pkt flows   drops   marks   overlmt hashcol tx-bytes        tx-packets
                        0       0       1100    0       0       0       0       158549          1417

```

I see a wider bandwidth when I'm running the 802.11ac mode:

```
root@quartz64a:/etc/hostapd# iw dev
phy#0
        Interface wlan0
                ifindex 3
                wdev 0x1
                addr 10:6f:d9:29:19:d7
                ssid nopea
                type AP
                channel 36 (5180 MHz), width: 80 MHz, center1: 5210 MHz
                txpower 3.00 dBm
                multicast TXQ:
                        qsz-byt qsz-pkt flows   drops   marks   overlmt hashcol tx-bytes        tx-packets
                        0       0       1144    0       0       0       0       169447          1475

```

I can also get a solid 100mbps down on my phone with the AC setup, and it's over a vpn so this is kinda expected even though I should get up  to a gigabit.  My VPN is limited to 500mbps I think.  I get about 15/15 on the ax AP, same card.

morrownr, 2022-10-25:
@desultory-zz 

I am studying what you posted. I haven't been working on AX AP mode lately as my attention was pulled to work on the Realtek drivers that are maintained here. It happens most of the time when new kernels are released for testing as users want to see what the status of the new kernels happens to be. I have a few other things to take care of and then I will start looking at AP mode again.

My thoughts at this point are that the community, me included, has some learning to do to optimize WiFi 6. I had to reread the thread but still am not clear on the specific hardware and which version of Armbian you are using. In looking at your hostapd.conf, I see a couple of things that could be tweaked for AC but are you trying to get AX going? If so, you are missing a lot of lines.

Nick

desultory-zz, 2022-10-25:
@morrownr 

Thanks for the response.  I was just testing this since I haven't touched it in a while and saw there was a much more recent armbian version.

I'm using a 4gb quartz64 model a
the card is just a mt7291 on a pcie board, I wanted this from factor to hold the antennas since I don't have a case for this.
Here's my uname -a
`Linux quartz64a 6.0.3-media #trunk.0080 SMP PREEMPT Sun Oct 23 18:40:39 UTC 2022 aarch64 aarch64 aarch64 GNU/Linux`
I recompiled hostapd with the same config I have above, the version is 2.10.  I got the sources from the project website.

morrownr, 2022-10-25:
> the card is just a mt7921 on a pcie board, 

I guess this setup would use the mt7921e driver. This is probably of interest to others here. I could even add a section to the in-kernel driver list. Can I talk you into info and links to the usb adapter that holds the cards?

> I recompiled hostapd with the same config I have above, the version is 2.10. I got the sources from the project website.

That is what I did as well.

desultory-zz, 2022-10-26:
Sure, I'll give it a shot

maheralbashek3, 2022-10-26:
Coins free for tiktok to maheralbashek3 9999999+ https://www.tiktok.com/@maher__albashek?_t=8WoUZ01UqQ5&_r=1

desultory-zz, 2022-10-26:
Hmmm, I plugged my old adapter `Bus 005 Device 006: ID 0e8d:7612 MediaTek Inc. MT7612U 802.11a/b/g/n/ac Wireless Adapter` into my new SBC, and it's giving these errors:
```
[ 1213.619960] usb 5-1: new high-speed USB device number 6 using xhci-hcd
[ 1213.770064] usb 5-1: New USB device found, idVendor=0e8d, idProduct=7612, bcdDevice= 1.00
[ 1213.770123] usb 5-1: New USB device strings: Mfr=2, Product=3, SerialNumber=4
[ 1213.770152] usb 5-1: Product: Wireless
[ 1213.770175] usb 5-1: Manufacturer: MediaTek Inc.
[ 1213.770197] usb 5-1: SerialNumber: 000000000
[ 1213.925156] usb 5-1: reset high-speed USB device number 6 using xhci-hcd
[ 1214.096091] mt76x2u 5-1:1.0: ASIC revision: 76120044
[ 1214.149612] mt76x2u 5-1:1.0: Direct firmware load for mt7662_rom_patch.bin failed with error -2
[ 1214.149675] mt76x2u 5-1:1.0: Falling back to sysfs fallback for: mt7662_rom_patch.bin
[ 1274.920463] mt76x2u: probe of 5-1:1.0 failed with error -110
```

Here's my mediatek firmware folder:
```
root@quartz64a:/lib/firmware/mediatek# ls -la
total 1196
drwxrwxr-x  3 root root   4096 Oct 25 06:25 .
drwxr-xr-x 23 root root   4096 Oct 23 16:39 ..
drwxrwxr-x  2 root root   4096 Oct 23 16:37 nvram
-rw-rw-r--  1 root root 113080 Sep 22 06:17 ROMv2_lm_patch_1_0_hdr.bin
-rw-rw-r--  1 root root  50296 Sep 22 06:17 ROMv2_lm_patch_1_1_hdr.bin
-rw-r--r--  1 root root  92192 Oct 25 06:25 WIFI_MT7961_patch_mcu_1_2_hdr.bin
-rw-rw-r--  1 root root 225840 Sep 22 06:17 WIFI_RAM_CODE_7623
-rw-r--r--  1 root root 716404 Oct 25 06:25 WIFI_RAM_CODE_MT7961_1.bin
-rw-rw-r--  1 root root     95 Sep 22 06:17 WMT_SOC.cfg
root@quartz64a:/lib/firmware/mediatek#
```

desultory-zz, 2022-11-06:
got a weird error on the mt7921, idk where else to report it, unplugging and replugging fixed it:
```
[16185.980960] xhci_hcd 0000:01:00.0: WARN: TRB error for slot 3 ep 7 on endpoint
[16185.981033] mt76x2u 2-2:1.0: tx urb failed: -84
[16185.981097] xhci_hcd 0000:01:00.0: WARN waiting for error on ep to be cleared
[16185.981105] mt76x2u 2-2:1.0: tx urb submit failed:-22
[16186.219558] xhci_hcd 0000:01:00.0: WARN waiting for error on ep to be cleared
[16186.219578] mt76x2u 2-2:1.0: tx urb submit failed:-22
[16186.254549] xhci_hcd 0000:01:00.0: WARN waiting for error on ep to be cleared
[16186.254557] mt76x2u 2-2:1.0: tx urb submit failed:-22
[16187.584434] xhci_hcd 0000:01:00.0: WARN waiting for error on ep to be cleared
[16187.584454] mt76x2u 2-2:1.0: tx urb submit failed:-22
[16187.678443] xhci_hcd 0000:01:00.0: WARN waiting for error on ep to be cleared
[16187.678452] mt76x2u 2-2:1.0: tx urb submit failed:-22
```

the network just kinda died, could not ping the interface from the host, and the link stayed up on devices but traffic was not passed.

morrownr, 2022-11-06:
Which kernel are you using?

desultory-zz, 2022-11-06:
the one packaged with the latest alpine, I recently reflashed this device and I'm using diskless mode
`5.15.55-0-rpi4 #1-Alpine SMP PREEMPT Mon Jul 18 10:51:25 UTC 2022 aarch64 Linux`

desultory-zz, 2022-11-06:
I'm considering rebuilding it because this one seems to have cpu governor issues https://gitlab.alpinelinux.org/alpine/aports/-/issues/14330

desultory-zz, 2022-11-06:
My bad, this is not the 7921, this is the 7612

morrownr, 2022-11-06:
> xhci_hcd 0000:01:00.0: WARN: TRB error for slot 3 ep 7 on endpoint

This seems to indicate a usb controller/port issue.

> mt76x2u 2-2:1.0: tx urb failed: -84

This is curious in that the mt76x2u driver is only for the mt7612u and mt7610u chipsets or so I think. I guess it can be investigated but one way or another, if the usb driver has a flat tire, a usb adapter is not going to get very far.

Oh, now I see the msg that this is a mt7612u chipset. My confusion is subsiding. However, I think you should concentrate first on the usb issue as it is probably causing the adapter problem.

desultory-zz, 2022-11-06:
I'm not sure how to look into this usb problem.  I'm also drawing a blank on how to set the module parameters for this kernel module.  I know I did it before when I compiled them myself, but this is working with in kernel drivers.  I believe it's some file under /etc/modules.load or /etc/modprobe?

desultory-zz, 2022-11-06:
I could open a new thread since it may be relevant, but I compiled hostapd on my new router instance, where the mt7612u is plugged in, and I was not able to get WPA3 support working.  I used the same exact config I currently have working on the 7921, like this with password omitted: 

```
ssid=nope
ignore_broadcast_ssid=0
macaddr_acl=0
auth_algs=1
wpa=2
#wpa_key_mgmt=WPA-PSK
wpa_key_mgmt=SAE WPA-PSK
#wpa_pairwise=TKIP
rsn_pairwise=CCMP-256 CCMP
#rsn_pairwise=CCMP
wpa_passphrase=

broadcast_deauth=1
eapol_key_index_workaround=0
eap_server=0
own_ip_addr=127.0.0.1
ieee80211w=1
sae_password=
sae_require_mfp=1

```

desultory-zz, 2022-11-06:
Well... nothing authed until i pulled the card out and reinstered it, nothing weird on dmesg other than it complaining when i abruptly removed the card.

morrownr, 2022-11-06:
> I'm also drawing a blank on how to set the module parameters for this kernel module.

I have a repo that is nothing but info on the mt7612u chipset:

https://github.com/morrownr/7612u

If you go down to Known Issue 2, it will include two ways to set the only module parameter available.

In-kernel drivers do not make much use of module parameters. They are basically like automatic transmissions.

> I'm using a 5v3a adapter for the pi4, I have it set to go up to 2.2ghz but I'm currently having issues because it's using the powersave governor and doesn't seem to want to scale the cores at all, so it's stuck at 1500mhz which is the minimum I set it to allow. I don't think the port is running out of power, but it could be. I have this and a tplink gbe adapter on the usb ports, and a asus usb n53 on the usb 2

I'm getting lost regarding the setup. I see you are using a Pi4 but I really need to know what distro and version plus anything else like external powered usb hub that you are using.

> And now wpa3 is working again, idk what to think of this. I guess I'll take it as working?

> auth_algs=1

Change this to `auth_algs=3` for SAE. Main Menu item 9 has an extensive guide for setting up an AP. The example `hostapd.conf` files it contains give a lot of information, including the specific settings for many adapters, including the mt7612u and mt7921au.

desultory-zz, 2022-11-06:
Ok, that is kinda what I was assuming with the kernel module parameters.  they are being loaded as modules, not baked into the kernel, I thought there was something I could still set on them.  I recall needing to turn off powersaving and force usb3 in one case. 

I'm using alpine linux.  I am not using any powered hubs or anything.  I have the my7612u plugged into the lower usb3 port because of the size, a tplink rt2800 gbe adapter on the upper port, and asus usb n53 on the upper remaining usb 2 port.'

The kernel I'm using is `5.15.76-0-rpi4 #1-Alpine SMP PREEMPT Mon Oct 31 00:54:37 UTC 2022 aarch64 Linux` which is part of the alpine raspberrypi-bootloader package I believe.

I'm currently using auth_algs=1 on the device and it is working.  That is also what I'm using my my mt7921.  I'm about to some more on that, but I believe I chose the option that alowed both wpa2 and 3?

desultory-zz, 2022-11-06:
A while back when I was first getting into making 802.11ac access points, I think I tried several cards before finding this repo and had to manually compile the drivers for some for the devices to work properly.  

Reading your repo about the 7612, I'm assuming I don't need to make that module change since the fix was applied to the raspberry pi kernel? I'm on the lastest rpi foundation kernel, I believe.

ibrtuyima, 2022-11-19:
finally I got 6GHZ AP mode working good with mt7921au (CF-953AX) attached to rpi (kernel - 5.19.4).
hostapd recompilation is needed to enable SAE and AX. 

uname -ra
Linux  5.19.14-v8 #1 SMP PREEMPT Fri Oct 7 09:09:24 UTC 2022 aarch64 GNU/Linux

lsusb -t
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=dwc2/1p, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
    |__ Port 2: Dev 2, If 0, Class=Wireless, Driver=btusb, 5000M
    |__ Port 2: Dev 2, If 1, Class=Wireless, Driver=btusb, 5000M
    |__ Port 2: Dev 2, If 2, Class=Wireless, Driver=, 5000M
    |__ **Port 2: Dev 2, If 3, Class=Vendor Specific Class, Driver=mt7921u, 5000M**
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M

Here is hostap conf file :

ssid=Bebo_Test
wpa_passphrase=_Password_

hw_mode=a

channel=21 (chose one of PSC channels in 6GHz band - 5, 21, 37, 53, 69, 85, 101, 117, 133, 149, 165, 181, 197, 213, and 229)
op_class=133 (This is required for bandwidth selection of 20Mhz (131), 40Mhz(132), 80Mz(133)

country_code=CA
ieee80211d=1

interface=wlan0
driver=nl80211

ieee80211n=0 (This is not required, but need to test only 6Ghz only so disable it)
ieee80211ac=0 (This is not required, but need to test only 6Ghz only so disable it)

auth_algs=3    
wpa=2
wpa_key_mgmt=SAE
rsn_pairwise=CCMP
ieee80211w=2

wmm_enabled=1
uapsd_advertisement_enabled=1
bss_load_update_period=50

chan_util_avg_period=600

ieee80211ax=1
he_bss_color=1
he_oper_centr_freq_seg0_idx=23 (This is required to set the center frequency for channel)

morrownr, 2022-11-20:
Hi @desultory-zz 

> he_oper_centr_freq_seg0_idx=23 (This is required to set the center frequency for channel)

I still don't have a second device that is 6 GHz capable but should soon. I have been running WiFi 6 though.

It is not clear from the hostapd docs if the above is necessary if we use `op_class=133`. Can you test this?

Thanks for posting this. At some point, it might be nice to start a new post with this info. I am working on a 6e hostapd.conf for my AP mode stuff and I plan on posting it once I have the hardware to actually see if it works. What kind of range and speed are you seeing with 6 GHz?

ibrtuyima, 2022-11-20:
in hostapd conf file, they clearly mention the following:

# On the 6 GHz band the center freq calculation starts from 5.950 GHz offset.
# For example idx=3 would result in 5965 MHz center frequency. **In addition,
# he_oper_chwidth is ignored, and the channel width is derived from the
# configured operating class or center frequency indexes (see
# IEEE P802.11ax/D6.1 Annex E, Table E-4).**

so looking for IEEE P802.11ax/D6.1 Annex E, Table E-4, to understand how WIFI operates in 6Ghz band
(https://mentor.ieee.org/802.11/dcn/20/11-20-0646-00-00ax-update-to-6ghz-operating-classes.docx)(Table E-4—Global operating classes)

Range is not so good, just few meters aways (3 - 6 meters).

Speed test as follows:

[image]

amisix, 2022-11-22:
I was able to get this working with on a Raspberry Pi 4B with @morrownr's extensive help using a Comfast CF-953AX. Thank you @ibrtuyima for your additional hostapd.conf settings. The speed isn't there yet but it's on the 6Ghz band!

[image]

[image]

laszlo-t, 2023-02-22:
> FYI: I just happened to reread the title of this issue:
>
>
> The mt7921 chipset that you can do WiFi 6 as in AX. It cannot do 6 GHz channels. WiFi 6e can do 6 GHz channels. You need a mt7921k (for [...]

@morrownr 
I bought a Fenvi MPE-AX1800H mini pci-e card (MT7921) and it works for me.
Access Point on 6Ghz (80Mhz, HU country code)

`OpenWrt SNAPSHOT r22117-7ae4716243 / LuCI Master git-23.039.28596-41e9b8d`

[image]

```
root@OpenWrt:~# lspci | grep MEDIATEK
04:00.0 Network controller: MEDIATEK Corp. MT7921 802.11ax PCI Express Wireless Network Adapter
```

I'm using the these firmware files:
File: mediatek/WIFI_MT7961_patch_mcu_1_2_hdr.bin
Version: 20230117170855a
File: mediatek/WIFI_RAM_CODE_MT7961_1.bin
Version: 20230117170942

```
root@OpenWrt:~# uname -a
Linux OpenWrt 5.15.94 #0 SMP Sun Feb 19 18:44:39 2023 x86_64 GNU/Linux
```

```
                Frequencies:
                        * 5955 MHz [1] (23.0 dBm)
                        * 5975 MHz [5] (23.0 dBm)
                        * 5995 MHz [9] (23.0 dBm)
                        * 6015 MHz [13] (23.0 dBm)
                        * 6035 MHz [17] (23.0 dBm)
                        * 6055 MHz [21] (23.0 dBm)
                        * 6075 MHz [25] (23.0 dBm)
                        * 6095 MHz [29] (23.0 dBm)
                        * 6115 MHz [33] (23.0 dBm)
                        * 6135 MHz [37] (23.0 dBm)
                        * 6155 MHz [41] (23.0 dBm)
                        * 6175 MHz [45] (23.0 dBm)
                        * 6195 MHz [49] (23.0 dBm)
                        * 6215 MHz [53] (23.0 dBm)
                        * 6235 MHz [57] (23.0 dBm)
                        * 6255 MHz [61] (23.0 dBm)
                        * 6275 MHz [65] (23.0 dBm)
                        * 6295 MHz [69] (23.0 dBm)
                        * 6315 MHz [73] (23.0 dBm)
                        * 6335 MHz [77] (23.0 dBm)
                        * 6355 MHz [81] (23.0 dBm)
                        * 6375 MHz [85] (23.0 dBm)
                        * 6395 MHz [89] (23.0 dBm)
                        * 6415 MHz [93] (23.0 dBm)
                        * 6435 MHz [97] (disabled)
                        * 6455 MHz [101] (disabled)
                        * 6475 MHz [105] (disabled)
                        * 6495 MHz [109] (disabled)
                        * 6515 MHz [113] (disabled)
                        * 6535 MHz [117] (disabled)
                        * 6555 MHz [121] (disabled)
                        * 6575 MHz [125] (disabled)
                        * 6595 MHz [129] (disabled)
                        * 6615 MHz [133] (disabled)
                        * 6635 MHz [137] (disabled)
                        * 6655 MHz [141] (disabled)
                        * 6675 MHz [145] (disabled)
                        * 6695 MHz [149] (disabled)
                        * 6715 MHz [153] (disabled)
                        * 6735 MHz [157] (disabled)
                        * 6755 MHz [161] (disabled)
                        * 6775 MHz [165] (disabled)
                        * 6795 MHz [169] (disabled)
                        * 6815 MHz [173] (disabled)
                        * 6835 MHz [177] (disabled)
                        * 6855 MHz [181] (disabled)
                        * 6875 MHz [185] (disabled)
                        * 6895 MHz [189] (disabled)
                        * 6915 MHz [193] (disabled)
                        * 6935 MHz [197] (disabled)
                        * 6955 MHz [201] (disabled)
                        * 6975 MHz [205] (disabled)
                        * 6995 MHz [209] (disabled)
                        * 7015 MHz [213] (disabled)
                        * 7035 MHz [217] (disabled)
                        * 7055 MHz [221] (disabled)
                        * 7075 MHz [225] (disabled)
                        * 7095 MHz [229] (disabled)
                        * 7115 MHz [233] (disabled)