morrownr/USB-WiFi #104: (Hot) Reports on adapters that use the new MT7921au chipset go here

morrownr, 2022-08-29:
I will update the report below as testing continues.

2022-08-29

USB WiFi Adapter: COMFAST CF-951AX

Client mode, 5 GHz, AC, 80 MHz channel width. Channel 60 (DFS - clear air), AMD64 CPU, kernel 6.0 rc3

```
$ iperf3 -c 192.168.1.1
Connecting to host 192.168.1.1, port 5201
local 192.168.1.185 connected to 192.168.1.1
Interval           Transfer     Bitrate         Retr  Cwnd
0.00-1.00   sec  72.2 MBytes   605 Mbits/sec    0   1.83 MBytes
1.00-2.00   sec  77.5 MBytes   650 Mbits/sec    0   1.83 MBytes
2.00-3.00   sec  77.5 MBytes   650 Mbits/sec    0   1.83 MBytes
3.00-4.00   sec  77.5 MBytes   650 Mbits/sec    0   1.83 MBytes
4.00-5.00   sec  77.5 MBytes   650 Mbits/sec    0   1.83 MBytes
5.00-6.00   sec  77.5 MBytes   650 Mbits/sec    0   1.83 MBytes
6.00-7.00   sec  77.5 MBytes   650 Mbits/sec    0   1.83 MBytes
7.00-8.00   sec  77.5 MBytes   650 Mbits/sec    0   1.83 MBytes
8.00-9.00   sec  77.5 MBytes   650 Mbits/sec    0   1.83 MBytes
9.00-10.00  sec  77.5 MBytes   650 Mbits/sec    0   1.83 MBytes   
- - - - - - - - - - - - - - - - - - - - - - - - -
 Bitrate               Retr
 646 Mbits/sec    0
```

Thoughts: The mt7921au chipset and mt7921u driver can push some data. Not bad.

One major problem has been noted and it has slowed testing. This adapter has refused to work with any extension cables that I have tried and I have several that are know to work. I can only get it to work if the adapter is plugged directly into an onboard USB port. I have tried on the 3 separate systems and it held true for all. It is not clear if this is an issue with this specific adapter or if all CF-951AX adapters have this problem. If you have this adapter, please report.

Testing with OpenWRT 22.03 rc6:

Once again, the adapter only works if plugged directly into one of the onboard USB ports. OpenWRT 22.03 has the mt7921u available for installation. The firmware is also available but it is currently part of the package that installs the mt7921e driver so you either have to additionally install that driver or manually install the firmware. A bug has been submitted to change the location of the firmware to a common file. I have tested AP mode 2.4 GHz and 5 GHz. Observations:

- The CF-951AX does not have long range.
- WPA3 works well.
- No 6 GHz channel support in AP mode. (needs investigating)
- AP mode works well in 2.4 and 5 GHz bands though the range is not long.
- No 5 GHz DFS channel support (is this going to be added?)
- No LED support. LED is not coming on as expected. (will this be added?)
- AX is shown as supported but I currently have no way to test.

Recommendation: I cannot recommend purchase at this time. We need more reports regarding whether the adapter will work with extension cables. With the exception of nano adapters, my opinion is that USB WiFi adapters must work well with extension cables in order to locate them for best performance. We shall see what other have to say as more testing is reported.

Nick

ayyyuki1, 2022-09-02:
I have images
[image]
[image]

iw phy >> https://pastebin.com/raw/gBCkBBji

[image]
[image]

6GHz AP needs manual edit on /etc/config/wireless
[image]
you might also need to edit regdb or just set it to a country with 6GHz enabled.
I cant verify 6GHz AP existence, all I know is that hostapd was able to bring it up

ishioni, 2022-09-02:
Any idea if the mt7921u would be backportable to older kernels? I have an odroid-c4 board which runs under a dedicated coreelec distro for kodi, and unfortunately it's bound to the BSP kernel because of unsupported (for now anyway, work is ongoing in linux-meson) video decoders. Right now it's on 4.9, but it will get 5.4 soonish.

morrownr, 2022-09-02:
Hi @ishioni 

> Any idea if the mt7921u would be backportable to older kernels?

It already has been backported to kernel 5.10:

https://github.com/openwrt/mt76

That repo is maintained by the OpenWRT devs, some of which are also Linux Kernel WiFi devs. It was backported for the very soon to be released OpenWRT 22.03 which uses kernel 5.10. The repo above is a downstream of the main Linux repo. 

I would prefer to be working on the in-kernel drivers instead of the Realtek drivers and I could do that if I someone would volunteer to take over some of the current work that I do around here.

ayyyuki1, 2022-09-02:
[image]
luci is offset by 190 in real channel, jow on irc said its iwinfo and thats right.

[image]
this is just visual anyway. 

I hope someone else can test and show if the 6GHz actually exists and seen by other devices. I only have one 6GHz capable device (this one) so I can't do that.

morrownr, 2022-09-02:
@ayyyuki1 

> 6GHz AP needs manual edit on /etc/config/wireless

What is the required edit?

ayyyuki1, 2022-09-02:
> What is the required edit?

[image]

morrownr, 2022-09-02:
@ayyyuki1 

> basically

> option band '6g'
> option channel '77'

I'm on 22.03 rc6 with a CF-951AX and this is not working. Luci won't show the 6g band. Which version of OpenWRT are you using?

ayyyuki1, 2022-09-02:
> I'm on 22.03 rc6 with a CF-951AX and this is not working. Luci won't show the 6g band. Which version of OpenWRT are you using?

show me 

iw phy phyX channels

morrownr, 2022-09-02:
@ayyyuki1 

> make sure first that you dont see "no IR"

I am seeing 'no IR' on all 6g band channels. That is a reg thing. I'm trying to think about what to do but I am less familiar with the internals of OpenWRT than I am with some other platforms. Any guidance?

ayyyuki1, 2022-09-02:
> @ayyyuki1
> 
> > make sure first that you dont see "no IR"
> 
> I am seeing 'no IR' on all 6g band channels. That is a reg thing. I'm trying to think about what to do but I am less familiar with the internals of OpenWRT than I am with some other platforms. Any guidance?

maybe its time for you to fly to DE :wink:

morrownr, 2022-09-02:
My trip to FR went well. Pleasant flight. Now the 'no IR' is gone. Now I need to come up with a way to test. The lack of 6g channel capability is a hindrance. Thanks for the push.

ayyyuki1, 2022-09-02:
> My trip to FR went well. Pleasant flight. Now the 'no IR' is gone. Now I need to come up with a way to test. The lack of 6g channel capability is a hindrance. Thanks for the push.

yeah it seems no Inflight Rockets on your current country.

Some observations
as for some tests I did just now, it seems 6GHz is pure SAE only

`Sat Sep  3 00:38:57 2022 daemon.err hostapd: Management frame protection is required in 6 GHz`
sae-mixed, 802.11w optional

`Sat Sep  3 00:49:47 2022 daemon.err hostapd: Invalid AKM suite for 6 GHz`
sae-mixed with 802.11w (required) still no go

thats some nice rules hey, and this is fine, WPA3-SAE predates 6GHz anyway

morrownr, 2022-09-02:
> Managed to bring up your AP from FR?

Not yet but seem to be making progress. Your comment about sae only seemed to have helped. 

Right now my problem is that I have several others things going and I can't concentrate on this. It may have to wait for a couple of days.

ayyyuki1, 2022-09-02:
ill help you out tomorrow, you might need to build wpad/hostapd with 11ax support, if you can build your own image then this is easy. to avoid technicalities you just build your openwrt sysupgrade with kmod-iwlwifi or kmod-mt7921e as that enables driver_config_11ax_support. I have built hostapd with this support for armv7l (part of wpad openssl full) but ifcourse, i recommend you to build your own image. im currently mobile so excuse tha bad formatting.

morrownr, 2022-09-02:
@ayyyuki1 

I appreciate the help. All indications are that it is working now. The problem is that I need to find a way to test as I currently do not have anything that supports the 6g band. It may be sometimes next week before I can find a way to test.

I went to the OpenWRT site and was checking with WiFi 6 and 6e routers are supported. The light came on for me. There are a few WiFi 6 routers supported but zero 6e at this point. Luci needs some serious work.

ayyyuki1, 2022-09-03:
Nice!, and yep, I do have the same problem too since this is my only 6GHz capable device heh. 
Well, I bought this anyway for monitor and its real good for that, headache free just like AR9271, and yup, 6GHz.

jackloomen, 2022-09-13:
Has anyone been able to get 6ghz meshpoints working with CF-951AX/CF-953AX? What speeds can you obtain?

ayyyuki1, 2022-09-14:
> Has anyone been able to get 6ghz meshpoints working with CF-951AX/CF-953AX? What speeds can you obtain?

no IBSS / 802.11s support yet, so the best you can do is AP+WDS and client+WDS. you can still do some "mesh" via batman but it won't be "mesh" like 802.11s+OLSR

amisix, 2022-09-17:
I'm unable to get the CF-953AX to come up in AP mode in OpenWrt 22.03 on a Raspberry Pi 4B. In Luci it says "Device is not active" & "wireless is not associated". I installed the mt7921e and mt7931u drivers. 

logread below @ 80Mhz channel width. Changing to 40Mhz does not resolve the issue, 

>Sat Sep 17 02:26:02 2022 daemon.notice netifd: radio1 (2203): Failed to create interface wlan0
Sat Sep 17 02:26:02 2022 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy0.conf (phy wlan0) --> new PHY
Sat Sep 17 02:26:02 2022 kern.info kernel: [  313.216398] br-lan: port 2(wlan0) entered blocking state
Sat Sep 17 02:26:02 2022 kern.info kernel: [  313.224484] br-lan: port 2(wlan0) entered disabled state
Sat Sep 17 02:26:02 2022 kern.info kernel: [  313.232448] device wlan0 entered promiscuous mode
Sat Sep 17 02:26:02 2022 daemon.err hostapd: Driver does not support configured HT capability [LDPC]
Sat Sep 17 02:26:02 2022 daemon.notice hostapd: wlan0: interface state UNINITIALIZED->DISABLED
Sat Sep 17 02:26:02 2022 daemon.notice hostapd: wlan0: AP-DISABLED
Sat Sep 17 02:26:02 2022 daemon.err hostapd: wlan0: Unable to setup interface.
Sat Sep 17 02:26:02 2022 daemon.notice hostapd: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Sat Sep 17 02:26:03 2022 kern.info kernel: [  313.282783] device wlan0 left promiscuous mode
Sat Sep 17 02:26:03 2022 kern.info kernel: [  313.291243] br-lan: port 2(wlan0) entered disabled state
Sat Sep 17 02:26:03 2022 daemon.err hostapd: rmdir[ctrl_interface=/var/run/hostapd]: Permission denied
Sat Sep 17 02:26:03 2022 daemon.notice hostapd: wlan0: CTRL-EVENT-TERMINATING
Sat Sep 17 02:26:03 2022 daemon.err hostapd: hostapd_free_hapd_data: Interface wlan0 wasn't started
Sat Sep 17 02:26:03 2022 daemon.notice netifd: radio1 (2203): Command failed: ubus call hostapd config_add {"iface":"wlan0", "config":"/var/run/hostapd-phy0.conf"} (Invalid argument)
Sat Sep 17 02:26:03 2022 daemon.notice netifd: radio1 (2203): Usage: ubus [<options>] <command> [arguments...]
Sat Sep 17 02:26:03 2022 daemon.notice netifd: radio1 (2203): Options:
Sat Sep 17 02:26:03 2022 daemon.notice netifd: radio1 (2203):  -s <socket>:             Set the unix domain socket to connect to
Sat Sep 17 02:26:03 2022 daemon.notice netifd: radio1 (2203):  -t <timeout>:            Set the timeout (in seconds) for a command to complete
Sat Sep 17 02:26:03 2022 daemon.notice netifd: radio1 (2203):  -S:                      Use simplified output (for scripts)
Sat Sep 17 02:26:03 2022 daemon.notice netifd: radio1 (2203):  -v:                      More verbose output
Sat Sep 17 02:26:03 2022 daemon.notice netifd: radio1 (2203):  -m <type>:               (for monitor): include a specific message type
Sat Sep 17 02:26:03 2022 daemon.notice netifd: radio1 (2203):                   (can be used more than once)
Sat Sep 17 02:26:03 2022 daemon.notice netifd: radio1 (2203):  -M <r|t>         (for monitor): only capture received or transmitted traffic
Sat Sep 17 02:26:03 2022 daemon.notice netifd: radio1 (2203):
Sat Sep 17 02:26:03 2022 daemon.notice netifd: radio1 (2203): Commands:
Sat Sep 17 02:26:03 2022 daemon.notice netifd: radio1 (2203):  - list [<path>]                  List objects
Sat Sep 17 02:26:03 2022 daemon.notice netifd: radio1 (2203):  - call <path> <method> [<message>]       Call an object method
Sat Sep 17 02:26:03 2022 daemon.notice netifd: radio1 (2203):  - subscribe <path> [<path>...]   Subscribe to object(s) notifications
Sat Sep 17 02:26:03 2022 daemon.notice netifd: radio1 (2203):  - listen [<path>...]                     Listen for events
Sat Sep 17 02:26:03 2022 daemon.notice netifd: radio1 (2203):  - send <type> [<message>]                Send an event
Sat Sep 17 02:26:03 2022 daemon.notice netifd: radio1 (2203):  - wait_for <object> [<object>...]        Wait for multiple objects to appear on ubus
Sat Sep 17 02:26:03 2022 daemon.notice netifd: radio1 (2203):  - monitor                                Monitor ubus traffic
Sat Sep 17 02:26:03 2022 daemon.notice netifd: radio1 (2203):
Sat Sep 17 02:26:03 2022 daemon.notice netifd: radio1 (2203): Device setup failed: HOSTAPD_START_FAILED
Sat Sep 17 02:26:03 2022 daemon.notice netifd: Wireless device 'radio1' set retry=0
Sat Sep 17 02:26:03 2022 daemon.crit netifd: Wireless device 'radio1' setup failed, retry=0
Sat Sep 17 02:26:03 2022 daemon.notice netifd: Wireless device 'radio1' is now down

morrownr, 2022-09-17:
> Sat Sep 17 02:26:02 2022 daemon.err hostapd: Driver does not support configured HT capability [LDPC]

You may have to manually get rid of `[LDPC}`.

amisix, 2022-09-17:
Awesome, thanks. No resolution after adding "option ldpc '0'" to /etc/config/wireless although I did progress further in the error chain per the below log. I had to add the following values to /etc/config/wireless to progress further.

>option ldpc '0'
option tx_stbc '0'
option rx_stbc '0'
option max_amsdu '0'

>Sat Sep 17 03:28:11 2022 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy0.conf (phy wlan0) --> new PHY
Sat Sep 17 03:28:11 2022 kern.info kernel: [ 4041.724030] br-lan: port 2(wlan0) entered blocking state
Sat Sep 17 03:28:11 2022 kern.info kernel: [ 4041.730968] br-lan: port 2(wlan0) entered disabled state
Sat Sep 17 03:28:11 2022 kern.info kernel: [ 4041.738007] device wlan0 entered promiscuous mode
Sat Sep 17 03:28:11 2022 daemon.notice hostapd: wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
Sat Sep 17 03:28:11 2022 daemon.err hostapd: Configured VHT capability [VHT_CAP_MAX_MPDU_LENGTH_MASK] exceeds max value supported by the driver (2 > 0)
Sat Sep 17 03:28:11 2022 daemon.notice hostapd: wlan0: interface state COUNTRY_UPDATE->DISABLED
Sat Sep 17 03:28:11 2022 daemon.notice hostapd: wlan0: AP-DISABLED
Sat Sep 17 03:28:11 2022 daemon.err hostapd: wlan0: Unable to setup interface.

morrownr, 2022-09-17:
That is interesting. I've only tested AP mode with OpenWRT on my ZyXEL NBG6817 wifi router, All I did was load the packages for mt7921e and mt7921u and configure the new interface with Luci. It just worked. Now, that is for WiFi 6. To get it to do WiFi 6e I had to mod /etc/config/wireless as Luci is not ready for 6e yet (22.03).

I'm not an expert on the internals of OpenWRT. Now, hostapd by itself, I can hold my own.

amisix, 2022-09-17:
Have you had a chance to test it with a CF-953AX yet or just the CF-951AX? (they should respond similar... right?) I have a CF-951AX here and can test it this weekend. Yeah, usually Luci handles most of it fine for me when using mt drivers,

(stupid question) How do you get your hostapd settings to persist through a restart in OpenWrt? Is everything done through uci?

morrownr, 2022-09-17:
> Have you had a chance to test it with a CF-953AX yet or just the CF-951AX?

I only have a 951AX.

> usually Luci handles it fine for me when using mt drivers

Luci handles WiFi 6 (AX) fine here with the 951AX. They problem comes in with the 6 GHz band. Luci is not yet upgraded to handle it. Luci can do WiFi 6 but not WiFi 6e. I'm sure it will happen but it is not there yet.

> How do you get your hostapd settings to persist through a restart in OpenWrt?

No clue.

Remember that there is some good 6 GHz setup info back up this thread. Note the need to set security to `sae` as 6 GHz doesn't do mixed or WPA2 or anything else, only WPA3.

amisix, 2022-09-17:
Tested it this morning with a fresh 22.03 system image & the 951AX produced identical errors on the Pi. I'll move this over to another thread.. 
Thx for the heads up re: 6e WPA3.

morrownr, 2022-11-17:
@dhewg 

The cover is easy to remove on my CF-951AX, not that I need to remove. Since the 953 have the external antennas, I'm curious as to your thoughts on the range?

My opinion of the range of the 951 is that it is shorter than I expected. I'm not saying it is really bad but a little shorter than I expected... however, I've been using Alfa adapters for a long time so my experience may not be typical.

dhewg, 2022-11-17:
I haven't yet tested it on 6g, just 5g as I'm still waiting for another 6g card to connect to it :)
And at least on the 5g band it's not yet stable for me, even with the latest and greatest driver using OpenWrt: https://github.com/openwrt/mt76/issues/711

So no judgement yet, but hopefully soon.

I did open it to see if I can connect external antennas once that thing is stable. But as you can see, the current ones are soldered. The pads aren't too small to solder, so I might just do that to increase the range even more, but we'll see...

dhewg, 2023-01-17:
Finally got around to solder antenna connectors to it:
[image]

With that I get these rates on close range:
[image]

Sweet! A few rooms/walls away I get better reception than with 5g using my ipq40xx router

morrownr, 2023-01-18:
Cool. Have you tried a fresh install with OpenWRT 22.03.3? Maybe the stability thing is history. I have a CF-951AX stuffed into the USB3 port on my Zyxel wifi router running 22.02.3 and it appears to be very stable. The devs finally took the firmware out of the mt7921e driver and gave it it's own package so all you have to these days is install the mt7921u package. No 6 GHz support in Luci yet but there is progress.

dhewg, 2023-01-18:
It's all latest OpenWrt master. As for the stability, it's now connected to my ipq4019 router. It's rock stable now, dunno why it wasn't on my ramips one. See also https://github.com/openwrt/mt76/issues/711 and https://github.com/openwrt/openwrt/issues/11796

But because usb sucks on ipq4019 those rate are theoretically only, I don't reach a throughput anywhere close. But the stability, range and non-populated freqs are there.

And I know about the changes, I did some of them :P
https://github.com/openwrt/openwrt/commit/6f729163b18fb5860f1aa5a5a0c8861a8e3f53ad
https://git.openwrt.org/?p=project/iwinfo.git;a=shortlog

What's left is the remainder of:
https://github.com/openwrt/openwrt/pull/11280
and
https://github.com/openwrt/luci/pull/6104

For nice and working 6g luci support.

morrownr, 2023-01-18:
@dhewg 

Thank you for your work.

dhewg, 2023-01-19:
Eh, no need to say thanks. Contributing back to things you've been using for years is always nice.
Time is sparse among core devs, so merging takes some time, but let's hope the remaining patches land soon.

While here, did you ever find a usb wifi adapter that support meshes/80211s?

morrownr, 2023-01-19:
> While here, did you ever find a usb wifi adapter that support meshes/80211s?

mesh is something that I actually know very little about and I do not have a mesh capable router/AP. I do have an extensive collection of adapters. How would I go about testing without a mesh capable router/AP?

dhewg, 2023-01-20:
You can just check `iw list`
My AP's devices have `mesh point` under `Supported interface modes:`, and CF-953AX doesn't.
With OpenWrt it's easy to setup a mesh, just set "Mode" to "802.11s"

ZerBea, 2023-01-22:
I would be interested to know how the adapter performs running hcxdumptool (because I don't have a MT7921 device to test):

prepare:
$ sudo systemctl stop NetworkManager.service
$ sudo systemctl stop wpa_supplicant.service

run test (please do not use iw to set monitor mode before)
$ sudo hcxdumptool -i INTERFACE --check_driver
$ sudo hcxdumptool -i INTERFACE --check_injection
$ sudo hcxdumptool -i INTERFACE --C

re-start WiFi network
$ sudo systemctl start NetworkManager.service
$ sudo systemctl start wpa_supplicant.service

Thank you in advance.

dhewg, 2023-08-25:
The patch has been merged to the mt76 repo on 06/22:
https://github.com/openwrt/mt76/commit/84ea1a24f5b57dc02e262d72ae3f41abc1b7aa3a
And that hit OpenWrt master on the same day:
https://github.com/openwrt/openwrt/commit/901af2718957fc878279073e9b917458144c75a4
As well as the 23.05 branch:
https://github.com/openwrt/openwrt/commit/649d9e359005377b9efc445466db58f54a547875
So you really shouldn't see the issue when using the device on OpenWrt.

But it looks like it hasn't been merged on mainline linux. But mentioning it here probably doesn't help in that regard.
In any case, it's still rock stable for me

saepurusass, 2023-08-25:
Sorry about the confusion.
I meant generic desktop/server linux and hostapd setup.

Looking at the latest kernel 6.4 source, the patch is not included there. And un-patched 6.4 kernel crashes your system.
Manual patching is required for any generic desktop/server linux.

I only mentioned OpenWRT because I found the patch there.

morrownr, 2023-08-25:
> Looking at the latest kernel 6.4 source, the patch is not included there. 

This patch is approved and is set to go into kernel 6.6.

It is frustrating that we sometimes wait longer than we would like for fixes to happen but it can happen. A few years ago some jackasses at some university managed to get some patches in the kernel that were BS. After that, Linus put the hammer down and patches such as this have to be checked rigorously.  Some patches can move through the system rapidly but others not so. 

This patch is interesting in that I have yet to experience the problem and I use AP mode a lot. If I recall correctly, this bug only shows up depending on how AP mode is set up. Nevertheless, the patch will be in soon, so hopefully the problem can fade away.