morrownr/USB-WiFi #105: Any USB adapter with embedded firmware?

bdantas, 2022-09-01:
I mostly deal with laptops, which almost always come with wireless network cards that require installing firmware. Getting a wired connection in order to install the missing firmware isn't always possible or convenient.

Do you know of any USB wireless adapter that works with in-kernel driver and **comes with embedded firmware**? I'm okay if it supports only 2.4 GHz and 802.11g. The sole purpose of such adapter would be to allow initial internet connection so that I can install firmware for the internal wireless network card to work.

morrownr, 2022-09-01:
Hi @bdantas 

> The sole purpose of such adapter would be to allow initial internet connection so that I can install firmware for the internal wireless network card to work.

The answer can depend on the distros of Linux that you use. If you use Debian or other distros that do not, by default, include wifi firmware, go for an adapter that uses an Atheros ar9271 chipset. This chipset is unique in that does not require firmware. The driver is fully open source,

If you deal with distros that do include wifi firmware by default, you might want to check out the adapters based on either the RT5370 or MT7601U chipsets.

Main Menu item 2 gives more information and links to adapters:

https://github.com/morrownr/USB-WiFi/blob/main/home/USB_WiFi_Adapters_that_are_supported_with_Linux_in-kernel_drivers.md

There are certainly more adapters than are listed so hopefully you can find an adapter that work well for you.

Nick

bdantas, 2022-09-01:
> The answer can depend on the distros of Linux that you use.
In my family we use Devuan (a Debian derivative) and Tiny Core Linux. Neither distro includes firmware by default.

> go for an adapter that uses an Atheros ar9271 chipset. This chipset is unique in that it does not require firmware.
I'm a big fan of Atheros because their chipsets are very freedom-friendly: both driver _and firmware_ are open source. 

Unfortunately, I'm finding that the AR9271 _does_ require external (albeit open source) firmware. 
I use one of these (https://www.amazon.com/dp/B07FVRKCZJ?psc=1&ref=ppx_yo2ov_dt_b_product_details) on a Tiny Core Linux box. Output of _lsusb_ confirms that it does, indeed, have the AR9271 chipset:
```
bruno@x200:/opt$ lsusb
Bus 002 Device 003: ID 0cf3:9271 Atheros Communications, Inc. AR9271 802.11n
---snip---
```
If I uninstall firmware-atheros.tcz and reboot, the adapter stops working. Maybe the chipset seems to not require external firmware to you because, by virtue of it being open source, the distro you use can include its firmware in its installation media without running afoul of licensing issues.
 
Can you think of any other option that might come with embedded firmware? Maybe I'm on an impossible quest because there is no such thing as a USB wireless adapter that comes with embedded firmware?

morrownr, 2022-09-01:
> Maybe the chipset seems to not require external firmware to you because, by virtue of it being open source, the distro you use includes its firmware in the base system rather than in an installable package?

You are likely correct. Debian is where I have tested this and Debian is against closed source firmware and they may include this since it is open source.

You may need to look at alternative solutions:

https://www.amazon.com/Powerline-Ethernet-Adapter-Extender-TP-Link/dp/B084CZMYNM/ref=sr_1_15?keywords=wifi+to+ethernet+adapter&qid=1662035435&sr=8-15

I have had good luck with Homeplug AV2 devices but your mileage will vary.

My understanding is that USB WiFi adapters all have a very limited internal firmware but it does not do the things we are looking for... such as be part of the device driver. It would be handy if the adapter did have flashable internal firmware so as to eliminate the need for kernel based firmware.

bdantas, 2022-09-01:
So it seems that I was on an impossible quest, after all. If _you_ don't know of something like this, that probably means that it doesn't exist :)
 
Powerline ethernet is an interesting thought. I'll look into it.
 
If you or one of your visitors ever finds a USB wireless network adapter that a) uses a driver that comes with linux by default (either as part of the kernel or as a kernel module) and b) comes with embedded firmware, please let me know by posting a reply here. Such a thing would sure take out a lot of pain from setting up new linux installations on laptops.

morrownr, 2022-09-01:
> Such a thing would sure take out a lot of pain from setting up new linux installations on laptops.

Moving the firmwre into flashable memory on the adapter would seem to be a very good solution. I can certainly leave your message here. I am aware of more than one company involved in the industry who has employees that stop by here so the message could make its way to where it needs to go.

bdantas, 2022-09-01:
Thank you. Fingers crossed!

bdantas, 2022-09-02:
While I wait for something like this to exist, a workaround is to have a USB wireless adapter and also a USB thumbdrive that has the necessary firmware file. In the case of my adapter with AR9271 chipset, the necessary firmware file is `htc_9271.fw`.
 
So on a fresh linux installation on a laptop whose internal wireless network isn't working (because of missing firmware) and where wired internet connection (to install the package containing the missing firmware) is unavailable or inconvenient, all I have to do is this:
```
$ sudo mkdir -p /lib/firmware
$ sudo cp /path/to/thumbdrive/htc_9271.fw /lib/firmware
```
Then plugin the AR9271 adapter. Now I can get connected to internet so that I can install the firmware for internet wireless network card.

bjlockie, 2022-09-12:
I want to compare driver and firmware sizes between different adapters.
I was using lsmod | grep mt but does that include the loaded firmware?

bdantas, 2022-09-13:
Hi, bjlockie. That command does not include the loaded firmware. Unfortunately there is no standard way to find the loaded firmware filename because different drivers use slightly different verbiage when they log the firmware loading step to the kernel ring buffer.
 
Your best bet is to do `sudo lshw -c network` to find the name of your adapter's driver. Look for the `driver=` field. My wireless network adapters (made by Intel, Atheros, and Mediatek), for example, use the `iwlwifi`, `ath9k`, and `mt76x0u` drivers, respectively.
 
To find the loaded firmware, try `dmesg | grep <driver>`, for example: 
```
$ dmesg | grep iwlwifi
---snip---
iwlwifi 0000:02:00.0: loaded firmware version 29.163394017.0 7265D-29.ucode op_mode iwlmvm
---snip---
$ dmesg | grep ath9k
---snip---
usb 2-2: ath9k_htc: Transferred FW: ath9k_htc/htc_9271-1.4.0.fw, size: 51008
---snip---
$ dmesg | grep mt76x0u
-> no hits having to do with firmware :(
```
Note how the relevant line for the Intel and Atheros firmware load are not following a standard pattern. Also notice how the Mediatek driver does not seem to generate a message when it loads the firmware.
 
My Atheros firmware filesize is tiny (49.8 KB) while the Intel firmware is quite a bit bigger (1012.4 KB). I'm not sure about the size of the Mediatek adapter's firmware because the `lshw` followed by `dmesg | grep` strategy did not give me the name of the firmware file. I hope this helps you compare the firmware sizes of your different adapters.

closed 2022-11-22 by morrownr