morrownr/USB-WiFi #106: rtl 8812au driver didnot compile in fedora 5.19.4-200.fc36 kernel update

biswajit-rajaguru-homepc, 2022-09-05:
hi,
i have linksys wusb6300 with rtl8812au chipset under fedora 36 workstation. till the last update 5.18.19 the driver through install.sh compiled, installed and worked fabulously. but the new 5,19.4-200.fc36 kernel update broke the wifi. i tried t remove the old driver by remove.sh and then installed using install.sh. but it did not compile and exited with error 2. i downgraded/rolled back to the previous kernel and it installed fine and working fabulous. thanks.

morrownr, 2022-09-05:
Hi @biswajit-rajaguru-homepc 

> i tried t remove the old driver by remove.sh and then installed using install.sh. but it did not compile and exited with error 2. 

You forgot a step. You needed to update the driver source code on your harddrive:

$ sudo ./remove-driver.sh
$ git pull
$ sudo ./install-driver.sh

This driver is updated constantly so if you are going to update the version of the kernel such as 5.18 to 5.19, you need to run the above before installing the new kernel.

FYI: The Main Menu for this site has a lot of info:

https://github.com/morrownr/USB-WiFi

Let me know how it goes.

Nick

morrownr, 2022-09-17:
Assuming this is fixed.

closed 2022-09-17 by morrownr