morrownr/USB-WiFi #107: (Hot) Post Bug Reports for the new mt7921au chipset and mt7921u driver here...

morrownr, 2022-09-06:
Bug: Support for Bluetooth that does not exist.

It appears that the mt7921u driver is hardwired to start Bluetooth support, even when their is no active Bluetooth hardware. There is no indication that the Comfast CF-951AX (mt7921au chipset) has Bluetooth support.

dmesg provides the following if the Bluetooth firmware is installed:

standby for up date

dmesg provides the following if the Bluetooth firmware is NOT installed:

usb 4-1: new SuperSpeed USB device number 27 using xhci_hcd
[  149.870815] usb 4-1: New USB device found, idVendor=0e8d, idProduct=7961, bcdDevice= 1.00
[  149.870824] usb 4-1: New USB device strings: Mfr=6, Product=7, SerialNumber=8
[  149.870828] usb 4-1: Product: Wireless_Device
[  149.870830] usb 4-1: Manufacturer: MediaTek Inc.
[  149.870833] usb 4-1: SerialNumber: 000000000
[  149.874493] bluetooth hci0: Direct firmware load for mediatek/BT_RAM_CODE_MT7961_1_2_hdr.bin failed with error -2
[  149.874502] Bluetooth: hci0: Failed to load firmware file (-2)
[  149.874508] Bluetooth: hci0: Failed to set up firmware (-2)
[  149.874511] Bluetooth: hci0: HCI Enhanced Setup Synchronous Connection command is advertised, but not supported.

More to come as I document the bugs I am seeing...

Post your bugs and I will bring them into this message so that we have a complete list.

deren, 2022-09-15:
Hi @morrownr 

I cannot reproduce the mess system log in in 6.0-rc3. Can you please show me full log & reproduce steps(if any special)?

Thanks,
Deren

morrownr, 2022-09-15:
Hi @deren 

I will retest on the original system plus two additional systems and report as soon as I can.

Regards,

Nick

deren, 2022-09-16:
Hi @morrownr 

I cannot reproduce Bug2, either. The BT function always working properly, even if plug&play or reload driver several times.

But the log is weird to me, it looks like missing fw file on filesystem. Can you please check this problem related to any specific device?
`[  149.874493] bluetooth hci0: Direct firmware load for mediatek/BT_RAM_CODE_MT7961_1_2_hdr.bin failed with error -2`

Regards,
Deren

morrownr, 2022-09-16:
Hi @deren 

> The BT function always working properly, even if plug&play or reload driver several times.

I have a laptop computer that uses a wifi card based on the mt7921 chipset. Bluetooth works well. However, the subject here is about a usb wifi adapter that uses the mt7921au chipset. I have found no evidence that this adapter supports bluetooth. Is the capability turned off in hardware? I don't know but am trying to find out.

> But the log is weird to me

What I posted is not information from a single log. The first section shows the log entries when the most up to date BT firmware is installed. The second section shows the log entries when I delete the BT firmware and reboot.

The first section of the log with the firmware installed makes me think the driver is trying to bring BT up but it is unable due to the lack of hardware. Bluetooth support is rare on usb wifi adapters. Could it be that the driver, mt7921u, is making an incorrect assumption that BT hardware is there to use?

My mt7921au based usb wifi adapter is showing no sign of support for BT whether the firmware is install or not. This adapter is a Comfast CF-951AX.

Regards,

Nick

deren, 2022-09-17:
Hi @morrownr 

> What I posted is not information from a single log. The first section shows the log entries when the most up to date BT firmware is installed. The second section shows the log entries when I delete the BT firmware and reboot.

Got it. I did not get the point.

> I have found no evidence that this adapter supports bluetooth. Is the capability turned off in hardware?

What I have is CF-953AX and can verify BT function working well in the card. Regarding the CF-951AX, there are some information about BT function. ( hope that is real :) )
https://www.sunsky-online.com/p/EDA003280201A/COMFAST-CF-951AX-1800Mbps-USB-3.0-WiFi6-Wireless-Network-Card-Black-.htm

> [   72.869871] Bluetooth: hci1: Opcode 0x c03 failed: -110
> [   74.882933] Bluetooth: hci1: Failed to read MSFT supported features (-110)
> [   76.896658] Bluetooth: hci1: AOSP get vendor capabilities (-110)

I guess you consider the three lines means the BT function not working properly, right? The log do not show up in my test environment(ubuntu2004+kernel 6.0-rc5 + CF-953AX) and I think this is related to BT protocol usage in your host system. Can you see BT still alive in your system, such as desktop UI? For example, I can see the device is running after CF-953AX plugged.

> \# hciconfig
> hci1:   Type: Primary  Bus: USB
        BD Address: XX:XX:XX:XX:XX:XX  ACL MTU: 1021:4  SCO MTU: 96:6
        UP RUNNING
        RX bytes:17035 acl:0 sco:0 events:2758 errors:0
        TX bytes:678011 acl:0 sco:0 commands:2756 errors:0

Regards,
Deren

morrownr, 2022-09-19:
Hi @deren 

> I guess you consider the three lines means the BT function not working properly, right?

I consider those 3 lines are possibly going to give a hint as to why BT is not working. And it is not working. I wish I was more familiar with BT but I am not so I am having to go slow with my troubleshooting. I also have a little nano BT adapter with a Braodcome chipset. I plugged it in and BT works with it. Mt test distro is Mint 21 (updated to kernel 6.0 rc3) and a short look at the forums tells me Mint is having problems with BT so I am going to suspend my report pending further investigation. I appreciate your help.

> Can you see BT still alive in your system,

Yes. It shows in the gui applet that supports BT. In fact, with the nano adapter plugged in, both show. The difference is that I can pair with other BT devices with the nano adapter but I can't get anything with the Mediatek based adapter. I'll continue trying to determine the problem.

Nick

morrownr, 2022-09-22:
@deren 

Question: Since you have a Comfast CF-953AX adapter, my question is have you tested AP mode 5 GHz band DFS channel support?

I bring this up due to the MANY users that make use of AP mode. Many products also include usb adapters and AP mode is sometimes an important part of the product. I am contacted at times to serve in a consulting roll so I see many use cases that would benefit greatly from this support... and the 5 Realtek drivers I have up here all support this and it works well. There is a gap in capability that needs to be closed and it doesn't seem to be working for me.

Thanks,

Nick

leezu, 2022-09-28:
On RasPi4B with Raspbian and [kernel 6.0.0-rc7-v8+](https://github.com/raspberrypi/linux/commit/3fb5ca89c844eb9fda924a5e76ffab9e8c068675) (as well as earlier kernels), CF-953AX in AP mode and the latest September 2022 Firmware, I can relatively consistently reproduce firmware hangs by running a speedtest and `apt upgrade` in parallel on the SC7180 Snapdragon 7c. (Having other devices connected and other traffic patterns sometimes also causes the hang).

After the hang, the AP will recover and the following (starting from 4440.067187) is printed in the kernel log.

```
[Wed Sep 28 12:48:10 2022] mt7921u 2-1:1.3: WM Firmware Version: ____010000, Build Time: 20220908211021
[Wed Sep 28 12:48:11 2022] mt7921u 2-1:1.3 wlxe0e1a934a6a9: renamed from wlan1
[Wed Sep 28 12:48:12 2022] Bluetooth: hci0: Opcode 0x c03 failed: -110
[Wed Sep 28 12:48:13 2022] IPv6: ADDRCONF(NETDEV_CHANGE): ap0: link becomes ready
[Wed Sep 28 12:48:13 2022] IPv6: ADDRCONF(NETDEV_CHANGE): wlxe0e1a934a6a9: link becomes ready
[Wed Sep 28 14:02:00 2022] mt7921u 2-1:1.3: Message 00020003 (seq 12) timeout
[Wed Sep 28 14:02:00 2022] wlxe0e1a934a6a9: failed to remove key (0, 8c:fd:f0:42:20:cf) from hardware (-110)
[Wed Sep 28 14:02:00 2022] mt7921u 2-1:1.3: timed out waiting for pending tx
[Wed Sep 28 14:02:00 2022] ------------[ cut here ]------------
[Wed Sep 28 14:02:00 2022] WARNING: CPU: 2 PID: 1158 at kernel/kthread.c:659 kthread_park+0xb4/0xd0
[Wed Sep 28 14:02:00 2022] Modules linked in: xt_mark nft_chain_nat ctr aes_arm64 aes_generic ccm xt_MASQUERADE iptable_nat ip6table_nat nf_nat tun mt7921u mt7921_common mt76_connac_lib mt76_usb mt76 mac80211 btusb btrtl btintel btbcm bluetooth ecdh_generic ecc libaes libarc4 sg vc4 snd_soc_hdmi_codec bcm2835_codec(C) drm_display_helper brcmfmac rpivid_hevc(C) cec drm_cma_helper bcm2835_v4l2(C) bcm2835_isp(C) brcmutil v3d drm_kms_helper bcm2835_mmal_vchiq(C) v4l2_mem2mem videobuf2_vmalloc videobuf2_dma_contig gpu_sched videobuf2_memops cfg80211 snd_soc_core drm_shmem_helper videobuf2_v4l2 snd_compress videobuf2_common snd_bcm2835(C) snd_pcm_dmaengine videodev rfkill raspberrypi_hwmon snd_pcm vc_sm_cma(C) mc snd_timer snd syscopyarea sysfillrect sysimgblt fb_sys_fops uio_pdrv_genirq nvmem_rmem uio ip6t_REJECT nf_reject_ipv6 xt_hl ip6_tables ip6t_rt ipt_REJECT nf_reject_ipv4 xt_comment xt_multiport nft_limit xt_limit xt_addrtype xt_tcpudp xt_conntrack nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4
[Wed Sep 28 14:02:00 2022]  nft_compat nf_tables nfnetlink drm fuse drm_panel_orientation_quirks backlight ip_tables x_tables ipv6
[Wed Sep 28 14:02:00 2022] CPU: 2 PID: 1158 Comm: kworker/u8:1 Tainted: G         C         6.0.0-rc7-v8+ #4
[Wed Sep 28 14:02:00 2022] Hardware name: Raspberry Pi 4 Model B Rev 1.4 (DT)
[Wed Sep 28 14:02:00 2022] Workqueue: mt76 mt7921_mac_reset_work [mt7921_common]
[Wed Sep 28 14:02:00 2022] pstate: 00000005 (nzcv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[Wed Sep 28 14:02:00 2022] pc : kthread_park+0xb4/0xd0
[Wed Sep 28 14:02:00 2022] lr : mt76u_stop_tx+0x278/0x330 [mt76_usb]
[Wed Sep 28 14:02:00 2022] sp : ffffffc00938bc50
[Wed Sep 28 14:02:00 2022] x29: ffffffc00938bc50 x28: 0000000000000000 x27: ffffff8049ee8848
[Wed Sep 28 14:02:00 2022] x26: 0000000000000000 x25: ffffff8049e42280 x24: ffffff8049ee2068
[Wed Sep 28 14:02:00 2022] x23: ffffff8049ee4820 x22: ffffff8049ee6020 x21: ffffff8049ee2048
[Wed Sep 28 14:02:00 2022] x20: ffffff8049368c00 x19: ffffff804c8f0000 x18: 0000000000000000
[Wed Sep 28 14:02:00 2022] x17: 0000000000000001 x16: ffffffdcef6b4e40 x15: 0018a9ea46410578
[Wed Sep 28 14:02:00 2022] x14: 000dd98ec5e42494 x13: 0000000000000213 x12: 00000000fa83b2da
[Wed Sep 28 14:02:00 2022] x11: 0000000000000213 x10: 0000000000001a90 x9 : ffffffdccfd9e8a8
[Wed Sep 28 14:02:00 2022] x8 : ffffff80401ed8f0 x7 : 0000000000000001 x6 : ffffffdcf0b0b0c0
[Wed Sep 28 14:02:00 2022] x5 : ffffffdcf09a9000 x4 : ffffffdcf09a90b0 x3 : 0000000000002800
[Wed Sep 28 14:02:00 2022] x2 : 0000000000000000 x1 : 0000000000000000 x0 : 0000000000000004
[Wed Sep 28 14:02:00 2022] Call trace:
[Wed Sep 28 14:02:00 2022]  kthread_park+0xb4/0xd0
[Wed Sep 28 14:02:00 2022]  mt76u_stop_tx+0x278/0x330 [mt76_usb]
[Wed Sep 28 14:02:00 2022]  mt7921u_mac_reset+0x88/0x2d8 [mt7921u]
[Wed Sep 28 14:02:00 2022]  mt7921_mac_reset_work+0xac/0x1a0 [mt7921_common]
[Wed Sep 28 14:02:00 2022]  process_one_work+0x1dc/0x450
[Wed Sep 28 14:02:00 2022]  worker_thread+0x154/0x450
[Wed Sep 28 14:02:00 2022]  kthread+0x104/0x110
[Wed Sep 28 14:02:00 2022]  ret_from_fork+0x10/0x20
[Wed Sep 28 14:02:00 2022] ---[ end trace 0000000000000000 ]---
[Wed Sep 28 14:02:00 2022] mt7921u 2-1:1.3: HW/SW Version: 0x8a108a10, Build Time: 20220908210919a

[Wed Sep 28 14:02:00 2022] mt7921u 2-1:1.3: WM Firmware Version: ____010000, Build Time: 20220908211021
[Wed Sep 28 14:02:05 2022] mt7921u 2-1:1.3: Message 00020003 (seq 6) timeout
[Wed Sep 28 14:02:06 2022] mt7921u 2-1:1.3: timed out waiting for pending tx
[Wed Sep 28 14:02:06 2022] mt7921u 2-1:1.3: HW/SW Version: 0x8a108a10, Build Time: 20220908210919a

[Wed Sep 28 14:02:06 2022] mt7921u 2-1:1.3: WM Firmware Version: ____010000, Build Time: 20220908211021
```

ayyyuki1, 2022-09-28:
maybe some issues again with usb scattering

leezu, 2022-10-01:
@ayyyuki1 possible. Based on above call trace, mt7921u driver does call a mt76_usb function. I'll try reproducing the issue with `/sys/module/mt76_usb/parameters/disable_usb_sg` set to `Y`

morrownr, 2022-10-02:
> Based on above call trace, mt7921u driver does call a mt76_usb function. I'll try reproducing the issue with `/sys/module/mt76_usb/parameters/disable_usb_sg` set to `Y`

My opinion is that I would like to see either the default setting for scatter-gather be changed or simple clean the code out of mt76. I have seen the problems that it causes with mt7612u based adapters and I have taken the time to do extensive tests to see if it increases speed in any worthwhile level. My conclusion is that I cannot see any increases in speed. I say dump it from the code.

bjlockie, 2022-10-02:
I think the problem with scatter gather was suspected to be a bug in the USB hardware of a Raspberry Pi.

leezu, 2022-10-02:
I was unable to reproduce this issue with `disable_usb_sg`. Raspberry Pi team has recently made progress on this/a related USB issue and established that it is a VL805 firmware bug. https://github.com/raspberrypi/linux/issues/4844 "VIA in Taiwan have reproduced the issue and are investigating. There's not likely to be another software workaround proposed here." "The fix as recommended by VIA is to disable bursts if this sequence of TRBs can occur." https://github.com/raspberrypi/linux/commit/3157603c3925a30c6b26dc8f7a1f2c23f7b7bb55 Currently it only applies to USB_CLASS_MASS_STORAGE devices. Here burst "defines how many single reads/writes [DMAengine is] allowed to do without the controller splitting the transfer into smaller sub-transfers." so I guess disabling burst may fix the scatter-gather codepath https://www.kernel.org/doc/html/latest/driver-api/dmaengine/provider.html (Also suggested at https://github.com/raspberrypi/linux/pull/5173#discussion_r974429018)

bjlockie, 2022-10-03:
Can the firmware of the VIA chip be upgraded?

leezu, 2022-10-04:
Yes.

```
% sudo rpi-eeprom-update
[...]
  VL805_FW: Using bootloader EEPROM
     VL805: up to date
   CURRENT: 000138a1
    LATEST: 000138a1
```

morrownr, 2022-10-08:
> On RasPi4B with Raspbian and [kernel 6.0.0-rc7-v8+](https://github.com/raspberrypi/linux/commit/3fb5ca89c844eb9fda924a5e76ffab9e8c068675) (as well as earlier kernels), CF-953AX in AP mode and the [...]

@leezu 

I'm looking to move this bug report up into message 1 with the other bugs. You mentioned the distro and kernel but did not mention:

- band/channel ?
- hostapd ? and hostapd.conf
- WPA3 ?
- wpa_supplicant version ?
- 32 bit ?
- hostapd.log showing anything ?

bjlockie, 2022-10-08:
I also want to know if is using an extension cable.
And is it using a powered USB hub.
I had to use a powered USB hub for the keyboard/mouse of my hdmi switch.

> On RasPi4B with Raspbian and kernel 6.0.0-rc7-v8+[https://github.com/raspberrypi/linux/commit/3fb5ca89c844eb9fda924a5e76ffab9e8c068675] (as well as earlier kernels), CF-953AX in AP mode and the [...]
* > band/channel ?
* > hostapd ? and hostapd.conf
* > WPA3 ?
* > wpa_supplicant version ?
* > 32 bit ?
* > hostapd.log showing anything ?

leezu, 2022-10-12:
@morrownr @bjlockie 

> I'm looking to move this bug report up into message 1 with the other bugs. You mentioned the distro and kernel but did not mention:

    band/channel: 36
    hostapd v2.10 and [hostapd.conf](https://github.com/morrownr/USB-WiFi/files/9765598/hostapd.conf.txt)
    WPA3: wpa_key_mgmt=WPA-PSK
    wpa_supplicant version: 2.9
    32 bit: running with `arm_64bit=1` RPi config but `getconf LONG_BIT` returns 32.
    hostapd.log showing anything: Please see the two journals regarding two separate occurrences attached: 
[journal-no-backtrace.txt](https://github.com/morrownr/USB-WiFi/files/9765771/journal-no-backtrace.txt)
[journal-with-backtrace.txt](https://github.com/morrownr/USB-WiFi/files/9765772/journal-with-backtrace.txt)

> I also want to know if is using an extension cable.
> And is it using a powered USB hub.
> I had to use a powered USB hub for the keyboard/mouse of my hdmi switch.

It uses this extension lead https://smile.amazon.com/gp/product/B082HQXRZ1/ as without extension lead, the CF-953AX covers up the LAN port.

gifter77, 2023-05-08:
Regarding bug 6., if I build a vanilla 6.4.rc linux kernel for my raspberry pi 4, I cannot reproduce the issue anymore.

morrownr, 2023-05-08:
Hi @gifter77 

Copy all. Will update.

gifter77, 2023-05-08:
Actually false alert @morrownr I reproduced even with vanilla 6.4-rc1 kernel :/

gifter77, 2023-05-08:
@morrownr thanks. I've added this patch to the latest rpi kernel (6.1) and unfortunately I can still reproduce the issue pretty easily with the same trace as @leezu below:
```
[  107.537594] usbcore: deregistering interface driver mt7921u
[  107.559823] wlan0: deauthenticating from <redacted> by local choice (Reason: 3=DEAUTH_LEAVING)
[  111.583809] mt7921u 2-1:1.0: Message 00020003 (seq 5) timeout
[  114.931814] mt7921u 2-1:1.0: timed out waiting for pending tx
[  115.060533] ------------[ cut here ]------------
[  115.060556] WARNING: CPU: 0 PID: 47 at kernel/kthread.c:659 kthread_park+0xb0/0xc8
[  115.060588] Modules linked in: xt_DSCP xt_tcpudp nft_compat nf_tables nfnetlink ctr aes_arm64 aes_generic libaes ccm mt7921u(-) mt7921_common mt76_connac_lib mt76_usb mt76 mac80211 libarc4 8021q garp stp llc imx708 cfg80211 dw9807_vcm vc4 rfkill cdc_acm spidev snd_soc_hdmi_codec drm_display_helper cec drm_dma_helper drm_kms_helper v3d gpu_sched snd_soc_core bcm2835_unicam drm_shmem_helper i2c_mux_pinctrl raspberrypi_hwmon i2c_mux v4l2_dv_timings snd_bcm2835(C) snd_compress v4l2_fwnode snd_pcm_dmaengine snd_pcm bcm2835_codec(C) rpivid_hevc(C) v4l2_async bcm2835_isp(C) i2c_brcmstb v4l2_mem2mem snd_timer videobuf2_dma_contig snd i2c_bcm2835 spi_bcm2835 syscopyarea sysfillrect sysimgblt fb_sys_fops uio_pdrv_genirq uio nvmem_rmem bcm2835_v4l2(C) bcm2835_mmal_vchiq(C) vc_sm_cma(C) videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common videodev drm mc fuse drm_panel_orientation_quirks backlight ip_tables x_tables ipv6
[  115.060892] CPU: 0 PID: 47 Comm: kworker/u8:2 Tainted: G         C         6.1.27-v8+ #1
[  115.060904] Hardware name: Raspberry Pi 4 Model B Rev 1.2 (DT)
[  115.060913] Workqueue: mt76 mt7921_mac_reset_work [mt7921_common]
[  115.060957] pstate: 00000005 (nzcv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[  115.060967] pc : kthread_park+0xb0/0xc8
[  115.060977] lr : mt76u_stop_tx+0x26c/0x330 [mt76_usb]
[  115.061003] sp : ffffffc0085bbc80
[  115.061008] x29: ffffffc0085bbc80 x28: 0000000000000000 x27: 0000000000000000
[  115.061026] x26: 0000000000000000 x25: ffffff8046d3d180 x24: 0000000000000100
[  115.061042] x23: ffffff804a762068 x22: ffffff804a762048 x21: ffffffde760ad838
[  115.061057] x20: ffffff804a550d80 x19: ffffff804a762020 x18: 0000000000000000
[  115.061072] x17: 0000000000000000 x16: ffffffde752ad6b0 x15: 0000007f7b7fcdd0
[  115.061087] x14: 0000000000000000 x13: 0000000000000000 x12: 0000000000000000
[  115.061102] x11: 0000000000000000 x10: 0000000000001a60 x9 : ffffffde3da9b1ac
[  115.061117] x8 : ffffff8040b33980 x7 : ffffffc0085bc000 x6 : ffffffc0085b8000
[  115.061131] x5 : ffffffc0085bc000 x4 : 0000000000000000 x3 : ffffff8046d3d414
[  115.061146] x2 : 0000000000000000 x1 : 0000000000000004 x0 : ffffff804349dc40
[  115.061161] Call trace:
[  115.061165]  kthread_park+0xb0/0xc8
[  115.061175]  mt76u_stop_tx+0x26c/0x330 [mt76_usb]
[  115.061197]  mt7921u_mac_reset+0x7c/0x278 [mt7921u]
[  115.061217]  mt7921_mac_reset_work+0xa0/0x198 [mt7921_common]
[  115.061245]  process_one_work+0x208/0x480
[  115.061263]  worker_thread+0x50/0x428
[  115.061273]  kthread+0xfc/0x110
[  115.061284]  ret_from_fork+0x10/0x20
[  115.061296] ---[ end trace 0000000000000000 ]---
```

whitslack, 2023-05-08:
> You very well may have run onto the same problem as @whitslack .

No. Very different stack trace. Nothing to do with SKB headroom underflow, which is what my patch fixes.

morrownr, 2023-05-08:
@whitslack

> No. Very different stack trace. Nothing to do with SKB headroom underflow, which is what my patch fixes.

My bad. I was in a hurry earlier today and did not take the time to investigate.

@gifter77 

I see you are using a RasPi4B v1.2. I have one of those.

Are you using the 64 bit 2023-05-03 release?

Is there anything in any USB port besides the adapter?

Are you using a USB3 port or USB2?

My Pi4B is working as an AP right now. 
Adapters: CF-951AX in USB3 port running WiFi 6 on channel 149, 5370 based WiFi 4 adapter in USB2 port on channel 6.
The only 2 things in USB ports are the 2 adapters.
The only upgrade to the software was a new compile of hostapd to v2.11-devel (2.9 won't do WiFi 6)
I use the AP guide on the Main Menu.

I can check the log to see if I am getting anything but it seems stable.

gifter77, 2023-05-09:
@morrownr yes I'm using the latest 64-bit release (kernel 6.1.27-v8+). The Pi is in my 3D printer so other ports have some serial devices connected to them and I'm not running an AP. My adapter is a CF-953AX. It doesn't matter if connected to USB2 or USB3 or even via a powered USB hub, the issue can be reproduced by hammering the connection with iperf for example. I've also tried to manually overwrite the firmware to the latest, doesn't change anything. I'm running on channel 52 at the moment but I guess this would change sometimes.
One slightly dodgy thing I can see if that the country seems to be set to Germany whereas I am in the UK and localisation settings are all set to GB:
```
global
country DE: DFS-ETSI
	(2400 - 2483 @ 40), (N/A, 20), (N/A)
	(5150 - 5250 @ 80), (N/A, 23), (N/A), NO-OUTDOOR, AUTO-BW
	(5250 - 5350 @ 80), (N/A, 20), (0 ms), NO-OUTDOOR, DFS, AUTO-BW
	(5470 - 5725 @ 160), (N/A, 26), (0 ms), DFS
	(5725 - 5875 @ 80), (N/A, 13), (N/A)
	(5945 - 6425 @ 160), (N/A, 23), (N/A), NO-OUTDOOR, AUTO-BW
	(57000 - 66000 @ 2160), (N/A, 40), (N/A)
```
I think this is a known issue of my Asus AX88U router broadcasting DE instead of UK for some reason but it shouldn't really matter.

morrownr, 2023-05-09:
> I'm using the latest 64-bit release (kernel 6.1.27-v8+).

The reason for the question has to do with firmware, not so much the driver. The new 20230503 release has very current firmware.

How do I check the firmware version in my system?

$ ethtool -i <interface name> 

Note: You may need to install the ethtool and iw packages depending on your distro.

> The Pi is in my 3D printer so other ports have some serial devices connected to them and I'm not running an AP.

I have seen many problems with the USB subsystem on Pi's over the last several years and from looking at your log, I immediately thought about the USB subsystem. If it were me, I would remove everything from the usb ports and start adding one thing at a time until the problem shows. I'll assume you need to add the CF-953AX first as you are probably using it to run headless. If that is case, pound the heck out of it with iperf3 and see what you get in both a USB2 and USB3 port. If you don't see the problem, add another usb device and test, and continue until you have a full report.

You mentioned a powered usb hub... make sure you add that back into the mix last. Powered hubs are generally a problem on Pi's and have been for many years.  This goes way back before the Pi4B. There is also the issue the Pi usb subsystem not having much power to offer devices. The usb subsystem is limited to 1200 mA.. By spec, it should be 2800 mA. Is it possible your devices are trying to pull more than 1200 mA when you throw a load on?

I've maintained 5 Realtek driver here for several years and also help users with Mediatek drivers. I would classify the RasPi usb subsystem as bad, The USB hub chipset is problematic. Well, I could go on but in this case, we really need to eliminate everything and add on device at a time until we see the problem. At that point, we can test differently or whatever is required to narrow down the issue. Your log certainly points to a usb issue but chasing around in code may not help as we need to do our best to determine the cause.

I am not seeing this. I just tested my Pi4B with iperf3. The only usb decive plugging in is the CF-951AX. I earned a long time ao that if you are going to use a usb wifi adapter, you have to be very careful adding other things to the usb subsystem. Mediatek adapters generally are more reliable in this situation as they use less power than Realtek adapters.

Let me know.

gifter77, 2023-05-09:
@morrownr 
```
ethtool -i wlan0
driver: mt7921u
version: 6.1.21-v8+
firmware-version: ____010000-20230331110939
expansion-rom-version: 
bus-info: 2-1:1.0
supports-statistics: yes
supports-test: no
supports-eeprom-access: no
supports-register-dump: no
supports-priv-flags: no
```

And btw I had to install this firmware version manually, my other printer with 64-bit Raspbian completely up to data has firmware ____010000-20220908211021.

I will try to reproduce with only the Wifi adapter attached.

gifter77, 2023-05-09:
Reproduced with nothing else attached to the Pi apart from the CF-953AX on USB3 port.

gifter77, 2023-05-09:
I'll do USB2 port by itself now. Btw it seems to be easier to reproduce when having the Pi as the iperf server, so traffic flowing to the Pi rather than from the Pi.

gifter77, 2023-05-09:
And reproduced as well on USB2 port.

morrownr, 2023-05-09:
> And btw I had to install this firmware version manually, my other printer with 64-bit Raspbian completely up to date has firmware ____010000-20220908211021.

That is why I asked if it was a fresh clean 2023-05-03 installation. Remember that firmware is not part of the kernel, it is part of the distros and there are hundreds, if not thousands, of firmware packages that support various drivers. Most distros, for the lack of bodies, don't do a particularly good job of keeping firmware current. Many update when a user posts an issue requesting the firmware be updated.

gifter77, 2023-05-09:
> That is why I asked if it was a fresh clean 2023-05-03 installation. Remember that firmware is not part of the kernel, it is part of the distros and there are hundreds, if not thousands, of firmware [...]

Fair enough. My latest tests are with the latest firmware anyway.

gifter77, 2023-05-09:
> Was the test with the 2023-05-03 release or an old one with upgraded firmware. I'm testing on the 2023-05-03 64 bit release.

Old one with upgraded firmware.

> Could it be something else in the distro that has since been upgraded that is causing the problem?

The distro is actually not clean Raspbian but MansailOs (3D printer distro) which is based on Raspbian. So yes there is definitely a difference in our distros.

> What firmware are your running in the Pi?
```
sudo rpi-eeprom-update
*** UPDATE AVAILABLE ***
BOOTLOADER: update available
   CURRENT: Thu  3 Sep 12:11:43 UTC 2020 (1599135103)
    LATEST: Wed 11 Jan 17:40:52 UTC 2023 (1673458852)
   RELEASE: default (/lib/firmware/raspberrypi/bootloader/default)
            Use raspi-config to change the release.

  VL805_FW: Dedicated VL805 EEPROM
     VL805: up to date
   CURRENT: 000138c0
    LATEST: 000138c0
```

> Do you have a solid power supply?
Yes.

morrownr, 2023-05-09:
> BOOTLOADER: update available
   CURRENT: Thu  3 Sep 12:11:43 UTC 2020 (1599135103)
    LATEST: Wed 11 Jan 17:40:52 UTC 2023 (1673458852)

Okay, do you think it might be a good idea to get your Pi firmware/eeprom some where close to up to date? A fix could be in there.

Mine:

BOOTLOADER: update available
   CURRENT: Tue  2 Aug 15:55:05 UTC 2022 (1659455705)
    LATEST: Wed 11 Jan 17:40:52 UTC 2023 (1673458852)

Okay, I need to update mine as well but at least it was better than yours.

morrownr, 2023-05-09:
I am updating mine as well. We can check to make sure both Pi's are on the same bootloader after finished.

We gotta eliminate as many little differences as we can.

gifter77, 2023-05-09:
```
BOOTLOADER: up to date
   CURRENT: Wed 11 Jan 17:40:52 UTC 2023 (1673458852)
    LATEST: Wed 11 Jan 17:40:52 UTC 2023 (1673458852)
   RELEASE: default (/lib/firmware/raspberrypi/bootloader/default)
            Use raspi-config to change the release.

  VL805_FW: Dedicated VL805 EEPROM
     VL805: up to date
   CURRENT: 000138c0
    LATEST: 000138c0
```

gifter77, 2023-05-09:
And.... reproduced again.

morrownr, 2023-05-09:
> Can I use this device with multiple SSIDs?

At this time, I can't think of a way. 

It is not clear why the limit of 1 AP in this driver when previous drivers certainly support more. Work is certainly ongoing with this driver. We need to make some noise.

morrownr, 2023-05-09:
> And.... reproduced again.

Is it time to burn a new sd card and try 2023-05-03?

gifter77, 2023-05-09:
So first observation, the firmware on clean 2023-05-03 installation is definitely not the latest:
```
[  654.717694] mt7921u 1-1.3:1.0: HW/SW Version: 0x8a108a10, Build Time: 20220908210919a

[  655.001882] mt7921u 1-1.3:1.0: WM Firmware Version: ____010000, Build Time: 20220908211021
```

morrownr, 2023-05-09:
My mind has turned to mush as I get older. I guess I jusr automatically upgrade it and forget I did so. Time to file a bug with the RasPi folks.

gifter77, 2023-05-09:
I cannot seem to reproduce on this fresh install though, even with this version of the firmware. This is very odd.

EasyNetDev, 2023-05-09:
OK. My setup is a RPI 3 with kernel 6.1.21+.
I've addeed the latest version of firmware for MT7921u I could found:

```
[280849.036366] usbcore: registered new interface driver mt7921u
[280849.046172] mt7921u 1-1.4:1.3: HW/SW Version: 0x8a108a10, Build Time: 20220908210919a
[280849.371419] mt7921u 1-1.4:1.3: WM Firmware Version: ____010000, Build Time: 20220908211021
[292744.962519] usbcore: deregistering interface driver mt7921u
[292822.280258] usbcore: registered new interface driver mt7921u
[292822.283238] mt7921u 1-1.4:1.3: HW/SW Version: 0x8a108a10, Build Time: 20230331110902a
[292822.299917] mt7921u 1-1.4:1.3: WM Firmware Version: ____010000, Build Time: 20230331110939
```
And here are details of the driver:
```
filename:       /lib/modules/6.1.21-v8+/kernel/drivers/net/wireless/mediatek/mt76/mt7921/mt7921u.ko.xz
license:        Dual BSD/GPL
author:         Lorenzo Bianconi <lorenzo@kernel.org>
firmware:       mediatek/WIFI_MT7961_patch_mcu_1_2_hdr.bin
firmware:       mediatek/WIFI_RAM_CODE_MT7961_1.bin
srcversion:     774ADC863988DEEFA8EF149
alias:          usb:v0E8Dp7961d*dc*dsc*dp*icFFiscFFipFFin*
depends:        mt76-connac-lib,mt76-usb,mt7921-common,mt76
intree:         Y
name:           mt7921u
vermagic:       6.1.21-v8+ SMP preempt mod_unload modversions aarch64
```

gifter77, 2023-05-09:
Oh I got a new one while trying to reproduce the other one:
```
[ 1178.838761] mt7921u 1-1.3:1.0: vendor request req:63 off:8154 failed:-71
[ 1178.838816] mt7921u 1-1.3:1.0: tx urb failed: -71
[ 1178.839302] mt7921u 1-1.3:1.0: tx urb failed: -71
[ 1178.839562] mt7921u 1-1.3:1.0: tx urb failed: -71
[ 1178.839826] mt7921u 1-1.3:1.0: tx urb failed: -71
[ 1178.840088] mt7921u 1-1.3:1.0: tx urb failed: -71
[ 1178.840350] mt7921u 1-1.3:1.0: tx urb failed: -71
[ 1178.840614] mt7921u 1-1.3:1.0: tx urb failed: -71
[ 1178.840876] mt7921u 1-1.3:1.0: tx urb failed: -71
[ 1178.841138] mt7921u 1-1.3:1.0: tx urb failed: -71
[ 1178.841428] mt7921u 1-1.3:1.0: tx urb failed: -71
[ 1178.841689] mt7921u 1-1.3:1.0: tx urb failed: -71
[ 1178.841952] mt7921u 1-1.3:1.0: tx urb failed: -71
[ 1178.842214] mt7921u 1-1.3:1.0: tx urb failed: -71
[ 1178.842472] mt7921u 1-1.3:1.0: tx urb failed: -71
[ 1178.842762] mt7921u 1-1.3:1.0: tx urb failed: -71
[ 1178.843025] mt7921u 1-1.3:1.0: tx urb failed: -71
[ 1178.843312] mt7921u 1-1.3:1.0: tx urb failed: -71
[ 1178.843570] mt7921u 1-1.3:1.0: tx urb failed: -71
[ 1178.843833] mt7921u 1-1.3:1.0: tx urb failed: -71
[ 1178.844094] mt7921u 1-1.3:1.0: tx urb failed: -71
[ 1178.844357] mt7921u 1-1.3:1.0: tx urb failed: -71
[ 1178.844621] mt7921u 1-1.3:1.0: tx urb failed: -71
[ 1178.844881] mt7921u 1-1.3:1.0: tx urb failed: -71
[ 1178.845157] mt7921u 1-1.3:1.0: tx urb failed: -71
[ 1178.845432] mt7921u 1-1.3:1.0: tx urb failed: -71
[ 1178.845694] mt7921u 1-1.3:1.0: tx urb failed: -71
[ 1178.845955] mt7921u 1-1.3:1.0: tx urb failed: -71
[ 1178.846224] mt7921u 1-1.3:1.0: tx urb failed: -71
[ 1178.846478] mt7921u 1-1.3:1.0: tx urb failed: -71
[ 1178.846788] mt7921u 1-1.3:1.0: tx urb failed: -71
[ 1178.847113] mt7921u 1-1.3:1.0: tx urb failed: -71
[ 1178.847466] mt7921u 1-1.3:1.0: tx urb failed: -71
[ 1178.847813] usb 1-1.3: USB disconnect, device number 5
[ 1178.847834] mt7921u 1-1.3:1.0: tx urb failed: -71
[ 1178.848111] mt7921u 1-1.3:1.0: tx urb failed: -71
[ 1178.848369] mt7921u 1-1.3:1.0: tx urb failed: -71
[ 1178.848960] mt7921u 1-1.3:1.0: tx urb failed: -71
[ 1178.867031] wlan0: deauthenticating from 04:d4:c4:51:88:8c by local choice (Reason: 3=DEAUTH_LEAVING)
[ 1179.382976] wlan0: failed to remove key (0, 04:d4:c4:51:88:8c) from hardware (-19)
[ 1179.388604] ------------[ cut here ]------------
[ 1179.388617] WARNING: CPU: 0 PID: 1953 at net/wireless/util.c:1388 cfg80211_calculate_bitrate_he+0x260/0x2b8 [cfg80211]
[ 1179.388761] Modules linked in: xt_DSCP xt_tcpudp nft_compat nf_tables nfnetlink ctr aes_arm64 aes_generic libaes ccm 8021q garp stp llc mt7921u mt7921_common mt76_connac_lib mt76_usb mt76 mac80211 imx708 dw9807_vcm libarc4 spidev cfg80211 rfkill vc4 snd_soc_hdmi_codec drm_display_helper cec drm_dma_helper drm_kms_helper bcm2835_codec(C) rpivid_hevc(C) bcm2835_unicam bcm2835_v4l2(C) snd_soc_core v4l2_dv_timings bcm2835_isp(C) v4l2_mem2mem bcm2835_mmal_vchiq(C) v4l2_fwnode snd_compress v4l2_async videobuf2_dma_contig videobuf2_vmalloc snd_pcm_dmaengine snd_bcm2835(C) videobuf2_memops videobuf2_v4l2 v3d snd_pcm gpu_sched videobuf2_common raspberrypi_hwmon i2c_mux_pinctrl drm_shmem_helper i2c_mux snd_timer videodev i2c_brcmstb snd spi_bcm2835 i2c_bcm2835 mc vc_sm_cma(C) syscopyarea sysfillrect sysimgblt fb_sys_fops nvmem_rmem uio_pdrv_genirq uio i2c_dev drm fuse drm_panel_orientation_quirks backlight ip_tables x_tables ipv6
[ 1179.388937] CPU: 0 PID: 1953 Comm: kworker/0:1 Tainted: G         C         6.1.21-v8+ #1642
[ 1179.388944] Hardware name: Raspberry Pi 4 Model B Rev 1.2 (DT)
[ 1179.388949] Workqueue: usb_hub_wq hub_event
[ 1179.388965] pstate: 20000005 (nzCv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[ 1179.388971] pc : cfg80211_calculate_bitrate_he+0x260/0x2b8 [cfg80211]
[ 1179.389047] lr : 0x1856ac15
[ 1179.389052] sp : ffffffc00953afc0
[ 1179.389055] x29: ffffffc00953afc0 x28: ffffff804f00db72 x27: ffffff8049600000
[ 1179.389065] x26: ffffff80496003a0 x25: ffffff8049697030 x24: ffffff8049697000
[ 1179.389074] x23: ffffff80449f6900 x22: 00000000000000d0 x21: ffffff80496970d0
[ 1179.389083] x20: ffffff804f68ea38 x19: ffffffefaaaad858 x18: 0cea122a0dac8927
[ 1179.389092] x17: 000000000b9f76c0 x16: 0675091506d65a47 x15: 0000000005cfbb60
[ 1179.389101] x14: 0240717102625a00 x13: 000000000206cc80 x12: 010f4471011f2ba0
[ 1179.389110] x11: 0000000000f42400 x10: 0000000000000004 x9 : 000063ff00008556
[ 1179.389119] x8 : 0000c80100018fff x7 : 000027ff00002c71 x6 : 00003201000042ab
[ 1179.389127] x5 : 0000180000001aac x4 : 0000000000000003 x3 : 0087a238008f9a27
[ 1179.389136] x2 : 00000000007a1200 x1 : 000014000000163a x0 : ffffff804f68ea38
[ 1179.389145] Call trace:
[ 1179.389148]  cfg80211_calculate_bitrate_he+0x260/0x2b8 [cfg80211]
[ 1179.389223]  cfg80211_calculate_bitrate+0x224/0x5d0 [cfg80211]
[ 1179.389297]  nl80211_put_sta_rate+0x6c/0x3d8 [cfg80211]
[ 1179.389370]  nl80211_send_station.isra.99+0x6e4/0xc00 [cfg80211]
[ 1179.389444]  cfg80211_del_sta_sinfo+0xbc/0x1c8 [cfg80211]
[ 1179.389517]  __sta_info_destroy_part2+0xd0/0x158 [mac80211]
[ 1179.389656]  __sta_info_flush+0x124/0x190 [mac80211]
[... 10 lines cut ...]
[ 1179.390433]  call_netdevice_notifiers_info+0x60/0xb0
[ 1179.390440]  __dev_close_many+0x68/0x158
[ 1179.390445]  dev_close_many+0x94/0x130
[ 1179.390450]  dev_close.part.117+0x50/0x80
[ 1179.390455]  dev_close+0x24/0x30
[ 1179.390461]  cfg80211_shutdown_all_interfaces+0x4c/0x108 [cfg80211]
[ 1179.390534]  ieee80211_remove_interfaces+0x5c/0x238 [mac80211]
[... cut ...]
[ 1179.390756]  device_release_driver_internal+0xe8/0x190
[ 1179.390762]  device_release_driver+0x20/0x30
[ 1179.390767]  bus_remove_device+0x130/0x140
[ 1179.390772]  device_del+0x184/0x3b8
[ 1179.390778]  usb_disable_device+0x94/0x160
[ 1179.390783]  usb_disconnect+0xc4/0x2c0
[ 1179.390791]  hub_event+0x6e0/0x1300
[ 1179.390797]  process_one_work+0x208/0x480
[ 1179.390805]  worker_thread+0x50/0x428
[ 1179.390811]  kthread+0xfc/0x110
[ 1179.390817]  ret_from_fork+0x10/0x20
[ 1179.390824] ---[ end trace 0000000000000000 ]---
```

whitslack, 2023-05-09:
> `usb 1-1.3: USB disconnect, device number 5`

That looks more like a USB power issue or a loose connector, broken cable, etc. It's not surprising that the driver will come to a very noisy halt when the device is effectively unplugged while in use.

gifter77, 2023-05-10:
> That looks more like a USB power issue or a loose connector, broken cable, etc. It's not surprising that the driver will come to a very noisy halt when the device is effectively unplugged while in use.

Ok then that is a side effect of me unplugging and replugging the adapter when I lost the connection.

gifter77, 2023-05-10:
@morrownr so it seems that the way to reproduce the issue on a fresh install is to get the Pi to download stuff using iperf and also starts some webrtc service on the pi at the same time to stream the Pi camera output. Then when viewing this stream from another computer, it triggers the wifi driver crash. This is extremely strange. Maybe the reproduction scenario could be simplified to downloading stuff on the Pi and sending out UDP multicast packets at the same time.

morrownr, 2023-05-10:
> also starts some webrtc service on the pi at the same time to stream the Pi camera output. 

> Then when viewing this stream from another computer, it triggers the wifi driver crash. This is extremely strange.

Agree. That is why I was being so picky about getting everything up to the latest version on a fresh configuration. We don't stand a change of providing a good bug report without having it narrowed down to something simply and repeatable.

You had not previously mentioned camera output. I am following along trying to duplicate the issue. What camera and app are you using? I have a webcam and can compile a streamer app but I need to see if I can get as close as possible to your setup.

Can you also make me a list of everything that is plugged into a usb port? And keep it updated as we go along. I have to see what you are doing to be able to try to duplicate the problem.

Devs have their own techniques for many things. When it comes to Pi's and usb, I am reluctant to just start digging in code as that has proved to be futile unless first I have a very simple, easy to reproduce scenario.

gifter77, 2023-05-10:
> You had not previously mentioned camera output. I am following along trying to duplicate the issue. What camera and app are you using? I have a webcam and can compile a streamer app but I need to see if I can get as close as possible to your setup.

I'm using the pi camera v3 so it is not plugged on a USB port but on the CSI port of the Pi. The WebRTC streamer is this one: https://github.com/ayufan/camera-streamer

> Can you also make me a list of everything that is plugged into a usb port? And keep it updated as we go along. I have to see what you are doing to be able to try to duplicate the problem.

It can be reproduced with only the adapter plugged on a USB2 or USB3 port.

I doubt this is specific to using a camera and streaming some WebRTC packets though. Maybe trying to send any UDP multicast packet from the Pi would work. Remember the original reporter of this bug in this thread (@leezu) could trigger it with some heavy load from a client connected to his AP.

morrownr, 2023-05-10:
@gifter77 

Status report:

```
[  5]   0.00-1.00   sec  65.3 MBytes   548 Mbits/sec    0   1.86 MBytes       
[  5]   1.00-2.00   sec  72.5 MBytes   608 Mbits/sec    0   1.86 MBytes       
[  5]   2.00-3.00   sec  75.0 MBytes   629 Mbits/sec    0   1.86 MBytes       
[  5]   3.00-4.00   sec  72.5 MBytes   608 Mbits/sec    0   1.86 MBytes       
[  5]   4.00-5.00   sec  73.8 MBytes   618 Mbits/sec    0   1.86 MBytes       
[  5]   5.00-6.00   sec  73.8 MBytes   619 Mbits/sec    0   1.86 MBytes       
[  5]   6.00-7.00   sec  72.5 MBytes   608 Mbits/sec    0   1.86 MBytes       
[  5]   7.00-8.00   sec  75.0 MBytes   629 Mbits/sec    0   1.86 MBytes       
[  5]   8.00-9.00   sec  73.8 MBytes   619 Mbits/sec    0   1.86 MBytes       
[  5]   9.00-10.00  sec  71.2 MBytes   598 Mbits/sec    0   1.86 MBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   725 MBytes   608 Mbits/sec    0             sender
[  5]   0.00-10.01  sec   724 MBytes   607 Mbits/sec                  receiver

```

The above uses a RasPi4B v1.2 with a lone CF-951AX in managed mode connected to my main wifi router.

The wifi router runs OpenWRT 22.03.5. Channel 100 DFS. No other AP's on the channel.

RasPiOS is 64 bit 2023-05-03. I have not upgrade the old firmware for the mt7921u driver but will. This takes a lot of time and I have other things going on.

The only thing in a port is the wifi adapter in a USB3 port. I have confirmed that it is running in USB3 mode but that was a waste of time as USB2 cannot do the speeds we see above.

The only ugliness I am seeing in `dmesg` is:

[   10.943065] Bluetooth: hci0: Opcode 0x c03 failed: -110
[   12.959031] Bluetooth: hci0: Opcode 0x c03 failed: -110

That is caused by RasPiOS detecting the bt support in the cf-951ax, even though Comfast turned the support off. I can get rid of this by deleting the bt firmware. I'll do that when I upgrade the firmware for the wifi.

Do you want me to do something like?

iperf3 -c 192.168.1.1 -t 600

...to see if heating things up will show something?

Of note: The speeds you see above would be higher, probably around 650 Mbps, but I am running headless so that takes some of the bandwidth.

What a thread! And we don't seem to be any closer to narrowing down the issue.

@morrownr

gifter77, 2023-05-10:
I think what you could try is getting the Pi much further from your router before running the test,. Mine is pretty far and the signal level is like -70dB. So I'm not getting more than 100Mb/s.

morrownr, 2023-05-10:
> I think what you could try is getting the Pi much further from your router

I moved the Pi. It was about 30 feet from the router (meters) but was line of sight. Now the distance is about 50 feet (16 meters or so) and has 3 walls in the way.

wavemon went from showing -38 to -52.

I ran...

iperf3 -c 192.168.1.1 -t 300

Average 505 Mbps. Speed was consistent. Nothing in the log other than the normal crap related to Broadcom's crap Linux support.

I can jack up the overclocking and run iperf3 longer?

morrownr, 2023-05-10:
So, time for a little abuse to see if I can trigger something.

I updated the wifi firmware and deleted the bt firmware.

```
over_voltage=6
arm_freq=2000
gpu_freq=750
```

Result:

```
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-300.00 sec  17.8 GBytes   509 Mbits/sec    2             sender
[  5]   0.00-300.01 sec  17.8 GBytes   508 Mbits/sec                  receiver
```

Still at about 16 meters and 3 walls. 

More abuse...

The Pi just dropped out on this attempt. Since I was running headless, now I need to find a solution to seeing what is in the log.

gifter77, 2023-05-11:
> The Pi just dropped out on this attempt. Since I was running headless, now I need to find a solution to seeing what is in the log.
You can plug an ethernet cable and check that way I guess.

gifter77, 2023-05-11:
Another way to get the connection back up is to unplug and replug the adapter. I'm guessing you have an udev rule for it.

morrownr, 2023-05-11:
> Another way to get the connection back up is to unplug and replug the adapter.

Ah, so now I need to trigger the problem again and see if this works. I do have ethernet cables but I'd prefer this easier solutions work. 

My router is Wave 2 capable and it appears the mt7921au chipset is Wave 2 capable as well. My router is also capable of DFS channels and it is on channel 100 (5 GHz) right now. Totally clear air. No other AP's on the channel at all. The router also does 160 MHz channel width and I have it set just because I can but I have no client devices that can do 160 MHz channel width.

I have a speed test up:

https://github.com/morrownr/USB-WiFi/blob/main/home/Speed_Comparison_Test.md

Now, let me see if I can trigger the drop out again and get a log.

morrownr, 2023-05-11:
Having trouble triggering the problem again:

```
$ iperf3 -c 192.168.1.1 -t 600
Connecting to host 192.168.1.1, port 5201
[  5] local 192.168.1.119 port 32992 connected to 192.168.1.1 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  52.6 MBytes   441 Mbits/sec    0   1.90 MBytes       
[  5]   1.00-2.00   sec  56.2 MBytes   472 Mbits/sec    0   1.90 MBytes       
[  5]   2.00-3.00   sec  56.2 MBytes   472 Mbits/sec    0   1.90 MBytes       
[  5]   3.00-4.00   sec  56.2 MBytes   472 Mbits/sec    0   1.90 MBytes       
[  5]   4.00-5.00   sec  56.2 MBytes   472 Mbits/sec    0   1.90 MBytes       
[  5]   5.00-6.00   sec  55.0 MBytes   461 Mbits/sec    0   1.90 MBytes       
[  5]   6.00-7.00   sec  56.2 MBytes   472 Mbits/sec    0   1.90 MBytes       
[  5]   7.00-8.00   sec  56.2 MBytes   472 Mbits/sec    0   1.90 MBytes       
[  5]   8.00-9.00   sec  56.2 MBytes   472 Mbits/sec    0   1.90 MBytes       
[  5]   9.00-10.00  sec  56.2 MBytes   472 Mbits/sec    0   1.90 MBytes       
[  5]  10.00-11.00  sec  55.0 MBytes   461 Mbits/sec    0   1.90 MBytes       
[  5]  11.00-12.00  sec  56.2 MBytes   472 Mbits/sec    0   1.90 MBytes       
[  5]  12.00-13.00  sec  56.2 MBytes   472 Mbits/sec    0   1.90 MBytes       
[  5]  13.00-14.00  sec  56.2 MBytes   472 Mbits/sec    0   1.90 MBytes       
[  5]  14.00-15.00  sec  57.5 MBytes   482 Mbits/sec    0   1.90 MBytes       
[  5]  15.00-16.00  sec  55.0 MBytes   462 Mbits/sec    0   1.90 MBytes       
[  5]  16.00-17.00  sec  56.2 MBytes   472 Mbits/sec    0   1.90 MBytes       
[  5]  17.00-18.00  sec  56.2 MBytes   471 Mbits/sec    0   1.90 MBytes       
[  5]  18.00-19.00  sec  32.5 MBytes   273 Mbits/sec    0   1.90 MBytes       
[  5]  19.00-20.00  sec  30.0 MBytes   252 Mbits/sec    0   1.90 MBytes       
[  5]  20.00-21.00  sec  31.2 MBytes   262 Mbits/sec    0   1.90 MBytes       
[  5]  21.00-22.00  sec  26.2 MBytes   220 Mbits/sec    0   1.90 MBytes       
[  5]  22.00-23.00  sec  26.2 MBytes   220 Mbits/sec    0   1.90 MBytes       
[  5]  23.00-24.00  sec  26.2 MBytes   220 Mbits/sec    0   1.90 MBytes       
[  5]  24.00-25.00  sec  25.0 MBytes   210 Mbits/sec    0   1.90 MBytes       
[  5]  25.00-26.00  sec  30.0 MBytes   252 Mbits/sec    0   1.90 MBytes       
[  5]  26.00-27.00  sec  36.2 MBytes   304 Mbits/sec    0   1.90 MBytes       
[  5]  27.00-28.00  sec  56.2 MBytes   472 Mbits/sec    0   1.90 MBytes       
[  5]  28.00-29.00  sec  61.2 MBytes   514 Mbits/sec    0   1.90 MBytes       
[  5]  29.00-30.00  sec  65.0 MBytes   545 Mbits/sec    0   1.90 MBytes       
[  5]  30.00-31.00  sec  66.2 MBytes   556 Mbits/sec    0   1.90 MBytes       
[  5]  31.00-32.00  sec  63.8 MBytes   534 Mbits/sec    0   1.90 MBytes       
[  5]  32.00-33.00  sec  63.8 MBytes   535 Mbits/sec    0   1.90 MBytes       
[  5]  33.00-34.00  sec  63.8 MBytes   535 Mbits/sec    0   1.90 MBytes       
[  5]  34.00-35.00  sec  65.0 MBytes   545 Mbits/sec    0   1.90 MBytes       
[  5]  35.00-36.00  sec  63.8 MBytes   535 Mbits/sec    0   1.90 MBytes       
[  5]  36.00-37.00  sec  63.8 MBytes   535 Mbits/sec    0   1.90 MBytes       
[  5]  37.00-38.00  sec  66.2 MBytes   556 Mbits/sec    0   1.90 MBytes       
[  5]  38.00-39.00  sec  63.8 MBytes   535 Mbits/sec    0   1.90 MBytes       
[  5]  39.00-40.00  sec  62.5 MBytes   524 Mbits/sec    0   1.90 MBytes       
[  5]  40.00-41.00  sec  62.5 MBytes   525 Mbits/sec    0   1.90 MBytes       
[  5]  41.00-42.00  sec  63.8 MBytes   535 Mbits/sec    0   1.90 MBytes       
[  5]  42.00-43.00  sec  65.0 MBytes   545 Mbits/sec    0   1.90 MBytes       
[  5]  43.00-44.00  sec  63.8 MBytes   535 Mbits/sec    0   1.90 MBytes       
[  5]  44.00-45.00  sec  63.8 MBytes   535 Mbits/sec    0   1.90 MBytes       
[  5]  45.00-46.00  sec  65.0 MBytes   545 Mbits/sec    0   1.90 MBytes       
[  5]  46.00-47.00  sec  62.5 MBytes   524 Mbits/sec    0   1.90 MBytes       
[  5]  47.00-48.00  sec  65.0 MBytes   545 Mbits/sec    0   1.90 MBytes       
[  5]  48.00-49.00  sec  66.2 MBytes   556 Mbits/sec    0   1.90 MBytes       
[  5]  49.00-50.00  sec  62.5 MBytes   524 Mbits/sec    0   1.90 MBytes       
[  5]  50.00-51.00  sec  66.2 MBytes   556 Mbits/sec    0   1.90 MBytes       
[  5]  51.00-52.00  sec  63.8 MBytes   535 Mbits/sec    0   1.90 MBytes       
[  5]  52.00-53.00  sec  63.8 MBytes   535 Mbits/sec    0   1.90 MBytes       
[  5]  53.00-54.00  sec  61.2 MBytes   514 Mbits/sec    0   1.90 MBytes       
[  5]  54.00-55.00  sec  63.8 MBytes   535 Mbits/sec    0   1.90 MBytes       
[  5]  55.00-56.00  sec  63.8 MBytes   535 Mbits/sec    0   1.90 MBytes       
[  5]  56.00-57.00  sec  65.0 MBytes   545 Mbits/sec    0   1.90 MBytes       
[  5]  57.00-58.00  sec  63.8 MBytes   535 Mbits/sec    0   1.90 MBytes       
[  5]  58.00-59.00  sec  67.5 MBytes   566 Mbits/sec    0   1.90 MBytes       
[  5]  59.00-60.00  sec  65.0 MBytes   545 Mbits/sec    0   1.90 MBytes       
[  5]  60.00-61.00  sec  66.2 MBytes   556 Mbits/sec    0   1.90 MBytes       
[  5]  61.00-62.00  sec  66.2 MBytes   556 Mbits/sec    0   1.90 MBytes       
[  5]  62.00-63.00  sec  66.2 MBytes   556 Mbits/sec    0   1.90 MBytes       
[... 525 lines cut ...]
[  5] 588.00-589.00 sec  56.2 MBytes   472 Mbits/sec    0   1.80 MBytes       
[  5] 589.00-590.00 sec  55.0 MBytes   461 Mbits/sec    0   1.80 MBytes       
[  5] 590.00-591.00 sec  56.2 MBytes   472 Mbits/sec    0   1.80 MBytes       
[... cut ...]
I started with 300 seconds on my first try. I did not trigger the problem. Then, as you can see above, I tried 600 seconds and am still not able to trigger the problem today. Next, 1200 seconds.

I have noted a periodic burst of retries which could be the result of many things. I wish I had something to analyze and test that goes more in-depth than iperf3. Suggestions?

After trying 1200 seconds, if it is not successful, then I may look at trying a non-DFS channel. Do you have high congestion levels on the channel you are using?

gifter77, 2023-05-11:
I let my router auto select channels. I'm on 52 at the moment using 80MHz band. There is no congestion on this channel.

morrownr, 2023-05-11:
While doing a 1200 second run, it happened again. Unplugging and replugging the adapter did not get me connected. I think I need to find a keyboard, mouse and video cable.

We are making progress. At least I think we are duplicating the problem.

gifter77, 2023-05-11:
> We are making progress. At least I think we are duplicating the problem.

Or you found a new one :) It's surprising that I only need a few minutes to reproduce it. Just need to turn on this webrtc service and connect to it while running iperf or doing a speedtest on the board.

[274 more replies]