morrownr/USB-WiFi #108: comfast wifi 6 adapter on amazon, buyer beware

ICRaptor, 2022-09-08:
https://www.amazon.com/dp/B09M3PKGXS?psc=1&ref=ppx_yo2ov_dt_b_product_details

purchased and received a comfast cf-957ax, on paper it's great, but further digging shows it's a RTL8832AU chipset.

sorry if this is in the wrong place

morrownr, 2022-09-09:
@ICRaptor 

This is a good place to let people know. Thanks for the post.

Right now there are only 2 products using the mt7921au chipset:

- Comfast CF-953AX
- Comfast CF-951AX

Word is that Alfa should start bringing adapters with the mt7921au chipset to market in the not so distant future. I suspect we will have several adapters to choose from within a few months.

My advice is to do the FREE Returns thing on the adapter that uses the rtl8832au chipset. The Linux driver is not only an out-of-kernel driver but it is a terrible driver. Consider the rtl8832au unsupported on Linux.

Nick

ICRaptor, 2022-09-10:
@morrownr 

Yes, instantly went through the return process on Amazon, further research shows there's no MT7921AU wifi adapter on Amazon 

I even checked on github for a repo for the RTL8832AU but couldn't find anything, only started looking for alternatives after kernel 5.19 stopped working with the cilynx/rtl88x2bu repo driver

morrownr, 2022-09-10:
> only started looking for alternatives after kernel 5.19 stopped working with the cilynx/rtl88x2bu repo driver

Hmmm... this site hosts several Realtek drivers, including one for the 8812bu chipset:

https://github.com/morrownr/88x2bu-20210702

Make sure you uninstall the previous driver. Let me know how it goes. Maybe that can get you by until there us better availability of adapters that use the new mt7921au chipset.

Nick

ICRaptor, 2022-09-10:
thanks, will try it out, instead i've just kept using kernel 5.18.x from fedora (the last one before 5.19)

ICRaptor, 2022-09-12:
As an update, the comfast cf-951ax worked out of the box with kernel 5.18.19 on fedora, no problems in more recent kernels

morrownr, 2022-09-12:
Thanks for the report. Keep in mind that if you want AP mode, you need to have at least kernel 5.19 and the June firmware update. There is a guide on the Main Menu for updating to the latest firmware.

A detailed report on your experience would be welcome by readers.

Nick