morrownr/USB-WiFi #109: Broken link in your documentation

Sami32, 2022-09-14:
https://github.com/morrownr/USB-WiFi/blob/main/iw_list/ALFA_AWUS036ACHM

I was [searching](https://github.com/morrownr/USB-WiFi/blob/main/home/USB_WiFi_Adapters_that_are_supported_with_Linux_in-kernel_drivers.md) what is the maximum TX power available with AWUS036ACHM and discovered this broken link instead :-(

I suspect that this model run stable and not hot using TX power = 30dBm but i've no idea if it can support 36 dBm (PA) and still be stable?

Thank you for your great work +1

morrownr, 2022-09-14:
Hi @Sami32 

The correct direct URL is:

https://github.com/morrownr/USB-WiFi/blob/main/home/iw_list/ALFA_AWUS036ACHM

I can't find where I have the bad URL on this site. Is the link from an off-site location?

Another way to get there is to select menu item 4 at the Main Menu and then select the adapter.

> I suspect that this model run stable and not hot using TX power = 30dBm but i've no idea if it can support 36 dBm (PA) and still be stable?

I've had the ACHM for a couple of years and have never seen any thermal problems. One thing to note about this adapter is that I think the TX power is fixed. I suspect it is a decision the engineers made based on the amp and antenna that are used. The TX power reading seem somewhat low based on historic norms but I think this is a result of the engineering. I have seen no problems with the low reading. In fact, I have conducted various head on test and the ACHM has the longest range of any modern dual band adapter that I have tested. Here is a comparison test:

https://github.com/morrownr/USB-WiFi/blob/main/home/Performance_Comparison.md

That test is not currently part of the Main Menu here.

Sami32, 2022-09-16:
Hello @morrownr

The page's link with the broken URL was provided in my "searching" hyperlink:
https://github.com/morrownr/USB-WiFi/blob/main/home/USB_WiFi_Adapters_that_are_supported_with_Linux_in-kernel_drivers.md

Thank you very much for the very interesting comparative link +1
I found 2 typos in it: "aggrivation" and "rtl881au"

You confirmed my searches and the Rokland compative test.
The funny part is that i discovered your page by searching for the Comfast MT7921au product.
Thought i wasn't aware of the DFS missing feature so thanks for making ne aware of this +1
The latest Linux kernel seem to have solved the TX power issue and OpenWRT is on the way to so it look pretty good to me :)

https://store.rokland.com/products/alfa-awus036achm-802-11ac-dual-band-high-power-ac1200-mediatek-wifi-usb-adapter
It seem that'll have to sort this out because they said that AWUS036ACHM do support DFS.

morrownr, 2022-09-17:
> It seem that'll have to sort this out because they said that AWUS036ACHM do support DFS.

Ah, it supports DFS channels in managed mode but not AP mode. WiFi used to be simple. It has become so complicated that we confuse ourselves on a regular basis.

> The funny part is that i discovered your page by searching for the Comfast MT7921au product.

I have a Comfast CF-951AX with the mt7921au. I'm not recommending this adapter yet but I can see that the chipset and driver are probably going to be real winners. I just learned today that the adapter has bluetooth support. I did not see that coming.

Sami32, 2022-09-17:
@morrownr Ohhh... OK, i thought it could be only because of the Windows and Linux drivers difference.
Damn...now i'll have to look at it and keep it in mind.
Thank you very much for pointing me to this +1

BTW, since you could be interested in the coming new Alfa AWUS036AXM by the end of october:

I'll be reading your testings as you became my new first source of informations for USB wireless chipsets/products ;)

Congratulations for all your great work +1

amisix, 2022-09-17:
>AWUS036AXM datasheet.pdf

Happy dance.

morrownr, 2022-09-17:
> I really hope that i'm wrong but i don't think that any Mediatek wireless chip with BT will provide an equal or better long range that the amazing AWUS036ACHM with the MT7610u.

If my understanding is correct, the chipset is not really what determines the range of an adapter. It is the amp and antennas that the adapter maker adds when building the adapter. ALFA has a history of selecting certain adapters to bless with very high quality amps and antennas. It is costly to do this and the adapters are not cheap but the results are good. 

> For some reason all their AX/BZ chipsets seem to have BT included and to me it is at the expense of the range :(

I'm not aware of ALFA ever having produced an adapter that has BT support. We will have to see the result. I'm not sure there will be a reduction in range given the form factor of their AXM adapter.

Thanks for the information about the ALFA AXM adapter. I had heard ALFA was going to come out with a couple of adapters based on the mt7921au but I had no details about the case. The case that we see in the pdf you provided tends to indicate to me that the AXM will be a MAX POWER or MAX RANGE (I forget exactly what ALFA calls it right now) version so we may get to see what range they can build right off the bat.

If you don't mind, I'll leave this issue open so others can read.

Thanks for the info,

Nick

Sami32, 2022-09-18:
The long range AWUS036AXHM model should come in february.

morrownr, 2022-09-19:
> The long range AWUS036AXHM model should come in february.

Do you have an informational pdf on it?

Sami32, 2022-09-23:
No, just my rought guess: MT7902U
[MT7902-Test-Report-MediaTek-ras-mt7902-ex-2-20.pdf](https://github.com/morrownr/USB-WiFi/files/9635618/MT7902-Test-Report-MediaTek-ras-mt7902-ex-2-20.pdf)
[WFA114717 (MT7902).pdf](https://github.com/morrownr/USB-WiFi/files/9635643/WFA114717.MT7902.pdf)
[5680405.pdf](https://github.com/morrownr/USB-WiFi/files/9635644/5680405.pdf)
[Test-Report-DFS-5680474.pdf](https://github.com/morrownr/USB-WiFi/files/9635645/Test-Report-DFS-5680474.pdf)

Whatever...i'll be waiting for an Alfa product using it ;)

closed 2022-09-23 by Sami32

morrownr, 2022-09-23:
The two primary wifi chipsets from Mediatek that I am aware of as far as WiFi 6e is concerned are the mt7921 and mt7922.

The usb chipset that is coming to market right now is the mt7921au which is the usb version the mt7921k which does 6 GHz channels. The only real difference with the mt7922 is that it does 160 MHz channel width. I hope we see a mt7922u chipset but I'm not sure it is that big of a deal. We will see.

Sami32, 2022-10-31:
https://www.alfa.com.tw/products/awus036axer?variant=39648597278792

Plenty of new products design should be available in 2023:
[ALFA Network_ModelStatement.pdf](https://github.com/morrownr/USB-WiFi/files/9901110/ALFA.Network_ModelStatement.pdf)

bjlockie, 2022-10-31:
Realtek :-(

morrownr, 2022-10-31:
That adapter is based on the rtl8832bu chipset. It is WiFi 6 only, no 6 GHz band unlike the mt7921au. And the Linux driver is out-of-kernel which leads to problems in the normal course of business even if you are a developer, let alone a regular user.

I'm writing that off as a no Linux support adapter and am waiting for the mt7921au and mt7922u based Alfa adapters.

bjlockie, 2022-10-31:
Will Alfa make mediatek adapters?
It was disappointing to see a Realtek adapter, not even 6e.

morrownr, 2022-10-31:
> Will Alfa make mediatek adapters?

Yes

Sami32, 2023-01-05:
[AWUS036AXML](https://www.alfa.com.tw/products/awus036axml?variant=39754360684616)
[AWUS036AX](https://www.alfa.com.tw/products/awus036ax?variant=39748815355976)

bjlockie, 2023-01-05:
https://store.rokland.com/products/alfa-awus036axml-802-11ax-wifi-6-1800-mbps-tri-band-wifi-usb-adapter-w-bluetooth
In stock late Feb, not a great price. :-(

On 2023-01-05 12:07, Samuel wrote:
>
> AWUS036AXML 
> <https://www.alfa.com.tw/products/awus036axml?variant=39754360684616>
>

morrownr, 2023-01-05:
> ### **[AWUS036AXML](https://www.alfa.com.tw/products/awus036axml?variant=39754360684616)**
> [AWUS036AX](https://www.alfa.com.tw/products/awus036ax?variant=39748815355976)

@Sami32 

Alright, what is this stuff with sticking this at the end of an old thread! Go start a NEW issue so people can see it.