morrownr/USB-WiFi #110: CF-951AX & CF-953AX not functioning in AP/Master mode in OpenWrt 22.03 using Luci

amisix, 2022-09-17:
I'm unable to get either Comfast adapter to initialize (device is not active/wireless is not associated) in OpenWrt 22.03.
I was able to progress past the error "daemon.err hostapd: Driver does not support configured HT capability [LDPC]"  by adding option ldpc '0' to /etc/config/wireless... Only to come up with 3 more errors. Adding the below options bypassed those errors:

**option ldpc '0'
option tx_stbc '0'
option rx_stbc '0'
option max_amsdu '0'**

Once you get past the above errors I receive this error:

**logread:**

>Sat Sep 17 02:26:02 2022 daemon.notice netifd: radio1 (2203): Failed to create interface wlan0
Sat Sep 17 02:26:02 2022 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy0.conf (phy wlan0) --> new PHY
Sat Sep 17 02:26:02 2022 kern.info kernel: [  313.216398] br-lan: port 2(wlan0) entered blocking state
Sat Sep 17 02:26:02 2022 kern.info kernel: [  313.224484] br-lan: port 2(wlan0) entered disabled state
Sat Sep 17 02:26:02 2022 kern.info kernel: [  313.232448] device wlan0 entered promiscuous mode
Sat Sep 17 02:26:02 2022 **daemon.err hostapd: Driver does not support configured HT capability [LDPC]**
Sat Sep 17 02:26:02 2022 daemon.notice hostapd: wlan0: interface state UNINITIALIZED->DISABLED
Sat Sep 17 02:26:02 2022 daemon.notice hostapd: wlan0: AP-DISABLED
Sat Sep 17 02:26:02 2022 daemon.err hostapd: wlan0: Unable to setup interface.
Sat Sep 17 02:26:02 2022 daemon.notice hostapd: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Sat Sep 17 02:26:03 2022 kern.info kernel: [  313.282783] device wlan0 left promiscuous mode
Sat Sep 17 02:26:03 2022 kern.info kernel: [  313.291243] br-lan: port 2(wlan0) entered disabled state
Sat Sep 17 02:26:03 2022 daemon.err hostapd: rmdir[ctrl_interface=/var/run/hostapd]: Permission denied
Sat Sep 17 02:26:03 2022 daemon.notice hostapd: wlan0: CTRL-EVENT-TERMINATING
Sat Sep 17 02:26:03 2022 daemon.err hostapd: hostapd_free_hapd_data: Interface wlan0 wasn't started
Sat Sep 17 02:26:03 2022 daemon.notice netifd: radio1 (2203): Command failed: ubus call hostapd config_add {"iface":"wlan0", "config":"/var/run/hostapd-phy0.conf"} (Invalid argument)
Sat Sep 17 02:26:03 2022 daemon.notice netifd: radio1 (2203): Usage: ubus [<options>] <command> [arguments...]
Sat Sep 17 02:26:03 2022 daemon.notice netifd: radio1 (2203): Options:
Sat Sep 17 02:26:03 2022 daemon.notice netifd: radio1 (2203):  -s <socket>:             Set the unix domain socket to connect to
Sat Sep 17 02:26:03 2022 daemon.notice netifd: radio1 (2203):  -t <timeout>:            Set the timeout (in seconds) for a command to complete
Sat Sep 17 02:26:03 2022 daemon.notice netifd: radio1 (2203):  -S:                      Use simplified output (for scripts)
Sat Sep 17 02:26:03 2022 daemon.notice netifd: radio1 (2203):  -v:                      More verbose output
Sat Sep 17 02:26:03 2022 daemon.notice netifd: radio1 (2203):  -m <type>:               (for monitor): include a specific message type
Sat Sep 17 02:26:03 2022 daemon.notice netifd: radio1 (2203):                   (can be used more than once)
Sat Sep 17 02:26:03 2022 daemon.notice netifd: radio1 (2203):  -M <r|t>         (for monitor): only capture received or transmitted traffic
Sat Sep 17 02:26:03 2022 daemon.notice netifd: radio1 (2203):
Sat Sep 17 02:26:03 2022 daemon.notice netifd: radio1 (2203): Commands:
Sat Sep 17 02:26:03 2022 daemon.notice netifd: radio1 (2203):  - list [<path>]                  List objects
Sat Sep 17 02:26:03 2022 daemon.notice netifd: radio1 (2203):  - call <path> <method> [<message>]       Call an object method
Sat Sep 17 02:26:03 2022 daemon.notice netifd: radio1 (2203):  - subscribe <path> [<path>...]   Subscribe to object(s) notifications
Sat Sep 17 02:26:03 2022 daemon.notice netifd: radio1 (2203):  - listen [<path>...]                     Listen for events
Sat Sep 17 02:26:03 2022 daemon.notice netifd: radio1 (2203):  - send <type> [<message>]                Send an event
Sat Sep 17 02:26:03 2022 daemon.notice netifd: radio1 (2203):  - wait_for <object> [<object>...]        Wait for multiple objects to appear on ubus
Sat Sep 17 02:26:03 2022 daemon.notice netifd: radio1 (2203):  - monitor                                Monitor ubus traffic
Sat Sep 17 02:26:03 2022 daemon.notice netifd: radio1 (2203):
Sat Sep 17 02:26:03 2022 daemon.notice netifd: radio1 (2203): Device setup failed: HOSTAPD_START_FAILED
Sat Sep 17 02:26:03 2022 daemon.notice netifd: Wireless device 'radio1' set retry=0
Sat Sep 17 02:26:03 2022 daemon.crit netifd: Wireless device 'radio1' setup failed, retry=0
Sat Sep 17 02:26:03 2022 daemon.notice netifd: Wireless device 'radio1' is now down

morrownr, 2022-09-18:
@amisix 

Just a couple of big picture thoughts:

- Your title fails to mention RasPi4B. Might be good to work that in. OpenWRT has different firmware for all of the different platforms for a reason... they are different. My experience with a CF-951AX with my Zyxel router running OpeWRT 22.03 has been very pleasant as far as WiFi 6 is concerned. Loci can't handle WiFi 6e yet I had to modify /etc/config/wireless to get the 6 GHz band going... well, I say going but I have no way to test yet. But it sure looks like it worked. If you ad RasPi4B to the title, it might attract of 4B users that have found a solution. You might also want to make a message in the appropriate OpenWRT forum to see if anyone there has tried this on a Pi.

- My RasPi4B is working another job right but it can go on vacation and see what I can figure out. I'll let you know.

amisix, 2022-09-18:
Done, thx. It's great to hear how well they work on certain devices, hopefully we can accomplish the same with the Raspberry Pi 4B, I'd be happy with Wifi 6 capabilities but 6e makes me drool, I've been following your other thread with excitement. 

Thanks for testing it out on your end. I will open a thread on the OpenWrt forum also.

morrownr, 2022-09-19:
@amisix 

Short follow on. I've been testing the 951AX on my ZyXEL NBG6817 router in AX in the 2.4 band today. It is strange to see anything other than N for the 2.4 band. No problems noted. So far OpenWRT 22.03 is working very well with the 951AX with the exception that Luci can't do 6 GHz yet 

Now I just need a mt7921au based adapter that actually has some range. This little Comfast adapter is good for testing but it is a medium range adapter. Come on Alfa, we are waiting on the high power adapter.

amisix, 2022-09-20:
@ayyyuki1 Yes, same issue with N, AC, and AX. Slightly different error message for each, but they have in common hostapd "invalid argument" and permission errors. Thanks. 

**N 2.4Ghz (20Mhz channel width):**
>Tue Sep 20 19:44:20 2022 daemon.notice netifd: radio1 (18306): Failed to create interface wlan0
Tue Sep 20 19:44:20 2022 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy0.conf (phy wlan0) --> new PHY
Tue Sep 20 19:44:20 2022 daemon.err hostapd: Driver does not support configured HT capability [SHORT-GI-40]
Tue Sep 20 19:44:20 2022 daemon.notice hostapd: wlan0: interface state UNINITIALIZED->DISABLED
Tue Sep 20 19:44:20 2022 daemon.notice hostapd: wlan0: AP-DISABLED
Tue Sep 20 19:44:20 2022 daemon.err hostapd: wlan0: Unable to setup interface.
Tue Sep 20 19:44:20 2022 daemon.notice hostapd: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Tue Sep 20 19:44:20 2022 daemon.err hostapd: rmdir[ctrl_interface=/var/run/hostapd]: Permission denied
Tue Sep 20 19:44:20 2022 daemon.notice hostapd: wlan0: CTRL-EVENT-TERMINATING
Tue Sep 20 19:44:20 2022 daemon.err hostapd: hostapd_free_hapd_data: Interface wlan0 wasn't started
Tue Sep 20 19:44:20 2022 daemon.notice netifd: radio1 (18306): Command failed: ubus call hostapd config_add {"iface":"wlan0", "config":"/var/run/hostapd-phy0.conf"} (Invalid argument)
Tue Sep 20 19:44:20 2022 daemon.notice netifd: radio1 (18306): Usage: ubus [<options>] <command> [arguments...]
Tue Sep 20 19:44:20 2022 daemon.notice netifd: radio1 (18306): Options:
Tue Sep 20 19:44:20 2022 daemon.notice netifd: radio1 (18306):  -s <socket>:            Set the unix domain socket to connect to
Tue Sep 20 19:44:20 2022 daemon.notice netifd: radio1 (18306):  -t <timeout>:           Set the timeout (in seconds) for a command to complete
Tue Sep 20 19:44:20 2022 daemon.notice netifd: radio1 (18306):  -S:                     Use simplified output (for scripts)
Tue Sep 20 19:44:20 2022 daemon.notice netifd: radio1 (18306):  -v:                     More verbose output
Tue Sep 20 19:44:20 2022 daemon.notice netifd: radio1 (18306):  -m <type>:              (for monitor): include a specific message type
Tue Sep 20 19:44:20 2022 daemon.notice netifd: radio1 (18306):                  (can be used more than once)
Tue Sep 20 19:44:20 2022 daemon.notice netifd: radio1 (18306):  -M <r|t>                (for monitor): only capture received or transmitted traffic
Tue Sep 20 19:44:20 2022 daemon.notice netifd: radio1 (18306):
Tue Sep 20 19:44:20 2022 daemon.notice netifd: radio1 (18306): Commands:
Tue Sep 20 19:44:20 2022 daemon.notice netifd: radio1 (18306):  - list [<path>]                 List objects
Tue Sep 20 19:44:20 2022 daemon.notice netifd: radio1 (18306):  - call <path> <method> [<message>]      Call an object method
Tue Sep 20 19:44:20 2022 daemon.notice netifd: radio1 (18306):  - subscribe <path> [<path>...]  Subscribe to object(s) notifications
Tue Sep 20 19:44:20 2022 daemon.notice netifd: radio1 (18306):  - listen [<path>...]                    Listen for events
Tue Sep 20 19:44:20 2022 daemon.notice netifd: radio1 (18306):  - send <type> [<message>]               Send an event
Tue Sep 20 19:44:20 2022 daemon.notice netifd: radio1 (18306):  - wait_for <object> [<object>...]       Wait for multiple objects to appear on ubus
Tue Sep 20 19:44:20 2022 daemon.notice netifd: radio1 (18306):  - monitor                               Monitor ubus traffic
Tue Sep 20 19:44:20 2022 daemon.notice netifd: radio1 (18306):
Tue Sep 20 19:44:20 2022 daemon.notice netifd: radio1 (18306): Device setup failed: HOSTAPD_START_FAILED
Tue Sep 20 19:44:20 2022 daemon.notice netifd: Wireless device 'radio1' set retry=0
Tue Sep 20 19:44:20 2022 daemon.crit netifd: Wireless device 'radio1' setup failed, retry=0
Tue Sep 20 19:44:21 2022 daemon.notice netifd: Wireless device 'radio1' is now down

**AC 5Ghz (40Mhz channel width):**
>Tue Sep 20 19:44:20 2022 daemon.notice netifd: radio1 (18306): Failed to create interface wlan0
Tue Sep 20 19:44:20 2022 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy0.conf (phy wlan0) --> new PHY
Tue Sep 20 19:44:20 2022 daemon.err hostapd: Driver does not support configured HT capability [SHORT-GI-40]
Tue Sep 20 19:44:20 2022 daemon.notice hostapd: wlan0: interface state UNINITIALIZED->DISABLED
Tue Sep 20 19:44:20 2022 daemon.notice hostapd: wlan0: AP-DISABLED
Tue Sep 20 19:44:20 2022 daemon.err hostapd: wlan0: Unable to setup interface.
Tue Sep 20 19:44:20 2022 daemon.notice hostapd: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Tue Sep 20 19:44:20 2022 daemon.err hostapd: rmdir[ctrl_interface=/var/run/hostapd]: Permission denied
Tue Sep 20 19:44:20 2022 daemon.notice hostapd: wlan0: CTRL-EVENT-TERMINATING
Tue Sep 20 19:44:20 2022 daemon.err hostapd: hostapd_free_hapd_data: Interface wlan0 wasn't started
Tue Sep 20 19:44:20 2022 daemon.notice netifd: radio1 (18306): Command failed: ubus call hostapd config_add {"iface":"wlan0", "config":"/var/run/hostapd-phy0.conf"} (Invalid argument)
Tue Sep 20 19:44:20 2022 daemon.notice netifd: radio1 (18306): Usage: ubus [<options>] <command> [arguments...]
Tue Sep 20 19:44:20 2022 daemon.notice netifd: radio1 (18306): Options:
Tue Sep 20 19:44:20 2022 daemon.notice netifd: radio1 (18306):  -s <socket>:            Set the unix domain socket to connect to
Tue Sep 20 19:44:20 2022 daemon.notice netifd: radio1 (18306):  -t <timeout>:           Set the timeout (in seconds) for a command to complete
[... cut ...]
Sat Sep 17 02:26:03 2022 daemon.notice netifd: radio1 (2203): - monitor Monitor ubus traffic
Sat Sep 17 02:26:03 2022 daemon.notice netifd: radio1 (2203):
Sat Sep 17 02:26:03 2022 daemon.notice netifd: radio1 (2203): Device setup failed: HOSTAPD_START_FAILED
Sat Sep 17 02:26:03 2022 daemon.notice netifd: Wireless device 'radio1' set retry=0
Sat Sep 17 02:26:03 2022 daemon.crit netifd: Wireless device 'radio1' setup failed, retry=0
Sat Sep 17 02:26:03 2022 daemon.notice netifd: Wireless device 'radio1' is now down

morrownr, 2022-09-21:
@amisix 

I installed the 64 bit, ext4, factory img of OpenWRT 22.03 for Raspberry Pi 4B today. I configured it to be a dumb ap. There are some serious problems. This is nowhere near the the smooth performance I saw on my Zyxel router.

As I have time this week, I will sort through the logs and see what I can figure out.

amisix, 2022-09-21:
@morrownr Same version as what I ran, yes. Thanks much for confirming & taking the time to test it out. The Zyxel experience sounded great when I read of it..

ayyyuki1, 2022-09-21:
@amisix 
if you have a pc or a laptop, you can boot openwrt from a usb to still try it ;) that way you can really see if pi is the issue

morrownr, 2022-09-21:
@amisix 

I decided to clean things out and get to the bottom of some of the things I am seeing. OpenWRT for the Pi4B has problems. Example:

https://github.com/openwrt/openwrt/issues/9600

I can't get 2.4 GHz going with the internal wifi and it appears nobody else can either. 

I did put the 951ax adapter in a usb3 port and load the drivers. Wow. That did not work well. It was nothing like what I saw on my ZyXEL NBG6817. 22.03 for the Pi4B appears to not be ready for prime time.

amisix, 2022-09-21:
@morrownr RE: onboard 2.4Ghz. Yeah... I kinda bumped into that issue too, thanks for linking the thread as it was exactly what I was looking for at the time. The latest poster is right about Legacy not even working like in did in previous RCs. And no fix in sight. 

It's a shame these issues are occurring given the support Raspberry Pi & OpenWrt have in the community. 
The 951AX & 953AX have a similar chipset to the upcoming Alfa adapter, I wonder if it will also be impacted..

morrownr, 2022-09-22:
> It's a shame these issues are occurring

Yes, it is. However, stuff happens. This stuff will be fixed. We have to remember that OpenWRT is designed for dedicated wireless routers and the Pi4B is not a wireless router, it is a general purpose computing device. We also have to remember that this edition of OpenWRT had some rather big changes. We are on the bleeding edge of the new OpenWRT distro. We are only the bleeding edge of the mt7921au chip and we are on the bleeding edge of the mt7921u driver. All problems should settle down over the next few months.

> The 951AX & 953AX have a similar chipset to the upcoming Alfa adapter, I wonder if it will also be impacted.

That is a good question. I don't have the answer.

amisix, 2022-09-22:
Eh, maybe I worded that poorly. No blame, just general disappointment. Making good things takes time. It's great that there's more than 1 person troubleshooting it. 

Maybe if the Alfa adapter is impacted it could put more eyes on obtaining a resolution.

morrownr, 2022-09-22:
I certainly was not expecting the Pi4B version of OpenWRT to be in this bad of shape. I don't see the mt7921u driver as the problem as it is working fine on my router. Something I have noticed when I stop by the OpenWRT forums is not much traffic by Pi users. I have to wonder if the Pi versions get much testing. 

I like to use my Pi4B as an AP but I use my own guide which uses hostapd.

ayyyuki1, 2022-09-22:
I tried running mt7921au on a ryzen system and on two routers (ipq40xx/armv7l and mt7621a/mips). All on OpenWrt with Linux 5.15.

Those worked fine, however on my intel laptop I got some issues with AP mode. Yup, we are really on the bleeding edge. This will take some time to settle down...

morrownr, 2022-09-22:
@ayyyuki1 @amisix 

> Those worked fine, however on my intel laptop I got some issues with AP mode. Yup, we are really on the bleeding edge. This will take some time to settle down...

Agree. I was pondering the best way to make a little progress with the Pi4B since amisix is trying to test on a Pi4B. I have an idea. I'm throwing together a simply guide to compiling kernel v6.0 on standard 64 bit RasPiOS installation. I can see a couple of other things we could test as well. The reason for this is that the 5.15 kernel in the standard RasPioS distro does not have the mt7921u driver yet. Okay, yes, still bleeding edge but could help with what we are trying to do.

morrownr, 2022-09-23:
$ uname -a
Linux RasPi4B 6.0.0-rc6-g8fc36980e861 #1 SMP PREEMPT Thu Sep 22 22:27:59 CDT 2022 aarch64 GNU/Linux

You also need to upgrade the 3 firmware files.

My Pi4B is running kernel 6.0.

amisix, 2022-09-24:
> I'm throwing together a simply guide to compiling kernel v6.0 on standard 64 bit RasPiOS installation. 

Please direct me to it when available and I will follow it. Thanks

morrownr, 2022-09-24:
Building the 64 bit RasPiOS kernel on the Pi3, Pi3+, Pi4, 400, Zero 2 W,
and Raspberry Pi Compute Modules 3, 3+ and 4

Note: Compiling the kernel on a RasPi can take a long time.

-----

1. Install the latest 64 bit version of Raspberry Pi OS on supported media. Boot your
Raspberry Pi, log in, configure the RasPiOS, reboot and ensure you’re connected
to the internet to give you access to the sources.

-----

2. Install Git and the build dependencies:

$ sudo apt install git bc bison flex libssl-dev make libncurses5-dev

Note: libncurses5-dev is required for menuconfig (not used in this guide yet.).

-----

3. Download the sources:

Note: change "rpi-6.0.y" to the branch you want in the command below.

$ git clone --depth=1 --branch rpi-6.0.y https://github.com/raspberrypi/linux

Note: Omitting --depth=1 will download the entire repository, including the
full history of all branches, but this takes much longer and occupies much
more storage.

Refer to the original GitHub repository for information about the available branches.

https://github.com/raspberrypi/linux

-----

4. Configure the kernel; as well as the default configuration, you may wish to configure your
kernel in more detail or apply patches from another source, to add or remove required
functionality.

If you want to apply the Default Configuration: (I'll add options later.)

Prepare the default configuration by running the following commands

$ cd linux
$ KERNEL=kernel8
$ make bcm2711_defconfig

-----

5. Building the Kernel

Build and install the kernel, modules, and Device Tree blobs; this step can take a long time
depending on the Raspberry Pi model in use.

For the 64-bit kernel:

$ make -j4 Image.gz modules dtbs
$ sudo make modules_install
$ sudo cp arch/arm64/boot/dts/broadcom/*.dtb /boot/
$ sudo cp arch/arm64/boot/dts/overlays/*.dtb* /boot/overlays/
$ sudo cp arch/arm64/boot/dts/overlays/README /boot/overlays/

If KERNEL was not set above, set it now:
$ KERNEL=kernel8

$ sudo cp arch/arm64/boot/Image.gz /boot/$KERNEL.img

Reboot. Your Raspberry Pi should be running your freshly-compiled kernel!

-----

Let me know how it goes.

morrownr, 2022-09-24:
@amisix 

After setting up new flash and update of the 64 bit RasPiOS, I noticed Chromium running like honey on a cold winter day. The fix:

Settings > System > Use hardware acceleration when available

Turn it off!

Oh, and the RasPiOS does need the 3 firmware files for the mt7921au updated. Remember that I have a checklist in the Main Menu.

Cheers

morrownr, 2022-09-24:
@amisix 

Heads up. New firmware files as of less than 24 hours ago:

525870 Sep 24 00:55 BT_RAM_CODE_MT7961_1_2_hdr.bin
  92192 Sep 24 00:49 WIFI_MT7961_patch_mcu_1_2_hdr.bin
716404 Sep 24 00:54 WIFI_RAM_CODE_MT7961_1.bin

morrownr, 2022-09-24:
Status update: My Pi4B now shows the below wifi interface using the CF-951AX adapter. RasPiOS is updated to kernel 6.0. The hot new firmware has been installed. The mt7921u driver is working well.

```
$ iw dev
phy#0
	Interface wlan0
		ifindex 3
		wdev 0x1
		addr e0:e1:a9:35:c5:1f
		type managed
		txpower 3.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
			0	0	0	0	0	0	0	0		0

```

What I am thinking about doing is working this until I have a tri-band wifi router or AP. Yes, 3 usb wifi adapters on the Pi4B. What are the issues we are likely to see:

- I suspect we will have to download and compile a new hostapd to make sure everything we need is turned on.
- I suspect we will have to download and compile a new wpa_supplicant to make sure everything we need is turned.

This could be a cool project. Should I start a new issue dedicated to this project?

amisix, 2022-09-25:
Step 5. 
$ make -j4 Image.gz modules dtbs

Produces this error:
*** No rule to make target 'Image.gz'. Stop. 

What am I doing incorrectly?

copyliu, 2022-11-09:
got my CF-951AX, and put it to a x86 OpenWRT, and load newest  firmware, and try to get to AP mode.

2.4 GHz band AP mode can bring up interface but client never see it,
5Ghz band AP mode client will immediately drop connection after authenticated

very strange

```root@OpenWrt:~# iw phy
Wiphy phy0
        wiphy index: 0
        max # scan SSIDs: 4
        max scan IEs length: 482 bytes
        max # sched scan SSIDs: 10
        max # match sets: 16
        Retry short limit: 7
        Retry long limit: 4
        Coverage class: 0 (up to 0m)
        Device supports AP-side u-APSD.
        Device supports T-DLS.
        Supported Ciphers:
                * WEP40 (00-0f-ac:1)
                * WEP104 (00-0f-ac:5)
                * TKIP (00-0f-ac:2)
                * CCMP-128 (00-0f-ac:4)
                * CCMP-256 (00-0f-ac:10)
                * GCMP-128 (00-0f-ac:8)
                * GCMP-256 (00-0f-ac:9)
                * CMAC (00-0f-ac:6)
                * CMAC-256 (00-0f-ac:13)
                * GMAC-128 (00-0f-ac:11)
                * GMAC-256 (00-0f-ac:12)
        Available Antennas: TX 0x3 RX 0x3
        Configured Antennas: TX 0x3 RX 0x3
        Supported interface modes:
                 * managed
                 * AP
                 * AP/VLAN
                 * monitor
        Band 1:
                Capabilities: 0x9ff
                        RX LDPC
                        HT20/HT40
                        SM Power Save disabled
                        RX Greenfield
                        RX HT20 SGI
                        RX HT40 SGI
                        TX STBC
                        RX STBC 1-stream
                        Max AMSDU length: 7935 bytes
                        No DSSS/CCK HT40
                Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
                Minimum RX AMPDU time spacing: No restriction (0x00)
                HT TX/RX MCS rate indexes supported: 0-15
                HE Iftypes: managed
                        HE MAC Capabilities (0x08011a000040):
                                +HTC HE Supported
                                Trigger Frame MAC Padding Duration: 2
                                OM Control
                                Maximum A-MPDU Length Exponent: 3
                                A-MSDU in A-MPDU
                        HE PHY Capabilities: (0x2270ce120dc0b306423f00):
                                HE40/2.4GHz
                                242 tone RUs/2.4GHz
                                Device Class: 1
                                LDPC Coding in Payload
                                HE SU PPDU with 1x HE-LTF and 0.8us GI
                                NDP with 4x HE-LTF and 3.2us GI
                                STBC Tx <= 80MHz
                                STBC Rx <= 80MHz
                                Full Bandwidth UL MU-MIMO
[... 554 lines cut ...]
                * [ BEACON_RATE_HT ]: HT beacon rate setting
                * [ BEACON_RATE_VHT ]: VHT beacon rate setting
                * [ FILS_STA ]: STA FILS (Fast Initial Link Setup)
                * [ CQM_RSSI_LIST ]: multiple CQM_RSSI_THOLD records
                * [ CONTROL_PORT_OVER_NL80211 ]: control port over nl80211
                * [ TXQS ]: FQ-CoDel-enabled intermediate TXQs
                * [ AIRTIME_FAIRNESS ]: airtime fairness scheduling
                * [ AQL ]: Airtime Queue Limits (AQL)
                * [ CONTROL_PORT_NO_PREAUTH ]: disable pre-auth over nl80211 control port support
                * [ SCAN_FREQ_KHZ ]: scan on kHz frequency support
                * [ CONTROL_PORT_OVER_NL80211_TX_STATUS ]: tx status for nl80211 control port support
root@OpenWrt:~# iw dev
phy#0
        Interface wlan0
                ifindex 6
                wdev 0x3
                addr e0:e1:a9:36:68:6d
                ssid CP_EAP2
                type AP
                channel 36 (5180 MHz), width: 40 MHz, center1: 5190 MHz
                txpower 3.00 dBm
                multicast TXQ:
                        qsz-byt qsz-pkt flows   drops   marks   overlmt hashcol tx-bytes        tx-packets
                        0       0       18      0       0       0       0       1712            20
```

morrownr, 2022-11-09:
Hi @copyliu 

Strange is what we do around here. <g>

Try limiting both radios to WPA2 AES. Make sure the country code is set for both radios. It should not matter for 2.4 but will for 5. Are you using AC or AX? If AX, try AC for now and see if it helps.

Do you have other wifi radios in the box?

Did you have the adapter in a usb3 port when you booted the os the first time as a new burn?

Luci is not 100% good with WiFi 6 yet and is a no-go with WiFi 6e so I am mostly only using ssh to change wireless setting instead of Luci. This will change with time but for now...

copyliu, 2022-11-09:
@morrownr 

I can see only 1 radio on  my openWRT, and country code is set.
I also set it to open network(no encrypt) but no luck, 5G client immediately drop after connected,

copyliu, 2022-11-10:
from logread
```
Wed Nov  9 17:44:04 2022 daemon.notice hostapd: Configuration file: /var/run/hostapd-phy0.conf (phy wlan0) --> new PHY
Wed Nov  9 17:44:04 2022 daemon.notice hostapd: wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
Wed Nov  9 17:44:04 2022 kern.info kernel: [ 4028.313618] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
Wed Nov  9 17:44:04 2022 daemon.notice hostapd: wlan0: interface state COUNTRY_UPDATE->ENABLED
Wed Nov  9 17:44:04 2022 daemon.notice hostapd: wlan0: AP-ENABLED
Wed Nov  9 17:44:05 2022 daemon.notice netifd: Wireless device 'radio0' is now up
Wed Nov  9 17:44:05 2022 daemon.notice netifd: Network device 'wlan0' link is up
Wed Nov  9 17:44:05 2022 daemon.notice netifd: Interface 'tempwlan' is enabled
Wed Nov  9 17:44:05 2022 daemon.notice netifd: Interface 'tempwlan' is setting up now
Wed Nov  9 17:44:05 2022 daemon.notice netifd: Interface 'tempwlan' is now up
Wed Nov  9 17:44:05 2022 daemon.notice netifd: Interface 'tempwlan' has link connectivity
Wed Nov  9 17:44:42 2022 daemon.debug hostapd: wlan0: STA 64:a2:f9:1c:15:23 IEEE 802.11: authentication OK (open system)
Wed Nov  9 17:44:42 2022 daemon.debug hostapd: wlan0: STA 64:a2:f9:1c:15:23 MLME: MLME-AUTHENTICATE.indication(64:a2:f9:1c:15:23, OPEN_SYSTEM)
Wed Nov  9 17:44:42 2022 daemon.debug hostapd: wlan0: STA 64:a2:f9:1c:15:23 MLME: MLME-DELETEKEYS.request(64:a2:f9:1c:15:23)
Wed Nov  9 17:44:42 2022 daemon.info hostapd: wlan0: STA 64:a2:f9:1c:15:23 IEEE 802.11: authenticated
Wed Nov  9 17:44:42 2022 daemon.debug hostapd: wlan0: STA 64:a2:f9:1c:15:23 IEEE 802.11: association OK (aid 1)
Wed Nov  9 17:44:42 2022 daemon.info hostapd: wlan0: STA 64:a2:f9:1c:15:23 IEEE 802.11: associated (aid 1)
Wed Nov  9 17:44:42 2022 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 64:a2:f9:1c:15:23
Wed Nov  9 17:44:42 2022 daemon.debug hostapd: wlan0: STA 64:a2:f9:1c:15:23 MLME: MLME-ASSOCIATE.indication(64:a2:f9:1c:15:23)
Wed Nov  9 17:44:42 2022 daemon.debug hostapd: wlan0: STA 64:a2:f9:1c:15:23 MLME: MLME-DELETEKEYS.request(64:a2:f9:1c:15:23)
Wed Nov  9 17:44:42 2022 daemon.debug hostapd: wlan0: STA 64:a2:f9:1c:15:23 IEEE 802.11: binding station to interface 'wlan0'
Wed Nov  9 17:44:42 2022 daemon.info hostapd: wlan0: STA 64:a2:f9:1c:15:23 RADIUS: starting accounting session 5384E144CD573907
Wed Nov  9 17:44:45 2022 daemon.notice hostapd: wlan0: AP-STA-DISCONNECTED 64:a2:f9:1c:15:23
Wed Nov  9 17:44:45 2022 daemon.debug hostapd: wlan0: STA 64:a2:f9:1c:15:23 IEEE 802.11: deauthenticated
```

it looks nothing wrong but client just drop connection

morrownr, 2022-11-10:
I took at look at my log file using this same adapter and I don't see the following line:

> 4:42 2022 daemon.info hostapd: wlan0: STA 64:**:**:**:**:23 RADIUS: starting accounting session 5384E144CD573907

Could it be a security related setting?

copyliu, 2022-11-10:
I think I was misconfig something before, and now I can bring up AP with a separate bridge,
now I try to bridge to my main bridge (I hope my OpenWRT machine run as a bridged AP) then got kernel panic

```[  126.456794] mt7921u 2-4:1.3: firmware: direct-loading firmware mediatek/WIFI_MT7961_patch_mcu_1_2_hdr.bin
[  126.456847] mt7921u 2-4:1.3: HW/SW Version: 0x8a108a10, Build Time: 20220908210919a
               
[  126.476462] mt7921u 2-4:1.3: firmware: direct-loading firmware mediatek/WIFI_RAM_CODE_MT7961_1.bin
[  126.476508] mt7921u 2-4:1.3: WM Firmware Version: ____010000, Build Time: 20220908211021
[  128.133306] mt7921u 2-4:1.3 wlxe0e1a936686d: renamed from wlan0
[  180.651960] br0: port 2(wlxe0e1a936686d) entered blocking state
[  180.651997] br0: port 2(wlxe0e1a936686d) entered disabled state
[  180.652812] device wlxe0e1a936686d entered promiscuous mode
[  181.227902] IPv6: ADDRCONF(NETDEV_CHANGE): wlxe0e1a936686d: link becomes ready
[  181.228036] br0: port 2(wlxe0e1a936686d) entered blocking state
[  181.228058] br0: port 2(wlxe0e1a936686d) entered forwarding state
[  186.305237] skbuff: skb_under_panic: text:ffffffffc0cbd4c5 len:128 put:4 head:ffff9caaf4292000 data:ffff9caaf4291ffc tail:0x7c end:0xc0 dev:wlxe0e1a936686d
[  186.305366] ------------[ cut here ]------------
[  186.305382] kernel BUG at net/core/skbuff.c:120!
[  186.305412] invalid opcode: 0000 [#1] PREEMPT SMP PTI
[  186.305430] CPU: 5 PID: 2689 Comm: mt76-tx phy1 Kdump: loaded Not tainted 6.0.0-2-amd64 #1  Debian 6.0.5-1
[  186.305455] Hardware name: QEMU Standard PC (Q35 + ICH9, 2009), BIOS 1.14.0-2 04/01/2014
[  186.305475] RIP: 0010:skb_panic+0x48/0x4a
[  186.305507] Code: 4f 70 50 8b 87 c0 00 00 00 50 8b 87 bc 00 00 00 50 ff b7 d0 00 00 00 4c 8b 8f c8 00 00 00 48 c7 c7 a8 8d 7e bd e8 7e ab fc ff <0f> 0b 48 8b 14 24 48 c7 c1 a0 75 52 bd e8 a6 ff ff ff 48 8b 14 24
[  186.305548] RSP: 0018:ffffa9918095fb70 EFLAGS: 00010246
[  186.305563] RAX: 000000000000008f RBX: ffff9caafcbcd100 RCX: 0000000000000000
[  186.305580] RDX: 0000000000000000 RSI: ffffffffbd77ed12 RDI: 00000000ffffffff
[  186.305602] RBP: ffff9caaf4342c90 R08: 0000000000000000 R09: ffffa9918095fa08
[  186.305618] R10: 0000000000000003 R11: ffffffffbded2128 R12: ffff9caac7fb2060
[  186.305632] R13: 0000000000000002 R14: 000000000000007c R15: 0000000000000002
[  186.306258] FS:  0000000000000000(0000) GS:ffff9caafed40000(0000) knlGS:0000000000000000
[  186.307201] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  186.307891] CR2: 00007f849da13a28 CR3: 00000000090b2003 CR4: 0000000000370ee0
[  186.308539] DR0: 0000000000000000 DR1: 0000000000000000 DR2: 0000000000000000
[  186.309149] DR3: 0000000000000000 DR6: 00000000fffe0ff0 DR7: 0000000000000400
[  186.309556] Call Trace:
[  186.309978]  <TASK>
[  186.310389]  skb_push.cold+0x10/0x10
[  186.311190]  mt7921_usb_sdio_tx_prepare_skb+0x105/0x180 [mt7921_common]
[  186.311817]  mt76u_tx_queue_skb+0x87/0x1d0 [mt76_usb]
[  186.312591]  __mt76_tx_queue_skb+0x41/0xa0 [mt76]
[  186.313314]  mt76_txq_schedule.part.0+0x1d3/0x3e0 [mt76]
[  186.314017]  ? mt76_get_min_avg_rssi+0x130/0x130 [mt76]
[  186.314685]  mt76_txq_schedule_all+0x19/0x30 [mt76]
[  186.315373]  mt7921_tx_worker+0x45/0xc0 [mt7921_common]
[  186.316002]  __mt76_worker_fn+0x49/0x80 [mt76]
[  186.316590]  kthread+0xe6/0x110
[  186.317265]  ? kthread_complete_and_exit+0x20/0x20
[  186.317895]  ret_from_fork+0x1f/0x30
[  186.318468]  </TASK>
[  186.319009] Modules linked in: ccm bridge stp llc intel_rapl_msr intel_rapl_common intel_pmc_core_pltdrv intel_pmc_core kvm_intel kvm irqbypass ghash_clmulni_intel btusb btrtl btbcm btintel aesni_intel btmtk mt7921u crypto_simd cryptd mt7921_common rapl bluetooth mt76_connac_lib mt76_usb iTCO_wdt mt76 intel_pmc_bxt jitterentropy_rng iTCO_vendor_support pcspkr virtio_rng sha512_ssse3 mac80211 watchdog binfmt_misc sha512_generic rng_core virtio_balloon libarc4 ctr virtio_console drbg cfg80211 ansi_cprng ecdh_generic rfkill bochs ecc drm_vram_helper drm_ttm_helper ttm drm_kms_helper evdev button joydev sg serio_raw drm fuse efi_pstore configfs qemu_fw_cfg ip_tables x_tables autofs4 ext4 crc16 mbcache jbd2 crc32c_generic dm_mod sd_mod hid_generic t10_pi usbhid hid crc64_rocksoft crc64 sr_mod crc_t10dif cdrom crct10dif_generic virtio_net ahci net_failover libahci xhci_pci virtio_scsi failover libata xhci_hcd crct10dif_pclmul crct10dif_common crc32_pclmul scsi_mod virtio_pci crc32c_intel
[  186.319107]  virtio_pci_legacy_dev usbcore virtio_pci_modern_dev i2c_i801 psmouse scsi_common i2c_smbus lpc_ich virtio usb_common virtio_ring```