morrownr/USB-WiFi #112: Power usage of Alfa adapters

Martin-Buchholz, 2022-09-21:
I found https://github.com/morrownr/USB-WiFi/blob/main/home/Performance_Comparison.md and was surprised to see there the same power usage for AWUS036ACHM and AWUS036ACM - 380mA.

One expects a higher usage for AWUS036ACHM since that is a "MAX Power" adapter while AWUS036ACM is merely "High Power"
(see https://store.rokland.com/collections/wi-fi-usb-adapters/products/alfa-awus036achm-802-11ac-dual-band-high-power-ac1200-mediatek-wifi-usb-adapter)

Also, I see the different value 420mA, not 380mA, at https://github.com/morrownr/USB-WiFi/blob/main/home/iw_list/ALFA_AWUS036ACHM .

Could it be that measuring on a Raspberry Pi forces a (misleading) power limit?

morrownr, 2022-09-22:
Hi @Martin-Buchholz 

Good question.

> Could it be that measuring on a Raspberry Pi forces a (misleading) power limit?

I don't think this is the case but I can't rule it out. One test was completed with a desktop system and the other with a RasPi4B. I was careful to make sure there was very little else pulling power from the usb subsystems when tests were conducted. The tests were conducted in different areas with different signal strengths. I was consistent as far as things on the same document go.

One factor that I think that can beef up the numbers of the Alfa ACM is that it is an AC1200 adapter whereas the ACHM is only an AC600 adapter. During the tests, the ACM would have been pumping a much greater volume of packets.

There is more that goes into range than "power." Remember that range has to be two way to maximize throughput. Having expensive, sensitive antennas to go along with a powerful amp is needed and the ACHM has both. The ACHM still amazes me with its range. Some folks want to go for the fastest thing going but the ACHM can sustain 200 Mb/s all day long and most homes and many business do not have a 200 Mb/s internet speed.

Also, related to the power consumption test, I use an inline meter and iperf3 set to max out the transfer capability on a clean DFS channel. While it may seem like 380 or 420 is a solid steady reading, it is not. The readings bounces up and down and I watch for a couple of minutes and more or less do an average in my head and then round it off to the nearest 10.

I have a new adapter based on the new mt7921au chipset. I would like to test it but have been unable to do so as the adapter will not function unless plugged into a port on the system. I think Comfast miscalculated something. I expect better from Alfa so we will see when they start releasing their adapters. My best guess right now is that the mt7921au chipset is using around 500 mA. It is very fast and the in-kernel mt7921u driver is impressive. In AC mode, clean DFS channel, 80 MHz chnnel width, it can hold 650 Mb/s when connected to my main router. I can't test AX or 6 GHz channels yet because this is the only thing I have that does AX or 6 GHZ channels.

I hope that helps. Let me know if you want me to do any additional testing with the Alfa adapters.

Nick

Martin-Buchholz, 2022-09-22:
Thanks, Nick.

On Wed, Sep 21, 2022 at 5:11 PM Nick ***@***.***> wrote:

> One factor that I think that can beef up the numbers of the Alfa ACM is
> that it is an AC1200 adapter whereas the ACHM is only an AC600 adapter.
> During the tests, the ACM would have been pumping a [...]
I have ordered the ACHM.  I also value range, reliability, low power.  And
I also have some Raspberry Pis.

> Also, related to the power consumption test, I use an inline meter and
> iperf3 set to max out the transfer capability on a clean DFS channel. While
> it may seem like 380 or 420 is a solid steady [...]
Benchmarking is tricky, even more so when concurrency and environmental
dependence are in play.

Consider mentioning in your docs that you are actually measuring power
using a USB inline power meter.

If I wanted to stress/heat up a wifi adapter, I might try putting all
machines next to each other so that signal strength is 100, and then run
iperf3 in *both* directions at the same time so that both receiver and
transmitter are busy with work.

---

Consider adding to your technical detail pages the MaxPower reported by the
device itself, in the output from `sudo lsusb -v`
That should be easy to do.

---

So many ways to investigate "power usage"!
- lsusb -v: MaxPower
- iwconfig: Tx-Power
- actual hardware power measuring device along with many variations of
iperf3-based workloads

I have a new adapter based on the new mt7921au chipset. I would like to
> test it but have been unable to do so as the adapter will not function
> unless plugged into a port on the system. I think
>

Did you mean "plugged into a USB 2 port"?

> Comfast miscalculated something. I expect better from Alfa so we will see
> when they start releasing their adapters. My best guess right now is that
> the mt7921au chipset is using around 500 mA. It is [...]

I expect to order an Alfa mt7921au adapter once you have blessed it!

> Message ID: ***@***.***>
>

Martin-Buchholz, 2022-09-22:
Hmmm .... gmail and/or github seems to have added unwanted quoting to parts of my message :

Consider adding to your technical detail pages the MaxPower reported by the
device itself, in the output from `sudo lsusb -v`
That should be easy to do.

So many ways to investigate "power usage"!
- lsusb -v: MaxPower
- iwconfig: Tx-Power
- actual hardware power measuring device along with many variations of
iperf3-based workloads

Martin-Buchholz, 2022-09-22:
Today I used iperf3 for the first time.  I learned:

- iperf3 offers `--reverse --bidir --parallel` to allow experiment variation without having to manage multiple iperf3 processes
- performance is extremely variable, at least in my own noisy wifi environment
- need to add port 5201 to firewall rules

morrownr, 2022-09-22:
> Did you mean "plugged into a USB 2 port"?

I guess that was not clear. What I mean is that the CF-951AX adapter will only work if plugged into a port on a system. It will not work with extension cables and it will not work with my inline power meter. I've seen issue with extension cables before, especially long extension cables. I have tested this adapter with all USB3 extension cables that I have and it will not work with any. I am looking forward to Alfa releasing adapters based on the mt7921au. Their quality and performance has historically been a step above the others. Every single Alfa adapter I have ever bought still works and works well. Most makers will kick out a periodic decent adapter but they also kick out some duds. I'm 

> I have ordered the ACHM.  I also value range, reliability, low power.  And I also have some Raspberry Pis.

My ACHM has been very stable on RasPi's. The cable allows you to position it for best reception. 

> Consider mentioning in your docs that you are actually measuring power using a USB inline power meter.

I guess I should make that clear. Some folks may think the readings are somehow related to txpw. What I am doing is trying to make sure folks know about how much electricity adapters use. The RasPi's usb subsystems are not capable of providing the standard amount of power... 2 x 500 mA (usb2) + 2 x 900 mA (usb3) = 2800 mA. The Pi subsystem is only capable of 1200 mA so having usb adapters that use less power is a good thing.

Thanks for the advice.

Nick

morrownr, 2022-09-23:
> in the real world there is always signal loss, so what happens depends on engineering tolerances.

Indeed. I'm fairly sure that the engineering tolerances of the CF-951AX are not allowing my CF-951AX to work with extension cables. I've seen this at times will certain cables and certain adapters on certain systems but I have never seen one that will simply not work with any cable I have. I guess the low price of this adapter should be a warning.

> There's a reason the low-end Kingston A400 is the most popular Raspberry Pi SSD.

I provide a lot of support for the Realtek drivers that I maintain here. It is astounding the amount of problems that Pi users have due to the limited overall power capability in the Pi4B USB subsystem.

User: my adapter does not work in my Pi4B.
Me: Tell me everything you have plugged into a USB port.
User: A Sandisk Extreme SSD, a rtl8814au based wifi adapter, ... and a bunch of other stuff.
Me: Unplug everything but the wifi adapter and reboot.
User: Hey, the adapter works.
Me: Yup, you want to know why?

Martin-Buchholz, 2022-09-23:
On Wed, Sep 21, 2022 at 5:11 PM Nick ***@***.***> wrote:

> One factor that I think that can beef up the numbers of the Alfa ACM is
> that it is an AC1200 adapter whereas the ACHM is only an AC600 adapter.
> During the tests, the ACM would have been pumping a much greater volume of
> packets.
>

There seems to be some confusion about the specs of the ACHM.  Alfa's own
page clearly claims it's AC1200, not AC600.
https://www.alfa.com.tw/products/awus036achm?variant=39477247082568
while rokland's page is ambiguous
https://store.rokland.com/products/alfa-awus036achm-802-11ac-dual-band-high-power-ac1200-mediatek-wifi-usb-adapter
Conversely, Alfa's page says "USB 2.0 plug and play" while rokland's says
"USB 3.0 compliant"
I'm guessing there was a quiet engineering change at Alfa, perhaps to
upgrade from USB 2 to USB 3, but the documentation hasn't caught up
(updating images is too much work!)

> Message ID: ***@***.***>
>

morrownr, 2022-09-23:
> There seems to be some confusion about the specs of the ACHM.  Alfa's own page clearly claims it's AC1200, not AC600.

Let me bring an end to the confusion: The ACHM is an AC600 level adapter. It is based on the mt7610u chipset.

I had not noticed the errors. Maybe you should make contact and ask them to clear things up.

Martin-Buchholz, 2022-09-24:
On Fri, Sep 23, 2022 at 12:44 PM Nick ***@***.***> wrote:

> There seems to be some confusion about the specs of the ACHM. Alfa's own
> page clearly claims it's AC1200, not AC600.
>
> Let me bring an end to the confusion: The ACHM is an AC600 level adapter.
> It is based on the mt7610u chipset.
>

You're right.  I just noticed the label on the device itself says "AC600".

> Message ID: ***@***.***>
>