morrownr/USB-WiFi #113: Netgear A8000 AX3000 USB Adapter - did not see this coming...

morrownr, 2022-09-23:
Link: https://dongknows.com/netgear-a8000-ax3000-usb-adapter-is-here/

The above links states that this adapter is based on the Intel AX210 Wi-Fi 6E chipset. If true, this is a surprise to me. I cannot recall Intel ever having supplied a USB chipset to the market. If Intel has merged USB support for this chipset into the kernel, I missed it. My bet is on this adapter using a mt7921au chipset and the Netgear rep simply does not know what he is talking about.

If anyone has any additional information, please post.

Nick

amisix, 2022-09-23:
Looks like it's $90 to pre-order one (if they were available).
[https://www.netgear.com/home/wifi/adapters/a8000/](https://www.netgear.com/home/wifi/adapters/a8000/)

morrownr, 2022-09-23:
I saw that price. I usually use a scale like the following when writing reviews of adapters:

Low price <= $25 USD
Mid price > $25 and <= $50
High price > $50 and <= $75
Too high > $75

I own a Netgear A6210. It has a mt7612u chipset and it not a bad adapter for certain use cases but Netscape did not do that good of  job with it. It uses what appears to be the same case as the new A8000. My experience with the A6210 is that its range is somewhat limited and it runs warmer than any other mt7612u based adapters that i have experience with. You won't see me getting in line to get this adapter even if it does turn out to have a mt7921au chipset. I would consider paying that much for a High Power Alfa adapter but I know Alfa uses good amps and antennas, I'm not so sure about Netgear.

amisix, 2022-09-24:
Yeah, I agree with your price points from a noobs perspective. I can't justify a $90 adapter for my i7 laptop let alone for a $45 Raspberry Pi. They're calling it the world's 1st Wifi6E USB adapter.. if only they had accounted for Chinese adapters, linux, and some creative hostapd hacking..

closed 2022-11-22 by morrownr