morrownr/USB-WiFi #115: (Hot) New Firmware files available for the new mt7921au chipset as of less than 24 hours ago...

morrownr, 2022-09-24:
For those testing adapters with the new mt7921au WiFi 6e chipset:

New firmware files as of less than 24 hours ago:

525870 Sep 24 00:55 BT_RAM_CODE_MT7961_1_2_hdr.bin
92192 Sep 24 00:49 WIFI_MT7961_patch_mcu_1_2_hdr.bin
716404 Sep 24 00:54 WIFI_RAM_CODE_MT7961_1.bin

Instructions found in Main Menu item 8.

leezu, 2022-09-27:
Unfortunately the updated firmware does not fix the AP hanging issues previously discussed at https://github.com/morrownr/USB-WiFi/issues/87#issuecomment-1207454600

```
[Tue Sep 27 13:33:42 2022] mt7921u 2-1:1.3: Message 00020002 (seq 1) timeout
[Tue Sep 27 13:33:42 2022] mt7921u 2-1:1.3: chip reset
[Tue Sep 27 13:33:42 2022] mt7921u 2-1:1.3: timed out waiting for pending tx
[Tue Sep 27 13:33:42 2022] ------------[ cut here ]------------
[Tue Sep 27 13:33:42 2022] WARNING: CPU: 3 PID: 15355 at kernel/kthread.c:659 kthread_park+0xb4/0xcc
[Tue Sep 27 13:33:42 2022] Modules linked in: ctr aes_arm64 ccm iptable_nat xt_MASQUERADE xt_mark nft_chain_nat ip6table_nat nf_nat tun mt7921u mt7921_common mt76_connac_lib mt76_usb mt76 mac80211 btusb btrtl btintel btbcm bluetooth libarc4 ecdh_generic ecc sg brcmfmac brcmutil vc4 bcm2835_codec(C) bcm2835_isp(C) bcm2835_v4l2(C) cfg80211 bcm2835_mmal_vchiq(C) rpivid_hevc(C) drm_display_helper v3d videobuf2_vmalloc v4l2_mem2mem cec videobuf2_dma_contig gpu_sched videobuf2_memops drm_cma_helper videobuf2_v4l2 drm_kms_helper videobuf2_common drm_shmem_helper snd_soc_core videodev rfkill raspberrypi_hwmon vc_sm_cma(C) mc snd_compress snd_pcm_dmaengine snd_pcm snd_timer snd syscopyarea sysfillrect sysimgblt fb_sys_fops uio_pdrv_genirq nvmem_rmem uio ip6t_REJECT nf_reject_ipv6 xt_hl ip6_tables ip6t_rt ipt_REJECT nf_reject_ipv4 xt_comment xt_multiport nft_limit xt_limit xt_addrtype xt_tcpudp xt_conntrack nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 nft_compat nf_tables nfnetlink drm fuse
[Tue Sep 27 13:33:42 2022]  drm_panel_orientation_quirks backlight ip_tables x_tables ipv6
[Tue Sep 27 13:33:42 2022] CPU: 3 PID: 15355 Comm: kworker/u8:2 Tainted: G         C        5.19.0-v8+ #3
[Tue Sep 27 13:33:42 2022] Hardware name: Raspberry Pi 4 Model B Rev 1.4 (DT)
[Tue Sep 27 13:33:42 2022] Workqueue: mt76 mt7921_mac_reset_work [mt7921_common]
[Tue Sep 27 13:33:42 2022] pstate: 60000005 (nZCv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[Tue Sep 27 13:33:42 2022] pc : kthread_park+0xb4/0xcc
[Tue Sep 27 13:33:42 2022] lr : mt76u_stop_tx+0x28c/0x34c [mt76_usb]
[Tue Sep 27 13:33:42 2022] sp : ffffffc009253c60
[Tue Sep 27 13:33:42 2022] x29: ffffffc009253c60 x28: 0000000000000000 x27: 0000000000000000
[Tue Sep 27 13:33:42 2022] x26: 0000000000000000 x25: ffffff8047ffd980 x24: ffffff804ad920e0
[Tue Sep 27 13:33:42 2022] x23: ffffff804ad944a0 x22: ffffff804ad95ca0 x21: ffffff804ad920c0
[Tue Sep 27 13:33:42 2022] x20: ffffff804656ec00 x19: ffffff8044bc0000 x18: 0000000000000000
[Tue Sep 27 13:33:42 2022] x17: 0000000000000001 x16: ffffffec592b3eb4 x15: 0029642c5919596e
[Tue Sep 27 13:33:42 2022] x14: 0000000000000000 x13: 0000000000000017 x12: ffffff80fb7beec0
[Tue Sep 27 13:33:42 2022] x11: 0000000000000017 x10: 0000000000001a80 x9 : ffffffebfc7bba40
[Tue Sep 27 13:33:42 2022] x8 : ffffff804284d8e0 x7 : 0000000000000001 x6 : ffffffec5a709480
[Tue Sep 27 13:33:42 2022] x5 : ffffffec5a5a9000 x4 : ffffffec5a5a9268 x3 : 0000000000002400
[Tue Sep 27 13:33:42 2022] x2 : 0000000000000000 x1 : 0000000000000000 x0 : 0000000000000004
[Tue Sep 27 13:33:42 2022] Call trace:
[Tue Sep 27 13:33:42 2022]  kthread_park+0xb4/0xcc
[Tue Sep 27 13:33:42 2022]  mt76u_stop_tx+0x28c/0x34c [mt76_usb]
[Tue Sep 27 13:33:42 2022]  mt7921u_mac_reset+0x8c/0x2f4 [mt7921u]
[Tue Sep 27 13:33:42 2022]  mt7921_mac_reset_work+0xb8/0xe7c [mt7921_common]
[Tue Sep 27 13:33:42 2022]  process_one_work+0x1dc/0x450
[Tue Sep 27 13:33:42 2022]  worker_thread+0x154/0x450
[Tue Sep 27 13:33:42 2022]  kthread+0x100/0x110
[Tue Sep 27 13:33:42 2022]  ret_from_fork+0x10/0x20
[Tue Sep 27 13:33:42 2022] ---[ end trace 0000000000000000 ]---
[Tue Sep 27 13:33:42 2022] mt7921u 2-1:1.3: HW/SW Version: 0x8a108a10, Build Time: 20220908210919a

[Tue Sep 27 13:33:42 2022] mt7921u 2-1:1.3: WM Firmware Version: ____010000, Build Time: 20220908211021
```

morrownr, 2022-09-27:
Hi @leezu 

Do you mind reporting this in issue #107 ? I'm trying to start a running list of issues for reference and reporting. I'll take what you put and start a Bug 5. It would be good if you can provide more information such as all hardware involved, especially everything plugged into a USB port and what you were doing with software.

Also, have you been able to duplicate this on non-RasPi4B hardware? The reason I ask is that I have wrestled with the RasPi4B USB subsystem on many occasions. My Pi4B is updated to kernel 6.0 and the firmware is up to date as well so let me see what happens here in AP mode. Was this with OpenWRT or the in-kernel driver?

leezu, 2022-09-28:
Hi @morrownr, it's possible that the bug is related to the RasPi4B USB subsystem. I've now updated to [kernel 6.0.0-rc7-v8+](https://github.com/raspberrypi/linux/commit/3fb5ca89c844eb9fda924a5e76ffab9e8c068675) and unplugged the USB cable connecting the RasPi4B to my monitor's inbuilt USB Hub. Besides the USB Hub, a SATA to USB 3.0 adapter is connected. Whenever I unplug an USB-C Displaylink device from my monitor, the monitor would switch it's USB Hub to be exposed to the RasPi4B instead of the now gone Laptop and thus make available keyboard, mouse, webcam etc devices connected to the monitor's USB Hub. Let's take a look if the issue reproduces in this reduced-USB-subsystem exposure setup. The frequency of occurrence is lower thanks to the previous firmware updates and the fixes in the recent kernels. If it occurs again, I'll report the issue in https://github.com/morrownr/USB-WiFi/issues/107

black-jmyntrn, 2022-10-01:
Model	Raspberry Pi 3 Model B Plus Rev 1.3
Architecture	ARMv7 Processor rev 4 (v7l)
Target Platform	bcm27xx/bcm2709
Firmware Version	OpenWrt 22.03.0 r19685-512e76967f / LuCI openwrt-22.03 branch git-22.245.77528-487e58a
Kernel Version	5.10.138

hanging issue or something... it has a mind of its own when it wants to work or not... 

i actually updated my blog earlier today on this with screenshots. 
https://black.jmyntrn.com/2022/09/17/comfast-cf-953ax-with-openwrt-on-a-raspberry-pi-3b/

morrownr, 2022-10-02:
Hi @black-jmyntrn 

> using the kmod-mt7921u driver 

Did you also install the kmod-mt7921e driver? The reason I ask is that the firmware files are in that package instead of being in a common package. Of course, you could have installed them manually.

Another person and I were testing CF-95x-AX adapters with RasPis and OpenWRT a couple of weeks ago. I have a CF-951AX and the other person has a CF-953AX. We were both using RasPi4Bs with OpenWRT 22.03 (initial stable release). Our results were uneven and we finally decided to wait for 22.03.1 to try again.

What I am doing right now with My RasPi4B is setting up to do AP mode testing using a fresh install of the RasPiOS 64 bit. I've been making a guide on how to upgrade the current RasPiOS to kernel 5.19.11. I can mod the guide to add RasPi3B support if you are interested.

bjlockie, 2022-10-03:
> Does this belong here?
> blueman-project/blueman#1868[https://github.com/blueman-project/blueman/issues/1868]

Are usb devices supposed to support Bluetooth?

LargePrime, 2022-10-03:
> Are usb devices supposed to support Bluetooth? Oct. 3, 2022 11:33:09 LargePrime ***@***.***>:

https://github.com/morrownr/USB-WiFi/discussions/88#discussioncomment-3213403
"MT7921AU supports Wi-Fi 6E and Bluetooth 5.2..."
Seems so?

morrownr, 2022-10-03:
> > Are usb devices supposed to support Bluetooth? Oct. 3, 2022 11:33:09 LargePrime _**@**_.***>:
> 
> [#88 (comment)](https://github.com/morrownr/USB-WiFi/discussions/88#discussioncomment-3213403) "MT7921AU supports Wi-Fi 6E and Bluetooth 5.2..." Seems so?

Since new wifi/BT chipsets are usually seen in laptops first and since USB WiFi adapters are a niche market from the big boy's perspective, we usually see new chipsets supporting wifi and BT. Often the BT is left out or turned off for USB WiFi adapters. However, you will see wifi only adapters, BT only adapters and adapters that have both. The first two adapters on the market that use the new mt7921au chipset support BT as well as wifi.

LargePrime, 2022-10-03:
> The first two adapters on the market that use the new mt7921au chipset support BT as well as wifi.

So BT should be working, right? Is it known to work?

morrownr, 2022-10-03:
@LargePrime 

> So BT should be working, right?

It should be working. I tested on Linux Mint 21. BT wasn't working even thought I had installed kernel 6.0 rc? and I had installed the Sep. firmware update. It simply was not working. I wrote it up as Bug 1 in issue 107. I decided to hold off as Mint was seeing a lot of BT issues that had nothing to do with this chipset/driver. I've been too busy to investigate further.

@deren stated in issue 107 that BT with his CF-953AX was working well on Ubuntu 22.04. What you can do is write up the specific details in issue 107. As I see more reports, maybe I can help determine what distros and what problems are happening and what the causes are.

@LargePrime 

It is very important to include distro name and version as well as kernel version and what firmware you are using in your reports. Is it possible you could burn Ubuntu 22.04 to a flash driver, boot it, run it and see if BT is working.

Right now I am working hard to determine how best to test WiFi 6 and WiFi 6e. I spent a good part of this day compiling kernel 5.19.11 on the RasPiOS 64 bit. I upgraded the firmware to the Sep. release. I installed hostapd and it was working wonderfully with my CF-051AX in AC AP mode 5 GHz. I then decided to try AX. It was a no go. The version of hostapd that RasPiOS currently uses is old. I had to download, compile and replace hostapd. Now AX comes up but I see problems. Next I will be downloading wpa_supplicant source and compiling it as the version of wpa_supplicant in  RasPiOS is also old. Some distros have old versions of Network Manager. It will cause problems trying to use AX and 6 GHz. 

We are in a situation where distros are all over the place with what they have upgraded or not. I saw the same type of issues with WPA3. Ubuntu is doing a pretty good job. Others, not so much. Even though Mint is based on Ubuntu, it has its own problems. I've used Mint as my primary dev os for the last 4 years but the recent release is driving me crazy. I have little tollerance for bugs. I have now been testing Xubuntu on the side and it is far more stable. I will be making the switch soon. The soon to be released Ubuntu and Xubuntu 22.10 will have the 5.19 kernel and new firmware as well as all of the utilities and apps that need to be updated. I don't use Ubuntu because I do not like Gnome... in case you were wondering but later this months, it you want a distro that is likely to work very well with adapters that have the mt7921au chipset, use Ubuntu or one of the official variations.

bjlockie, 2022-10-03:
How do you know what firmware version is being used?

Can you you use a more recent firmware with an older kernel?

morrownr, 2022-10-03:
> How do you know what firmware version is being used?

I check the file sizes to know if the most recent files that are available have different file sizes than the files in the distro.

https://github.com/morrownr/USB-WiFi

Select Main Menu item 8.

 Can you you use a more recent firmware with an older kernel?

Yes but let's say you use the Sep. firmware files and kernel 5.15. Your firmware is ready to the support the mt7921u driver but the 5.15 kernel won't have the mt7921u driver. You need to get up to at least kernel 5.18 for managed and monitor mode and att least kernel 5.19 if you also want AP mode.

As distros upgrade to kernel 5.19 or later and the latest firmware... and the various other parts required for WiFi 6 and WiFi 6e, these headaches will start going away. We are on the bleeding edge. My CF-951AX is working very well as a very fast WiFi 5 (AC) adapter in everything I bring up to kernel 5.19 and Sep. firmware. Getting on up to WiFi 6 or 6e is still a challenge. My main router runs OpenWRT 22.03 which includes the driver and firmware. On OpenWRT, AX works well. Luci, the graphical interface can't handle 6 Ghz yet so you have to get dirty to set 6 GHz and depending on where you are located, AP mode may require a trip to France.

FWIW, other platorms are having problems with WiFi 6 and especially 6e. These are MAJOR changes that are very complex. In the list that shows the adapters that are supported with in-kernel drivers, I have tried to make it clear that users looking to but an adapter right now, if the are not technically knowledgable, they should skip to the AC1200/1300 section as the products and drivers and firmware are mature.

black-jmyntrn, 2022-11-06:
> Hi @black-jmyntrn
>
>
> Did you also install the kmod-mt7921e driver? The reason I ask is that the firmware files are in that package instead of being in a common package. Of course, you could have [...]

ahhh so use 64 bit huh? okay, will compile one tonight, but will try the firmware selector first. I used it last night to get the latest and added packages and built custom, it worked out of the box yet still buggy... thats because 22.03.02 from the online tool has kernal 5.10.146.... will work to get one with 5.19 like you mentioned. 

FWIW, I bought an AX210 on amazon for cheap and upgraded an old thinkpad t450s to get wifi6 and thats how im testing along with a note s20 ultra 5g.

black-jmyntrn, 2022-11-07:
> Did you also install the kmod-mt7921e driver? The reason I ask is that the firmware files are in that package instead of being in a common package. Of course, you could have installed them manually.

funny you ask, in the begining i tried each combination of mt driver from a page of yours i found for all usb devices youve covered, no success. each time new build... finally when I added all mt drivers it worked, I saw a command i cant recall which(this all still new to me) that showed kmod-mt7921u and the mt7x2x was causing san issue so only went with kmod-mt7921u and buggy. the kmod-mt7921e i started added and things magically got better but it was like these end days of covid... is it over, is it working? are things reliable... lol and here we are. 

I started with rpi3b+ the rpi4 was my main server and rpi3b+ was the main router in my multi wan carLab. I feel like I remember the rpi3b+ working the most stable with the 935, when I finally moved all components to the rpi4 taking a dive makes me glad I'm not doing this on the road and home not stressing about getting internet and good wifi back running for my overlanding prius. heh id be cranky pants if i was doing this driving through canada to alaska and having to do rebuilds on one of the machines in here. that said im planning a supercomputer build using mini gaming pcs with the rayzen9 5900 and cant wait to see how long it takes to make a custom openwrt firmware file... my positivity thinks less than 10 minutes easily!

morrownr, 2022-11-07:
> things magically got better but it was like these end days of covid... is it over, is it working? are things reliable... lol and here we are.

I appreciate that the OpenWRT devs backported the mt7921u driver but not getting the firmware in a package that is loaded when you install the driver is puzzling because it is like no driver was installed. A bug has been filed but has not been acted on yet.

Growing pains. I'm pretty sure the mt7921u driver for the mt7921au chipset is the first time in the history of usb wifi drivers in the Linux kernel where the driver was in the mainline kernel before the hardware was available on the market. It is astounding that it works well for the most part given that two major changes to wifi are in progress at the same time... WiFi 6 and WiFi 6e. Which is less complicate? Building a new spaceship to go to Mars or this WiFi stuff? Tough call. As time passes, we will get the rough edges worked out.

black-jmyntrn, 2022-11-12:
hi @morrownr 
 
where can i download 5.19 kernel for openwrt? all i see is 5.15 when I try to make a build file myself.

morrownr, 2022-11-14:
> where can i download 5.19 kernel for openwrt? all i see is 5.15 when I try to make a build file myself.

5.15 is as far as the OpenWRT devs have gone at this point so downloading a 5.19 kernel is not going to happen yet.

Let me backup for a moment. What are you trying to do that requires 5.19?