morrownr/USB-WiFi #121: MediaTek mt7662u: AVM FRITZ!WLAN WiFi USB-A 3.0 Stick AC 860 2.4GHz/5GHz

mendel5, 2022-09-30:
I would like to make you aware of an USB WiFi stick that has a MediaTek `mt7662u` chipset:
- AVM FRITZ!WLAN USB Stick AC 860, 2.4GHz/5GHz WLAN, USB-A 3.0

This USB WiFi stick is popular in Europe and is manufactured by the German company AVM.

Here are some relevant links:
- Technical specifications by the manufacturer: https://en.avm.de/products/mesh-wi-fi/fritzwlan-stick-ac-860/technical-specifications/
- Technical specifications by a price comaprison web site: https://geizhals.eu/avm-fritz-wlan-usb-stick-ac-860-20002687-20002724-a1082383.html
- Linux Wireless Mailing List with a reference to the "Avm FRITZ!WLAN AC860": https://patchwork.kernel.org/project/linux-wireless/patch/20180701141231.25323-20-lorenzo.bianconi@redhat.com/
- AVM GmbH (manufacturer): https://en.wikipedia.org/wiki/AVM_GmbH

morrownr, 2022-10-02:
Hi @mendel5 

I appreciate the information. Are you suggesting that this adapter be added to the list in Main Menu item 2?  If so...

Can you confirm that it works well?
Can you confirm which driver that it uses? mt7612u?
Can you provide at least 2 links to online dealers where users can purchase this adapter?
Can you do a review to include with the listing?

Thanks for your help.

Nick

mendel5, 2022-10-02:
Hi @morrownr

> Are you suggesting that this adapter be added to the list in Main Menu item 2?

Yes, I think we could add this adapter.

Regarding your questions:
- Can you confirm that it works well? - I have this adapter and for me it works well with Linux.
- Can you confirm which driver that it uses? mt7612u? - Yes, it uses the `mt76x2u` driver which I think is identical for both the `mt7612u` and the `mt7662u`. For a reference please see the kernel.org link I posted in the first comment.
- Can you provide at least 2 links to online dealers where users can purchase this adapter? - Yes, I can add the links. However, the online dealers would be located in Europe because this adapter is not directly sold in the US (as far as I know).
- Can you do a review to include with the listing? - Yes, I can do that as well. Unfortunately I don't have this adapter with me right now because it is at a different location. Providing the information might take a couple months because it depends on when I will be at the other location again...

morrownr, 2022-10-02:
Hi @mendel5 

> However, the online dealers would be located in Europe because this adapter is not directly sold in the US (as far as I know).

This is not a US or North American site. This is a worldwide site. There are many links to sellers around the world in the list. 

Can you go ahead and send some links to the adapter on sellers web sites?

mendel5, 2022-10-02:
> Can you go ahead and send some links to the adapter on sellers web sites?

Sure, @morrownr !
- https://geizhals.eu/avm-fritz-wlan-usb-stick-ac-860-20002687-20002724-a1082383.html - Geizhals is a European price comparison website
- https://www.amazon.de/dp/B017BD21R8/ - Amazon Germany (de)
- https://www.amazon.fr/dp/B019E28V4G/ - Amazon France (fr)
- https://www.amazon.it/dp/B019E28V4G/ - Amazon Italy (it)

morrownr, 2022-10-02:
Hi again, @mendel5 

I have added a section for this adapter in the AC1200/1300 section:

https://github.com/morrownr/USB-WiFi/blob/main/home/USB_WiFi_Adapters_that_are_supported_with_Linux_in-kernel_drivers.md

Please take a look and make suggestions. I had trouble with the French link. Hopefully you can do a review to be posted at some point. I need somebody in Europe to mail me one of these for testing.. They look cool.

Thanks for the info.

Nick

mendel5, 2022-10-02:
Hi @morrownr 

> I have added a section for this adapter in the AC1200/1300 section

Thank you!

> Please take a look and make suggestions

For the AVM Fritz AC 860 it says `Note: This is a single-state adapter`. May I ask how you know this?

morrownr, 2022-10-02:
> For the AVM Fritz AC 860 it says Note: This is a single-state adapter. May I ask how you know this?

I did a little reading and could find no evidence that it is multistate. Plus it has been on the market for a few years. Once you have access to your adapter, I can show you how to test and confirm.

Nick

mendel5, 2022-10-03:
Hi @morrownr 

> Once you have access to your adapter, I can show you how to test and confirm.

Alright, I'm looking forward to it 👍

morrownr, 2022-10-27:
> would it be possible to add the Mediatek MT7662u chipset to the list?

Done

mendel5, 2022-10-28:
> Done

Thank you, @morrownr !

morrownr, 2022-10-28:
@mendel5 

I'm pretty sure that the information that I added to the chipset list is correct, but there is an issue you should be aware of:

While the mt7662u chipset is capable of USB3, it seems that most or all implementations that I can find only use USB2. Also, I can only find 2 external usb wifi adapters that have been built with the chipset. The Fritz and...

https://www.benq.com/en-us/business/signage/wdr02u.html

It also appears that the only difference between the mt7662u and mt7612u is that the mt7662u included bluetooth support. The mt7662u chipset has been used in several products but mostly in internal use cases, not for external usb wifi adapters. The mt7612u seems to have been popular for external usb wifi use cases.

You are welcome to use any of this info when you do a review of the Fritz!. Some folks might find it to be interesting and you should probably check the Fritz! to see if it is usb2 or usb3 as people will be interested in that.

Nick

ralaud, 2022-10-30:
I would like to buy this AVM FRITZ!WLAN USB Stick AC 860, but I fear there will be no driver updates in near future. Due to it was released in 2014 I think. 

I have to use it on linux and windows, so I need a Wlan stick which also has good driver for windows. 

A short review would be really helpful for me.

@Off-topic
Does some one know if a AX version is planned from AVM?

morrownr, 2022-10-30:
Hi @ralaud 

> I would like to buy this AVM FRITZ!WLAN USB Stick AC 860, but I fear there will be no driver updates in near future. Due to it was released in 2014 I think.

The mt7662u is the version of the chipset that includes bluetooth. The mt7612u is the version of the chipset without bluetooth. So both use the same driver for wifi. The mt76x2u is AC1200 and the driver is in the Linux kernel so there is no driver to install. In-kernel driver meet Linux Wireless Standards and they tend to be maintained in the kernel for a LONG time. Keep in mind that Linus is just now starting to talk about removing Intel 486 support from the kernel. Yes, that 486 CPU that was popular 32 years ago.

> I have to use it on linux and windows, so I need a Wlan stick which also has good driver for windows.

I generally don't have Windows available but I got a new little laptop a few months ago and I perserved the Windows 11 on it just in case you were to come along and ask this question. I put a mt7612u adapter in one of the usb ports and poof, it was active and ready to go. The Windows driver is in the Linux kernel. No need to install anything. I can't say how long the driver will stay in Windows but it is there now.

> A short review would be really helpful for me.

Hey @mendel5 , where is that review?

> Does some one know if a AX version is planned from AVM?

You might contact AVM and ask if they have plans to make a usb adapter with either the mt7921au or mt7922u chipsets. Those are both WiFi 6e and the driver is already in the Linux kernel. Encourage them to do so if they are not sure.

Nick

mendel5, 2022-10-31:
@morrownr @ralaud 

My AVM AC 860 adapter is currently at a friends place and I will only get there by Christmas. So unfortunately it will take some time until I can test it.

> While the mt7662u chipset is capable of USB3, it seems that most or all implementations that I can find only use USB2.

If I remember correctly, the USB connector of my AVM AC 860 has a blue color. This usually indicates that it uses USB 3.0. But of course I can test this in more detail.

> A short review would be really helpful for me.

Unfortunately there is no review yet. I will post it around Christmas time. Please excuse the delay.

> You might contact AVM and ask if they have plans to make a usb adapter with either the mt7921au or mt7922u chipsets. Those are both WiFi 6e and the driver is already in the Linux kernel. Encourage them to do so if they are not sure.

Yes, it's a very good idea to contact AVM about this. AVM has a speical web page for contacting them about these questions. You can find it here:
* https://avm.de/kontakt/anregungen-zu-fritz-produkten/

ralaud, 2022-11-02:
Hi, I asked AVM if they are planning a new USB wifi stick.
Their answer was this:

> Good day,

> Thank you for your request to AVM Support.

> You had asked:

> > Is a new WLAN USB stick with Wifi 6 (AX) support already being planned?

> There is currently no information on this.

> Unfortunately, we in Support do not know anything about future products, functions and extensions of product development or about dates for their release. Product management reserves the right to publish such information.

> _A notice_:
> _You can be informed about new products, functions and updates via newsletter. Here you can sign up:_

> https://avm.de/service/newsletter/newsletter-anmeldung/

> Kind regards from Berlin

AVM has already updated all of their router series for at least wifi 6 standard. But sadly it seems like they are not planning to update their USB wifi stick.

mendel5, 2022-12-23:
Hi @morrownr , I now have my AVM AC 860 adapter. Currently I'm in the process of testing it.

A question regarding the tests:
In the file [NETGEAR_A6210.txt](https://github.com/morrownr/USB-WiFi/blob/main/home/iw_list/NETGEAR_A6210.txt) it says at the end:
```
usb-modeswitch not required. This is a single-state device.

WPA-3 SAE: yes

Power requirement: Heavy load: ~420 mA
```

How can I test the AVM AC 860 adapter for these three properties?

morrownr, 2022-12-23:
Hi @mendel5 

Good to see you.

> usb-modeswitch not required. This is a single-state device.

To prove that the adapter is a single-state adapter, deactivate usb_modeswitch as follows and reboot. If the adapter shows then it is single-state:

```
How to deactivate usb_modeswitch

$ sudo nano /etc/usb_modeswitch.conf

Change

DisableSwitching=0

to

DisableSwitching=1

then reboot
```

The above info is also available at the bottom of the below link:

https://github.com/morrownr/USB-WiFi/blob/main/home/How_to_Modeswitch.md

> WPA-3 SAE: yes

Set your router to WPA3 only, no mixed, and connect to it. If everything works then WPA3 is good. With that said, I know the driver for this adapter does WPA3 so if it doesn't work, the issue is likely that your distros contains an old Network Manager or wpa_supplicant.

> Power requirement: Heavy load: ~420 mA

This is going to be harder. The numbers you see for power usage come from an inline usb power measuring adapter that I have. You would have to buy one. It is okay to leave that blank.

Nick

mendel5, 2022-12-24:
Hi @morrownr
Thank you, I'm happy to hear from you as well!

I have now tested my AVM AC 860 USB WiFi adapter. The results are very interesting:
- I've taken the PCB of the AVM AC 860 out of its casing and the IC marking on the chip clearly states `MT7612UN`. So it's a `mt7612u` (or `mt7612un`?) chip and **not (unlike previously assumed) a `mt7662u`**.
- Regarding usb-modeswitch: It is a multi-state device.
- The color of the USB port on the adapter is blue, indicating that USB 3.0 is supported.
- Unfortunately I was not able to test my AVM AC 860 for WPA3-SAE and how much power it consumes.

More detailed information about my tests can be found here: https://github.com/mendel5/wifi-sandbox . I will later create a pull request to add this information to your repository.

Does this provide all the information you are interested in? Is there anything else I can do?

morrownr, 2022-12-25:
> I've taken the PCB of the AVM AC 860 out of its casing and the IC marking on the chip clearly states MT7612UN

Interesting. I have made the appropriate change on the list. This means the adapter does not do bluetooth. Previous research indicated the 7662u chipset was used so maybe AVM has made a change at some point.

> Regarding usb-modeswitch: It is a multi-state device.

So, it would not work when you disabled usb_modeswitch? That is also interesting. I get the feeling AVM has made some changes to this product at some point. List changed.

> Unfortunately I was not able to test my AVM AC 860 for WPA3-SAE and how much power it consumes.

Every single adapter that I have tested that uses an in-kernel driver works well with WPA3-SAE and I'm sure this adapter is no exception, All mt7612u based adapters that I have tested use betwen 360 and 400 mA when being pressed hard so this adapter probably falls into that category.

> More detailed information about my tests can be found here: https://github.com/mendel5/wifi-sandbox . I will later create a pull request to add this information to your repository.

Looks good and sounds great.

> Is there anything else I can do?

Can you write a short review in your own words to post with the entry in the list? Main Menu item 2 list,

Thanks for everything.

Nick

mendel5, 2022-12-25:
@morrownr 

> I have made the appropriate change on the list.

Thank you!

> Previous research indicated the 7662u chipset was used so maybe AVM has made a change at some point.

As far as I know the AVM AC 860 was released in March 2014. I think I bought mine in 2016.

I know that on other websites, forums, etc. it is stated that the AVM AC 860 uses the `mt7662u` chipset but I'm not sure what the origin of this claim is. Maybe there are versions of the AVM AC 860 that have a `mt7662u` chipset or maybe this information is incorrect and all of them use the `mt7612u` / `mt7612un` chipset.
Further research is needed on this question :smile: 

> So, it would not work when you disabled usb_modeswitch?

When I disabled usb_modeswitch the AVM AC 860 adapter did not work at first. But after about 5 minutes my operating system recognized it anyway and then I could use it.

morrownr, 2022-12-27:
> I know that on other websites, forums, etc. it is stated that the AVM AC 860 uses the mt7662u chipset but I'm not sure what the origin of this claim is...

As far as I can tell, the only difference between the mt7612u and mt7662u chipsets is that the later supports bluetooth. The wifi is the same in both and uses the same driver. There may have been versions at some point that had bluetooth support.

> When I disabled usb_modeswitch the AVM AC 860 adapter did not work at first. But after about 5 minutes my operating system recognized it anyway and then I could use it.

Then the adapter is a single-state adapter. If you disable usb_modeswitch, then all you will get is an icon on the desktop that either shows a CD or a Flash driver if the adapter is a multi-state adapter. A multi-state adapter just does not start working as a wifi adapter without usb_modeswitch telling it to change state. I'll change this again.

mendel5, 2022-12-27:
@morrownr 

> Then the adapter is a single-state adapter. If you disable usb_modeswitch, then all you will get is an icon on the desktop that either shows a CD or a Flash driver if the adapter is a multi-state [...]

Ok, I will test this again.

I have created two pull requests and would appreciate it if you could merge them :)

mendel5, 2022-12-31:
Copy and paste from [here](https://github.com/morrownr/USB-WiFi/pull/172#issuecomment-1368040004):
> By morrownr:
> Sorry for the delay in merging this. The holidays are slowing me down.
> Have you had time to do the review yet?
> Nick

mendel5, 2022-12-31:
Hi @morrownr , no worries.

>  Have you had time to do the review yet?

Yes! You can find my review of the AVM AC 860 here: https://github.com/mendel5/wifi-sandbox/issues/1

Please give me feedback on things that could be improved. After having included your feedback I will add the review through a pull request.

mendel5, 2023-01-02:
- The review of the AVM AC 860 was added here: https://github.com/morrownr/USB-WiFi/commit/7e2ef3bfa8ad468ceafd1e62a6ebc62fdbac22ff
- An update to the description of the AVM AC 860 was made here: https://github.com/morrownr/USB-WiFi/commit/41f6c27b99a1760a9ae96625200966e862c98d09

mendel5, 2023-01-02:
Hi @morrownr , I opened three pull requests:
- https://github.com/morrownr/USB-WiFi/pull/175
- https://github.com/morrownr/USB-WiFi/pull/176
- https://github.com/morrownr/USB-WiFi/pull/177

I'd appreciate it if you could take a look at them. Thanks!

mendel5, 2023-01-06:
Copy and paste from `Fix formatting of AVM AC 860 review`, https://github.com/morrownr/USB-WiFi/pull/174 :

> Thank you for the update. As time passes, if you want to amend your review, feel free to do so.
> 
> I realize I have not responded yet to all of your correspondence but I will. I am simply behind on answering a lot of messages. I am very busy right now.
> 
> Nick

mendel5, 2023-01-06:
> As time passes, if you want to amend your review, feel free to do so.

All right 👍