morrownr/USB-WiFi #123: Add more Github topics to the repository

mendel5, 2022-10-02:
Hi @morrownr , can you add the following Github topics to this repository?
- https://github.com/topics/wifi
- https://github.com/topics/wireless
- https://github.com/topics/linux-kernel
- https://github.com/topics/linux-wifi
- https://github.com/topics/linux-wireless
- https://github.com/topics/usb
- https://github.com/topics/adapter
- https://github.com/topics/hardware
- https://github.com/topics/wifi-hacking

morrownr, 2022-10-03:
I'm not sure what you are asking me to do. I'm all ears.

mendel5, 2022-10-03:
Hi @morrownr 

On the page https://github.com/morrownr/USB-WiFi you can see the section **About** on the top right. Within this section are tags in blue circles, e.g. `linux`. These are the Github topics of this repository.

I think it would be great to add the topics I listed in the first comment. So, the new topics would be `wifi`, `wireless`, etc. The following documentation page by Github explains how to add topics to a repository: https://docs.github.com/en/repositories/managing-your-repositorys-settings-and-features/customizing-your-repository/classifying-your-repository-with-topics

morrownr, 2022-10-03:
Okay. I understand. I added some of your suggestions. Thanks.