morrownr/USB-WiFi #124: News: Netgear A8000 (AX3000) USB WiFi Adapter Annouced...

morrownr, 2022-10-05:
Netgear A8000 AX3000 USB Adapter - chipset: mt7921au
https://dongknows.com/netgear-a8000-ax3000-usb-adapter-is-here/

I am just annoucing this product, not recommending it. The price, $90 USB, seems high to me. For that price, it should clean house and do the dishes in addition to its wifi job.

Nick

bjlockie, 2022-10-05:
Is a 160 MHz wide channel even useful in the real world?
It seems pretty wishful thinking.
Is there a newer Mediatek chipset that will support 160 MHz wide channels?

Even the Comfast ones are priced higher than the Realtek ones.

morrownr, 2022-10-05:
> Is a 160 MHz wide channel even useful in the real world?

Depends on the situation. If you have to ask the question, it is not useful for you.

> Is there a newer Mediatek chipset that will support 160 MHz wide channels?

Yes, the mt7922u. The driver is in the mainline kernel. Actually it is supported by the same driver as the mt7921au... mt7921u.

Here is the Mediatel USB WiFi Roadmap as best I can tell:

MT7921AU
(Filogic 330)
USB 3.0
2x2:2
2.4Gbps
Wi-Fi 6E
WPA3
OFDMA
Zero DFS
ESR
MU-MIMO
1024QAM
HE80
LNA/PA
BT 5.2
0e8d:7961

MT7922U
(Filogic 330)
USB 3.0
2x2:2
2.4Gbps
Wi-Fi 6E
WPA3
OFDMA
Zero DFS
ESR
MU-MIMO
1024QAM
HE160
LNA/PA
BT 5.2
0e8d:7922

MT7938A
(Filogic 380)
USB 3.0
abgn+ac+ax+be
2x2x2:2
6.5Gbps
Wi-Fi 7
WPA3
2.4/5/6GHz
DBDC
BW 360MHz
4K QAM
MLO
BT 5.3 LE
MRC
MRU
HDT
6 nm

> Even the Comfast ones are priced higher than the Realtek ones.

For WiFi 6? I think you need to look again.

pbotros, 2023-01-10:
@morrownr et al, any confirmation that you know of that this Netgear A8000 / AX3000 works with the mt7921u driver?

I have this adapter, but am not getting it to work. I'm running a modified version of OpenWRT 22.03.3 + Xilinx Petalinux on an embedded device (an Ultra96v2 from AVNet), Linux 5.10.161, and have managed to get the mt7921u driver properly built and loaded. Once I added this device's vendor/device ID to the `mt7921/usb.c`, I got Linux to at least try to communicate with it, but then it fails:

```
...
[   48.847295] mt7921u 2-1.2:1.0: WM Firmware Version: ____000000, Build Time: 20210122002730
[   52.015918] mt7921u 2-1.2:1.0: Message 000021ed (seq 3) timeout
[   55.343928] mt7921u 2-1.2:1.0: Message 00000010 (seq 4) timeout
[   55.349870] mt7921u 2-1.2:1.0: Failed to get patch semaphore
root@u96v2-sbc-base-2021-2:~# [   58.671907] mt7921u 2-1.2:1.0: Message 00000010 (seq 5) timeout
[   58.677843] mt7921u 2-1.2:1.0: Failed to get patch semaphore
[   61.999906] mt7921u 2-1.2:1.0: Message 00000010 (seq 6) timeout
[   62.005840] mt7921u 2-1.2:1.0: Failed to get patch semaphore
[   65.327905] mt7921u 2-1.2:1.0: Message 00000010 (seq 7) timeout
[   65.333832] mt7921u 2-1.2:1.0: Failed to get patch semaphore
[   68.655896] mt7921u 2-1.2:1.0: Message 00000010 (seq 8) timeout
[   68.661814] mt7921u 2-1.2:1.0: Failed to get patch semaphore
[   71.983905] mt7921u 2-1.2:1.0: Message 00000010 (seq 9) timeout
[   71.989837] mt7921u 2-1.2:1.0: Failed to get patch semaphore
[   75.311932] mt7921u 2-1.2:1.0: Message 00000010 (seq 10) timeout
[   75.317960] mt7921u 2-1.2:1.0: Failed to get patch semaphore
[   78.639900] mt7921u 2-1.2:1.0: Message 00000010 (seq 11) timeout
[   78.645910] mt7921u 2-1.2:1.0: Failed to get patch semaphore
[   81.967900] mt7921u 2-1.2:1.0: Message 00000010 (seq 12) timeout
[   81.973908] mt7921u 2-1.2:1.0: Failed to get patch semaphore
[   82.097424] mt7921u 2-1.2:1.0: hardware init failed
...
```

I might've messed something up when merging the OpenWRT + Xilinx linux builds, hence why I'm reaching out to see if you know of any success stories for this adapter.

bjlockie, 2023-01-10:
Could you try live distro with a 6x kernel?

morrownr, 2023-01-10:
> @morrownr Any ETA when ALFA will release adapter based on 7921u?

@usama7628674 

Finally, I can answer. The answer is Feb. 23. You can see the initial product on the Alfa webpage now. The best info I have at this point is that this first product is a "regular power" adapter. I am not recommending it at this point, similar to the Netgear adapter that started this thread, because not enough information is known.

The information that has been discussed in a couple of place here at the site is that Alfa's second product will be their "high power" (long range) adapter which is the one I will be interested in.

Nick

morrownr, 2023-01-10:
@pbotros 

>  any confirmation that you know of that this Netgear A8000 / AX3000 works with the mt7921u driver?

Your report is the first report I have seen. I was wondering if the product was vaporware. The output of `lsusb` does not depend on a driver so if you have access to a very recent distro that, like @bjlockie said, maybe you can boot from a flash drive and see what happens. Ubuntu 22.10 runs kernel 5.19 and I have verified that it has new enough firmware to let you see what is going on.

I got one report from a user that has a Comfast CF-952AX adapter. That adapter appeared initially to use a rtl8832xu chipset but the using making the report said it has an mt7921au chipset but he had to add a new device is to the mt7921u driver and recompile it for the adapter to work. He has since submitted the id to linux-wireless to be included in the kernel.

If we could at least see what `lsusb` has to say, that would help.

Nick

bjlockie, 2023-01-10:
2 of the Alfa adapters use Realtek chips.
The third one appears to be usb-c but maybe the cable is detachable.

morrownr, 2023-01-10:
> 2 of the Alfa adapters use Realtek chips.

Where are you looking?

bjlockie, 2023-01-10:
https://www.alfa.com.tw/collections/wifi-6-6e-usb

pbotros, 2023-01-10:
Spun up latest Ubuntu (22.10) and did a `lsusb -v`; dump below. The Netgear A8000 / AX3000 does indeed use a MediaTek chip, but Ubuntu (22.10) didn't immediately recognize it. I might try to see if I can compile the newest mt7921u for Ubuntu 22.10 with the appropriate USB vendor/device ID.

```
Bus 002 Device 003: ID 0846:9060 NetGear, Inc. Wireless_Device
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.20
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         9
  idVendor           0x0846 NetGear, Inc.
  idProduct          0x9060 
  bcdDevice            1.00
  iManufacturer           2 MediaTek Inc.
  iProduct                3 Wireless_Device
  iSerial                 4 000000000
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0087
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          5 Config_01
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              160mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           9
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              1 WiFi_If
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x85  EP 5 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x08  EP 8 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
[... 73 lines cut ...]
  bNumDeviceCaps          2
  USB 2.0 Extension Device Capability:
    bLength                 7
    bDescriptorType        16
    bDevCapabilityType      2
    bmAttributes   0x0000f11e
      BESL Link Power Management (LPM) Supported
    BESL value      256 us 
    Deep BESL value    61440 us 
  SuperSpeed USB Device Capability:
    bLength                10
    bDescriptorType        16
    bDevCapabilityType      3
    bmAttributes         0x00
    wSpeedsSupported   0x000e
      Device can operate at Full Speed (12Mbps)
      Device can operate at High Speed (480Mbps)
      Device can operate at SuperSpeed (5Gbps)
    bFunctionalitySupport   1
      Lowest fully-functional device speed is Full Speed (12Mbps)
    bU1DevExitLat          10 micro seconds
    bU2DevExitLat         180 micro seconds
Device Status:     0x0000
  (Bus Powered)
```

jiribrejcha, 2023-01-10:
It uses MT7921AUN chip.

morrownr, 2023-01-10:
> ID 0846:9060 NetGear, Inc. Wireless_Device

Well, of course Netgear had to use their own device id so we need to add it, compile, test and submit a patch to Linux-wireless. Any volunteers to do this?

@pbotros 

You may know how to do this but I have had some users asking for a checklist. Can I post the checklist and get you to follow it, and suggest corrections?

> It uses MT7921AUN chip.

@jiribrejcha 

How is the performance? Is it a good enough product to add to the in-kernel adapter list? (with a note that users have to compile their own driver until the device id is mainlined).

pbotros, 2023-01-10:
> You may know how to do this but I have had some users asking for a checklist. Can I post the checklist and get you to follow it, and suggest corrections?

@morrownr sure! I can commit a little bit of free time in the next couple weeks.

> Well, of course Netgear had to use their own device id so we need to add it, compile, test and submit a patch to Linux-wireless. Any volunteers to do this?

That said, a word of caution - per my first note, I _did_ add the right device/vendor ID to the mt7921u driver from the [mt76 repo from OpenWRT](https://github.com/openwrt/mt76), and `dmesg` reflected that it did try to use the mt7921u driver for the adapter. However, I received the timeouts stated above, so it might end up being non-trivial to get this working.

bjlockie, 2023-01-10:
Paul, would you be able to try the adapter on an x86?
The 7921au adapters work on Arm so that probably won't show anything but it might. :-)

morrownr, 2023-01-10:
> I can commit a little bit of free time in the next couple weeks.

@pbotros 

Here you go. Let me know what needs to be worked on.

-----

Checklist-Compiling_Linux_Mainline_Kernel

Purpose: Facilitate Testing or Adding new Device IDs

2023-01-10

Prerequisites:

A system running Linux (only Ubuntu is tested).

Access to the terminal/command line.

A user account with sudo/root privileges.

12GB of available space on the hard drive.

-----

Note: This checklist has only been tested on Ubuntu 22.10
but will likely work, possibly with minor modifications,
on numerous distros, however you should check with your
distro to confirm.

-----

1. Download the Source Code tarball from:

https://www.kernel.org/

linux-6.2-rc2tar.xz

Note: The above name will change depending on the
version of the kernel you decide to compile. Adjust
accordingly.

Note: make a directory called `src` in my home directory
and that is what I download the file to. Anywhere in your
home directory should work.

-----

2. Extract the Source Code:

$ tar xvf linux-6.2-rc2.tar.xz

-----

3. Install Required Packages:

$ sudo apt-get install git fakeroot build-essential ncurses-dev xz-utils libssl-dev bc flex libelf-dev bison

Note: The above line is specific to Debian based distros such
as Ubuntu.

-----

4. Configure Kernel:

Navigate to the `linux-6.2-rc2` directory using the cd command:

$ cd linux-6.2-rc2

Copy the existing configuration file using the cp command:

$ cp -v /boot/config-$(uname -r) .config

To make changes to the configuration file, there are two options:

Option 1: Run make:

$ make menuconfig

The command launches several scripts that open the configuration menu.

The configuration menu includes options such as firmware, file system, network, and
memory settings. Use the arrows to make a selection or choose Help to learn more about the
options. When you finish making the changes, select Save, and then exit the menu.

Option 2: Edit .config directly with a text edition:

I use `geany`:

$ geany .config

If you are intested in turning on the new support
for the below listed new Mediatek driver, make sure
the following line is as below:

Search for `MT76`.

CONFIG_MT7921U=m

Save file and exit.

If you are intested in turning on the new support for the
below listed Realtek drivers:

Search for `RTW88`.

To turn on support for the rtl8822bu, rtl8812bu, rtl8821cu and rtl8811cu
chipsets, make sure the following lines are as below:

CONFIG_RTW88=m
CONFIG_RTW88_CORE=m
CONFIG_RTW88_USB=m
CONFIG_RTW88_8822B=m
CONFIG_RTW88_8821C=m
CONFIG_RTW88_8822BU=m
CONFIG_RTW88_8821CU=m

Save file and exit.
---

5. Prevent errors related to certs.

If you are compiling the kernel on Ubuntu or on any distros based on Ubuntu,
you may receive the following error that interrupts the building process:

No rule to make target 'debian/canonical-certs.pem

To prevent the above problem, disable the conflicting security certificates
by executing the two commands below:

$ scripts/config --disable SYSTEM_TRUSTED_KEYS
$ scripts/config --disable SYSTEM_REVOCATION_KEYS

-----

6. Now is the time to patch any files if you are going to test patches.

-----

7. Compile the kernel.

$ make -j$(nproc)

The process of compiling the Linux kernel takes some time to complete.

Note: If the compile stops, hit Enter until it starts again (this is
related to the cert stuff above).

---

8. Install the required modules:

$ sudo make modules_install

---

9. Install the kernel:

$ sudo make install

---

10. Reboot and check kernel:

$ uname -r

-----

For uninstalling the kernel installed from source, run:

$ sudo rm -rf /lib/modules/kernel_version
Example: $ sudo rm -rf /lib/modules/linux-6.2-rc2

$ sudo rm -f /boot/vmlinuz-kernel_version*
Example: $ sudo rm -f /boot/vmlinuz-linux-6.2-rc2*

$ sudo rm -f /boot/initrd.img-kernel_version*
Example: $ sudo rm -f /boot/initrd.img-linux-6.2-rc2*

$ sudo rm -f /boot/config-kernel_version*
Example: $ sudo rm -f /boot/config-linux-6.2-rc2*

$ sudo rm -f /boot/System.map-kernel_version*
Example: $ sudo rm -f /boot/System.map-linux-6.2-rc2*

Finally, after uninstalling the kernel, run:

$ sudo update-grub

to clean the grub menu.

-----

Happy compiling!