morrownr/USB-WiFi #125: Need help: Has anyone with a CF-953AX or CF-851AX been able to get Bluetooth going?

morrownr, 2022-10-07:
Bug 1 in #107 gives some details of what I am seeing. I have yet to get Bluetooth going with any distros or in any computer. The 951AX that I have is recognized as having BT capability but I cannot get it to actually work. Does anyone have Bluetooth working with either adapter?

Nick

gacragg-github, 2022-10-09:
Are there any specific commands you would want results for?  It seems to work in kismet - it picks up BTLE and BR devices.
```
hci1:   Type: Primary  Bus: USB
        BD Address: E0:E1:A9:35:C4:48  ACL MTU: 1021:6  SCO MTU: 240:8
        UP RUNNING INQUIRY
        RX bytes:3411 acl:0 sco:0 events:256 errors:0
        TX bytes:4124 acl:0 sco:0 commands:213 errors:0

hci0:   Type: Primary  Bus: USB
        BD Address: E0:D0:45:A0:7A:43  ACL MTU: 1021:4  SCO MTU: 96:6
        UP RUNNING INQUIRY
        RX bytes:23105 acl:0 sco:0 events:538 errors:0
        TX bytes:3800 acl:0 sco:0 commands:202 errors:0
```
hci1 matches the CF951:
```
admin1@dinm9:~$ lsusb
Bus 002 Device 002: ID 0e8d:7961 MediaTek Inc. Wireless_Device

```
4: wlan1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP mode DEFAULT group default qlen 1000
    link/ieee802.11/radiotap e0:e1:a9:35:c4:47 brd ff:ff:ff:ff:ff:ff
```

Ndx      Iface   Phy       Driver      Mode   Up?       Channel Width   Center    Packets Adapter
  1      wlan1  phy1      mt7921u   monitor     Y   5 (5975MHz) 80MHz 5985 MHz     827512  MediaTek Inc. Wireless_Device
```

The other adapter is an Intel AX210 which is picking up more devices.

morrownr, 2022-10-09:
@gacragg-github 

> Are there any specific commands you would want results for?

I am not an expert on bluetooth (yet). My expertise is with wifi. So, all I would like to see is my computer, that has the CF-951AX plugged into a usb port, be able to pair with my phone so I can transfer files or do the other bluetooth stuff.

Here is the output from the system the adapter currently plugged into:

$ hciconfig
(usb nano Broadcom chipset based adapter) (it works)
hci0:	Type: Primary  Bus: USB
	BD Address: 00:02:72:D9:EC:1E  ACL MTU: 1021:8  SCO MTU: 64:1
	UP RUNNING PSCAN ISCAN 
	RX bytes:3601 acl:0 sco:0 events:444 errors:0
	TX bytes:40884 acl:0 sco:0 commands:443 errors:0

(CF-951AX) (not ability to pair or see my phone and my phone can't see it)
hci1:	Type: Primary  Bus: USB 
	BD Address: E0:E1:A9:35:C5:20  ACL MTU: 1021:6  SCO MTU: 240:8
	UP RUNNING PSCAN ISCAN 
	RX bytes:1186 acl:0 sco:0 events:118 errors:0
	TX bytes:7729 acl:0 sco:0 commands:118 errors:0

I have upgraded to the latest BT firmware for this chipset (Sep 22). I am using kernel 6.0.

I have tested this adapter on several computers at this point and have not been able to pair with anything. I've been looking at some online troubleshooting guides but so far, nothing. That is why I am searching for people with adapters that have the mt7921au chipset.

Thanks,

Nick

morrownr, 2022-10-10:
Update: I also have a modern laptop computer that has a mt7921 based card. It uses the mt7921e driver. I just finished testing to for basic function with 2 phones and everything worked. I was able to pair and do bluetooth things.

The laptop has Xubuntu 22.04 on it and runs kernel 5.15. I first tested with the older bluetooth firmware and then with the newer Sep 22 firmware. It worked both times. Could it be that there is a problem with my CF-951AX adapter?

bjlockie, 2022-10-10:
I am not an engineer but I wouldn't be surprised if that adapter doesn't support Bluetooth.
They weren't even advertising triband.

morrownr, 2022-10-10:
@bjlockie 

I could live with that because I wasn't looking for Bluetooth in the first place. However, note this:

$ hciconfig
hci1:	Type: Primary  Bus: USB
	BD Address: E0:E1:A9:35:C5:20  ACL MTU: 1021:6  SCO MTU: 240:8
	UP RUNNING PSCAN ISCAN 
	RX bytes:1258 acl:0 sco:0 events:130 errors:0
	TX bytes:7827 acl:0 sco:0 commands:130 errors:0

The adapter is advertising its Bluetooth capability, the driver is loading and everything comes to life... but it can't discover other Bluetooth adapters and it doesn't seem to be advertising itself as I can't pick it up on anything else. If a maker wants to shut down a capability on a chipset, they generally shut it down and it cannot be detected. This is a mystery to me.

gacragg-github, 2022-10-10:
My ipad was able to see the AX210 BT adapter, but similar preparation of the interface with bluetoothctl using the mt7921u usb adapter did not produce results - the ipad could not pick up the interface.  I would say this is consistent with your result.  I am on latest FW and 5.18/x64.

bjlockie, 2022-10-11:
I suspect the Bluetooth is not even wired in the USB adapters.
My 7921k (RZ608) shows the Bluetooth and is pair able.
I suspect those adapters have problems but they are the right chipset at least. :-)

morrownr, 2022-10-11:
@gacragg-github 

> similar preparation of the interface with bluetoothctl using the mt7921u usb adapter did not produce results - the ipad could not pick up the interface. I would say this is consistent with your result.

Yes, this sounds familiar. What distros, kernel and interface prep do you do?

morrownr, 2022-10-11:
@bjlockie 

> My 7921k (RZ608) shows the Bluetooth and is pair able.

Same here with my laptop that has a 7921 internal card. Bluetooth just worked with Xubuntu 22.04 and kernel 5.15.

Funny thing about the 7921au based adapter is that the system detects bluetooth when I plug the adapter. Everything seems normal yet it is imply will not see other bluetooth signals and other bluetooth adapters cannot see it. This is a mystery to solve.

bjlockie, 2022-10-15:
I got my Comfast CF-953AX

I need to find a USB3 extension cable, it didn't work with the cable I was suspicious about anyways.

I wonder if the 79**6**1 is significant in the lsusb:
`Bus 001 Device 002: ID 0e8d:79**6**1 MediaTek Inc. Wireless_Device`

Wifi seems to work (I need to leave it up for a while.
I just used a nightly lubuntu (beta of 22.10) with kernel 5.19.0

bjlockie, 2022-10-15:
I have it working with my USB3 extension cable. :-)
```
        signal: -42 dBm
        rx bitrate: 1200.9 MBit/s 80MHz HE-MCS 11 HE-NSS 2 HE-GI 0 HE-DCM 0
        tx bitrate: 1200.9 MBit/s 80MHz HE-MCS 11 HE-NSS 2 HE-GI 0 HE-DCM 0
```
I don't know if that is good signal but the link rates are good.
The AP is one floor below.
It is 5GHz.

I tried to install the adapter on Windows to see if Bluetooth or the LED worked.
I was going to ask on the linux wireless mailing list.
I couldn't make the driver work at all and I don't see any way to get support from Comfast. :-(
In most cases the Windows driver is better but for Mediatek the Linux drivers are better.

morrownr, 2022-10-15:
Hi @bjlockie 

> I need to find a USB3 extension cable, it didn't work with the cable I was suspicious about anyways.

I have several USB3 extention cables. My CF-951AX would not work with any of them. It does work with the below:

https://www.amazon.com/dp/B07S6B5X76

Maybe that can help depending on what you are doing.

> I just used a nightly lubuntu (beta of 22.10) with kernel 5.19.0

Looking forward to your reports.

morrownr, 2022-10-15:
> I don't know if that is good signal but the link rates are good.

That is very good.

> I tried to install the adapter on Windows to see if Bluetooth or the LED worked. I was going to ask on the linux wireless mailing list.

Hmmm... the Linux kernel wireless mailing list is a bad place to ask about Windows drivers. I'll take a look for you.

Look here:

http://en.comfast.com.cn/index.php?m=content&c=index&a=lists&catid=30

bjlockie, 2022-10-16:
If the led and Bluetooth worked in Windows then I was going to ask for help on the list about getting it to work in Linux.

black-jmyntrn, 2022-10-25:
953ax LED works in windows , just not bluetooth for me.

lestcape, 2022-10-30:
I have now an CF-953AX device since yesterday in my Ubuntu 22.04. It works for me in AP mode in 5 GHz and 2.4 GHz using https://github.com/lakinduakash/linux-wifi-hotspot. I have a custom linux kernel linux-6.1-rc2 where i have enable:

```
CONFIG_MT7921_COMMON=m
CONFIG_MT7921E=m
CONFIG_MT7921S=m
CONFIG_MT7921U=m
```

I also update the firmware to the last version from: https://github.com/morrownr/USB-WiFi/blob/main/home/How_to_Install_Firmware_for_Mediatek_based_USB_WiFi_adapters.md

**The Bluetooth don't work on linux:**

```
[    8.899035] kernel: Bluetooth: hci0: Opcode 0x c03 failed: -110
[   10.915095] kernel: Bluetooth: hci0: Failed to read MSFT supported features (-110)
[   12.931065] kernel: Bluetooth: hci0: AOSP get vendor capabilities (-110)
[   13.159293] kernel: Bluetooth: hci0: Device setup in 121899 usecs
[   13.159310] kernel: Bluetooth: hci0: HCI Enhanced Setup Synchronous Connection command is advertised, but not supported.
```
I have installed the driver in Windows 11 (21H2 and 22H2) from http://en.comfast.com.cn/index.php?m=content&c=index&a=show&catid=13&id=182 on the tab (Product Related), that is supposed to be both drivers (wifi and bluetooth), but after install it the Bluetooth also not work and is recognized wrong (as another wifi 6 card) at the list of devices. While i was trying to find a solution, i downloaded the Lenovo bluetooth driver: https://support.lenovo.com/mx/es/downloads/ds552065-mediatek-mt7921-bluetooth-driver-for-windows-10-version-2004-or-later-thinkpad-l14-gen-2-l15-gen-2 and after install it, the unrecognized card was recognized as a Mediatek Bluetooth device finally, but the driver installation required a reboot. After the PC reboot with driver installed, the pc hangs and the only solution to fix the issue is uninstall the driver.

**The led don't work on linux**, but i have not tested in on Windows 11 yet.

ibrtuyima, 2023-01-03:
I think the following kernel config option will help to get Bluetooth working:
CONFIG_BT_HCIBTUSB_MTK=y

morrownr, 2023-01-04:
@ibrtuyima 

I will check that when I next compile a kernel. I don't think this is the issue. What I think the issue is happens to be something that has been around for a long time and forces adapters makers into a decision. USB3 cables, wires and connections emit a radio signal in the 2.4 GHz range which interferes with BT so for many years, adapter makers have had to chose between USB3 wifi OR USB2 wifi plus BT. I suspect Comfast thought that deactivating BT so that the wifi could run with USB3 is the option they took... and I agree with them. for my use case.

The optimal solution would be a software/driver solution that lets users chose either USB3 wifi or USB2 wifi plus BT. USB2 with the mt7921au chipset can sustain 300 Mb/s but that is less than half of what it can do in WiFi 5 if USB3 is active. For many folks 300 Mb/s plus BT would be the preferred solution but for many, USB3 and maximun performance is the preferred solution.

We shall see how this works out. I wish we had more information at this point.

ibrtuyima, 2023-01-04:
may be you are right, but based on my testing when was trying to build 6Ghz AP using RPIB4 along with 7921au adapter, initially throughput was not  good but after doing some tweaks such as enabling RPS and overclocking CPU to 2.2GHz, throughput reaches a good level (414Mb/s for download- in AP mode, mt7921 supports only 40 MHz for TX ) and (650 Mb/s for upload- supports 80 Mhz for RX), for Bluetooth, I have complied a kernel with that option and working okay.

morrownr, 2023-01-04:
@ibrtuyima 

Tell me what the following shows:

lsusb -t

ibrtuyima, 2023-01-04:
@morrownr 

Here you go
[image]

morrownr, 2023-01-04:
@ibrtuyima 

Wow. Never seen BT running in USB3 mode. Which adapter do you have?

ibrtuyima, 2023-01-04:
@morrownr 

 CF-953AX

ibrtuyima, 2023-01-04:
@morrownr 

had created service unit file with name of btagent.service in the /etc/systemd/system directory with following content

[Unit]
Description=btagent
Requires=bluetooth.target
After=bluetooth.target
BindsTo=bluetooth.service

[Service]
ExecStartPre=/bin/sh -c "/usr/bin/sleep 10"
ExecStart=/bin/sh -c "/usr/bin/bt-agent --capability=NoInputNoOutput"
ExecStartPost=/bin/sh -c "/usr/bin/bt-adapter -a hci0 --set Alias `hostname`"
ExecStartPost=/bin/sh -c "/usr/bin/bt-adapter -a hci0 --set PairableTimeout 0"
ExecStartPost=/bin/sh -c "/usr/bin/bt-adapter -a hci0 --set DiscoverableTimeout 0"
ExecStartPost=/bin/sh -c "/usr/bin/bt-adapter -a hci0 --set Powered 1"
ExecStartPost=/bin/sh -c "/usr/bin/bt-adapter -a hci0 --set Discoverable 1"
ExecStartPost=/bin/sh -c "/usr/bin/bt-adapter -a hci0 --set Pairable 1"
ExecStartPost=/bin/sh -c "/usr/bin/bt-network -d -s nap br0"
TimeoutStopSec=5
Type=simple

[Install]
WantedBy=multi-user.target

then run the following commands

systemctl daemon-reload
systemctl enable btagent.service 
systemctl start btagent.service

fishBone000, 2023-10-10:
I wasn't able to get the Bluetooth on my CF-953AX running.  
I confirmed that `CONFIG_BT_HCIBTUSB_MTK` is enabled on my kernel (6.5.5, Arch official kernel), and have run commands in ibrtuyima's `btagent.service`, but the Bluetooth on my CF-953AX is still as usual: commands in bluetoothctl can be executed, but have no actual effect at all.

morrownr, 2023-10-10:
> I wasn't able to get the Bluetooth on my CF-953AX running.

You won't. Some adapters based on the mt7921au chipset incorrectly say BT support but none actually have it. The chipset actually has the capability but the makers have to turn it off or wifi will be limited to USB2 speeds and that would make no sense at all. It seems the engineers and marketers were not talking to each other.

This is a USB3 design problem and has been around a LONG time. It does not affect PCIe cards with the same chipsets but it does affect USB adapters. On USB adapters, USB3 cables will emit a signal that interfers with BT. What you can do if you want both is get a PCIe card. I have a PCIe card that uses the mt7922 chipset. Both WiFi and BT work well together and the mt7922 does 160 Mhz channel width if that is of any use.

Nick