morrownr/USB-WiFi #129: Suggest usb-wifi dongle for Linux RT-patch

erichyc7035, 2022-10-12:
Hi, 

I found you have list some usb-wifi dongle for Linux.
But I'm not sure  if they are suitable for Linux RT-patch.
I'm wondering if you have any suggestion if we want a usb-wifi dongle for Linux RT-patch.

Thanks.
Eric

morrownr, 2022-10-12:
Hi @erichyc7035 

Eric, My suggestion is a test. You tell me a distro with an RT-patch kernel. I install it on a test system. You give some test to run and I provide the results from various adapters.

Nick

erichyc7035, 2022-10-13:
Hi @morrownr ,

We are based on Yocto 3.0.4 and 5.2 kernel with full preempt.
I'm not sure if the information is enough for you.

Our test is easy.  Connect to internet by usb-wifi dongle and can ping ip(google ip, for example), and see if dongle working fine without any panic.

Eric

morrownr, 2022-10-13:
Eric,

I did a little reading on Yocto. Let me ask some questions and rework my recommendation.

Are you looking for 2.4 GHz, 5 GHz or both?
What size adapter are you looking for?
What throughput are you looking for?
What range are you looking for?
What modes do you need? Managed, monitor, AP?
What are you looking to spend?

With kernel 5.2, you would have in-kernel drivers available for all adapters listed in Main Menu item 2 with the exception of the latest mt7921au WiFi 6e chipsets. Recommend you go take a look. 

Nick

erichyc7035, 2022-10-14:
Hi @morrownr ,

Let me answer your question first.

1. Only 2.4GHz is ok. 5GHz is optional.
2.The size similar to Panda Mid Range 150Mbps is ok to me, has a antenna outside.
3.Throughtput and range is not important.
4.Managed mode is necessarily at least.

The most important is it can works well with Linux RT, and user can buy it easily.
It's very helpful if you can try to find some adapters.

And you mention the Main Menu item 2 in kernel 5.2, did you mean in menuconfig? 
Device Drivers -> Network device support -> Wireless Lan
Or other place?

Eric

bjlockie, 2022-10-14:
I think he means menu item #2 on his github list.
I don't know where in the kernel menuconfig the driver is.

morrownr, 2022-10-14:
Hi @erichyc7035 

> And you mention the Main Menu item 2 in kernel 5.2, did you mean in menuconfig?

Sorry about being unclear. The Main Menu for this site is in the following repo:

https://github.com/morrownr/USB-WiFi

Go there, scroll down and select menu item 2. You will find a VERY long list of solid adapters that use in-kernel adapters and given your requirements and kernel level, you need adapters that use in-kernel drivers. What I think would work best is if you go over the list and come back with specific questions. I think the best way to discover the answer is for you to pick a low cost adapter and order it.

Nick

morrownr, 2022-10-14:
@erichyc7035 

Oh, I forgot to mention, all adapters on the list with the exception of the new mt7921au chipset will have drivers in the 5.2 kernel. I do not know if the mt7921u driver will be backported to 5.2. It has been baclported to 5.10 but you needs appear to be simple so a AX3000 adapter is probably not on your list anyway.

Nick

closed 2022-11-22 by morrownr