morrownr/USB-WiFi #134: ubuntu: override udev usb names?

bjlockie, 2022-10-17:
Does anyone know how to override the udev usb names in Ubuntu?
lsusb says:
0e8d:7961 MediaTek Inc. Wireless_Device
for my Comfast CF-953AX.

I have an Alfa Networks AWUS036ACM that shows in lsusb as:
0e8d:7612 MediaTek Inc. MT7612U 802.11a/b/g/n/ac Wireless Adapter

Is the name lsusb shows embedded in the device (Comfast) or the chipset (MediaTek)?

I am suspicious MediaTek don't embed a detailed name in the chipset and just use "Wireless_Device".

I found:
usb:v0E8Dp7612*
 ID_MODEL_FROM_DATABASE=MT7612U 802.11a/b/g/n/ac Wireless Adapter
in /lib/udev/hwdb.d/20-usb-vendor-model.hwdb.
So I removed that.
That didn't change anything so I also tried to override the name in /etc/udev/hwdb.d/
Neither worked.

I copied that directive to make one for 0E8D:7961 but that didn't work either. 🙁
usb:v0E8Dp7961*
 ID_MODEL_FROM_DATABASE=MT7612U 802.11a/b/g/n/ac/ax Wireless Adapter

morrownr, 2022-10-18:
> Does anyone know how to override the udev usb names in Ubuntu? lsusb says: 0e8d:7961 MediaTek Inc. Wireless_Device

That is not the greatest. Maybe @bjlockie can submit something better:

https://opensource.com/article/20/8/usb-id-repository

bjlockie, 2022-10-22:
I submitted stuff before.
The IDs went in but the names didn't.
I asked on the usb mailing list and basically they said the device should have better naming in the firmware.
I suspect MediaTek never did. :-(

closed 2022-11-22 by morrownr