morrownr/USB-WiFi #137: How to install mt7921u on ubuntu 22.10 Raspberry Pi 4

ViRb3, 2022-10-22:
Hello,

The new Ubuntu 22.10 with kernel 5.19 finally dropped and I am eager to test this adapter. Sadly, it seems like the official image for Raspberry Pi does not include the driver — `/usr/lib/modules/<your kernel version>/kernel/drivers/net/wireless/mediatek/mt76/mt7921/mt7921u.ko` is missing. In fact, the entire `mediatek` folder is missing. Strangely the arm64 generic ISO has those. All of the firmware files are already in `/lib/firmware/mediatek` so this doesn't make much sense. Do you have any idea how I can get this working?

Thanks!

bjlockie, 2022-10-22:
You need to install the package linux-modules-extra-raspi
Ya, ubuntu makes it painful.

ViRb3, 2022-10-22:
Thanks a ton! Works wonderfully. @morrownr getting 413/267Mbit about 10 meters away from my Gigabit 5Ghz AP. The receiver works perfectly in USB2, USB3, and USB3 hub over USB3. Finally a worthy upgrade for my Raspberry!

ViRb3, 2022-10-22:
Hmm, under sustained load, it seems like performance slowly crumbles.

- speedtest-go: 490/95Mbit (shortest test)
- librespeed-cli: 217/207Mbit (longer test)
- wget https://speed.hetzner.de/10GB.bin: between 64 and 144 Mbit

Any way to optimize it?

morrownr, 2022-10-22:
@ViRb3 @bjlockie 

Okay. Two issues in one issue and the need for clarification...

> The new Ubuntu 22.10 with kernel 5.19 finally dropped

I installed Xubuntu 22.10 (amd64 desktop) on a test system yesterday. It was a clean install, no upgrade. I plugged a CF-951AX into a USB3 port. It was instantly recognized, connected to my main router and has performed flawlessly in managed mode so far. I have not checked to see if the firmware needs to be updated (there was a Sep release of firmware that might not have made it).

I did not look at Ubuntu for the RasPi4B. I tried it once last year for about 20 minutes then burned the sd card to something else. It was like pouring honey on a cold winter morning. Not sure what Canonical is thinking using Gnome on a Pi. They might want to make Xubuntu the lead distro for Pi's.

> getting 413/267Mbit about 10 meters away from my Gigabit 5Ghz AP. Still not ideal, since my MacBook, iPhone, and Windows PC can easily reach 600/600 from the same place.

My 951AX is showing 600-650 Mbp/s with 5 GHz, AC, DFS channel, clean air. This is measured with iperf3. Are you using a DFS or non-DFS channel and how are you measuring the speed?

> The receiver at least works perfectly in powered USB3 hub as well as USB3 extension cable in powered USB3 hub.

@ViRb3 Which adapter are you using? This is encouraging news. The problems I have had with my 951AX could simply be a problem with my adapter.

ViRb3, 2022-10-22:
I am using the CF-953AX (with antenna flippers). On Ubuntu 22.10 arm64 Desktop (not Server) on my MacBook Pro 2021, it was also plug-and-play for me. I guess it's just the Server/Raspberry release that needs a separate package installed, but that's no bother.

I am measuring the speed using three different methods as detailed above:
- [speedtest-go](https://github.com/showwin/speedtest-go)
- [librespot-cli](https://github.com/librespeed/speedtest-cli)
- `wget https://speed.hetzner.de/10GB.bin`

My router is one of the generic Nokia Gbit routers provided by ISPs. Here's some specs about the AP from my macOS GUI:

[image]

bjlockie, 2022-10-22:
I did that wget.
It started out over 4MB/s but went down to less than 1 but then back up to 4 and then down to 1ish, then back up.
You could compare it to a wifi5 mediatek adapter.
I suspect the driver has a lot of improvements for kernel 6 but I don't know how to get the extra-modules for v6 on ubuntu.

morrownr, 2022-10-22:
>  I don't know how to get the extra-modules for v6 on ubuntu.

I'm not 100% sure I understand what you are saying but if you are saying that you would like to try kernel version 6.0 on amd64 Ubuntu 22.10, then...

2022-10-22

Upgrading the Linux Kernel on Ubuntu based amd64 systems

Note: Turn secure sode off before installing. These kernels
do not support secure mode.

Go to: 

https://kernel.ubuntu.com/~kernel-ppa/mainline/

Note: This guide is an example for kernel 6.0.3. Please
make adjustments for other kernels.

Download into a clean directory: (no other .deb files)

linux-headers-6.0.3-060003-generic_6.0.3-060003.202210211149_amd64.deb
linux-headers-6.0.3-060003_6.0.3-060003.202210211149_all.deb
linux-image-unsigned-6.0.3-060003-generic_6.0.3-060003.202210211149_amd64.deb
linux-modules-6.0.3-060003-generic_6.0.3-060003.202210211149_amd64.deb

Install:

sudo dpkg -i *.deb

-----

Remove:

Note: To remove a kernel you installed this way, you will
need to boot the computer to a different kernel.

sudo apt remove linux-headers-6.0.3*
sudo apt remove linux-image-unsigned-6.0.3*
sudo apt remove linux-modules-6.0.3*

Note: Once complete, reboot and the default kernel will be
the kernel you installed. To select another kernel version
to boot, use the grub menu.

bjlockie, 2022-10-22:
I can get the kernel for ubuntu but not the package of the MediaTek modules for v6 of the kernel..
I guess I could maybe get a kernel that has the driver built-in but would that work with ubuntu.

morrownr, 2022-10-22:
> I can get the kernel for ubuntu but not the package of the MediaTek modules for v6 of the kernel.

I learn something everyday. I'm sure you do as well. Here is your lesson for today.

in-kernel means in-kernel. The driver you will install per the instructions above include the 6.0.3 driver... mt7921u.ko.

The lesson is that the open source in-kernel part of the drivers in Linux are PART of the KERNEL. The only way for the kernel not to provide the driver is if the distro managers turn it off...which many do for server distros.

On the other hand, the firmware part of the driver is not part of the kernel and is kept in a place in your installation that does not change when you upgrade kernels. You can check to see if your firmware is the latest by using my installation instructions about upgrading Mediatek Firmware here in the Main Menu.

morrownr, 2022-10-22:
> I guess I could maybe get a kernel that has the driver built-in but would that work with ubuntu.

Check the url in the instructions I provided. It is an Ubuntu url. It is not the Linux Mainline kernel, it is the Ubuntu Mainline kernel.

morrownr, 2022-10-23:
> I think I clarified all of your questions above, you have may have missed my second post and edits.

Okay, I see. You are right. I had started mt reply last night but did not finish it. I'm not feeling well.

> I am using the CF-953AX (with antenna flippers).

So, you are seeing the 953AX work well with extension cables and powered hubs. That is very encouraging.

> On Ubuntu 22.10 arm64 Desktop (not Server) on my MacBook Pro 2021,

It seems to me that server versions of almost all distros do not install the wireless stuff. It is like they think all servers will have an ethernet connection...and they may be close to being correct. Anyway, it is just something us wireless users need to know and deal with.

Are you using an A to C usb converter to attach the 953 to the MacBook? Also, what OS are you running on the MacBook? You say MacOS, which version?

lwkeya119, 2022-10-25:
HI @ViRb3 and @morrownr While I am testing mt7921u on my rspi4 ubuntu server , I could only see mediatek in lsusb output, but no interface created. Please give me some tips, Thanks in advance!
pi@pi-ty:~$ uname -a
Linux pi-ty 5.19.0-1006-raspi #13-Ubuntu SMP PREEMPT Mon Oct 17 19:01:51 UTC 2022 aarch64 aarch64 aarch64 GNU/Linux
pi@pi-ty:~$ lsusb
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 004: ID 3574:6211 MediaTek Inc. Wireless_Device
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
pi@pi-ty:/lib/firmware/mediatek$ lsusb -t -v
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=dwc2/1p, 480M
    ID 1d6b:0002 Linux Foundation 2.0 root hub
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
    ID 1d6b:0003 Linux Foundation 3.0 root hub
    |__ Port 2: Dev 2, If 0, Class=Vendor Specific Class, Driver=, 5000M
        ID 3574:6211
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 480M
    ID 1d6b:0002 Linux Foundation 2.0 root hub
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        ID 2109:3431 VIA Labs, Inc. Hub

bjlockie, 2022-10-25:
Do you have linux-modules-extras-raspi installed?

lwkeya119, 2022-10-25:
> Do you have linux-modules-extras-raspi installed?

yes, I have installed it by "sudo apt install linux-modules-extra-raspi". Any additional config I missed?

morrownr, 2022-10-25:
@bjlockie 

Does that package @lwkeya119 installed include the firmware?

@lwkeya119 

We are all learning here. A key to understanding Linux in-kernel (built-in drivers) is that all of the wireless in-kernel drivers have two parts. If either of those parts is missing, it simply looks like no driver is installed... you won't have an interface to use.

Part one of the driver for the mt7921au chipset is the open-source, in-kernel driver called mt7921u. I'm assuming that the package @bjlockie suggested included mt7921u.ko. Most desktop distros either are or will be making the mt7921u driver a standard part of their distros and you will need to do nothing as time passes. The driver and the mt7921au chipset are both pretty new so are working their way through the system.

Part two of the driver for the mt7921au chipset is the firmware. The firmware files are closed source so as to limit competitors access to internal company information. Firmware files are not stored in the kernel and thus new versions are not made available when you or your package manager upgrades your kernel. The package manager may provide updated firmware if the managers of the distros decide to make them available but it is a separate issue from the in-kernel driver.

Below are the firmware files required by the mt7921au chipset:

-rw-rw-r-- 1 morrow morrow 525870 Sep 27 00:12 BT_RAM_CODE_MT7961_1_2_hdr.bin
-rw-rw-r-- 1 morrow morrow  92192 Sep 27 00:12 WIFI_MT7961_patch_mcu_1_2_hdr.bin
-rw-rw-r-- 1 morrow morrow 716404 Sep 27 00:12 WIFI_RAM_CODE_MT7961_1.bin

Yes, 3. The first one if for bluetooth support as this chipset supports bluetooth. The next 2 are for wifi. The filenames do not change. The way to check if you have the latest is the file size. Go to:

https://github.com/morrownr/USB-WiFi

... and select menu item 8 for instuctions on how to install or upgrade the firmware files.

> Bus 002 Device 004: ID 3574:6211 MediaTek Inc. Wireless_Device

What is this wireless adapter?

Nick

bjlockie, 2022-10-25:
What does sudo dmesg say?

Try sudo modprobe mt7921u
What is the output?

lwkeya119, 2022-10-25:
> sudo modprobe mt7921u

yes, @morrownr I have tried cp those firmware you mentioned here, but nothing changed.
Thw wireless adapter is COMFAST CF-952AX 

> What does sudo dmesg say? Try sudo modprobe mt7921u What is the output?
pi@pi-ty:~$ sudo modprobe mt7921u
pi@pi-ty:~$ sudo dmesg | grep usb
[    0.119907] usbcore: registered new interface driver usbfs
[    0.119956] usbcore: registered new interface driver hub
[    0.120021] usbcore: registered new device driver usb
[    1.764573] usbcore: registered new interface driver lan78xx
[    1.764650] usbcore: registered new interface driver smsc95xx
[    2.703040] usb_phy_generic phy: supply vcc not found, using dummy regulator
[    2.720953] usb_phy_generic phy: dummy supplies not allowed for exclusive requests
[    2.755637] usb usb1: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 5.19
[    2.755657] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    2.755664] usb usb1: Product: xHCI Host Controller
[    2.755670] usb usb1: Manufacturer: Linux 5.19.0-1006-raspi xhci-hcd
[    2.755675] usb usb1: SerialNumber: 0000:01:00.0
[    2.773978] usb usb2: New USB device found, idVendor=1d6b, idProduct=0003, bcdDevice= 5.19
[    2.773999] usb usb2: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    2.774006] usb usb2: Product: xHCI Host Controller
[    2.774012] usb usb2: Manufacturer: Linux 5.19.0-1006-raspi xhci-hcd
[    2.774017] usb usb2: SerialNumber: 0000:01:00.0
[    2.827973] dwc2 fe980000.usb: supply vusb_d not found, using dummy regulator
[    2.828268] dwc2 fe980000.usb: supply vusb_a not found, using dummy regulator
[    3.031240] usb 1-1: new high-speed USB device number 2 using xhci_hcd
[    3.035318] dwc2 fe980000.usb: EPs: 8, dedicated fifos, 4080 entries in SPRAM
[    3.035534] dwc2 fe980000.usb: DWC OTG Controller
[    3.035556] dwc2 fe980000.usb: new USB bus registered, assigned bus number 3
[    3.035584] dwc2 fe980000.usb: irq 40, io mem 0xfe980000
[    3.035755] usb usb3: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 5.19
[    3.035769] usb usb3: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    3.035776] usb usb3: Product: DWC OTG Controller
[    3.035782] usb usb3: Manufacturer: Linux 5.19.0-1006-raspi dwc2_hsotg
[    3.035787] usb usb3: SerialNumber: fe980000.usb
[    3.181894] usb 1-1: New USB device found, idVendor=2109, idProduct=3431, bcdDevice= 4.21
[    3.181926] usb 1-1: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[    3.181941] usb 1-1: Product: USB2.0 Hub
[    3.307533] usb 2-2: new SuperSpeed USB device number 2 using xhci_hcd
[    3.329385] usb 2-2: New USB device found, idVendor=3574, idProduct=6211, bcdDevice= 1.00
[    3.329405] usb 2-2: New USB device strings: Mfr=2, Product=3, SerialNumber=4
[    3.329420] usb 2-2: Product: Wireless_Device
[    3.329432] usb 2-2: Manufacturer: MediaTek Inc.
[    3.329443] usb 2-2: SerialNumber: 000000000

[   10.411152] usbcore: registered new interface driver brcmfmac
[   86.772585] usbcore: registered new interface driver mt7921u

morrownr, 2022-10-25:
Hi @lwkeya119 

You are really not going to like what I am about to say,,,

> The wireless adapter is COMFAST CF-952AX

Model Number: CF-952AX wifi 6 adapter
Wifi 6 adapter chip: RTL8832AU and PA

The CF-952AX uses a Realtek chipset. Realtek only makes out-of-kernel drivers for their modern usb chipsets. I actually have the source and it is set up in a repo here but it is not public like the other Realtek drivers. The reason: It is a terrible driver so I had to make the decision to not open the repo to the public. My suggestion is to return the adapter for your money back or sell it to someone (which may be your best option.) The only adapters that are confirmed to use the new mt7921au chipset are the CF-953AX and CF-951AX. It appears that the new Netgear A8000 also uses the mt7921au chipset but I do not have confirmation and, this is a big AND, Netgear is asking $89 USD for the adapter. That is outrageous in my humble opinion.

Nick

lwkeya119, 2022-10-25:
> Hi @lwkeya119
>
> You are really not going to like what I am about to say,,,
>
>
> Model Number: CF-952AX wifi 6 adapter Wifi 6 adapter chip: RTL8832AU and PA
>
> The CF-952AX uses a Realtek chipset. Realtek [...]

Got it, Thanks a lot. Just order a new CF-951AX.

dovandung, 2022-11-02:
I just buy a CF-952AX from ComFAST , It is MT7961U but the PID and VID is differ with CF-951AX( VID_0E8D&PID_7961 vs VID_3574&PID_6211)  , so the Linux driver cannot detect and install driver . It think is we add VID and PID to kernel driver maybe it should be work

morrownr, 2022-11-02:
Hi @dovandung 

> I just buy a CF-952AX from ComFAST , It is MT7961U...

Many sites list incorrect information. Here is the correct info for the CF-952AX:

Model Number: CF-952AX wifi 6 adapter
WiFi 6 adapter chip: RTL8832AU and PA

I wish there was a good solution in this case but there is not. The only real option for Linux users is to send it back and get your money back. I have the source for the Realtek out-of-kernel for the rtl8832au chipset. I had hoped to make it available publicly  but the driver is so bad that I can't make it available.

Regarding Main Menu item 2, the list of adapters that use in-kernel drivers: I made that list the way I did because what has happened in your case was happening far too often due to the bad information that is all over the place. That list takes a lot of work but it has helped many Linux users find a good solution.

Nick

dovandung, 2022-11-04:
I add VID and PID to MT7961 kernel driver( file usb.c), rebuild kernel and using NEW firmware WIFI_MT7961U_patch_mcu_1_2_hdr.bin from windows drivers ( rename it to WIFI_MT7961_patch_mcu_1_2_hdr.bin and put in /lib/firmware/mediatek ) , It works
 |mode: Managed, connected to: 5C:92:5E:0D:02:AC, time: 11:32m, inactive: 0.1s                        │
│freq: 2412 MHz, ctr1: 2422 MHz, channel: 1 (width: 40 MHz)                                          │
│rx rate: 300.0 Mbit/s MCS 15 40MHz short GI, tx rate: 270.0 Mbit/s MCS 15 40MHz                     │
│beacons: 6750, avg sig: -25 dBm, interval: 0.1s, DTIM: 1                                            │
│power mgt: on,  tx-power: 3 dBm (2.00 mW)                                                           │
│retry: short limit 7,  rts/cts: off,  frag: off

981213, 2023-01-04:
@morrownr 

> Many sites list incorrect information. Here is the correct info for the CF-952AX:
> 
> Model Number: CF-952AX wifi 6 adapter WiFi 6 adapter chip: RTL8832AU and PA

I recently purchased a Comfast CF-952AX (in China) and it's a MT7921U dongle with VID=3574&PID=6211. By adding the corresponding VID&PID to the linux driver everything works just fine.
Also, the CF-952AX driver downloaded from the comfast Chinese website only contains a driver for the Mediatek chips, not a Realtek one. Maybe Comfast is selling different card under the same model name outside China?
Here are some pictures of the card I purchased.
[image]
[image]

morrownr, 2023-01-04:
> HI @ViRb3 and @morrownr While I am testing mt7921u on my rspi4 ubuntu server , I could only see mediatek in lsusb output, but no interface created. Please give me some tips

```

$ lsusb

 Device 004: ID 3574:6211 MediaTek Inc. Wireless

```

@lwkeya119 @ViRb3 @bjlockie @dovandung @981213 

After having seen the below thanks to @981213 , I did some more searching.

```
Comfast CF-952AX is a MT7921 based USB WiFi dongle with custom
VID/PID. Add an entry for it.

---
 drivers/net/wireless/mediatek/mt76/mt7921/usb.c | 3 +++
 1 file changed, 3 insertions(+)

diff --git a/drivers/net/wireless/mediatek/mt76/mt7921/usb.c b/drivers/net/wireless/mediatek/mt76/mt7921/usb.c
index 5321d20dcdcb..a0778ecdb995 100644
--- a/drivers/net/wireless/mediatek/mt76/mt7921/usb.c
+++ b/drivers/net/wireless/mediatek/mt76/mt7921/usb.c
@@ -15,6 +15,9 @@
 static const struct usb_device_id mt7921u_device_table[] = {
        { USB_DEVICE_AND_INTERFACE_INFO(0x0e8d, 0x7961, 0xff, 0xff, 0xff),
                .driver_info = (kernel_ulong_t)MT7921_FIRMWARE_WM },
+       /* Comfast CF-952AX */
+       { USB_DEVICE_AND_INTERFACE_INFO(0x3574, 0x6211, 0xff, 0xff, 0xff),
+               .driver_info = (kernel_ulong_t)MT7921_FIRMWARE_WM },
        { },
 };

--
2.39.0
```

The following ad specifically says the chipset is RTL8832AU.

https://www.aliexpress.com/i/3256804530921070.html?gatewayAdapt=4itemAdapt

 I realize ads are often wrong but why would Comfast make another adapter almost the same as the CF-953AX with the same chipset and with a different Device ID? To use as white box distribution? I certainly don't know. Is this adapter a multi-state adapter?

This does not make sense to me but if I was wrong, I was wrong and will suck it up. Is this an adapter that should go in the list once its ID is mainlined?

Nick

981213, 2023-01-04:
> Is this adapter a multi-state adapter?

No. There's only one configuration and it doesn't have a mass-storage mode with windows driver.
It seems that there's also no bluetooth interface available.
Here's the output of lsusb:
```
Bus 008 Device 002: ID 3574:6211 MediaTek Inc. Wireless_Device
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.20
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         9
  idVendor           0x3574 
  idProduct          0x6211 
  bcdDevice            1.00
  iManufacturer           2 MediaTek Inc.
  iProduct                3 Wireless_Device
  iSerial                 4 000000000
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0087
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          5 Config_01
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              160mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           9
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              1 WiFi_If
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x84  EP 4 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x85  EP 5 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0400  1x 1024 bytes
        bInterval               0
        bMaxBurst               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x08  EP 8 OUT
        bmAttributes            2
[... 76 lines cut ...]
  bNumDeviceCaps          2
  USB 2.0 Extension Device Capability:
    bLength                 7
    bDescriptorType        16
    bDevCapabilityType      2
    bmAttributes   0x0000f11e
      BESL Link Power Management (LPM) Supported
    BESL value      256 us 
    Deep BESL value    61440 us 
  SuperSpeed USB Device Capability:
    bLength                10
    bDescriptorType        16
    bDevCapabilityType      3
    bmAttributes         0x00
    wSpeedsSupported   0x000e
      Device can operate at Full Speed (12Mbps)
      Device can operate at High Speed (480Mbps)
      Device can operate at SuperSpeed (5Gbps)
    bFunctionalitySupport   1
      Lowest fully-functional device speed is Full Speed (12Mbps)
    bU1DevExitLat          10 micro seconds
    bU2DevExitLat         180 micro seconds
Device Status:     0x0000
  (Bus Powered)
```

morrownr, 2023-01-04:
> > Is this adapter a multi-state adapter?
> 
> No. There's only one configuration and it doesn't have a mass-storage mode with windows driver.

That is good.

> It seems that there's also no bluetooth interface available.

Many Linux users, including me, would prefer single-function devices so BT not being support is actually a plus.

> Here's the output of lsusb:
> 
> ```
> Bus 008 Device 002: ID 3574:6211 MediaTek Inc. Wireless_Device

Glad you submitted the device ID to linux-wireless.

I see the following link in the Shen Zhen store on AliExpress:

 https://www.aliexpress.us/item/3256804530869798.html?spm=a2g0o.productlist.main.1.4d0858cbNFpmcz&algo_pvid=66ab124e-1b43-4780-a871-c7c96699633f&algo_exp_id=66ab124e-1b43-4780-a871-c7c96699633f-0&pdp_ext_f=%7B%22sku_id%22%3A%2212000030215335847%22%7D&pdp_npi=2%40dis%21USD%2128.63%2122.9%21%21%21%21%21%40211beca116728575457385710d070f%2112000030215335847%21sea&curPageLogUid=MUNNle6CzQTK

It avoids the question of what chipset and has no reviews. The link I sent previously to the Comfast store on AliExpress said the chipset is rtl8832au. And they were no reviews there either.

Do you think I should add it to the in-kernel list? Does it work well? Is it a reasonable good adapter for the price?

I would need to add a note about the device ID just now working its way into the kernel.

981213, 2023-01-06:
> Do you think I should add it to the in-kernel list? Does it work well? Is it a reasonable good adapter for the price?

I've been using it for two days now and it works pretty well in sta mode.
On Comfast Taobao shop they sells two cards under the CF-952AX listing and the one I got is the cheaper one (86CNY with discount, or about 13 USD).
The other one is marketed as driverless. The picture of the driverless one is CF-962AX instead. I don't know what chip that one uses, and 'driverless' usually means more work for us Linux users.
I think, if you add it to the list, it should be noted that the driverless one should be avoided if there are two version listed together.

morrownr, 2023-01-06:
I think I am going to delay adding it to the list for now because the device id is not even in the kernel yet.

geerlingguy, 2023-09-22:
FYI using a custom kernel build and `Method 1` in the `Netgear A8000` section on https://github.com/morrownr/USB-WiFi/blob/main/home/USB_WiFi_Adapters_that_are_supported_with_Linux_in-kernel_drivers.md — I was able to get 6 GHz WiFi running on a Pi under Debian 12. I am still documenting a bit about the setup, but it was considerably easier than I expected!

Only able to get 80 MHz channel widths right now, haven't had time to fully debug things, but it's a bit faster than onboard WiFi already.

morrownr, 2023-09-22:
Hi @geerlingguy 

> it was considerably easier than I expected!

Mediatek has been doing wonderful things for wifi on Linux.

> Only able to get 80 MHz channel widths right now...

The Netgear A8000 uses the mt7921au chipset (wifi 6e) that has a maximum capability of 80 Mhz channel width. The upcoming mt7925 chipset (wifi 7) will have 160 Mhz channel width capability:

https://github.com/morrownr/USB-WiFi/issues/308

There is a mt7922 (wifi 6e) wifi chipset available and has 160 Mhz channel width capability but we have so far only seen versions of the chipset for cards. I use a mt7922e chipset in my PCIe wifi card. A USB version of the mt7922 appeared to be planned but it is not clear if it will be released or if it will be skipped since the mt7925 may be available soon.

> Also noting that Pi OS includes Mediatek firmware and the kernel driver in it's default build, nice!

Yes, as of the 2023-05-05 release of the RasPiOS, adapters based on the mt7921au chipset are plug and play. This chipset has been subject to many firmware updates as modern wifi is getting really complicated. To ensure your firmware is the latest, go to the site Main Menu:

https://github.com/morrownr/USB-WiFi

...and select menu item 7.

Cheers

morrownr, 2023-09-22:
Hi again @geerlingguy 

> and Method 1 in the Netgear A8000 section 

That should no longer be necessary once you get to kernel 6.3 or 6.4 as the patch went in last spring. In fact, I added a new adapter to the plug and play list yesterday, #309  , and it uses the default vid/pid for this chipset. In fact, it appears that adapter makers for this chipset are catering to Linux users in that all known adapters with this chipset are single-state (no windows driver onboard), single-function (no bt) and all recent new adapters use the default vid/pidt.

Using usb wifi adapters is much easier than it used to be and it looks like it will continue to become easier as we go into the future.