morrownr/USB-WiFi #138: COMFAST CF-953AX with DietPi (Debian Bullseye 11)

fmunteanu, 2022-10-23:
I have a Raspberry Pi 4B+ running with DietPi (Debian Bullseye 11). Can you please let me know what steps I need the CF-953AX adapter working?

bjlockie, 2022-10-23:
>Debian 11 (Bullseye) was released on 14 August 2021. It is based on the Linux 5.10 LTS kernel and will be supported for five years.
You need kernel-5.19

I would upgrade to testing if you are NOT afraid of breading your system.
I'm use a new SD card.
I'm not sure it uses a recent kernel.
https://wiki.debian.org/DebianTesting

Burn something recent to a new SD card.

morrownr, 2022-10-23:
Hi @fmunteanu 

I can't be specific with DietPi because I am not familiar with it but I can show you what I did to the RasPiOS:

 2022-10-15

Upgrade_RasPiOS_Kernel

Using: 64 bit RasPiOS kernel on a RasPi Pi4B

Original Source:

https://www.raspberrypi.com/documentation/computers/linux_kernel.html#building

Note: Compiling the kernel on a RasPi can take a long time.

Note: This guide is detailed and is specifically for a RasPi4B
using the Raspberry Pi OS (64 bit) 2022-09-22. You may need to
modify it based on your setup or desires.

-----

Step 1: Prepare RasPiOS

Note: Make sure that you have internet access, I plug an ethernet
cable into my Pi4B.

Using the Raspberry Pi Imager (1.72 in use for this guide)

Erase 32 GB sd card

Set Advanced Options so that SSH is available on first boot. You may need
to set a wifi interface to start on first boot if you do not have an
ethernet cable to provide internet access during setup

Select and Burn Raspberry Pi OS (64 bit) 2022-09-22

Boot sd in RasPi4B

Check IP address of Pi4B with OpenWRT or by other means

Start Putty or other SSH app and enter Pi4B iP address

Log into Pi4B with login ID and password as set with Imager Advanced Options

Run:

$ sudo raspi-config

Interface Options > VNC > Yes

We should now be able to reboot and continue with intial setup with VNC.

Log into the RasPi4B with VNC and run:

Preferences > Raspberry Pi Configuration

You may configure other items but make sure to set the following:

Localization > Set WLAN Country to your country code (or FR if you are on the way to FR)

Remember to update, upgrade and autoremove before rebooting:

$ sudo apt update
$ sudo apt upgrade ($ sudo apt full-upgrade will get a firmware upgrade)
$ sudo apt autoremove (to clean house)
$ sudo reboot

-----

Step 2: Install Git and the build dependencies

$ sudo apt install git bc bison flex libssl-dev make libncurses5-dev

Note: libncurses5-dev is required for menuconfig (not used in this guide yet.)

Note: git, bc and make will show as already installed for the RasPiOS version
that I am using in this guide but that is okay.

-----

Step 3: Download the sources

Note: change "rpi-5.19.y" to the branch you want in the command below.

$ git clone --depth=1 --branch rpi-5.19.y https://github.com/raspberrypi/linux

Note: Omitting --depth=1 will download the entire repository, including the
full history of all branches, but this takes much longer and occupies much
more storage.

Note: Refer to the original GitHub repository for information about the
available branches.

https://github.com/raspberrypi/linux

-----

Step 4: Prepare the kernel configuration

Prepare the default configuration by running the following commands.

Note: I'll add additional configuration options in the future.

Warning: Once you have moved into the linux subdirectory below, you
need to stay there until complete.

$ cd linux
$ KERNEL=kernel8
$ make bcm2711_defconfig

Note: This is the point where you can open .config and make changes or
modify the kernel source before compiling. [1]

-----

Step 5: Build the Kernel

$ make -j4 Image.gz modules dtbs

-----

Step 6: Install the kernel, modules, and Device Tree blobs

$ sudo make modules_install

$ sudo cp arch/arm64/boot/dts/broadcom/*.dtb /boot/
$ sudo cp arch/arm64/boot/dts/overlays/*.dtb* /boot/overlays/
$ sudo cp arch/arm64/boot/dts/overlays/README /boot/overlays/

$ sudo cp arch/arm64/boot/Image.gz /boot/kernel8.img

Reboot. Your Raspberry Pi should be running your freshly-compiled kernel!

Remember to upgrade the firmware!

-----

fmunteanu, 2022-11-05:
I'm going to close this, I created https://github.com/morrownr/USB-WiFi/issues/144, with a kernel supported ComFast CF-926AC V2 adapter.

closed 2022-11-05 by fmunteanu