morrownr/USB-WiFi #142: APB: Check your mt7612u based adapter with Linux kernel 6.1 rc* (if you have time)

morrownr, 2022-10-31:
This reason for this message is that I am seeing a system lockup when trying to use mt7612u based adapters. Yes, device drivers can do that. I was seeing it kernel 6.1 rc1 and am still seeing it in 6.1 rc3. Would like confirmation. This only started with kernel 6.1.

Thanks,

Nick

morrownr, 2022-10-31:
Note: This could be something else going on in this specific test system so I am asking other that have the ability to upgrade to kernel 6.1 to test before I spend time gathering info to file a bug report.

lestcape, 2022-10-31:
I have that problem too, but i was thinking was another patch i was tested. Can this be related some how?

https://gitlab.freedesktop.org/drm/amd/-/issues/2236

morrownr, 2022-10-31:
@lestcape 

> Can this be related some how?

I don't know but I don't think so. The problem I am seeing is only with adapters based on the mt7612u chipset. My adapter with a mt7921au chipset is working better than ever on the same kernel 6.1. In fact, I have several adapters with various chipsets and I am only seeing this lock up problem with mt7612u chipset based adapters on kernel 6.1. 

Hopefully we have some other folks roll in with additional information so we can see if this widespread and speciifc to the chipset and kernel we are talking about. There could have been a patch that is not working well with the mt7612u chipset.

bjlockie, 2022-11-01:
This was before I associated:
```
$ uname -a
Linux me-aspiretc281 6.1.0-060100rc3-generic #202210301931 SMP PREEMPT_DYNAMIC Sun Oct 30 23:40:22 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

[  354.214752] usb 5-3: New USB device strings: Mfr=2, Product=3, SerialNumber=4
[  354.214758] usb 5-3: Product: Wireless 
[  354.214762] usb 5-3: Manufacturer: MediaTek Inc.
[  354.214766] usb 5-3: SerialNumber: 000000000
[  354.488160] usb 5-3: reset SuperSpeed USB device number 3 using xhci_hcd
[  354.514207] mt76x2u 5-3:1.0: ASIC revision: 76120044
[  354.660293] mt76x2u 5-3:1.0: ROM patch build: 20141115060606a
[  354.876618] mt76x2u 5-3:1.0: Firmware Version: 0.0.00
[  354.876632] mt76x2u 5-3:1.0: Build: 1
[  354.876637] mt76x2u 5-3:1.0: Build Time: 201507311614____
[  356.321129] ieee80211 phy1: Selected rate control algorithm 'minstrel_ht'
[  356.323917] usbcore: registered new interface driver mt76x2u
[  356.424442] mt76x2u 5-3:1.0 wlx00c0caaa9c2b: renamed from wlan0
```
That is the Alfa.

I associated and the system locked up within seconds.

morrownr, 2022-11-01:
> I associated and the system locked up within seconds.

Exactly what I am seeing. Within seconds of association, hard lock, pull the plug.

The only combination I am seeing this with is kernel 6.1 and mt7612u chipset.
Distro: Mint 21 (based on Ubuntu 20.04)
Kernel 6.1 rc1-rc3
Intel i7
Did not see this problem with kernel 6.0 and earlier so it appears that a patch that went into 6.1 is the cause.

We really don't want 6.1 (LTS) released with this issue. @bjlockie , do you have time to post a [bug - mt7612u] to linux-wireless? You can reference this thread.

morrownr, 2022-11-01:
More testing: This time with a mt7610u chipset based adapter (am Alfa ACHM to be exact). 

With kernel 6.1, I am seeing the same problem-- about 3-4 seconds after association with an AP, hard lock, plug the plug.

So this issue appears to not just be with the mt7612u chipset. It appears with both the mt7612u and mt7610u chipsets.

morrownr, 2022-11-01:
@bjlockie 

Thanks for posting the bug. I guess with recent guidance it should have been [BUG] wifi: MT7612U.

It sure looks to me at this point that some patch for 6.1 is causing a problem.

morrownr, 2022-11-06:
@bjlockie 

This morning I saw the following patch on linux-wireless. I see it is cc'ed to you but others might be interested. We really need this problem fixed before 6.1 is released.

```
[PATCH wireless] wifi: mac8021: fix possible oob access in
ieee80211_get_rate_duration
Lorenzo Bianconi <lorenzo@kernel.org> Sun, Nov 6, 2022 at 4:30 AM
To: linux-wireless@vger.kernel.org
Cc: bjlockie@lockie.ca, toke@toke.dk, johannes@sipsolutions.net, nbd@nbd.name
Fix possible out-of-bound access in ieee80211_get_rate_duration routine
as reported by the following UBSAN report:
UBSAN: array-index-out-of-bounds in net/mac80211/airtime.c:455:47
index 15 is out of range for type 'u16 [12]'
CPU: 2 PID: 217 Comm: kworker/u32:10 Not tainted 6.1.0-060100rc3-generic
Hardware name: Acer Aspire TC-281/Aspire TC-281, BIOS R01-A2 07/18/2017
Workqueue: mt76 mt76u_tx_status_data [mt76_usb]
Call Trace:
<TASK>
show_stack+0x4e/0x61
dump_stack_lvl+0x4a/0x6f
dump_stack+0x10/0x18
ubsan_epilogue+0x9/0x43
__ubsan_handle_out_of_bounds.cold+0x42/0x47
ieee80211_get_rate_duration.constprop.0+0x22f/0x2a0 [mac80211]
? ieee80211_tx_status_ext+0x32e/0x640 [mac80211]
ieee80211_calc_rx_airtime+0xda/0x120 [mac80211]
ieee80211_calc_tx_airtime+0xb4/0x100 [mac80211]
mt76x02_send_tx_status+0x266/0x480 [mt76x02_lib]
mt76x02_tx_status_data+0x52/0x80 [mt76x02_lib]
mt76u_tx_status_data+0x67/0xd0 [mt76_usb]
process_one_work+0x225/0x400
worker_thread+0x50/0x3e0
? process_one_work+0x400/0x400
kthread+0xe9/0x110
? kthread_complete_and_exit+0x20/0x20
ret_from_fork+0x22/0x30
Reported-by: bjlockie@lockie.ca
Fixes: db3e1c40cf2f ("mac80211: Import airtime calculation code from mt76")

---
net/mac80211/airtime.c | 3 +++
1 file changed, 3 insertions(+)
diff --git a/net/mac80211/airtime.c b/net/mac80211/airtime.c
index 2e66598fac79..4ed05988131d 100644
--- a/net/mac80211/airtime.c
+++ b/net/mac80211/airtime.c
@@ -452,6 +452,9 @@ static u32 ieee80211_get_rate_duration(struct ieee80211_hw
*hw,
(status->encoding == RX_ENC_HE && streams > 8)))
return 0;
+ if (WARN_ON_ONCE(idx >= MCS_GROUP_RATES))
+ return 0;
+
duration = airtime_mcs_groups[group].duration[idx];
duration <<= airtime_mcs_groups[group].shift;
*overhead = 36 + (streams << 2);
--
2.38.1

```

bjlockie, 2022-11-07:
Thanks, I missed that because I delete all the patches on that list. :-)
I'll try rc4. :-)

morrownr, 2022-11-07:
I don't think it will be in RC4. There are follow on messages now where another guy is questioning whether mac80211 is the right place to fix the problem. I'm reading between the lines here but I get the feeling that a mac80211 patch has broken a few drivers and Lorenzo is patching mac80211 as a short term solution because the real fix may require firmware patches which come from the company. I suspect there are more than MT76 drivers that are broken. In fact, if I have time this week, I'm going to test everything I have here which is a significant job. I don't like hard lock, pull the plug situations.

bjlockie, 2022-11-15:
Is the patch definitely in 6.1rc5?

morrownr, 2022-11-15:
No, the patch is not in rc5. Not sure what the status is.

randomcodepanda, 2022-12-23:
So i'm using a MT7610U and updated to 6.1.1 in Arch Linux and it's a complete lockup a few seconds after connecting as other people have said, i'll use the LTS kernel (5.15) in the meanwhile but this is probably going to get a lot of people that use distros with newer kernels.

bjlockie, 2022-12-23:
I had hoped it would be fixed by now.
I'm surprised the primary developer hasn't seen it, it is easy to reproduce.

morrownr, 2022-12-23:
@bjlockie 

> I had hoped it would be fixed by now.

Me too. I'm reopening this issue so folks can see it.

Are you able to grab your old report to linux-wireless and add the above report to it while pointing out that this is a show stopper?

morrownr, 2022-12-23:
@bjlockie 

I think the proper title for a report like something like this would be:

[bug report} wifi: mt76: mt7612u/mt7610u 6.1.x - hard locking systems

morrownr, 2022-12-23:
@bjlockie 

If you are busy, I can send the report. Let me know.

morrownr, 2022-12-23:
@bjlockie 

I have some time this afternoon so I'll go ahead and report this.

bjlockie, 2022-12-24:
I'm slow, thanks for doing that.
I see there is a kernel patch.
Does the patch only get in a released kernel only after someone verifies the patch fixes the problem?

morrownr, 2022-12-24:
No problem. It is a time of year when most of us are busy to one degree or another.

> Does the patch only get in a released kernel only after someone verifies the patch fixes the problem?

I've been looking at the patch. The patch is not to the mt76 driver. It is a patch to mac80211 which affects almost all wireless so it has to be approved.

I'm traveling so it is not a good time for me to test the patch.

morrownr, 2022-12-24:
@randomcodepanda 

The proposed patch has to be approved by Johannes Berg as it is not a patch to the mt76 driver. It is a patch to the mac80211 driver which is a system wide wifi driver so care has to be taken. The patch was provided by a Mediatek dev. 

I can't say how much time this will take but the devs are very aware of it.

Nick

randomcodepanda, 2022-12-25:
That's great to hear @morrownr and @bjlockie , you guys do some awesome work for the Linux wifi users here.  
I'll update if there's a kernel update in the meanwhile.

morrownr, 2022-12-26:
@bjlockie @randomcodepanda 

I had time to apply the patch that needed to be tested. It worked. I could not see any side effects so we'll just have to see.

I'll recommend to stay with kernel 6.0 or earlier until such time as that patch works its way into 6.1 and later.

For what its worth. That patch is not to the Mediatek driver. That patch is to the mac80211 driver which is part of the wifi supporting stack. The patch was written by a Mediatek dev. 

Nick

bjlockie, 2022-12-27:
Why did the bug only affect mediatek devises?

morrownr, 2022-12-27:
It may not just affect Mediatek devices. I looked at the patch and it is not a trivial patch. We need more testing. I'm wondering if it would help if I posted a guide that would give specific instructions how to test.

bjlockie, 2022-12-28:
If mediatek devices are the only ones that cause a crash then wouldn't the developers know?
Can this be tested in a VM?

morrownr, 2022-12-29:
> If mediatek devices are the only ones that cause a crash then wouldn't the developers know?

Depends on what the devs are working on and whether what caused the problem is something that came from them. This problem does not appear to have been caused by a patch from a Mediatek dev.

> Can this be tested in a VM?

It probably can but I'm the wrong person to answer as I run 100% bare metal here.

morrownr, 2022-12-30:
V5 of the patch was submitted today. Not sure when it will be accepted:

[PATCH v5] wifi: mac80211: fix initialization of rx->link and rx->link_sta

```

There are some codepaths that do not initialize rx->link_sta properly. This
causes a crash in places which assume that rx->link_sta is valid if rx->sta
is valid.
One known instance is triggered by __ieee80211_rx_h_amsdu being called from
fast-rx. It results in a crash like this one:

 BUG: kernel NULL pointer dereference, address: 00000000000000a8
 #PF: supervisor write access in kernel mode
 #PF: error_code(0x0002) - not-present page PGD 0 P4D 0
 Oops: 0002 [#1] PREEMPT SMP PTI
 CPU: 1 PID: 506 Comm: mt76-usb-rx phy Tainted: G            E      6.1.0-debian64x+1.7 #3
 Hardware name: ZOTAC ZBOX-ID92/ZBOX-IQ01/ZBOX-ID92/ZBOX-IQ01, BIOS B220P007 05/21/2014
 RIP: 0010:ieee80211_deliver_skb+0x62/0x1f0 [mac80211]
 Code: 00 48 89 04 24 e8 9e a7 c3 df 89 c0 48 03 1c c5 a0 ea 39 a1 4c 01 6b 08 48 ff 03 48
       83 7d 28 00 74 11 48 8b 45 30 48 63 55 44 <48> 83 84 d0 a8 00 00 00 01 41 8b 86 c0
       11 00 00 8d 50 fd 83 fa 01
 RSP: 0018:ffff999040803b10 EFLAGS: 00010286
 RAX: 0000000000000000 RBX: ffffb9903f496480 RCX: 0000000000000000
 RDX: 0000000000000000 RSI: 0000000000000000 RDI: 0000000000000000
 RBP: ffff999040803ce0 R08: 0000000000000000 R09: 0000000000000000
 R10: 0000000000000000 R11: 0000000000000000 R12: ffff8d21828ac900
 R13: 000000000000004a R14: ffff8d2198ed89c0 R15: ffff8d2198ed8000
 FS:  0000000000000000(0000) GS:ffff8d24afe80000(0000) knlGS:0000000000000000
 CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
 CR2: 00000000000000a8 CR3: 0000000429810002 CR4: 00000000001706e0
 Call Trace:
  <TASK>
  __ieee80211_rx_h_amsdu+0x1b5/0x240 [mac80211]
  ? ieee80211_prepare_and_rx_handle+0xcdd/0x1320 [mac80211]
  ? __local_bh_enable_ip+0x3b/0xa0
  ieee80211_prepare_and_rx_handle+0xcdd/0x1320 [mac80211]
  ? prepare_transfer+0x109/0x1a0 [xhci_hcd]
  ieee80211_rx_list+0xa80/0xda0 [mac80211]
  mt76_rx_complete+0x207/0x2e0 [mt76]
  mt76_rx_poll_complete+0x357/0x5a0 [mt76]
  mt76u_rx_worker+0x4f5/0x600 [mt76_usb]
  ? mt76_get_min_avg_rssi+0x140/0x140 [mt76]
  __mt76_worker_fn+0x50/0x80 [mt76]
  kthread+0xed/0x120
  ? kthread_complete_and_exit+0x20/0x20
  ret_from_fork+0x22/0x30

Since the initialization of rx->link and rx->link_sta is rather convoluted
and duplicated in many places, clean it up by using a helper function to
set it.

Fixes: ccdde7c74ffd ("wifi: mac80211: properly implement MLO key handling")
Fixes: b320d6c456ff ("wifi: mac80211: use correct rx link_sta instead of default")

---
v5: fix sdata link assignment, fix sdata when receiving on multiple interfaces
v4: fix regression in handling mgmt frames with AP_VLAN
v3: include crash log
v2: fix uninitialized variable
 net/mac80211/rx.c | 222 +++++++++++++++++++++-------------------------
 1 file changed, 99 insertions(+), 123 deletions(-)

diff --git a/net/mac80211/rx.c b/net/mac80211/rx.c
index 7e3ab6e1b28f..6ccc487bad7f 100644
--- a/net/mac80211/rx.c
+++ b/net/mac80211/rx.c
@@ -4049,6 +4049,58 @@ static void ieee80211_invoke_rx_handlers(struct ieee80211_rx_data *rx)
 #undef CALL_RXH
 }
[... 364 lines cut ...]
                        prev_sta = sta;
                }

                if (prev_sta) {
-                       if ((status->link_valid &&
-                            !ieee80211_rx_is_valid_sta_link_id(&prev_sta->sta,
-                                                               link_id)) ||
-                           (!status->link_valid && prev_sta->sta.mlo))
+                       rx.sdata = prev_sta->sdata;
+                       if (!ieee80211_rx_data_set_sta(&rx, &prev_sta->sta,
+                                                      link_id))
                                goto out;

-                       rx.link_id = status->link_valid ? link_id : -1;
-                       rx.sta = prev_sta;
-                       rx.sdata = prev_sta->sdata;
+                       if (!status->link_valid && prev_sta->sta.mlo)
+                               goto out;

                        if (ieee80211_prepare_and_rx_handle(&rx, skb, true))
                                return;
--
2.38.1

```

randomcodepanda, 2023-01-02:
Updated today to 6.1.2 and it's still freezing, hopefully it makes it on the next release.

morrownr, 2023-01-02:
It sure would be good for the patch to go into 6.1 and 6.2 as soon as possible but we need to respect that this is going into a part if the kernel that is heavily used by many drivers so a mistake would not be good.

giga, 2023-01-04:
Could this apply to mt7921 as well? I have a recently acquired wifi adapter that I can't get to work. I'm trying to use it in ap mode, hostapd runs fine but when another computer tries to connect the AP freezes.

morrownr, 2023-01-04:
> Could this apply to mt7921 as well? I have a recently acquired wifi adapter that I can't get to work. I'm trying to use it in ap mode, hostapd runs fine but when another computer tries to connect the AP freezes.

@giga 

I have not seen this problem with the mt7921. Recommend you start a new issue and describe what you are seeing. A new issue with a good title will attract users that may be able to help.

sha5672, 2023-01-09:
@morrownr 

Hello, I am using Arch Linux 6.1.4 with a new Alfa AWUS036ACM and am experiencing a crash within a few seconds of logging in. What can I do until the issue is fixed? Is there any way that I can downgrade to 6.0?

morrownr, 2023-01-09:
Hi @sha5672 

> Hello, I am using Arch Linux 6.1.4 with a new Alfa AWUS036ACM and am experiencing a crash within a few seconds of logging in

Yes, this is an ugly issue. Programmers call it a null pointer dereference. On my systems, it is pull the plug time...and remove the adapter before rebooting. The problem is happening with the 7612u and 7610u chipsets. This started with kenel 6.1 by a patch from a Intel dev. A patch I will link below has been submitted by a Mediatel dev but the patch is not to a Mediatek driver, it is to mac80211 which is a Linux Wireless system stack driver and that is delaying the merging to mainline as a gatekeeper has to be satisfied it it not going to cause system wide problems. In looking at the patch and the reasons behind it, I tend to agree with what is being done. It would have helped if it had not happened during the holiday season. Hopefully the patch will be merged soon.

> What can I do until the issue is fixed?

Downgrade to any kernel prior to 6.1.

> Is there any way that I can downgrade to 6.0?

I'm sure there is but my expertise is with Debian/Ubuntu based systems so you don't want me to help you downgrade your kernel. You probably should go to the appropriate Arch forum and ask how to downgrade.

Below is a link to the patch:

https://lore.kernel.org/all/20221230200747.19040-1-nbd@nbd.name/

I have tested it on 6.1 and 6.2-rc2 and it works. I hope it is merged soon as this could get ugly as more distros upgrade to 6.1.

Nick

randomcodepanda, 2023-01-10:
> @morrownr
> 
> Hello, I am using Arch Linux 6.1.4 with a new Alfa AWUS036ACM and am experiencing a crash within a few seconds of logging in. What can I do until the issue is fixed? Is there any way that I can downgrade to 6.0?

If you are not on some bleeding edge hardware that needs newer kernels the best bet for stability in Arch right now would be to install the linux-lts package and update your bootloader so you can boot with linux-lts which is on 5.15.

It is possible to downgrade to 6.0 if you still have the old package on your cache or if you download it from the arch archive and install it with pacman -U, but then you would have to add the linux package to the ignorepkg line in pacman.conf so it doesn't get upgraded on a regular upgrade.

It's easier to just use the linux-lts kernel if that is compatible with the rest of your hardware.

adelias, 2023-01-30:
This is now fixed in Debian kernel 6.1.8-1. From the changelog:
 `- wifi: mac80211: fix initialization of rx->link and rx->link_sta
      (Closes: #1029816)`