morrownr/USB-WiFi #143: Support Req: MT7921K over m.2 usb

fhteagle, 2022-11-05:
Greetings all. I purchased an inexpensive m.2 form factor MT7921K card, but am having issues getting the interface to actually show up. Google search, looking through the issues here gave me some ideas, but no success. Time to ask for help.

- Hardware: Framework mainboard, Intel i5-1135G7 CPU
- OS: EndeavourOS (arch linux based), fully updated
- Kernel: linux 6.0.6-arch1-1  (not -zen or any other variant, just basic linux package kernel)
- lsusb: https://pastebin.com/0biVZQNu   (Note, the USB vendor ID of this mediatek board is different than the RZ608)
- journalctl lines where usb card is detected: https://pastebin.com/GrD2TfsQ
- tried force loading mt7921e kernel module on boot using systemd, adding udev rule, and modprobe alias: https://pastebin.com/QySu6HGM
- lsmod output shows the mt7921e module and several dependencies loaded at boot: https://pastebin.com/41PkHdv7

but the adapter does not show up in ```iw list``` or ```ip addr show``` lists

Not sure what I am missing, forgetting, etc. Let me know what else you might need for troubleshooting.

End goal is to make a tri band wireless AP using this card, but first I need it to actually show up in interfaces. If this is the wrong hardware for this, let me know as I am well within the return window and can send it back and get something else.

Thanks.

joshschmelzle, 2022-11-05:
0e8d:0608 from your paste is the Bluetooth radio of the module. The Wi-Fi radio uses PCIe not USB.

bjlockie, 2022-11-05:
I think my mt7921k pci is actually a rz608 m.2 card in a pci adapter.
My lspci shows it:
04:00.0 Network controller: MEDIATEK Corp. MT7921K (RZ608) Wi-Fi 6E 80MH

As Nick said, the usb is the bluetooth part.
Bus 003 Device 003: ID 0e8d:0608 MediaTek Inc. Wireless_Device
0e8d is MediaTek and 0608 is an rz608.
MediaTek annoyingly seem to label everything usb as "Wireless_Device".
I didn't get anywhere with trying to get the file of usbids updated.
The response seems to be "get the vendor to stop being lazy".

If it doesn't show up in lspci then try reseating it.
If the m.2 socket has never been used, try compressed ir on the socket.

Something should show up in lspci.
I use kernel 6.1rc3 but I did have 5.19 and it was fine.

fhteagle, 2022-11-05:
I am not seeing the mt7921k in my lspci: https://pastebin.com/GZ5VeNw8

joshschmelzle, 2022-11-05:
Here is what filtering dmesg on the mt7921k slot looks like for me. Perhaps search dmesg for `14c3:0608`, `14c3`, or `0608` and check results for `pci` and perhaps any relevant errors to look at.

```
dmesg | grep 05:00.0
[    1.348896] pci 0000:05:00.0: [14c3:0608] type 00 class 0x028000
[    1.348990] pci 0000:05:00.0: reg 0x10: [mem 0x00000000-0x000fffff 64bit pref]
[    1.349056] pci 0000:05:00.0: reg 0x18: [mem 0x00000000-0x00003fff 64bit pref]
[    1.349120] pci 0000:05:00.0: reg 0x20: [mem 0x00000000-0x00000fff 64bit pref]
[    1.349422] pci 0000:05:00.0: supports D1 D2
[    1.349445] pci 0000:05:00.0: PME# supported from D0 D1 D2 D3hot D3cold
[    1.350540] pci 0000:05:00.0: BAR 0: assigned [mem 0x600300000-0x6003fffff 64bit pref]
[    1.350599] pci 0000:05:00.0: BAR 2: assigned [mem 0x600400000-0x600403fff 64bit pref]
[    1.350658] pci 0000:05:00.0: BAR 4: assigned [mem 0x600404000-0x600404fff 64bit pref]
[    9.178027] mt7921e 0000:05:00.0: enabling device (0000 -> 0002)
[    9.201673] mt7921e 0000:05:00.0: ASIC revision: 79610010
[    9.291220] mt7921e 0000:05:00.0: HW/SW Version: 0x8a108a10, Build Time: 20220908210919a
[    9.548453] mt7921e 0000:05:00.0: WM Firmware Version: ____010000, Build Time: 20220908211021
```

fhteagle, 2022-11-05:
Not seeing anything relevant in dmesg: https://pastebin.com/x36bP4rA

Starting to wonder if this particular unit is a defective card?

joshschmelzle, 2022-11-05:
What does `over m.2 usb` mean? Is the mt7921k in a m.2 slot on the host? or in some sort of m.2 to USB dongle and connected via USB?

fhteagle, 2022-11-05:
in the m.2 slot on the mainboard, yes.

joshschmelzle, 2022-11-05:
Very odd... do you happen to have any other Wi-Fi modules available to try and check if they show up in lspci?

fhteagle, 2022-11-05:
Yeah, I can swap in an AX210 into the slot, and try the mediatek in a different Framework mainboard as well. Probably tomorrow before I can get to it, though. 

Thanks for the help.

bjlockie, 2022-11-05:
On 2022-11-04 21:28, D.H wrote:
>
> Not seeing anything relevant in dmesg: https://pastebin.com/x36bP4rA
>
> Starting to wonder if this particular unit is a defective card?
>
Except the Bluetooth part of it shows up.

bjlockie, 2022-11-05:
Where did you plug it in?
I couldn't find the manual.
Maybe you plugged it in a USB4 M.2 slot and maybe you need a PCI M.2 slot?
https://frame.work/ca/en/products/mainboard?v=FRANFG000A
Has a community support forum.

fhteagle, 2022-11-05:
So I swapped the known good AX210 into the development Framework mainboard, and moved the MT7921K from the dev board into my regular Framework laptop. AX210 in dev board did not show up in ```lspci``` either. MT7921K in regular laptop is up and working as expected out of the box in arch 6.0.7 kernel. Looks like the Framework dev board I received has a defective m.2 slot. 

So, support request completed, and thank you all very much for your time and perspective to help troubleshoot this!

Last thing before I close this, I will need to switch to a USB dongle to make the access point for this project. Any good hardware recommendations? I vastly prefer 6E capable AP, with swappable u.fl or (rp-)sma antenna connections, but do not care regarding USB-A or USB-C form factor. I see the Comfast CF-953AX mentioned elsewhere here (it is capable of AP on 6GHZ, correct?), but are there any other good choices? 

Thanks

bjlockie, 2022-11-05:
The Comfast CF-953ax has attached antennas.
The 952ax has built-in antennas.
There is not much out there yet.
Hopefully in 2023.

morrownr, 2022-11-05:
> The 952ax has built-in antennas.

The 952ax is based on the rtl8832au chipset. Bad chipset for Linux users,

> are there any other good choices?

Unless you are in a big hurry, wait for the Alfa adapters.

fhteagle, 2022-11-06:
Thanks. I have a CF-953AX coming for now, will keep an eye out for ALFA to release more hardware in the future. I have been pretty pleased with their stuff in the past.

closed 2022-11-06 by fhteagle