morrownr/USB-WiFi #144: ComFast CF-926AC V2 with OpenWRT

fmunteanu, 2022-11-05:
First, thank you for the useful documentation. Based on suggestions found here, I purchased another adapter, [ComFast CF-926AC V2](https://www.aliexpress.com/item/32972863958.html) (MT7612U chip with in-kernel 4.19+ drivers). It will be used with a Raspberry Pi 4B.

ComFast CF-926AC V2 (top) vs. ComFast CF-953AX (bottom):

[image]

On RaspiOS Lite (64bits), the new `wlan1` interface is detected right away on a clean install, no need for any drivers:

```sh
# ip a show wlan1
11: wlan1: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state DOWN group default qlen 1000
    link/ether e0:e1:a9:17:37:2f brd ff:ff:ff:ff:ff:ff

# inxi -EN
Network:   Device-1: MediaTek MT7612U 802.11a/b/g/n/ac Wireless Adapter type: USB driver: mt76x2u
Bluetooth: Message: No Bluetooth data was found.
```

Next, I installed `openwrt-22.03.2-bcm27xx-bcm2711-rpi-4-squashfs-factory.img.gz` for [Pi 4B](https://downloads.openwrt.org/releases/22.03.2/targets/bcm27xx/bcm2711/openwrt-22.03.2-bcm27xx-bcm2711-rpi-4-squashfs-factory.img.gz) and configured the initial setup (wwan and wireless to update the OS):

/etc/config/network

```sh
# cat /etc/config/network
config interface 'loopback'
	option device 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'
	option ula_prefix 'fdd2:5e70:072e::/48'

config device
	option name 'br-lan'
	option type 'bridge'
	list ports 'eth0'

config interface 'lan'
	option device 'br-lan'
	option proto 'static'
	option ipaddr '10.10.100.1'
	option netmask '255.255.255.0'
	option ip6assign '60'
	option force_link '1'

config interface 'wwan'
	option proto 'dhcp'
	option peerdns '0'
	option dns '1.1.1.1'
```

Set `option input` to `ACCEPT` for `wan` zone in /etc/config/firewall:

```
# grep "option name 'wan'" -B1 -A8 /etc/config/firewall
config zone
	option name 'wan'
	option input 'ACCEPT'
	option output 'ACCEPT'
	option forward 'REJECT'
	option masq '1'
	option mtu_fix '1'
	list network 'wan'
	list network 'wan6'
	list network 'wwan'
```

OS update, note the weird output for `0e8d:2870` which is the MediaTek MT7612U WiFi adapter:

```
# opkg update
# opkg install kmod-mt76x2u kmod-usb3 usbutils
# reboot

# lsusb
Bus 001 Device 003: ID 0e8d:2870 Љ Љ
Bus 002 Device 001: ID 1d6b:0003 Linux 5.10.146 xhci-hcd xHCI Host Controller
Bus 001 Device 002: ID 2109:3431  USB2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux 5.10.146 xhci-hcd xHCI Host Controller
```

The WiFi adapter `wlan1` is missing:

```
# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq master br-lan state UP qlen 1000
    link/ether e4:5f:01:a0:55:36 brd ff:ff:ff:ff:ff:ff
3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP qlen 1000
    link/ether e4:5f:01:a0:55:37 brd ff:ff:ff:ff:ff:ff
    inet 192.168.2.16/27 brd 192.168.2.31 scope global wlan0
       valid_lft forever preferred_lft forever
    inet6 fe80::e65f:1ff:fea0:5537/64 scope link
       valid_lft forever preferred_lft forever
4: br-lan: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP qlen 1000
    link/ether e4:5f:01:a0:55:36 brd ff:ff:ff:ff:ff:ff
    inet 10.10.100.1/24 brd 10.10.100.255 scope global br-lan
       valid_lft forever preferred_lft forever
    inet6 fdd2:5e70:72e::1/60 scope global noprefixroute
       valid_lft forever preferred_lft forever
    inet6 fe80::e65f:1ff:fea0:5536/64 scope link
       valid_lft forever preferred_lft forever
```

I'm stuck, I don't know how to proceed. Thank you for your help.

morrownr, 2022-11-05:
Hi @fmunteanu 

> note the weird output for 0e8d:2870 which is the MediaTek MT7612U WiFi adapter

The above indicates that the 926AC V2 is a multi-state adapter. With multi-state adapters, a nice little utility to automatically switch state is needed. In OpenWRT 22.03.2, look for the package called:

usb-modeswitch

Install it and reboot. Comfast tends to use the same multi-state info over and over so once you reboot, the adapter should then show as a wifi adapter. A little background info can be found at Main Menu item 7:

https://github.com/morrownr/USB-WiFi/blob/main/home/How_to_Modeswitch.md

You should not need to follow the above instrunctions as usb-modeswitch should work without anything from you except installing the package. The reason it worked on RasPiOS is that the RasPiOS includes usb-modeswitch whereas OpenWRT has to hold back on what is installed by default because it has to run on many routers that have very little space.

Something to ask in return: So far we do not have a review on the CF-953AX in the main list of in-kernel adapters:

https://github.com/morrownr/USB-WiFi/blob/main/home/USB_WiFi_Adapters_that_are_supported_with_Linux_in-kernel_drivers.md

Would you mind writing a review?

Also, if you think the CF-926AC V2 would be something that should be included in the list, let me know.

Nick

fmunteanu, 2022-11-05:
@morrownr Nick, sure thing, I will write a review for CF-953AX. Can you please provide an OS name that I can install on Pi 4B and has the kernel 5.19+? This will allow me to test the adapter out of the box.

I thought that ComFast CF-926AC V2 model is a single-state, like CF-953AX. Are there any ComFast single-state adapters with MT7612U chip?

fmunteanu, 2022-11-05:
@morrownr I installed `usb-modeswitch` and looked at your guide, but `usb-modeswitch-data` is not available in OpenWRT. Also the binary is called `submode`:

```sh
# usbmode -K -W -v 0e8d -p 2870
usbmode: unrecognized option: K
Usage: usbmode <command> <options>
Commands:
	-l		List matching devices
	-s		Modeswitch matching devices

Options:
	-v		Verbose output
	-c <file>	Set configuration file to <file> (default: /etc/usb-mode.json)

# usbmode -l
Found device: 0e8d:2870 (Manufacturer: "?", Product: "?", Serial: "?")
```

Nothing changed after reboot.

morrownr, 2022-11-05:
>  I will write a review for CF-953AX

Thanks. A lot of people would appreciate it. USB-WiFi averages about 600 hits per day.

> Can you please provide an OS name that I can install on Pi 4B and has the kernel 5.19+? This will allow me to test the adapter out of the box.

I assume you are talking about the 953AX? If so, with OpenWRT, you don't need kernel 5.19+. There is overlap as some of the Linux kernel Mediatek devs and also OpenWRT devs. Said devs have already backported the mt7921u driver to OpenWRT 22.03.

kmod-mt7921u

Something to note: The firmware happens to be in package kmod-mt7921e so you either have to install that package also or manually install the firmware ( see issue 135 here - How to install latest mt7921au firmware on OpenWRT ). I can say that my CF-951AX works well with OpenWRT 22.03.2 on my Zyxel router. I have tested the 2.4 GHZ and 5 GHz bands and they work well but the range is only average. I'm waiting for Alfa to kick a high power adapter with the mt7921au and I hear that could happen in the first quarter of next year. I have played with 6 GHz but I do not currently have any other devices that do 6 GHz so I really don't know if it is working. Be aware that Luci is not ready for 6 GHz... you have to do the: 

nano /etc/config/wireless

... thing to configure for now.

fmunteanu, 2022-11-05:
@morrownr Nick, I experience this weird bug installing packages:

```sh
# opkg install kmod-mt7921u
Unknown package 'kmod-mt7921u'.
```

It happened before for `kmod-mt76x2u`, after wiping the MicroSD card and starting fresh, the package was there

So I will start again from scratch and install the package, to use it with CF-953AX adapter. I hope everything will work after that, my goal was to use anyways the CF-953AX adapter. Do I need the `mod-usb3` package, in OpenWRT?

List of packages to install with CF-953AX adapter:

```sh
opkg update
opkg install kmod-mt7921u usbutils
reboot
```

morrownr, 2022-11-05:
>  I installed usb-modeswitch and looked at your guide, but usb-modeswitch-data is not available in OpenWRT. Also the binary is called usbmode:

Opps. Did not think about that. The OpenWRT devs have to be very careful about size so often they supply reduced size versions and in this case, it looks like we need to make our own data file. The regular data file is big. I have a multi-state Comfast mt7612u adapter and I run OpenWRT 22.03 so let me do some investigating.

morrownr, 2022-11-05:
> opkg update
> opkg install kmod-mt7921u usbutils
> reboot

I would suggest:

 opkg install kmod-mt7921u kmod-mt7921e

The need to install kmod-mt7921e is only because of firmware and I pointed you to issue 135 if you want to do it manually. There is a bug already filed to get this firmware issue taken care of. Not sure when it will be worked.

fmunteanu, 2022-11-05:
For CF-953AX review, should I create a new issue? If you have some sort of guidelines how to test everything, please let me know, so I follow them.

morrownr, 2022-11-05:
Something might come in handy with your project:

https://www.amazon.com/dp/B07S6B5X76

I use them and recommend them.

fmunteanu, 2022-11-05:
I have them. 😄 

[image]

morrownr, 2022-11-05:
Cool pic!

> For CF-953AX review, should I create a new issue?

You can or to put in with the entry for the adapter in Main Menu item 2, you could do a Pull Request.

> If you have some sort of guidelines how to test everything, please let me know, so I follow them.

I don't have an official guide (yet) but here are some things to talk about if able:

- Your impression of the quality of the adapter.

- Performance in managed, master and monitors mode if possible, as those are the most heavily used modes.

- Are the antenna removable?

- How is the range?

- Does it work well with a Raspberry Pi? If so, which Pi and what distro?

- Any problems that you have encountered?

- Overall thoughts about the adapter. What do want people to know.

If you need helping testing anything, ask.

morrownr, 2022-11-05:
Oh, and you don't have to do a complete review all at once. You add enough to be handy for folks and then add as you want.

If anyone else wants to do a separate review, that is welcome also.

morrownr, 2022-11-14:
I seem to have missed this question.

> Are there any ComFast single-state adapters with MT7612U chip?

Not that I am aware of. Comfast has several adapters with Mediatek chipset but the only two that I am aware of that are single state are the CF-953AX and CF-951AX.

I've been sick for a few days and am behind. How are doing with your project at this point?

fmunteanu, 2022-11-14:
I got everything working with OpenWrt, so far us very stable. There’s still the issue of the CF-953AX being disabled at startup but I fixed that with a custom service. I’ll update the discussion, I hope you are feeling better soon.

closed 2022-11-14 by fmunteanu

morrownr, 2022-11-15:
Thanks. Slowly improving. I'm curious as to what is causing the 953 to not come up on startup. I'm not seeing that on OpenWRT but then I am on a ZyXEL router so this might be specific to the Pi distro you are using. Have you thought about chasing down the maintainer of the Pi distro? I know you can file a bug but sometimes a chat can help things along.

fmunteanu, 2022-11-16:
You mean the OpenWrt maintainer? I don’t know how to reach him, should I post a forum thread into Developer forums?