morrownr/USB-WiFi #152: WiFi for Linux devices with the best range

girl313, 2022-11-18:
Hi hi,

I need to connect two RPi4b far apart via Wi-Fi, I am thinking about "802.11n (Wi-Fi 4) 2,4GHz" vs "802.11ax (Wi-Fi 6E) 2,4GHz" 
But I don't know if Wi-Fi 6E on 2.4Ghz has a better range than Wi-Fi 4 2.4Ghz (maybe a negligible difference).
For the distance I need, I tried to connect these two devices: "RT-AC53" and "AWUS036ACM" the connection has been established but is very unstable.

"RT-AC53" and "AWUS036ACM" from both these devices, AWUS036ACM has the better range.

1,
I got an idea, buy the same device "AWUS036ACM" and connect them together via "802.11n (Wi-Fi 4) 2,4GHz"

And since I don't need 5Ghz, is it possible to somehow reset these devices to only 2.4Ghz (optimize)?
I can't tell if one antenna is 2.4Ghz and the other is 5Ghz.

2,
Or can you please recommend some two devices that I can use long distance and that support Linux well?

[image]

thanks all.

morrownr, 2022-11-18:
Hi @girl313 

> I got an idea, buy the same device "AWUS036ACM"

Based on what you are saying, not the direction I would go. The ACM is a fine general purpose adapter but it is not a "very long range" adapter.

> And since I don't need 5Ghz, is it possible to somehow reset these devices to only 2.4Ghz (optimize)?

The radio's in these adapters can be single band or dual band or triple band. There is nothing like reseting. The software tells them what to do.

> Or can you please recommend some two devices that I can use long distance and that support Linux well?

Sure. I'll throw some info out but I really need to know what throughput capable you are looking for and what your budget?

To think about:

See Main Menu: The Short List

Very long range:

ALFA AWUS036ACHM
EDUP EP-MS8551 (no AP mode)
Panda PAU08

See Main Menu: item 2 as it contains more details on each of the above adapters.

It would help if I had more info about what you are trying to accomplish.

Regards

morrownr, 2022-11-18:
> I am trying to connect two RPi4b at this distance:

I can't say that the distance would be a big problem with good adapters but unless you get both antennas outside with open air between them, you are probably looking at doing the impossible due to the numerous walls, floors and ceilings that will cause signal problems. Are you allowed to mount antennas externally?

girl313, 2022-11-18:
According to the video here, these basic models should have the best range: https://www.youtube.com/watch?v=lIM9yC8U42E
AWUS036ACH = RTL8812AU = 2x Antenne
AWUS036ACHM  = MT7610U = 1x Antenne

but I dont know.. RTL8812AU I wonder if it is a problem to start this firmware on ARM Raspberry Pi OS and ARM Manjaro.

girl313, 2022-11-18:
> I can't say that the distance would be a big problem with good adapters but unless you get both antennas outside with open air between them, you are probably looking at doing the impossible due to [...]

good question, my mistake for not writing it sorry. No, it's not possible allowed to mount antenna external example: balcony.

morrownr, 2022-11-18:
> No, it's not possible allowed to mount antenna external example: balcony.

If that is the case, then I can't see a wifi solution to your problem. You may have to rethink solutions for what you are trying to accomplish.

Regards

girl313, 2022-11-18:
thanks thanks.
the last stupid question, is one of the solutions teoretic better?
https://i.imgur.com/B2FIRVo.png

girl313, 2022-11-18:
I think I found the answer to my question: https://superuser-com.translate.goog/questions/1595771/why-is-the-range-of-a-mobile-wi-fi-hotspot-shorter-than-that-of-a-router?_x_tr_sl=en&_x_tr_tl=cs&_x_tr_hl=cs&_x_tr_pto=sc

I decided to buy two devices "AWUS036ACHM" and one "ID36ACH" = (Realtek RTL8812AU) for testing.
I will try to use this: https://github.com/morrownr/8812au-20210629

If I had a problem with the drivers, can I ask you Master? :-)

riker65, 2022-11-22:
HI, will AWUS051NH work?
I am confused about the chipset
Looks like AWUS051NH  is having Realtek RTL8188EUS

Thanks

closed 2022-11-25 by morrownr