morrownr/USB-WiFi #154: Realtek RTL8822bu in USB_WiFi_Chipsets.md

OS2User, 2022-11-27:
The Realtek RTL8822bu is listed in your table with a USB3 / 5 Gbps Interface.
According to   www.realtek.com/en/products/communications-network-ics/item/rtl8822bu   the "Host interface" is "USB2.0 for WLAN and BT controller"
Please kindly update that entry.

Thank you very much for all the hard work you are putting into this site, which is very much appreciated.
I just came across it, when searching for driver support for 2 Realtek RTL8812bu WiFi5 adapters, since I was under the apparently wrong impression, that they already got in-kernel driver support around Linux 5.17+, yet is still missing in 6.0.9.

morrownr, 2022-11-27:
Hi @OS2User 

> Please kindly update that entry.

Done. Please confirm.

> I just came across it, when searching for driver support for 2 Realtek RTL8812bu WiFi5 adapters, since I was under the apparently wrong impression, that they already got in-kernel driver support around Linux 5.17+, yet is still missing in 6.0.9.

I monitor the linux-wireless mailing list so I can update you. This effort has been under way since early this year. The developer that is working on it is a community developer, not a Realtek employee. There is one Realtek employee who is offering some help but his job is actually to work on the drivers for the cards (PCIe, SDIO).  It would be nice if Realtek were interested in doing things according to Linux Wireless Standards but at this point, that is not the case. With that said, a significant drop happened last week and it seems progress is being made.

One thing that should really help us is when more usb wifi adapters based on the mt7921au chipset are available and they are coming. It is an immpressive chipset and driver:

https://github.com/morrownr/USB-WiFi/issues/139

Let me know if you find any more corrections.

Nick

OS2User, 2023-01-20:
Hi Nick,

sorry for the long delay in responding.

> > Please kindly update that entry.
> 
> Done. Please confirm.

Yep, looks good ;-)

> I monitor the linux-wireless mailing list so I can update you.

That would be much appreciated. Thank you.

> One thing that should really help us is when more usb wifi adapters based on the mt7921au chipset are available and they are coming. 
 
Indeed and as a special bonus it additionally provides BT5.2, which will be very handy, if one needs to substitute crappy RTL8188EE WiFi(n) 2.4GHz 1x1:1 with no BT at all, like in some cheap HP Atom-based laptops, with a USB-based solution.

The upcoming Mediatek mt7938A promises to be even more amazing: WiFi7(be)+BT5.3

Unfortunately at the time of purchasing there wasn't any Mediatek based USB WiFi adapter that I could find, which fit my requirements other than the RT8812bu based ones.
I wanted something (2x2:2, USB3.0, AC1200 or better) for <$10 and rather compact for permanent use in one of the above mentioned el-cheapo HP laptops, like this tiny 20.5x14.5x7mm [one](www.jacobsparts.com/items/DONGLE-G). 
Also, in a slightly bigger format the requirement was an additional activity LED indicator, like in the 42x19x8mm [EDUP EP-AC1689](www.szedup.com/product-item/edup-1300mbps-high-power-11ac-usb-wifi-dongle-adapter-ep-1689/#tab-id-2-active).

If anyone knows about comparable Mediatek based AC1200 or better adapters for <$15, I am all ears.

> Let me know if you find any more corrections.

Will do ;-)

OS2User

morrownr, 2023-01-23:
>> One thing that should really help us is when more usb wifi adapters based on the mt7921au chipset are available and they are coming.

> Indeed and as a special bonus it additionally provides BT5.2, which will be very handy, if one needs to substitute crappy RTL8188EE WiFi(n) 2.4GHz 1x1:1 with no BT at all, like in some cheap HP Atom-based laptops, with a USB-based solution.

Let me encourage you to temper your enthusiasm for multi-function (wifi and bt) adapters. The history of the last 10 years shows us that mutli-function adapters are limited to USB2 capability because USB3 cables, wires and connections emit a radio transmission that interferes with bluetooth which is right in the 2.4 GHz are where the interfering signal is heaviest. Thus the reason that makers need to limit multi-function adapters to either USB3 wifi only or USB2 wifi and bt.

Therefore, if an adapter maker advertises bluetooth, check closely to see if the adapter is limited to USB2 speeds. This won't bother some people but others that are expecting USB3 speeds with wifi, might get a surprise.

An example of this: Adapters based on the rtl8821bu chipset (wifi and bt) run only at USB2 speeds while adapters based on the rtl8812bu (wifi only) run a USB3 speeds.

This is a problem cause by the design and lack of adequate testing of USB3 and is not something adapter makers are doing to cause us problems. Blame those behind USB3.

> The upcoming Mediatek mt7938A promises to be even more amazing: WiFi7(be)+BT5.3

It is cool but people are still working on getting everything ironed out with WiFi 6 and 6e.

> If anyone knows about comparable Mediatek based AC1200 or better adapters for <$15, I am all ears.

I try to keep some low priced adapters in all categories if possible, In the mt7612u section curently there is the following:

https://www.walmart.com/ip/Lomubue-M-1200M-Wireless-Network-Card-High-speed-Anti-interference-Driver-free-USB3-0-MT7612U-Dual-Band-WiFi-Transceiver-for-Router/731709450?athbdg=L1400

A really good low cost adapter that I currently don't have a link for that you might find is the:

TEROW ROW02CD

I think the first adapter link above is a rebranded TEROW ROW02CD.

Now, I know you said < $15. Before the pandemic, that was possible. It may be again someday but right now that is almost impossible.

Nick

OS2User, 2023-01-23:
The good news first, I am now on kernel 6.1.0-1004-oem and my WiFi5 adapter is working nicely. Still in the progress of composing a reply to your previous message in that matter.

> One thing that should really help us is when more usb wifi adapters based on the mt7921au chipset are available and they are coming
>
They then better come with the mt7922u chipset, which is supposed to be same but with WiFi channel bandwidth increased from 80 to 160MHz.

> Thus the reason that makers need to limit multi-function adapters to either USB3 wifi only or USB2 wifi and bt.
> 
Then we probably have to wait forWiFi6+ USB combi adapters to become available with USB3.1 and/or USB4 interface, since that should move their main emission spectrum out of the way of BT.
> Therefore, if an adapter maker advertises bluetooth, check closely to see if the adapter is limited to USB2 speeds. This won't bother some people but others that are expecting USB3 speeds with wifi, might get a surprise.
>
Aah, that explains a lot. During my research I was wondering why the combi dongles only used the older chipsets and USB2, since my main requirement was USB3 . Of course also getting BT would have been nice, but was not that high a priority compared to size/activity LED and price.
Just wondering, shouldn't this issue then not affect 2.4GHz WiFi(n) as well? Or is it operating just off the peak emissions of USB3 and BT being right smack in the middle of it? If memory serves, then 2.4GHz WiFi and BT sometimes even share the same Rx/Tx antennas.
Either way, yet another prime example of glorious ivory tower engineering without considering how their waste emissions might affect existing legit spectrum uses. 

> TEROW ROW02CD
> I think the first adapter link above is a rebranded TEROW ROW02CD.

Aahh, one of those big ones with detachable antennas. I was also looking at some of them with swiveling antenna wings, but they are all too long. 
My laptop is rarely sitting on a desk but being moved around all the time and frequently placed on less than ideal surfaces, therefore any adapter sticking out further than absolutely necessary, doesn't seem to be a good choice for the sake of the adapter and the laptop's USB ports. 
In most of my use cases high gain and long distance connection are less critical than a compact design to not cause damage due to the large lever momentum.

morrownr, 2023-01-24:
> They then better come with the mt7922u chipset, which is supposed to be same but with WiFi channel bandwidth increased from 80 to 160MHz.

I am seeing cards with the mt7922 chipset but no usb adapters so far. It is normal that usb adapters are released later than cards.

> Just wondering, shouldn't this issue then not affect 2.4GHz WiFi(n) as well?

I think it does. USB3 is not mankind's greatest creation. It has been clear for several years now that there was not adequate testing before release.

> In most of my use cases high gain and long distance connection are less critical than a compact design 

Maybe the ANDDEAR - MT761003 ?

https://github.com/morrownr/USB-WiFi/blob/main/home/USB_WiFi_Adapters_that_are_supported_with_Linux_in-kernel_drivers.md

Scroll down.