morrownr/USB-WiFi #157: Achm vs acm

George-Kaloudis, 2022-12-01:
I have a raspberry pi 4 and use to sdcard one with kali and one with raspbian. Is it correct to think that the best one would the achm? (Im planning on upgrading to a laptop in the future)

morrownr, 2022-12-02:
Hi @George-Kaloudis 

The one that is best really depends on your use case. Since you are using Kali, I will assume you are doing some monitor mode work and having very good range is handy for that use. The ACHM has better range than the ACM. I did a performance test back in early 2021:

https://github.com/morrownr/USB-WiFi/blob/main/home/Performance_Comparison.md

I don't have a menu entry for that test because some of the info a little dated now but it is pretty good. The good range of the ACHM is also handy for AP mode or even managed mode. What the ACHM won't do burn up the racetrack with high throughput. It is an AC600 class adapter but that is actually still pretty fast and should handle a lot of use cases for a long time.

George-Kaloudis, 2022-12-02:
Thank you very much. Im studying cse and learning some pentesting on my own. I was looking mostly towards the achm as well. I agree that the 130 + mb/s are enough for most cases. Also have you tested them on a portable rpi setup? Any problems with power?

morrownr, 2022-12-02:
>  I agree that the 130 + mb/s are enough for most cases.

That test was intentionally difficult so that I could get some separation in the numbers. The ACHM can get you up to around 200 Mb/s. Here is a recent test that has both the ACHM and ACM plus more and it is a speed test:

https://github.com/morrownr/USB-WiFi/issues/139

> Also have you tested them on a portable rpi setup?

On a regular Pi#B and Pi4B, I test 24/7/365. Well, maybe the test period is over. Both adapters work well with Pi's.

> Any problems with power?

Not with the ACHM and ACM. They both use about 380 mA at full speed. The Pi usb subsystem can supply up to 1200 mA which is not that much but both of these adapters are frugal with power usage. I can run both in my Pi4B at the same time... dual band 4B AP/router ya know. 

I searching for something to complain about: The LED's don't work. It isn't supported in the driver yet. One of the Mediatek devs replied to a bug report the other day that he going to investigate and see about taking care of that. So if you need the LED on, right now you might want to avoid these adapters... I'm just kidding. These adapters won't win races but they run cool, don't use much power and there uptime is measured in months...maybe years.

George-Kaloudis, 2022-12-02:
Thanks for all the information @morrownr. You've been really helpful!