morrownr/USB-WiFi #158: Repository for 8852bu

alkisg, 2022-12-02:
@morrownr thank you for your rtl88xx drivers, you've been doing an awesome work with them!
I got an 8852bu-based adapter and I tested it with [this 8852bu 1.15.11-36 driver version](https://drive.google.com/file/d/1r-cDn3Ou5y16juWpjXpeGfG6XCI-GLJj/view?usp=share_link), released on 12 Sep 2022.
It seemed stable and pretty fast to me, I guess the driver matured from the earlier 15.1 version.
I tested it and it worked fine on 10 different versions of Debian and Ubuntu (x86_64/x86_32, from 2012 and on),
and on both the armhf and arm64 versions of Rasperry Pi OS Bullseye.
On Ubuntu 22.04 Raspberry Pi edition, I saw a glitch where it stopped downloading after 1 GB; with rmmod/modprobe it continued and successfully downloaded another 50 GB with no other issues. It might have been caused by undervoltage issues though so I'm not sure I can blame only the driver/adapter.

Unfortunately it only supports kernels up to 5.17, so I can use it with e.g. Ubuntu 22.04 but not with Ubuntu 22.10. If you think you can create a repository with this driver and maybe add a few patches for newer kernel support, I'd be glad to test it in more distributions! Thanks again!

bjlockie, 2022-12-02:
Where did the driver come from?
Does it work with the 8832au I have (Dlink dwa-x1850).

morrownr, 2022-12-03:
@alkisg 

Keep an eye out for an email inviting you to the repo. Post an issue to let me know you are there.

Nick

alkisg, 2022-12-03:
@bjlockie, it came with my adapter which is 8852bu-based. I've no idea if it'll work on 8832au, I don't have an 8832au adapter.
@morrownr, thank you very much for the invite, I'm there! :)

closed 2022-12-03 by alkisg

morrownr, 2022-12-03:
@bjlockie 

> Where did the driver come from?

I got it from Edimax. They are probably the best at posting Realtek Linux drivers. However, they only post for the chipsets they use in adapters.

> Does it work with the 8832au I have (Dlink dwa-x1850).

No. it is specific to the 8852/32bu. Edimax never made an adapter with the 8852/32au chipset so they don't post a driver.

bjlockie, 2022-12-03:
I hoped Realtek just added support to the au driver to support the bu and drastically improved the au driver as well. :-)
Oh well.
Does the bu driver look as cr*ppy as the au driver?

morrownr, 2022-12-03:
> Does the bu driver look as cr*ppy as the au driver?

I haven't had the amount of time with the bu driver that I did with the au driver but so far, what I can say is that it does not look better.

Having worked with the dual band Realtek drivers that I have available here, I have seen a lot of issues and we should not forget the the Realtek driver are not standards compliant, which is a big deal, but I have 5 Realtek dual band drivers up that will get the job done in most situations. It is not clear to me why but Realtek's  WiFi 6 out-of-kernel drivers so far are really bad. On the other hand, Mediatek's mt7921u driver is not bad at all. Sure there is a bug or two here and there and we all seem to be struggling to figure out how to do things like WiFi 6e AP mode but that is more a lack of knowledge and a need for some system components to be upgraded rather than poor driver performance.

Fishwaldo, 2023-04-18:
@alkisg Do you have a copy of that driver? Ive managed to find 1.15.7 version, so your link has a slightly updated driver I'd like to try.