morrownr/USB-WiFi #161: Issue with CF-953AX

MiriJ15, 2022-12-09:
Hello,

I have just received my Bluetooth adapter and it is not working even though I have installed the latest firmware. I am using Manjaro, kernel 6.0.11, all drivers were present in the MediaTek folder, but the adapter was not working, so I followed your guide to update/install the firmware manually. lsusb is not detecting it either. Additionally, the LED light is not blinking/working.

Is it a driver/device issue or I am missing something important?

MiriJ15, 2022-12-09:
Thank you so much for your prompt reply!

lsusb with and without wifi adapter connected to my laptop
```
~  lsusb                                                                                                                                                                                                                                                                    
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 005: ID 13d3:56a2 IMC Networks USB2.0 HD UVC WebCam
Bus 003 Device 009: ID 1044:7a3f Chu Yuen Enterprise Co., Ltd USB-HID Keyboard
Bus 003 Device 007: ID 8087:0029 Intel Corp. AX200 Bluetooth
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
 ~  lsusb                                                                                                                                                                                                                                                                    
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 005: ID 13d3:56a2 IMC Networks USB2.0 HD UVC WebCam
Bus 003 Device 009: ID 1044:7a3f Chu Yuen Enterprise Co., Ltd USB-HID Keyboard
Bus 003 Device 007: ID 8087:0029 Intel Corp. AX200 Bluetooth
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

MiriJ15, 2022-12-09:
Very interesting issue, I also have a Netgear wifi adapter based on Realtek chipset and it works flawlessly after following your guide for Realtek-based adapters, whereas the COMFAST MediaTek adapter with driver shipped in the latest kernel does not.

morrownr, 2022-12-09:
Do you have another computer that you could plug the adapter into? Does not matter if it is a Windows or Linux system. I would like to rule out the adapter being DOA.

MiriJ15, 2022-12-09:
I have tried it on a Mac and it was not working as well (did not install any drivers), I will try to test it tomorrow on windows/another mac.

morrownr, 2022-12-09:
It is a good idea to try it on others systems to rule out that it is not DOA. DOA does happen and so far, I have seen no evidence to disprove that the adapter is DOA. Even if there is no driver or firmware, we should see a device ID when we run lsusb and we are not getting anything.

MiriJ15, 2022-12-09:
Very sad. I will keep you updated!

MiriJ15, 2022-12-13:
Update: I tested it on a few windows PCs, and the machines could still identify the device (no led light). I am not an expert, but it looks like the modules are not loaded. For example:
```
sudo modinfo mt76                                                                                                                                                                                                                                       
filename:       /lib/modules/6.0.11-1-MANJARO/kernel/drivers/net/wireless/mediatek/mt76/mt76.ko.zst
license:        Dual BSD/GPL
srcversion:     DDAAA4639C5B8C4859DD4C7
depends:        mac80211,cfg80211
retpoline:      Y
intree:         Y
name:           mt76
vermagic:       6.0.11-1-MANJARO SMP preempt mod_unload 
sig_id:         PKCS#7
signer:         Build time autogenerated kernel key
sig_key:        2E:FE:EE:95:76:3E:4E:97:E5:7F:5F:94:E5:3A:3F:BD:32:17:4A:2F
sig_hashalgo:   sha512
signature:      30:66:02:31:00:92:9C:17:A9:AA:97:31:34:1D:50:23:1D:3C:DF:0C:
                04:07:9E:61:70:58:A7:19:54:BF:63:A3:9B:01:A1:36:69:2F:CA:A3:
                1C:2E:EC:93:B9:1F:44:73:EB:4E:9A:FA:CD:02:31:00:B1:89:82:32:
                51:61:D0:2B:76:0A:BF:F0:7A:3A:DA:24:BB:00:B6:D2:7C:94:FA:6D:
                85:26:2E:DF:EB:72:94:32:CD:9C:C3:40:0A:AD:80:91:81:ED:65:03:
                62:DB:9E:D8
sudo modinfo mt79                                                                                                                                                                                                                                               
modinfo: ERROR: Module mt79 not found.
```

travlytle, 2022-12-13:
Does dmesg give any errors / clues when you plug it in? Maybe provide full dmesg if you can.

MiriJ15, 2022-12-13:
```[416627.115349] usb 3-2.2: new high-speed USB device number 56 using xhci_hcd
[416627.310635] usb 3-2.2: New USB device found, idVendor=0e8d, idProduct=7961, bcdDevice= 1.00
[416627.310639] usb 3-2.2: New USB device strings: Mfr=6, Product=7, SerialNumber=8
[416627.310640] usb 3-2.2: Product: Wireless_Device
[416627.310641] usb 3-2.2: Manufacturer: MediaTek Inc.
[416627.310642] usb 3-2.2: SerialNumber: 000000000
[416637.828268] Bluetooth: hci1: Execution of wmt command timed out
[416637.828272] Bluetooth: hci1: Failed to send wmt patch dwnld (-110)
[416637.828298] Bluetooth: hci1: Failed to set up firmware (-110)
[416637.828299] Bluetooth: hci1: HCI Enhanced Setup Synchronous Connection command is advertised, but not supported.
[416638.005366] usb 3-2.2: reset high-speed USB device number 56 using xhci_hcd
[416643.334643] usb 3-2.2: device descriptor read/64, error -110
[416658.907296] usb 3-2.2: device descriptor read/64, error -110
[416659.190620] usb 3-2.2: reset high-speed USB device number 56 using xhci_hcd
[416664.453704] usb 3-2.2: device descriptor read/64, error -110
[416680.033028] usb 3-2.2: device descriptor read/64, error -110
[416680.319649] usb 3-2.2: reset high-speed USB device number 56 using xhci_hcd
[416685.402778] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[416690.732822] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[416690.939166] usb 3-2.2: device not accepting address 56, error -62
[416691.112529] usb 3-2.2: reset high-speed USB device number 56 using xhci_hcd
[416696.278968] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[416701.615378] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[416701.825322] usb 3-2.2: device not accepting address 56, error -62
[416701.825600] mt7921u: probe of 3-2.2:1.3 failed with error -5
[416701.825775] usb 3-2.2: USB disconnect, device number 56
[416702.002000] usb 3-2.2: new high-speed USB device number 57 using xhci_hcd
[416707.331795] usb 3-2.2: device descriptor read/64, error -110
[416722.907748] usb 3-2.2: device descriptor read/64, error -110
[416723.191030] usb 3-2.2: new high-speed USB device number 58 using xhci_hcd
[416728.454170] usb 3-2.2: device descriptor read/64, error -110
[416744.030140] usb 3-2.2: device descriptor read/64, error -110
[416744.140216] usb 3-2-port2: attempt power cycle
[416744.840053] usb 3-2.2: new high-speed USB device number 59 using xhci_hcd
[416750.036563] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[416755.373010] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[416755.582898] usb 3-2.2: device not accepting address 59, error -62
[416755.756237] usb 3-2.2: new high-speed USB device number 60 using xhci_hcd
[416760.919397] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[416766.249164] xhci_hcd 0000:00:14.0: Timeout while waiting for setup device command
[416766.459098] usb 3-2.2: device not accepting address 60, error -62
[416766.459231] usb 3-2-port2: unable to enumerate USB device
```

travlytle, 2022-12-13:
What is the host platform? PC? Embedded device? (if so which board).  I have seen this issue before on an NXP board. If you try this adapter on another linux does the dmesg show the same issue?

MiriJ15, 2022-12-13:
I am on a laptop, I assume board is made by manufacturer itself (Gigabyte) 
```Handle 0x0002, DMI type 2, 15 bytes
Base Board Information
        Manufacturer: Default string
        Product Name: AORUS 15P XD
        Version: Default string
        Serial Number: Default string
        Asset Tag: Default string
        Features:
                Board is a hosting board
                Board is replaceable
        Location In Chassis: Default string
        Chassis Handle: 0x0003
        Type: Motherboard
        Contained Object Handles: 0
```
I do not have any other linux machines with me at the moment

travlytle, 2022-12-13:
The USB is having a hard time enumerating (if this is the best word for it). It may work just fine on another platform, it may not if something is wrong with the adapter. I've seen resistors replaced on a usb dongle that fixed this issue, but I would definitely check it on a different platforms first before making any hardware modification. Raspberry Pi is a good platform to check on if you have access to one. WLAN-Pi has the drivers installed by default so makes it easy to check on.  Maybe someone else has seen this issue and knows a fix.

Does your laptop have a usb-c port? If so maybe try a usb-c to usb-a cable.

MiriJ15, 2022-12-13:
I will get one tomorrow. When I plug this device (in any USB port or extender) lsusb command freezes and I have to wait for a minute or 2 for it to show full output (or disconnect the device and the output comes out immediately)

morrownr, 2022-12-14:
@MiriJ15 

After looking at your dmesg snippet, it sure looks like a firmware problem. Can we get you to post the output of:

$ ls -l /lib/firmware/mediatek

MiriJ15, 2022-12-14:
```
total 14304
-rwxr-xr-x 1 root root  505558 Nov 22 10:28 BT_RAM_CODE_MT7922_1_1_hdr.bin
-rw-r--r-- 1 root root  527150 Dec  8 20:17 BT_RAM_CODE_MT7961_1_2_hdr.bin
-rw-r--r-- 1 root root  137632 Nov 22 10:28 WIFI_MT7922_patch_mcu_1_1_hdr.bin
-rw-r--r-- 1 root root   92192 Dec  8 20:17 WIFI_MT7961_patch_mcu_1_2_hdr.bin
-rw-r--r-- 1 root root  823748 Nov 22 10:28 WIFI_RAM_CODE_MT7922_1.bin
-rw-r--r-- 1 root root  718772 Dec  8 20:17 WIFI_RAM_CODE_MT7961_1.bin
-rw-r--r-- 1 root root   80680 Nov 22 10:28 mt7610e.bin
-rw-r--r-- 1 root root   80288 Nov 22 10:28 mt7610u.bin
-rw-r--r-- 1 root root  122644 Nov 22 10:28 mt7615_cr4.bin
-rw-r--r-- 1 root root  457992 Nov 22 10:28 mt7615_n9.bin
-rw-r--r-- 1 root root   11102 Nov 22 10:28 mt7615_rom_patch.bin
-rw-r--r-- 1 root root  300376 Nov 22 10:28 mt7622_n9.bin
-rw-r--r-- 1 root root   82110 Nov 22 10:28 mt7622_rom_patch.bin
-rwxr-xr-x 1 root root   78158 Nov 22 10:28 mt7622pr2h.bin
-rw-r--r-- 1 root root  382212 Nov 22 10:28 mt7650e.bin
-rw-r--r-- 1 root root   91412 Nov 22 10:28 mt7662u.bin
-rw-r--r-- 1 root root   20686 Nov 22 10:28 mt7662u_rom_patch.bin
-rw-r--r-- 1 root root  335072 Nov 22 10:28 mt7663_n9_rebb.bin
-rw-r--r-- 1 root root  441688 Nov 22 10:28 mt7663_n9_v3.bin
-rw-r--r-- 1 root root  278558 Nov 22 10:28 mt7663pr2h.bin
-rw-r--r-- 1 root root  209230 Nov 22 10:28 mt7663pr2h_rebb.bin
-rwxr-xr-x 1 root root  170990 Nov 22 10:28 mt7668pr2h.bin
-rw-r--r-- 1 root root    3584 Nov 22 10:28 mt7915_eeprom.bin
-rw-r--r-- 1 root root    3584 Nov 22 10:28 mt7915_eeprom_dbdc.bin
-rw-r--r-- 1 root root  144544 Nov 22 10:28 mt7915_rom_patch.bin
-rw-r--r-- 1 root root  116368 Nov 22 10:28 mt7915_wa.bin
-rw-r--r-- 1 root root 1260960 Nov 22 10:28 mt7915_wm.bin
-rw-r--r-- 1 root root    4096 Nov 22 10:28 mt7916_eeprom.bin
-rw-r--r-- 1 root root    8288 Nov 22 10:28 mt7916_rom_patch.bin
-rw-r--r-- 1 root root  464848 Nov 22 10:28 mt7916_wa.bin
-rw-r--r-- 1 root root 1634080 Nov 22 10:28 mt7916_wm.bin
-rw-r--r-- 1 root root    4096 Nov 22 10:28 mt7986_eeprom_mt7975_dual.bin
-rw-r--r-- 1 root root    4096 Nov 22 10:28 mt7986_eeprom_mt7976.bin
-rw-r--r-- 1 root root    4096 Nov 22 10:28 mt7986_eeprom_mt7976_dbdc.bin
-rw-r--r-- 1 root root    4096 Nov 22 10:28 mt7986_eeprom_mt7976_dual.bin
-rw-r--r-- 1 root root   11744 Nov 22 10:28 mt7986_rom_patch.bin
-rw-r--r-- 1 root root   11744 Nov 22 10:28 mt7986_rom_patch_mt7975.bin
-rw-r--r-- 1 root root  505008 Nov 22 10:28 mt7986_wa.bin
-rw-r--r-- 1 root root 2221772 Nov 22 10:28 mt7986_wm.bin
-rw-r--r-- 1 root root 2221772 Nov 22 10:28 mt7986_wm_mt7975.bin
drwxr-xr-x 1 root root      36 Dec  6 23:49 mt8173
drwxr-xr-x 1 root root      14 Dec  6 23:49 mt8183
drwxr-xr-x 1 root root      14 Dec  6 23:49 mt8186
drwxr-xr-x 1 root root      14 Dec  6 23:49 mt8192
drwxr-xr-x 1 root root      14 Dec  6 23:49 mt8195
drwxr-xr-x 1 root root     108 Dec  6 23:49 sof
drwxr-xr-x 1 root root     182 Dec  6 23:49 sof-tplg
```

morrownr, 2022-12-14:
> -rw-r--r-- 1 root root  527150 Dec  8 20:17 BT_RAM_CODE_MT7961_1_2_hdr.bin

> -rw-r--r-- 1 root root   92192 Dec  8 20:17 WIFI_MT7961_patch_mcu_1_2_hdr.bin

> -rw-r--r-- 1 root root  718772 Dec  8 20:17 WIFI_RAM_CODE_MT7961_1.bin

Those are the latest firmware files. Same ones I am using here. I'm also using kernel 6.0 and a cf-951ax that has the same chipset as yours. I'm out of ideas.

golinski, 2023-12-19:
I've just received a CF-953AX and I have similar issues: when the dongle is plugged into my desktop computer (Arch Linux, latest kernel), I get a hanged `lsusb` for a while, then the following in `dmesg`:

```
[18625.243838] usb 2-1: new SuperSpeed USB device number 2 using xhci_hcd
[18625.261976] usb 2-1: New USB device found, idVendor=0e8d, idProduct=7961, bcdDevice= 1.00
[18625.261979] usb 2-1: New USB device strings: Mfr=6, Product=7, SerialNumber=8
[18625.261981] usb 2-1: Product: Wireless_Device
[18625.261983] usb 2-1: Manufacturer: MediaTek Inc.
[18625.261984] usb 2-1: SerialNumber: 000000000
[18625.270290] Bluetooth: hci1: HW/SW Version: 0x008a008a, Build Time: 20231109191416
[18627.882333] Bluetooth: hci1: Device setup in 2553293 usecs
[18627.882339] Bluetooth: hci1: HCI Enhanced Setup Synchronous Connection command is advertised, but not supported.
[18630.063751] Bluetooth: hci1: Opcode 0x0c03 failed: -110
[18632.197075] Bluetooth: hci1: Failed to read MSFT supported features (-110)
[18634.330353] Bluetooth: hci1: AOSP get vendor capabilities (-110)
[18639.663669] xhci_hcd 0000:08:00.1: Timeout while waiting for setup device command
[18644.996958] xhci_hcd 0000:08:00.1: Timeout while waiting for setup device command
[18645.203597] usb 2-1: device not accepting address 2, error -62
[18650.543592] xhci_hcd 0000:08:00.1: Timeout while waiting for setup device command
[18655.876854] xhci_hcd 0000:08:00.1: Timeout while waiting for setup device command
[18656.083490] usb 2-1: device not accepting address 2, error -62
[18661.423495] xhci_hcd 0000:08:00.1: Timeout while waiting for setup device command
[18666.760097] xhci_hcd 0000:08:00.1: Timeout while waiting for setup device command
[18666.966744] usb 2-1: device not accepting address 2, error -62
[18672.303372] xhci_hcd 0000:08:00.1: Timeout while waiting for setup device command
[18677.636677] xhci_hcd 0000:08:00.1: Timeout while waiting for setup device command
[18677.843296] usb 2-1: device not accepting address 2, error -62
[18677.858077] usb 2-1: USB disconnect, device number 2
[18677.858242] mt7921u: probe of 2-1:1.3 failed with error -5
[18677.858316] usbcore: registered new interface driver mt7921u
[18683.186613] xhci_hcd 0000:08:00.1: Timeout while waiting for setup device command
[18688.516551] xhci_hcd 0000:08:00.1: Timeout while waiting for setup device command
[18688.723196] usb 2-1: device not accepting address 3, error -62
[18694.063170] xhci_hcd 0000:08:00.1: Timeout while waiting for setup device command
[18699.396454] xhci_hcd 0000:08:00.1: Timeout while waiting for setup device command
[18699.606428] usb 2-1: device not accepting address 4, error -62
[18699.613149] usb usb2-port1: attempt power cycle
[18705.583077] xhci_hcd 0000:08:00.1: Timeout while waiting for setup device command
[18710.919694] xhci_hcd 0000:08:00.1: Timeout while waiting for setup device command
[18711.129661] usb 2-1: device not accepting address 5, error -62
[18716.462979] xhci_hcd 0000:08:00.1: Timeout while waiting for setup device command
[18721.796257] xhci_hcd 0000:08:00.1: Timeout while waiting for setup device command
[18722.002896] usb 2-1: device not accepting address 6, error -62
[18722.009795] usb usb2-port1: unable to enumerate USB device
```

I've tried different ports (on the case, on the backplate, with a cable), results are similar.

The desktop is built on Asus Crosshair VIII Dark Hero motherboard. I've tried making the dongle work with Windows 11 on the same machine: even after installing drivers, it would be recognized as a Mediatek device, but an error was displayed and I don't think it worked.

The same dongle works perfectly in my HP Omen laptop (Arch Linux as well).

morrownr, 2023-12-19:
> The same dongle works perfectly in my HP Omen laptop (Arch Linux as well).

That seems to rule out a manufacturing problem with the adapter. There is something different between the two systems.

> Bluetooth: hci1: HW/SW Version: 0x008a008a, Build Time: 20231109191416

The adapter likely has BT turned off. You can get rid of the BT error in the log by deleting the bluetooth firmware file for this adapter. If you need help, let me know.

Concerning wifi, it appears there is something going on with usb. Exactly what is not clear. You might look around in the bios to see what usb related settings you have available.

heinwol, 2024-01-12:
I have the same issue as @golinski. The adapter worked fine initially, but it broke on the second day of me using it. I'm on nixos-unstable (kernel 6.6.8-zen1), the dmesg log looks exactly like golinski's one. I tried to use this on windows and installed the official driver. It didn't work either, throwing `cannot start device (code 10)`. I contacted the aliexpress store support, after a few days they told me to reinstall the driver (what an unexpected turn of events!). **However**, it worked for some reason! I could use this device on windows. I tried using it on another machine with nixos-23.11 (kernel 6.1.x) and it worked fine again. But here I'm writing this on my first computer and no, dmesg log looks exactly the same. All in all, this _might_ be a hardware issue, since something strange is happening.

Oh, and yeah, I tried using the adapter on the first machine with kernel 6.1.x, and had no luck either.

morrownr, 2024-01-12:
@heinwol 

Have you tried other USB ports?

USB3 is not mankind's greatest creation. Switching ports often helps.

ricardoper, 2024-05-11:
Exactly the same issue in Ubuntu with kernel 6.8. The same dmesg logs. Sometimes gets stuck in BIOS POST.

Works flawless in the same hardware with Windows and after reboots I don't have any BIOS POST issues.

This is a very strange behaviour.

morrownr, 2024-05-13:
Hi @ricardoper 

Try the following:

$ rmmod btusb

ricardoper, 2024-05-16:
Hello @morrownr and thanks for the reply.

After the command `rmmod btusb` the adapter works as expected (WiFi & BT) and no more BIOS POST issues. Many thanks for the help.

My sensations about the CF-953AX adapter. Comparing this adapter in Windows or with other WIFI AC COMFAST adapters in Linux, the Speedtest results are under my expectations. With this adapter I can reach 240/280Mbps (Down/Up) and in Windows I can reach more than 300Mbps easily (the same for other COMFAST AC adapters in Linux). Download latencies are a bit higher too. I notice some stability problems with the connections, sometimes I have connection drops.

```
OS: Ubuntu 22.04.3 LTS
Kernel: 6.5.0-28-generic

Base Board Information
	Manufacturer: Gigabyte Technology Co., Ltd.
	Product Name: Z370M D3H-CF
	Version: x.x
	Serial Number: Default string
	Asset Tag: Default string
	Features:
		Board is a hosting board
		Board is replaceable
	Location In Chassis: Default string
	Chassis Handle: 0x0003
	Type: Motherboard
	Contained Object Handles: 0
```

Many thanks again for the help and for your good work.

Regards.

morrownr, 2024-05-16:
It appears that the problem is with BT and not WIFI. I guess not a problem with BT per se but some patch has gone into BT somewhere and we are paying the price. I'll keep an eye out.

ricardoper, 2024-05-21:
After a few days the same issue :(
Frozens in BIOS POST when this adapter is plugged. When removed, it works perfectly. Once Ubuntu booted and plug the adapter again, all the USB devices stop working.
Is hard for me to consider a hardware fault because in Windows I don't have these issues and I don't understand why the BIOS POST doesn't freeze once the adapter works in Windows.
BIOS POST is OS independently, right? If it is an USB3 issue, BIOS POST always fail, right?
Very confused...

```
/var/log/syslog

kernel: [ 1572.600006] usb 1-2: new high-speed USB device number 19 using xhci_hcd
kernel: [ 1572.750083] usb 1-2: New USB device found, idVendor=0e8d, idProduct=7961, bcdDevice= 1.00
kernel: [ 1572.750100] usb 1-2: New USB device strings: Mfr=6, Product=7, SerialNumber=8
kernel: [ 1572.750109] usb 1-2: Product: Wireless_Device
kernel: [ 1572.750115] usb 1-2: Manufacturer: MediaTek Inc.
kernel: [ 1572.750121] usb 1-2: SerialNumber: 000000000
systemd[1]: Starting Load/Save RF Kill Switch Status...
systemd[1]: NetworkManager-dispatcher.service: Deactivated successfully.
kernel: [ 1576.012754] Bluetooth: hci0: Device setup in 3177214 usecs
kernel: [ 1576.012769] Bluetooth: hci0: HCI Enhanced Setup Synchronous Connection command is advertised, but not supported.
kernel: [ 1578.016078] Bluetooth: hci0: Opcode 0x0c03 failed: -110
kernel: [ 1580.032077] Bluetooth: hci0: Failed to read MSFT supported features (-110)
kernel: [ 1582.048193] Bluetooth: hci0: AOSP get vendor capabilities (-110)
kernel: [ 1582.176746] usb 1-2: reset high-speed USB device number 19 using xhci_hcd
kernel: [ 1587.392248] usb 1-2: device descriptor read/64, error -110
systemd[1]: systemd-rfkill.service: start operation timed out. Terminating.
systemd[1]: systemd-rfkill.service: Failed with result 'timeout'.
systemd[1]: Failed to start Load/Save RF Kill Switch Status.
systemd[1]: Starting Load/Save RF Kill Switch Status...
systemd[1]: Started Load/Save RF Kill Switch Status.
kernel: [ 1603.008464] usb 1-2: device descriptor read/64, error -110
kernel: [ 1603.244466] usb 1-2: reset high-speed USB device number 19 using xhci_hcd
systemd[1]: systemd-rfkill.service: Deactivated successfully.
kernel: [ 1608.384529] usb 1-2: device descriptor read/64, error -110
```

morrownr, 2024-05-22:
@ricardoper 

> After a few days the same issue :(

I think the problem has to do with a bad patch that went into the kernel. Given what I have seen, It is likely either a bad bluetooth patch or a bad usb patch. I'd recommend you try a different kernel. Something non-destructive that you could try is throw a copy of Ubuntu 24.04 on a flash drive and boot up. See if it works. If it works, then the kernel 6.5 version you are using appears to still have a problem and you might want to backup the important stuff and do a clean install of 24.04.

ricardoper, 2024-05-26:
Hello,

Fresh Kubuntu 24.04 LT with 6.8.0-31 Kernel. Nothing changes... Same issue, the same logs.

Yesterday I see a new Kernel release, 6.9.2, and revising the changelog:

> Bluetooth: btusb: Fix the patch for MT7920 the affected to MT7921
> 
> Because both MT7920 and MT7921 use the same chip ID.
> We use the 8th bit of fw_flavor to distingush MT7920.
> The original patch made a mistake to check whole fw_flavor,
> that makes the condition both true (dev_id == 0x7961 && fw_flavor),
> and makes MT7921 flow wrong.
> 
> In this patch, we correct the flow to get the 8th bit value for MT7920.
> And the patch is verified pass with both MT7920 and MT7921.

I changed the system to use this Kernel version and had no luck. The same behavior.

When I do `rmmod btusb && modprobe btusb` sometimes I can use WiFi & BT together but the BT is useless because it cannot detect my devices only noise.

I need a solution to use my old BT adapter and using the `rmmod btusb` command it's impossible to use it so I made a few fixes.

Maybe I can help another person, so I will share my solution to use Comfast CF-953AX only in WiFi mode and using another BT adapter.
```
sudo nano /etc/modprobe.d/blacklist.conf
...
# Comfast CF-953AX Fix
blacklist btusb
...

sudo nano /usr/lib/systemd/system/cf953ax-bfix.service
...
[Unit]
Wants=network-online.target
After=network.target network-online.target

[Service]
Type=oneshot
ExecStartPre=/usr/bin/sleep 10
ExecStart=/usr/lib/systemd/cf953ax-bfix.sh

[Install]
WantedBy=default.target

sudo nano /usr/lib/systemd/cf953ax-bfix.sh
...
#!/usr/bin/env bash

MAC_ADDR="AA:AA:AA:AA:AA:AA"  <---- Put your Comfast adapter MAC address

HCI=`hcitool dev | grep ${MAC_ADDR} | cut -f2`
RFK_IDX=`rfkill list | grep ${HCI} | cut -d: -f1`

modprobe btusb
rfkill block ${RFK_IDX}
hciconfig ${HCI} down

sudo chmod +x /usr/lib/systemd/cf953ax-bfix.sh
sudo systemctl enable cf953ax-bfix.service
reboot
```

Maybe it will not be the best solution but it works for me. If you have a better solution to avoid the use of the `rmmod btusb` command and disable the BT module in the Comfast adapter, let me know.

Thanks.

morrownr, 2024-05-28:
I don't know that I have a better solution. My problem is that I can't reproduce this. I've been watching and I do have two adapters with the same chip but I am not seeing this. I see the patch you posted and that does sound like a player for this issue but you are reporting no success. My question is: which kernel are you testing? Is is the Linux mainline 6.9.2 or the Ubuntu mainline 6.9.2. They are different and Ubuntu uses their own patch philosophy. If you install Ubuntu mainline, you may not have that patch.

This bug is very unfortunate and I hope it is a lesson for the maintainers and devs to be more careful.

ricardoper, 2024-05-28:
Hello @morrownr ,

Sorry if I don't give to much information but is the data that I have with my knowledge. 

I tested with two kernels: 
6.8.0-31 from a fresh Kubuntu 24.04 LTS install
6.9.2 from Ubuntu mainline kernel webpage

Both kernels give the same results, the same logs that I share in my last responses.

When Linux boot up get stuck when try to detect the BT driver of the adapter. If I blacklist the 'btusb' the Linux boot process ignores the adapter BT and loads the WiFi driver and works as expected but I don't have BT.

When I load the 'btusb' after the boot, Linux can detect the BT driver correctly and it works but... Can detect only a 2 or 3 devices that are basically noise. It can't detect any of my devices like headphones.

I tried with USB 3 and 2 and it's the same. In a USB Hub with or without external power and no changes at all.

My fix is to used another BT adapter because this one is useless.

Thanks for trying to solve this issue. Many many thanks. 

Can you give a little of details about your environment? Like OS, kernel, USB 2 or 3, with good USB cable extender or not, USB Hub with or without external power?

Once again, thanks for your effort.

morrownr, 2024-05-29:
> Can you give a little of details about your environment?

Sure. My primary dev box runs Debian 12 with kernel  6.6. It is an 11 years old i7 with 4 cores and 12 gb of RAM. I think it is a really fast system but then I started with CP/M on a Z80 (as far a microcomputers are concerned.)

I do have a little lab with various systems that I use for testing. I try to have one of each chipset that Linux users could be interested in. Since I maintain several Realtek out-of-kernel drivers and help with efforts to bring drivers to the Linux kernel, I have a large variety of adapters. Sometimes retailers send adapters to me in the hopes that I will test and write a report or add the adapter to the Plug and Play List. Sometimes this works...if my tests show me that all of the criteria that I set is met. I won't post multi-state adapters, period. The adapters have to be reasonably available and well made. The adapters have to be plug and play. 

> Like OS, kernel, USB 2 or 3, with good USB cable extender or not, USB Hub with or without external power?

I generally use and test with the usb port that will give the best performance. If it does not matter, like with WiFi 4 adapters, then I go with USB2 as USB2 is more stable. USB3 is not mankind's greatest invention. It was not well tested. I know situations to avoid.

There are good USB3 cables out there. The cable that comes with the Alfa AXML is simply outstanding. Beyond that, you need to read a lot of be propared to send the junk back. It would be cool if I had the time and resources to test and recommend cables but that could be hard as cables come and go. The same can be said for the USB L adapters which are very handy in many situations. I use them as well.

> USB Hub with or without external power?

If you find an external USB hub that is worth owning, let me know. I generally sidestep that by using PCIe USB cards to add ports. 

I will continue to keep an eye out for anything that might help. I have seen some other with this problem but have not been able to determine what the command factor is. You might try to delete the firmware that supports bluetooth for the mt7921.