morrownr/USB-WiFi #162: Article: Building your own remote sensor with 6 GHz support

morrownr, 2022-12-09:
Some of you may find this article interesting so here is the link:

https://www.intuitibits.com/2022/12/01/building-your-own-remote-sensor-with-6-ghz-support/

amisix, 2022-12-10:
I really like this project. It's possible to image a CM4 and make it even smaller (although headless). 
 
[image]

jiribrejcha, 2022-12-10:
We also built a conplete unit with built-in display and buttons, PoE and M.2 Wi-Fi 6E adapter. We call it WLAN Pi M4. 

Check: www.wlanpi.com

[image]

amisix, 2022-12-10:
How are the Mcuzone carrier board(s) working out for you?

jiribrejcha, 2022-12-10:
They are great value and provide just about the right features for our project. 

Pros:
- balanced feature set
- great form factor with E-Key M.2 slot for Wi-Fi adapters and built-in PoE module

Cons:
- their QA process could be slightly improved, but the failure rate is below 5% so far
- it comes without 40-pin GPIO header
- it doesn’t support USB OTG on the USB-C (or other) port

Long term, USB OTG, USB 3.1 and 2.5 GbE would be great additions.

amisix, 2022-12-12:
I like the board, and I do like the idea of being able to run this with two mt7921 adapters. The cons aren't bad, 
The MagSafe battery is a clever way of attaching the power source.

Do you think you'll offer the case/fascia for the M4 at some point like you do the R4?

jiribrejcha, 2022-12-12:
The case for M4 is not as easy to assemble as the R4, and there are quite a few components required. We have no plans to sell it separately at the moment. 

[image]

jiribrejcha, 2022-12-12:
Just for reference and to give him credit, Nick from www.badgerwifi.co.uk is our chief case designer.

amisix, 2022-12-12:
Clearly a lot of thought went into it - hats off to Nick.  

I was looking at the specs of the Anker battery recommended for the R4 and it's only capable of 5V @ 2A. Could that be an issue when plugging in an external USB adapter or other peripherals?

amisix, 2022-12-14:
Well, either way, it's a neat little box. I'm tempted to use a [Waveshare CM4 to Pi4B carrier board](https://www.aliexpress.us/item/3256804835628328.html?spm=a2g0o.order_list.order_list_main.17.17e61802rJk9BS&gatewayAdapt=glo2usa&_randl_shipto=US) with the R4 because it has POE and 4x USB3 ports..

jiribrejcha, 2022-12-14:
@amisix the Waveshare board needs an additional HAT to support PoE. The Mcuzone board has PoE built in. The standard size HDMI port is also useful for troubleshooting.

In the current chip shortages times, it also seems to be easier to purchase RPi4 rather than CM4.

amisix, 2022-12-15:
I see. I didn't even think much about the standard size HDMI port.. so you don't have to carry around/lose a dongle. 

Yeah, the chip shortage has been terrible, but I was able to pick up a couple CM4s months ago and figured I could press one into service with this. Looks like I'm back to looking for an RPi4B. 

Thanks for sharing your knowledge with me