morrownr/USB-WiFi #163: Add new dongle: TP-Link Archer T1U - AC450 - mt7610u

Anthony96922, 2022-12-10:
It might be worth adding a section for this dongle since it is even smaller than the Linksys AE6000. I have one and while the range is even less than the AE6000 it is compact enough to be used in travel WiFi routers. I know you have a no TP-Link rule but this should be an exception.

jiribrejcha, 2022-12-10:
Why fhe “no TP-Link” rule? A personal preference or a design flaw?

morrownr, 2022-12-10:
@Anthony96922 

> Why fhe “no TP-Link” rule? A personal preference or a design flaw?

Neither. As explained in Main Menu item 1, Tp-Link has a long history or changing chipsets in their products while keeping the model number the same which makes our job of maintaining a list of good adapters, such as Main Menu item 2, almost impossible.

We can list the Alfa ACM because the Alfa ACM has only had the mt7612u chipset since it was first produced. This makes for very little confusion whether what TP-Link and D-Link so makes for a lot of confusion. 

My opinion is that the highest goal of this site it is to get the best, most accurate, information possible into the hands of Linux users. It takes a lot of time and adding TP-Link produces into the mix would do nothing but add a lot of confusion. As I said above, if the right exception comes along, I am good with that.

Nick

Anthony96922, 2022-12-10:
This is a discontinued product. There is one available second-hand on eBay for $12 total although cheaper ones may show up occasionally. I bought mine months ago for $10. There are 2 revisions and both contain the mt7610u.

jiribrejcha, 2022-12-10:
@morrownr agreed, that makes perfect sense 👍🏻 Thanks for the explanation.   Changing chipsets in the same model is bad.

morrownr, 2022-12-10:
@Anthony96922 

> There are 2 revisions and both contain the mt7610u.

That does appear to be the case which would limit confusion. I would prefer that TP-Link pull down their posted driver from the Dark Ages and just post a statement that says the adapter works with a Linux in-kernel driver from kernel 4.19 and later. But TP-Link does not even try to understand Linux support so they are who they are.

Can I get you to post the output of:

$ lsusb

... so we can see the adapter ID.

Amazon is out and we probably cannot expect any more. I see this link on eBay:

https://www.ebay.com/itm/392631860422

...but only 3 more for sale and they will probably be bought by users reading this thread before I can get a listing included in the main list.

Tell me if you think it is worth the time and will be of long term benefit to the users here? I ask because I am skeptical. Not that I do not appreciate your input because I do. Consider the following:

During 2023, the list of adapters using in-kernel drivers, Main Menu item 2, is likely to explode. Reasons:

- Many more adapters using the new mt7921au chipset should come on the market,

- New in-kernel drivers for the rtl8822bu and rtl8821cu chipsets are going into the kernel. Watch for these drivers in kernel 6.2 when it is released for testing here in a couple of weeks. I will be using the same criteria that I currently use for adding adapters to the list plus I will eliminate any multi-state adapters and will probably eliminate adapters that also support bluetooth, especially the rtl8821cu, as that has been know to cause issues.

Nick

Anthony96922, 2022-12-10:
Here's the output of lsusb on my Raspberry Pi 2 running a self compiled 5.15.78 kernel:
```
anthony@Sicso:~$ lsusb
Bus 001 Device 005: ID 2357:0105 TP-Link Archer T1U 802.11a/n/ac Wireless Adapter [MediaTek MT7610U]
Bus 001 Device 004: ID 13b1:003e Linksys AE6000 802.11a/b/g/n/ac Wireless Adapter [MediaTek MT7610U]
Bus 001 Device 003: ID 0424:ec00 Microchip Technology, Inc. (formerly SMSC) SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Microchip Technology, Inc. (formerly SMSC) SMC9514 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

The reason I'm wishing this adapter was listed in your repo is because some might be looking for a WiFi AC adapter that can be left plugged in (such as for custom travel WiFi routers). I personally used the AE6000 (bought it based off your recommendation) with my RasPi on a trip and it works very well. But it's more expensive than the T1U. I'm looking forward to nano AX adapters arriving and becoming cheap enough so I can upgrade.

morrownr, 2022-12-11:
@Anthony96922 

You have made your case. Go to Main Menu item 2 and go to the section on mt7610u adapter and see if you agree with what I did.

Anthony96922, 2022-12-11:
Looks good. Thank you.

closed 2022-12-11 by Anthony96922