morrownr/USB-WiFi #165: CF953AX iw data

codebling, 2022-12-13:
Posting this since it is slightly different than [home/iw_list/COMFAST_CF-951AX.txt](https://github.com/morrownr/USB-WiFi/blob/bb72653fbf0f91452a94ee78534b7b53213f65a7/home/iw_list/COMFAST_CF-951AX.txt). 

Firmware date is 2022-11-09, 7 days before the most recent (relevant) Mediatek firmware update. Can provide update after upgrading. 

Cursory trials show 6Ghz frequencies appear to work despite being labelled "disabled" but have not confirmed.

---

2022-12-12

COMFAST_CF953AX

Chipset: mt7921au

$ lsusb
ID 0e8d:7961 MediaTek Inc. Wireless_Device

$ $ uname -sr
Linux 6.0.9-arch1-1

$ iw list
Wiphy phyX
	wiphy index: X
	max # scan SSIDs: 4
	max scan IEs length: 482 bytes
	max # sched scan SSIDs: 10
	max # match sets: 16
	Retry short limit: 7
	Retry long limit: 4
	Coverage class: 0 (up to 0m)
	Device supports AP-side u-APSD.
	Device supports T-DLS.
	Supported Ciphers:
		* WEP40 (00-0f-ac:1)
		* WEP104 (00-0f-ac:5)
		* TKIP (00-0f-ac:2)
		* CCMP-128 (00-0f-ac:4)
		* CCMP-256 (00-0f-ac:10)
		* GCMP-128 (00-0f-ac:8)
		* GCMP-256 (00-0f-ac:9)
		* CMAC (00-0f-ac:6)
		* CMAC-256 (00-0f-ac:13)
		* GMAC-128 (00-0f-ac:11)
		* GMAC-256 (00-0f-ac:12)
	Available Antennas: TX 0x3 RX 0x3
	Configured Antennas: TX 0x3 RX 0x3
	Supported interface modes:
		 * managed
		 * AP
		 * AP/VLAN
		 * monitor
	Band 1:
		Capabilities: 0x9ff
			RX LDPC
			HT20/HT40
			SM Power Save disabled
			RX Greenfield
			RX HT20 SGI
			RX HT40 SGI
			TX STBC
			RX STBC 1-stream
			Max AMSDU length: 7935 bytes
			No DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: No restriction (0x00)
		HT TX/RX MCS rate indexes supported: 0-15
		HE Iftypes: managed
			HE MAC Capabilities (0x08011a000040):
				+HTC HE Supported
				Trigger Frame MAC Padding Duration: 2
[... 604 lines cut ...]
		 * mesh point: 0xb0 0xc0 0xd0
		 * P2P-client: 0x40 0xd0
		 * P2P-GO: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * P2P-device: 0x40 0xd0
	Supported extended features:
		* [ RRM ]: RRM
		* [ SET_SCAN_DWELL ]: scan dwell setting
		* [ BEACON_RATE_LEGACY ]: legacy beacon rate setting
		* [ BEACON_RATE_HT ]: HT beacon rate setting
		* [ BEACON_RATE_VHT ]: VHT beacon rate setting
		* [ FILS_STA ]: STA FILS (Fast Initial Link Setup)
		* [ CQM_RSSI_LIST ]: multiple CQM_RSSI_THOLD records
		* [ CONTROL_PORT_OVER_NL80211 ]: control port over nl80211
		* [ TXQS ]: FQ-CoDel-enabled intermediate TXQs
		* [ AIRTIME_FAIRNESS ]: airtime fairness scheduling
		* [ AQL ]: Airtime Queue Limits (AQL)
		* [ CONTROL_PORT_NO_PREAUTH ]: disable pre-auth over nl80211 control port support
		* [ SCAN_FREQ_KHZ ]: scan on kHz frequency support
		* [ CONTROL_PORT_OVER_NL80211_TX_STATUS ]: tx status for nl80211 control port support
		* [ BEACON_RATE_HE ]: HE beacon rate support (AP/mesh)

usb-modeswitch not required. This is a single-state device.

archlinux kernel: mt7921u 2-1:1.3: HW/SW Version: 0x8a108a10, Build Time: 20220908210919a
archlinux kernel: mt7921u 2-1:1.3: WM Firmware Version: ____010000, Build Time: 20220908211021

travlytle, 2022-12-13:
This is on a RPi using wlan pi.

```
lsusb 
Bus 002 Device 006: ID 0e8d:7961 MediaTek Inc. Wireless_Device

uname -a
Linux wlanpi-cc2 5.19.14-v8-wlanpi+ #1 SMP PREEMPT Fri Oct 7 09:09:24 UTC 2022 aarch64 GNU/Linux

Wiphy phy4
        wiphy index: 4
        max # scan SSIDs: 4
        max scan IEs length: 482 bytes
        max # sched scan SSIDs: 10
        max # match sets: 16
        Retry short limit: 7
        Retry long limit: 4
        Coverage class: 0 (up to 0m)
        Device supports AP-side u-APSD.
        Device supports T-DLS.
        Supported Ciphers:
                * WEP40 (00-0f-ac:1)
                * WEP104 (00-0f-ac:5)
                * TKIP (00-0f-ac:2)
                * CCMP-128 (00-0f-ac:4)
                * CCMP-256 (00-0f-ac:10)
                * GCMP-128 (00-0f-ac:8)
                * GCMP-256 (00-0f-ac:9)
                * CMAC (00-0f-ac:6)
                * CMAC-256 (00-0f-ac:13)
                * GMAC-128 (00-0f-ac:11)
                * GMAC-256 (00-0f-ac:12)
        Available Antennas: TX 0x3 RX 0x3
        Configured Antennas: TX 0x3 RX 0x3
        Supported interface modes:
                 * managed
                 * AP
                 * AP/VLAN
                 * monitor
        Band 1:
                Capabilities: 0x9ff
                        RX LDPC
                        HT20/HT40
                        SM Power Save disabled
                        RX Greenfield
                        RX HT20 SGI
                        RX HT40 SGI
                        TX STBC
                        RX STBC 1-stream
                        Max AMSDU length: 7935 bytes
                        No DSSS/CCK HT40
                Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
                Minimum RX AMPDU time spacing: No restriction (0x00)
                HT TX/RX MCS rate indexes supported: 0-15
                HE Iftypes: managed
                        HE MAC Capabilities (0x08011a000040):
                                +HTC HE Supported
                                Trigger Frame MAC Padding Duration: 2
                                OM Control
                                Maximum A-MPDU Length Exponent: 3
                                A-MSDU in A-MPDU
                        HE PHY Capabilities: (0x2270ce120dc0b306423f00):
                                HE40/2.4GHz
                                242 tone RUs/2.4GHz
                                Device Class: 1
                                LDPC Coding in Payload
                                HE SU PPDU with 1x HE-LTF and 0.8us GI
                                NDP with 4x HE-LTF and 3.2us GI
                                STBC Tx <= 80MHz
                                STBC Rx <= 80MHz
[... 588 lines cut ...]
                 * IBSS: 0x40 0xb0 0xc0 0xd0
                 * managed: 0x40 0xb0 0xd0
                 * AP: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
                 * AP/VLAN: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
                 * mesh point: 0xb0 0xc0 0xd0
                 * P2P-client: 0x40 0xd0
                 * P2P-GO: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
                 * P2P-device: 0x40 0xd0
        Supported extended features:
                * [ RRM ]: RRM
                * [ SET_SCAN_DWELL ]: scan dwell setting
                * [ BEACON_RATE_LEGACY ]: legacy beacon rate setting
                * [ BEACON_RATE_HT ]: HT beacon rate setting
                * [ BEACON_RATE_VHT ]: VHT beacon rate setting
                * [ FILS_STA ]: STA FILS (Fast Initial Link Setup)
                * [ CQM_RSSI_LIST ]: multiple CQM_RSSI_THOLD records
                * [ CONTROL_PORT_OVER_NL80211 ]: control port over nl80211
                * [ TXQS ]: FQ-CoDel-enabled intermediate TXQs
                * [ AIRTIME_FAIRNESS ]: airtime fairness scheduling
                * [ AQL ]: Airtime Queue Limits (AQL)
                * [ CONTROL_PORT_NO_PREAUTH ]: disable pre-auth over nl80211 control port support
                * [ SCAN_FREQ_KHZ ]: scan on kHz frequency support
                * [ CONTROL_PORT_OVER_NL80211_TX_STATUS ]: tx status for nl80211 control port support
                * [ BEACON_RATE_HE ]: HE beacon rate support (AP/mesh)
```

codebling, 2022-12-13:
Ok so same frequencies as morrownr's 951 dump but with the same "EHT Iftypes" as mine. Wonder if "disabled" in channels might just be a driver quirk with 6.0 kernel. I can check next time I restart my system.

morrownr, 2022-12-22:
@codebling @travlytle 

> Wonder if "disabled" in channels might just be a driver quirk with 6.0 kernel.

I don't think it is a driver quirk but rather the reg db. WiFi 6e has 3 different power levels that are identified and it is not clear to me if the rules and regulations are fully in place in all countries. If I take a vacation to a country where everything seems to be sorted out, such as FR or DE, then things show up closer to what we would expect. I'm just going to keep an eye on the situation.

codebling, 2022-12-22:
@morrownr you are correct about the reg db. Once I set my region, the frequencies no longer showed up as disabled.

Do you (or anyone) know what "EHT" is in the iw info? Could it be "EHT" from the 802.11be (Wifi 7) draft?

I had vague plans to homebrew a new router with the 953ax until I realized that I would likely have to backport the driver to OpenWRT. I haven't used the device much but I can give it a more serious test and perhaps write a review. I do get an initialization error related to Bluetooth, but then the endpoints show up, so I suspect they might work. I can give that a try

I see that one of your main cons for 951 was that it wouldn't work with USB extensions or hubs. I had no issues with this with the 953. I have in the past experienced limitations with the physical length of cabling for USB devices, but I'm guessing this is not the case for you.

morrownr, 2022-12-22:
> I had vague plans to homebrew a new router with the 953ax until I realized that I would likely have to backport the driver to OpenWRT.

Hold on. Will you be using OpenWRT 22.03.x or a current snapshot? The reason I ask is that the packages you need to install are slightly different owing to a change in the location of the firmware but the bottom line is that the OpenWRT devs have already backported the driver to earlier kernels for OpenWRT. 

> perhaps write a review

That would be great.

> I do get an initialization error related to Bluetooth, but then the endpoints show up, so I suspect they might work. I can give that a try

It is not clear to me that Comfast actually intended bluetooth support. For now, ignore bluetooth and report on the wifi.

> I see that one of your main cons for 951 was that it wouldn't work with USB extensions or hubs. I had no issues with this with the 953. I have in the past experienced limitations with the physical length of cabling for USB devices, but I'm guessing this is not the case for you.

That is good info to know and would be good info for a review. 

Let me know if you need any help setting up OpenWRT, Recommend you set up WiFi 6 (AX) and test it first. WiFi 6e (6 GHz) is a different subject. Luci works well with WiFi 6 but not 6e yet so it has to be setup by editing files.

Nick

codebling, 2022-12-22:
> Will you be using OpenWRT 22.03.x or a current snapshot?

Whatever works. I did a quick search for the driver but didn't see it. It's great if they've already backported. I've built OpenWRT myself once and it made me very sad. 

I'd eventually like to test performance between two 6ghz devices, since I live in an area with relatively congested wifi. I suppose I don't need a router for that, would give me an excuse to test out AP mode

morrownr, 2022-12-22:
>> Will you be using OpenWRT 22.03.x or a current snapshot?

> Whatever works. I did a quick search for the driver but didn't see it.

Tell me what hardware you will install OpenWRT on.

codebling, 2022-12-22:
> Tell me what hardware you will install OpenWRT on.

I'll likely wait until Raspberry Pi become more available late January and install on that. I also have an old Buffalo WZR-HP-AG300H that I could install on or a Tenda RX9/TX9/AX12/AX3000, but the latter neither has a USB port nor is directly supported by OpenWRT at this time (though the chipset likely is, its firmware seems to be based on OpenWRT but I haven't succeeded it getting it to accept any of my custom firmware yet)

morrownr, 2022-12-22:
> Buffalo WZR-HP-AG300H

Does this Buffalo have a usb port?

codebling, 2022-12-22:
It does!

morrownr, 2022-12-22:
>> Does this Buffalo have a usb port?

> It does!

Well, you could be in business if it is a Buffalo WZR-HP-AG300H v1. Look at the bottom of the following list:

https://openwrt.org/toh/views/toh_fwdownload?dataflt%5B0%5D=supported+current+rel_%3D22.03.2&dataflt%5BBrand*%7E%5D=Buffalo

To me, it does not matter if it is a USB2 or USB3 port. It will allow you get the 953ax going.

I don't know your experience level with OpenWRT but do be careful and read the instructions because you can brick a router if not careful.

codebling, 2022-12-23:
I am currently running OpenWrt 18.06.1 (kernel 4.9) on that router!  I see that mt7921u [driver](https://github.com/openwrt/openwrt/blob/275c51bf4002eefd85d66fd82d2f2f93da31e8b8/package/kernel/mt76/Makefile#L248) and [firmware](https://github.com/openwrt/openwrt/blob/275c51bf4002eefd85d66fd82d2f2f93da31e8b8/package/kernel/mt76/Makefile#L235) are available on 22. Might try it, but probably not until I get some form of backup

morrownr, 2022-12-23:
> but probably not until I get some form of backup router

Good idea. No backup could make for a bad day.

The files you need to install have changed between 22.03 and the next edition (Snapshot).

For 22.03.x:

kmod-mt7921u
kmod-mt7921e     (the firmware was put in this file for 22.03 for some reason but has been corrected in snapshot) 

For Snapshot:

kmod-mt7921u
kmod-mt7921-firmware

I see that the usb port is usb2 so there will be a limitation of speed but it is no big deal. You can still check out WiFi 6/6e and get used to things.

Good luck.

codebling, 2022-12-23:
Thanks for all of the help and suggestions with OpenWRT!  By the way, 22.03's source code does show a **[firmware package](https://github.com/openwrt/openwrt/blob/openwrt-22.03/package/kernel/mt76/Makefile#L171)**, so maybe 22.03 and snapshot are now aligned, but I don't know as I'm only saying what I see in the source.

By the way, I've updated the `iw` output in initial description with the reg db correctly set. Here's the diff of the CF-953AX (left) with the CF-951AX (right):

```
104,111d103
< 		EHT Iftypes: managed
< 			EHT MAC Capabilities (0x0000):
< 			EHT PHY Capabilities: (0x0000000000000000):
< 			EHT MCS/NSS: (0x):
< 			EHT bw=20 MHz, max NSS for MCS 0-7: Rx=0, Tx=0
< 			EHT bw=20 MHz, max NSS for MCS 8-9: Rx=0, Tx=0
< 			EHT bw=20 MHz, max NSS for MCS 10-11: Rx=0, Tx=0
< 			EHT bw=20 MHz, max NSS for MCS 12-13: Rx=0, Tx=0
154,161d145
< 		EHT Iftypes: AP
< 			EHT MAC Capabilities (0x0000):
< 			EHT PHY Capabilities: (0x0000000000000000):
< 			EHT MCS/NSS: (0x):
< 			EHT bw=20 MHz, max NSS for MCS 0-7: Rx=0, Tx=0
< 			EHT bw=20 MHz, max NSS for MCS 8-9: Rx=0, Tx=0
< 			EHT bw=20 MHz, max NSS for MCS 10-11: Rx=0, Tx=0
< 			EHT bw=20 MHz, max NSS for MCS 12-13: Rx=0, Tx=0
235d218
< 		VHT extended NSS: supported
294,300d276
< 		EHT Iftypes: managed
< 			EHT MAC Capabilities (0x0000):
< 			EHT PHY Capabilities: (0x0000000000000000):
< 			EHT MCS/NSS: (0x):
< 			EHT bw <= 80 MHz, max NSS for MCS 8-9: Rx=0, Tx=0
< 			EHT bw <= 80 MHz, max NSS for MCS 10-11: Rx=0, Tx=0
< 			EHT bw <= 80 MHz, max NSS for MCS 12-13: Rx=0, Tx=0
343,349d318
< 		EHT Iftypes: AP
< 			EHT MAC Capabilities (0x0000):
< 			EHT PHY Capabilities: (0x0000000000000000):
< 			EHT MCS/NSS: (0x):
< 			EHT bw <= 80 MHz, max NSS for MCS 8-9: Rx=0, Tx=0
< 			EHT bw <= 80 MHz, max NSS for MCS 10-11: Rx=0, Tx=0
< 			EHT bw <= 80 MHz, max NSS for MCS 12-13: Rx=0, Tx=0
446,452d414
< 		EHT Iftypes: managed
< 			EHT MAC Capabilities (0x0000):
< 			EHT PHY Capabilities: (0x0000000000000000):
< 			EHT MCS/NSS: (0x):
< 			EHT bw <= 80 MHz, max NSS for MCS 8-9: Rx=0, Tx=0
< 			EHT bw <= 80 MHz, max NSS for MCS 10-11: Rx=0, Tx=0
< 			EHT bw <= 80 MHz, max NSS for MCS 12-13: Rx=0, Tx=0
495,501d456
< 		EHT Iftypes: AP
< 			EHT MAC Capabilities (0x0000):
< 			EHT PHY Capabilities: (0x0000000000000000):
< 			EHT MCS/NSS: (0x):
< 			EHT bw <= 80 MHz, max NSS for MCS 8-9: Rx=0, Tx=0
< 			EHT bw <= 80 MHz, max NSS for MCS 10-11: Rx=0, Tx=0
< 			EHT bw <= 80 MHz, max NSS for MCS 12-13: Rx=0, Tx=0
673c628,629
< 		* [ BEACON_RATE_HE ]: HE beacon rate support (AP/mesh)
```

morrownr, 2022-12-25:
I've been so bust on other things that I have not had to time get smart on some of the WiFi 6/6e stuff but will be taking a harder look at it after the first of the year. Keep me up on your progress.

Nick

fhteagle, 2023-01-05:
Just a follow up after a bit more testing on x86_64 host, using arch linux under 6.1.1 kernel, hostapd 2.11-devel from github:

CF-953AX still crashes 100% of the time for me (entire kernel network stack frozen) if hostapd is told to put this adapter into an existing bridge. I do not see this crash if the adapter is not bridged. I do not see the same crash if bridging an m.2 MT7921K . I need to try bringing up the hostapd instance and moving it into the bridge afterwards, but I have limited opportunities to do testing these days.

travlytle, 2023-01-17:
> @codebling @travlytle
> 
> Would either you like to write a review of cf-953ax? Main Menu item 2 is the list of adapters with in-kernel drivers and the 953ax sure could use a review or two. I have a 951ax and I posted a review but I don't have a 953ax.
> 
> Nick

Hi @morrownr , just seeing this, not sure why I didn't get an email notification. But I'm going to have to pass on the review, it might be seen as bias :stuck_out_tongue_winking_eye:.