morrownr/USB-WiFi #168: HOT: If you have a 7612u or 7610u based adapter and can test a proposed patch, read this issue...

morrownr, 2022-12-24:
Starting with kernel 6.1.x, adapters with the mt7612u or mt7610u chipset are causing a hard system lock when connection is made.

A patch has been proposed but needs to be tested. I will report any results posted here back to linux-wireless and maybe that can help speed the approval process along:

Can you please try the patch below?

https://patchwork.kernel.org/project/linux-wireless/patch/20221217085624.52077-1-nbd@nbd.name/

Note: This issue only came about starting with kernel 6.1 rc1 so you will need to upgrade your system to kernel 6.1 and you will need to patch in the proposed fix and compile.

Thanks

Danct12, 2023-01-02:
Using a CF-WU785AC with this patch applied on top of kernel 6.1.2 from Arch Linux, I can confirm that this fixed my kernel panic issue.

[image]

morrownr, 2023-01-02:
Thanks @Danct12 

I saw your report in linux-wireless. I saw a version 5 patch from Felix today so I suspect it won't be long before we see it in 6.1 and 6.2.

bdantas, 2023-01-21:
I have an ALFA AWUS036ACHM which I use to create a 802.11ac AP on a homebrewed router running Tiny Core Linux. After upgrading TCL to version 14 (which uses linux 6.1.2), router immediately becomes unresponsive when first client authenticates.

I applied version 5 patch from Felix and rebuilt the mac80211 kernel module. I can confirm that problem is solved with the patched module.

morrownr, 2023-01-21:
@bdantas 

Good to hear. The patch has been approved and should show in releases soon.

Danct12, 2023-01-21:
The latest kernel on Arch Linux has the fix backported as well.

bdantas, 2023-01-21:
@morrownr I had actually come here to ask you for help troubleshooting the crasehes, but this thread led me to the solution. Your github repository is a precious resource. Thank you, Nick.

morrownr, 2023-01-22:
@bdantas 

> Your github repository is a precious resource. 

Thanks but to be clear, it takes a lot more than me to make this site work. Those that help know who they are my thanks goes to them and to the users who stop by.

Nick

morrownr, 2023-02-08:
FYI: This issue appears to be fixed in mainline kernel 6.1.9.

I have had time to circle back and test kernel 6.2.x yet  but I suppose if it is fixed in 6.1.9, it is likely fixed in 6.2.

For those that read this in the future: The problematic bug this issue is speaking did not exist before kernel 6.1. Now that it is fixed, the problem will showly start fading away...

bjlockie, 2023-02-09:
I totally forgot about this.
I've been using a 7612 in 6.1.9 on my raspberry pi 24x7 for a week.
I just put in my 7921 and it was way better link speed. :-)