morrownr/USB-WiFi #169: WPA3-SAE Transition mode   or   WPA3-AES Transition mode?

mendel5, 2022-12-24:
Hi @morrownr ,

I was browsing through this repository and noticed something that I'm not sure about. It's about this line:
```
# wpa_key_mgmt=SAE WPA-PSK is required for WPA3-AES Transitional
```
This line can be found in the following five files:
- https://github.com/mendel5/USB-WiFi/blob/main/home/AP_Mode/Bridged_Wireless_Access_Point.md
- https://github.com/mendel5/USB-WiFi/blob/main/home/AP_Mode/hostapd-2g.conf#L48
- https://github.com/mendel5/USB-WiFi/blob/main/home/AP_Mode/hostapd-5g.conf#L58
- https://github.com/mendel5/USB-WiFi/blob/main/home/AP_Mode/hostapd-6eg.conf#L75
- https://github.com/mendel5/USB-WiFi/blob/main/home/AP_Mode/hostapd-6g.conf##L74

Would it be possible that the line was actually meant to be written as the following one?
```
# wpa_key_mgmt=SAE WPA-PSK is required for WPA3-SAE Transitional
```

So should it be `WPA3-AES` like in the first code block or `WPA3-SAE` like in the second code block?

morrownr, 2022-12-25:
> So should it be WPA3-AES like in the first code block or WPA3-SAE like in the second code block?

Everything inside the AP_Mode folder has to do with using hostapd. The hostapd documentation has to be followed carefully so please don't modify anything there unless you are reading the hostapd docs and see a mistake.

mendel5, 2022-12-25:
Yes, I understand that.

My question is not about changing the code,  it's about changing the comment that describes the code. As far as I know there is no such thing as `WPA3-AES Transition mode`.

mendel5, 2022-12-25:
Hi @morrownr
Yes, I understand that.

My question is not about changing the code,  it's about changing the comment that describes the code. As far as I know there is no such thing as `WPA3-AES Transition mode`.

morrownr, 2022-12-25:
Hi @mendel5 

Ah. I misunderstood.

You are welcome to correct the terminology in the comments and you are correct, I am not doing a good job of the terminology.

> As far as I know there is no such thing as WPA3-AES Transition mode.

Correct. It should say:

WPA2/WPA3 Transitional

...and WPA3 should say:

WPA3 Personal

If you have time and want to submit a PR, I would appreciate it.

Thanks,

Nick

mendel5, 2022-12-31:
> Ah. I misunderstood.

No worries!

I had opened this pull request to fix the issue: https://github.com/morrownr/USB-WiFi/pull/172

Thank you for merging the pull request!

closed 2022-12-31 by mendel5