morrownr/USB-WiFi #178: Adding new section to the in-kernel supported adapter list: rtl8811cu chipset

morrownr, 2023-01-02:
##### ```chipset - Realtek rtl8811cu (version without bluetooth support) - supported in-kernel since Linux kernel 6.2 (2023)```

I will begin adding adapters as time permits. I plan to add only adapters that are single-state and single-function (no bluetooth). Versions of the chipset that include bluetooth support (rtl8821cu) have some interesting problems but the rtl8811cu does not. I have tested the in-kernel driver in kernel 6.2. It works but performance is not good right now. I am sure that performance will increase as more eyes are focused on the issue. Also, I am getting close to releasing a new out-of-kernel rtl8821cu driver that is VERY good (even MU-MIMO works) so I am convinced that users who buy single-state usb wifi adapters based on the rtl8811cu chipset will have solid support many years into the future. If you have suggestions for adapters that should be included, please let me know.

bjlockie, 2023-01-04:
Could your out of kernel driver be submitted to replace the in-kernel one?

morrownr, 2023-01-04:
> Could your out of kernel driver be submitted to replace the in-kernel one?

No. The Realtek out-of-kernel drivers have far too much code that does not meet Linux Wireless Standards.

I am confident that over time, the in-kernel driver will have much better performance and eventually we can retire the out-of-kernel driver. My experience with maintaining the 8821cu driver over the last few years reminds me to let users know:

- Do not buy MULTI-STATE adapters based on the rtl8811cu. There are plenty of single-state adapters available.

- Do not buy MULTI-FUNCTION adapters based on the rtl8821cu chipset. Buy single-function adapters based on the rtl8811cu adapter.

There is some junk out there so be careful to buy good quality, single-state, single-function adapters based on the rtl8811cu chipset. They should provide good stable service for many years. 

I still think Mediatek based adapters are best for most users due to the very strong support Mediatek provides regarding in-kernel driver support but there are parts of the world where Mediatek based adapters are currently hard to find.

Nick