morrownr/USB-WiFi #181: First Alfa Mediatek adapter based on MT7921 has been released!

usama7628674, 2023-01-11:
@morrownr Your wait is over. 
https://www.alfa.com.tw/products/awus036axml?variant=39754360684616

It's there on their website but I could not find any online retailer where it is available. What's your thought on this?

userabcd0, 2023-01-11:
Interesting, thanks for that info.

Listed  on Alfa product page they mention bluetooth, wonder if it is working on this device. Bluetooth is not working on the comfast 953AX 7921AU device on both windows and linux with old and new drivers installed.

usama7628674, 2023-01-11:
@userabcd0 It's not available yet to buy so hopefully by the time it arrives bluetooth issue will be fixed.

morrownr, 2023-01-11:
> @morrownr Your wait is over. https://www.alfa.com.tw/products/awus036axml?variant=39754360684616

Hey @usama7628674 

Well, I'm actually waiting on the "high power", as Alfa calls it, version to be released. I have an ACH and ACHM and am kinda used to that long range capability so I'm going to wait.
 
> It's there on their website but I could not find any online retailer where it is available. What's your thought on this?

The only pre-order I see currently on an Alfa dealer site is:

https://store.rokland.com/collections/wifi-6-6e/products/alfa-awus036axml-802-11ax-wifi-6-1800-mbps-tri-band-wifi-usb-adapter-w-bluetooth

I'm going to add some extra replies here to save some writing...

> Bluetooth is not working on the comfast 953AX 7921AU device on both windows and linux with old or new drivers installed.

I have a Comfast CF-951AX so I am familiar with the bluetooth issue. When I first got the adapter, I tried to get BT going. One day I remembered something from about 10 years ago and did some research. What I remembered was Intel releasing a paper regarding USB3. USB3 is not mankind's best creation. USB2 has proven to be solid over the years but USB3 could have used some more testing and there are issues. One of the issues is that USB3 cables, wires and connections emit a radio signal in the 2.4 GHz range. Well, the BT frequency is right in that range as is the 2.4 GHz wifi band. Opps...

A little more research: Take a look at the rtl8822bu chipset. It is a AC1200 class chipset that is popular but guess what? It is USB2 only. Before you say... waa waa wait a minute, my rtl8812bu adapter can do USB3. That is correct, the wifi only version of the chipset, the rtl8812bu, can do USB3. However, the multifunction version, wifi and BT,  the rtl8822bu, cannot do USB3. Anyone wanna bet that it is the USB3 vs. BT interference problem that is the reason?

So, is that the reason we are seeing so much confusion with the mt7921au based adapters? It could be. I'll bet Comfast turned the BT off so they could activate USB3. This is all speculation on my part so we will have to wait to find out.

> See also on websites many people with computers with the Mediatek 7921 wifi/BT card having trouble with wifi and bluetooth connections and are removing them to replace with intel wifi/bluetooth cards.

I have a laptop that is about a year old with a mt7921 card. WiFi and BT both work fine.

Last thoughts: I've been around computers for a long time. I've seen disruptive changes and additions to the industry many times. What we have seen over the last few years is 3 MAJOR disruptive factors that are all having to be sorted out more or less at the same time and this is just with WiFi.

- WPA3
- WiFi 6 (AX)
- WiFi 6e (6 GHz)

The drivers have become incredibly complex. Has anyone noticed that the in-kernel driver for the mt7921 came out in kernel 5.12. It only had managed and monitor mode support. USB support appeared in kernel 5.18. AP mode support appeared in kernel 5.19 and last week I saw initial support for P2P and some other things being sent in. I am very surprised at how stable the mt7921u and mt7921e drivers are. I have tried to make the rtl8852a/bu out-of-kernel drivers available but they are really bad. The increase in size over the previous generation is incredible and I don't see how Realtek of Mediatek keep up with the code in these WiFi 6 drivers.

Enough for today. Cheers,

Nick