morrownr/USB-WiFi #183: Issue CF-WU785 AC

userabcd0, 2023-01-12:
Received this comfast device yesterday, it has the 7612u chipset.
Test with the cf-785ac plugged in after pc is booted and pc is rebooted with the usb wifi conneted:
Plugged it into a usb hub connected to a usb port on the pc already booted and the device works correctly in both windows and linux.

Reboot the pc and the cf-785ac connects and works correctly.

Test with cf-785 connected to pc via usb hub from cold boot:
Shut down the pc with the cf-785ac wifi connected via the usb hub. Cold boot and the pc gets stuck during boot in windows or linux. Unplug the cf-785ac and the pc continues the boot process.

Tried with different usb hubs and it is the same, the boot is stuck until the cf wifi device is unplugged then the boot cycle continues. The wifi device can be plugged into the usb hub and it works normally.

Test with the cf-785ac plugged directly into a pc usb 2 or usb 3 port:
PC boots correctly with the cf-785ac wifi device connected in windows and linux.

Do not have this situation with cf953ax, pc boots normally with usb wifi connected via a usb hub, nor with any realtek wifi devices.

Anyone have the same or could suggest something for a fix when the cf-785ac is connected via a usb hub, it prevents the pc booting from a cold boot?

morrownr, 2023-01-12:
@userabcd0 

There is an issue in Linux kernels 6.1 and 6.2 rcx where adapters based on the mt7612u and mt7610u are broken. Seriously broken as in lock your system broken. A patch provided by Intel broke the drivers. A new patch by Mediatek fixes things but the fix is to the mac80211 driver which is part of the system stack so needs approval before it can be applied. Hopefully that happens very soon.

In the meantime, use kernels <= 6.0.

Nick

morrownr, 2023-01-12:
@userabcd0 

If you are using kernel 6.1 or greater and this is your problem, I can point you to the patch if you want.

userabcd0, 2023-01-13:
> 

Thanks a lot for that explanation.

I am on kernel 6.1.

I will first try downgrade the kernel to 5.15 and give that a try, if that does not work then I will ask for the patch.

Thanks for your help.

morrownr, 2023-01-13:
@userabcd0 

> When the wifi device is plugged in to a usb hub connected to a usb port on the pc, the pc is prevented from booting (cold boot) Remove the wifi adapter and the pc boot cycle continues and can then plug and use the wifi adapter.

I did not real close enough the first time. The problem I wrote about above would happen the same if plugged into a hub or directly into a port on the PC. 

Is the hub a powered hub?

morrownr, 2023-01-13:
> I will first try downgrade the kernel to 5.15 and give that a try, if that does not work then I will ask for the patch.

Well, maybe I misread the 2nd time. Long day.

One way or another, the mt7612u and mt7610u chipsets are not working with kernels greater than 6.0 so testing with 5.15 is good idea.

I really hope the patch goes in soon. Right now we see a trickle of users reporting problems in the various distro forums but it is going to turn into a tidal wave as more distros move to the new LTS (6.1).

morrownr, 2023-01-13:
FYI: I have a CF-951AX. The problem does not bother the mt7921au chipset. Of course, this chipset is not yet supported in the 5.15 kernel so what we need for distros like Ubuntu, is for them to release a HWE 5.19 kernel and all would be good... until we decide to play with the latest toy...

userabcd0, 2023-01-13:
> @userabcd0
>
>
> I did not real close enough the first time. The problem I wrote about above would happen the same if plugged into a hub or directly into a port on the PC.
>
> Is the hub a powered hub?

The hub has a power point on it but I have not connected the power supply. 

When I connect the CF-WU785AC wifi device directly to the USB port 2 or 3 on the PC, it boots no problems in windows and linux and with kernel 5.15 and 6.1. The problem is the Comfast device is huge and it takes 2 usb ports on the PC.

When I connect the CF785AC via the usb hub the boot gets stuck, cannot enter BIOS as it seems to be stuck before that. When the boot is stuck and I just remove the wifi dongle, then the boot proceeds and I can enter the BIOS, replug the wifi dongle and it is working.

Strangely, I just bought a vention USB C hub, I connect the CF785 wifi dongle to the vention hub and plug the vention hub into my original usb hub giving boot problems and all is now booting correctly.

I think for me the matter is solved with the device connected via the vention hub connected to my usb hub.

userabcd0, 2023-01-13:
> FYI: I have a CF-951AX. The problem does not bother the mt7921au chipset. Of course, this chipset is not yet supported in the 5.15 kernel so what we need for distros like Ubuntu, is for them to release a HWE 5.19 kernel and all would be good... until we decide to play with the latest toy...

I also have a 953AX (7921AU) and this device is giving no boot problems, it works on all usb hubs in linux on kernels 5.19 and 6.1.

morrownr, 2023-01-14:
See #186