morrownr/USB-WiFi #189: WARNING: The Netgear A8000 USB WiFi Adapter does us a mt7921au chipset but ...

morrownr, 2023-01-19:
Warning: This Netgear A8000 uses a device ID that is not yet in the Linux kernel driver, mt7921u, and until that happens, users will have to add the device ID (ID 0846:9060 NetGear, Inc) and compile the driver. This is not a trival task and should only be undertaken by experienced developers or with a mentor to help.

This information is based on a user report.

What we need to happen is that someone that has this adapter needs to post a PATCH to linux-wireless so that the device ID goes into the kernel. If you have this adapter and know how to submit a PATCH after confirming the above info is correct, please do so. If you have this adapter and do not know how to submit a PATCH or test a PATCH, please say so and we will try to find someone to work with you.

Nick

russeree, 2023-01-21:
Just went out and purchased this adapter to replace the TP Link T9UH that was frequently disconnecting. I can confirm that device ID is 0846:9060 NetGear, Inc. Wireless_Device and is not initializing in Ubuntu Kinetic running 5.19

morrownr, 2023-01-23:
@russeree 

 Issues #63 , starting with the 2nd message goes through the process of getting the ID in the proper file and compiling. Actually, the user compiled and installed the latest 6.2-rc kernel but unless you enjoy some of the issues that may come with a test version, maybe using the most recent stable like 6.1.7 would be better.

What needs to happen is a PATCH message needs to go to linux-wireless so as to get the info in the Linux kernel. I actually started a PATCH message today but did not finish. I should have some time this week unless someone else comes along and wants to get it in.

Nick

russeree, 2023-01-23:
I have submitted a kernel patch for the AXE3000 (A8000) USB VID:PIDs . 

https://lore.kernel.org/linux-mediatek/20230123090555.21415-1-git@qrsnap.io/T/#u

morrownr, 2023-01-23:
@russeree 

Thanks for submitting the device ID. It needed to be done.

I have added the A8000 to the in-kernel list (menu item 2). Any possibility of getting you to do a review?

Nick

Nick

russeree, 2023-01-23:
Yeah email me a patch file and I will gladly test.

morrownr, 2023-01-24:
@russeree 

I wasn't talking about a patch. I added a section about the A8000 to the following section of the site and am looking for someone that owns the adapter to take a look and maybe write a short review for others.

https://github.com/morrownr/USB-WiFi/blob/main/home/USB_WiFi_Adapters_that_are_supported_with_Linux_in-kernel_drivers.md

You have to scroll down to see the entry for the A8000.

russeree, 2023-01-25:
Yeah no problem, I will checkout and create a PR, for this. One note though... is the A8000 actually in kernel if the device is not currently plug and play without recompiling the kernel with the correct USB VID:PID combo?

morrownr, 2023-01-25:
> One note though... is the A8000 actually in kernel if the device is not currently plug and play without recompiling the kernel with the correct USB VID:PID combo?

It is not. That is why I put the detailed warning with the listing. My reasoning, after seeing that at least 3 people had already bought the adapter with no listing, was that maybe I should list it with a warning.

russeree, 2023-01-25:
Perfect. This page led me to get the A8000. With that said after patch the adapter blew my expectations away.

Yeah I am writing a review now. Any review guidelines to follow??

witcheslive, 2023-02-02:
just got one of these to use on a gentoo box that i am using as a router. planning on getting this installed and working on friday morning, and I got kernel 6.1.9 ready but i'm not sure if that includes that device id patch or not. i know how to apply it if it does still need that, but the wiki mentions needing to compile a driver, too? which one?

will be happy to write up a review/my experiences getting this working

russeree, 2023-02-02:
> just got one of these to use on a gentoo box that i am using as a router. planning on getting this installed and working on friday morning, and I got kernel 6.1.9 ready but i'm not sure if that [...]

The AXE3000 (A8000) device ids are not included. Please use my kernel patch file to get support for the device.  The device will register and be usable immediately after adding the device ids. No additional drivers necessary for me.

morrownr, 2023-02-02:
Remember that Linux in-kernel drivers have 2 parts: The part that we usually call a driver, in this case mt7921u, and the firmware which may consist of one or more binary blobs.

Firmware. Go to the Main Menu:

https://github.com/morrownr/USB-WiFi

Go to the 2nd section in menu item 8. Follow the instructions for downloading the 2 firmware files for the mt7921au. Install them in the indicated location. Done.

Driver. mt7921u: the driver is part of the kernel. If a kernel file needs to be patched, you apply the patch to the source files or files that need the patch and then compile either the driver or the entire kernel. I prefer just to compile the whole thing. The file that needs to be modified is:

drivers/net/wireless/mediatek/mt76/mt7921/usb.c

Here is a modified copy. Lines 18-20 are the ones that have been added.

[usb.c.tar.gz](https://github.com/morrownr/USB-WiFi/files/10564213/usb.c.tar.gz)

Let me know how it goes and a full report would be very welcome.

Nick

morrownr, 2023-02-02:
@russeree 

> Also I have gotten no feedback on the mailing lists and wanted to make sure my patch was submitted correctly. 

You can check on it at Patchwork:

https://patchwork.kernel.org/project/linux-wireless/list/

witcheslive, 2023-02-02:
lines 16 and 17 of my extant usb.c looked different than the patch file or that example but just saying this for informational purposes as adding the new lines worked anyway. i was able to successfully compile the kernel with this change and load the firmware, and the device is indeed showing up as a wlan device

what i am having issues with now is this is my first rodeo with hostapd, and i'm very unsure what to set some of these settings.

for instance, the hardware mode:

```
# Operation mode (a = IEEE 802.11a (5 GHz), b = IEEE 802.11b (2.4 GHz),
# g = IEEE 802.11g (2.4 GHz), ad = IEEE 802.11ad (60 GHz); a/g options are used
# with IEEE 802.11n (HT), too, to specify band). For IEEE 802.11ac (VHT), this
# needs to be set to hw_mode=a. For IEEE 802.11ax (HE) on 6 GHz this needs
# to be set to hw_mode=a. When using ACS (see channel parameter), a
# special value "any" can be used to indicate that any support band can be used.
# This special case is currently supported only with drivers with which
# offloaded ACS is used.
# Default: IEEE 802.11b
hw_mode=g?a?idk?????
```

any tips on configuring this for hostapd would be welcome as not only have i never done this before, trying to find any examples for a 6e device is proving elusive

morrownr, 2023-02-02:
> this is my first rodeo with hostapd...

Now you tell us.

Okay, go to:

https://github.com/morrownr/USB-WiFi

Select:

9. How to build a bridged wireless access point

That document has a lot of good info in. There is an example hostapd.conf in the document. Recommend you grab it and start there.

I can't be more specific because you did not tell us what you are trying to do.

witcheslive, 2023-02-02:
it is a very simple home network setup where i just need this device to be a wireless switch on the network. the network setup itself is just the same gentoo box the netgear device is connected to serving as a router running dnsmasq and iptables.

devices on the wireless network run the gamut from things like the old chromecast or the wii u, to the steam deck, to very recent smartphones with ax support.

mostly what i'm not sure about here is how to configure this for maximum usage of the wifi 6 features. there are several devices that can use wifi 6, and although nothing yet uses 6e/the 6 ghz band, it would be nice to not have to revisit this when i do get one if at all possible!

i am working off of https://wiki.gentoo.org/wiki/Hostapd mostly but as i mentioned above, i'm stuck at the parts that involve setting the bands and modes as that is what there are no extant examples of how to configure this device in particular.

morrownr, 2023-02-02:
Okay. Fair enough.

I am going to strongly suggest you get the new adapter going as a 5 GHz AP first. Once that is working well. Let's try AX (WiFi 6.) Once that is going... you do have something to test with, right? ...we can then try to get WiFi 6e going. 

I have a sample WiFi 6e hostapd.conf file and I'll show you where it is once we get there. I have not been able to test it yet as I only have one mt7921au adapter so far and my wifi router stole it from me to add WiFi 6 capability to the router.

The 5 GHz hostapd.conf example in the document I pointed you to is a really good place to start.

# mt7921au
ht_capab=[LDPC][HT40+][HT40-][GF][SHORT-GI-20][SHORT-GI-40][TX-STBC][RX-STBC1][MAX-AMSDU-7935]

# mt7921au (VHT Cap. 339071b2)
vht_capab=[MAX-MPDU-11454][RXLDPC][SHORT-GI-80][TX-STBC-2BY1][RX-STBC-1][SU-BEAMFORMEE][BF-ANTENNA-4][MAX-A-MPDU-LEN-EXP7][RX-ANTENNA-PATTERN][TX-ANTENNA-PATTERN]

witcheslive, 2023-02-02:
OK was able to get it lit up, will report back with anything interesting, though so far the only thing of note is that the samsung/android wifi interface does not show a 6 in the wifi icon. 

The only issues so far are just me being a hobbyist network administrator and getting confused by a few things along the way, and my annoyance that apparently the zflip 4 only has wifi 6 but the zfold 4 has 6e that was cheap samsung!

witcheslive, 2023-02-05:
One of the laptops -- and the most important one to have good wifi -- regularly disconnects with logs showing the device is regularly disconnected for "deauthenticated code 1" and even after updating the wifi driver on it to the most recent (its a 2021 high end laptop, it also has a mediatek chip in it as I found out) and every so often it seems like the driver crashes and any attempt to restart hostapd brings down the entire network so badly that I have to reboot the entire router for it to work again.

I do not recommend this device at this time for use as an AP. It is extremely unstable.

morrownr, 2023-02-05:
Let me remind you...

> any tips on configuring this for hostapd would be welcome as not only have i never done this before...

I understand that you were in a hurry. What I don't think you understood is that setting up a stable AP with hostapd takes knowledge and experience... for the 2.4 GHz band. It gets more complicated with 5 GHz and then it goes off scale with WiFi 6 and 6 GHz is currently about like learning to fly an F-16 in one weekend.

I have a CF-951AX (same chipset and driver as the A8000) running on OpenWRT in AX (WiFi 6) very stable on my main router. I have played with trying to set up 6 GHz AP mode but wow... I am not there yet. I've used hostapd for years and I used Linux since 1994. I know what I am up against. Remember that 6 GHz mandates WPA3 use so what os you are setting hostapd on has to have everything totally lined out for WPA3. Not all distros are there.

Anyway, I don't have an A8000 so I can't speak to the quality but some of the things you seem to be blaming on the adapter may be because of other issues. There are experienced people here that can offer guidance on setting up an AP with a usb wifi adapter but it is something that needs time to work well.

Good luck.

witcheslive, 2023-02-06:
Update that while I am hesitant to declare victory yet, it does seem like the problem was due to the wireless adapter getting the same IP address as the bridge connecting all the interfaces. I just had to give it a different IP address. My fiancee plays a lot of flight sims and I am to understand that changing an ip address in a conf file is markedly less complicated than flying an F-16.

witcheslive, 2023-02-06:
Spoke too soon, crashed overnight after more issues staying connected to the laptop. Would be nice to try if actually configuring it correctly with the 6e bands make it more stable, but I guess that is forbidden knowledge without a pilot's license, so time to get an ubiquiti or whatever.

morrownr, 2023-02-07:
> Would be nice to try if actually configuring it correctly with the 6e bands make it more stable, but I guess that is forbidden knowledge without a pilot's license, so time to get an ubiquiti or whatever.

Running on the 6e band in AP mode is not forbidden knowledge, it is just hard to find knowledge. If I had a solution that would work, I would share it.

My adapters that has the mt7921au chipset is earning a living as an AP currently. It is plugged into a USB3 port on my main wifi router that runs OpenWRT. It uses the 5 GHz band and AX and is really stable. OpenWRT is really no ready for 6 GHz yet as the gui interface can't handle it yet. Once the RasPiOS is updated to kernel 6.1, I will give 6 GHz a try.

I'll be glad to look over your hostapd.conf file if you want me to take a look. Of course, there is more to it than that but I don't really know how you are setup.

bjlockie, 2023-02-07:
If you're using Raspbian you can upgrade to 6.1.9 now.
$ sudo rpi-update next

I did that but I haven't tried my CF-953 yet plus I don't run an AP anymore.

notmatthewa, 2024-04-30:
Just got one of these bad boys after i somehow broke the drivers on my linux machine, hoping that it would allow me to connect to the internet and run `sudo apt update`. However, i'm clearly not understanding how to get the driver compiled on my machine. I followed the instructions from the readme for the mt7921au chipset (i think that's the correct one), but whenever i do `sudo modprobe -v mt7921u`, i get the error that the mt7921u driver cannot be found in my kernal. I see a lot of places online mentioning that i can just downgrade my kernal, but i can't do that without internet.

This usb is my only hope. Do you guys have any advice?

morrownr, 2024-04-30:
Post the results of the following command with the adapter plugged in:

$ lsusb