morrownr/USB-WiFi #191: Building Driver for 5.19.0-28-generic Lacks GCC-12

OS2User, 2023-01-21:
I am using Linux Mint 21.1 (Vera), which is based on Ubuntu 21.04 Jammy and until recently only offered kernels 5.15.0-xx. 
Since a few days now there also is the first 5.19.0-28-generic kernel available from the repository.

For the last 3 kernel updates within 5.15.0-xx the DKMS driver update worked flawlessly :-)

Now after also installing the first 5.19.0 variant, I just rebooted and tried to build the RTL8812bu driver for it as well.

So here comes the first of my questions:
1.) How can I maintain drivers for both kernel lines, 5.15.0-x and now additionally 5.19.0-x, , so that I can fall back if needed, when install.sh always seems to remove everything for the older kernel line?
Like right now, when building the driver for 5.19.0 failed, but the "settings" and modules for 5.15.0 are already removed from `/var/lib/dkms/rtl88x2bu/5.13.1`, thus rebooting with 5.15.0 wouldn't allow the USB-WiFi adapter to work either, unless the driver is built yet again.

2.) From where is '$ sudo ./remove-driver.sh' supposed to be run? 
From within the source directory '*/src/88x2bu-20210702' or its clone driver directory '/src/rtl88x2bu-5.13.1'?

3.) Now with building the driver for kernel 5.19.0-28 install.sh is having an issue with a missing GCC-12. I only have 
`$ gcc -v 
 gcc version 11.3.0 (Ubuntu 11.3.0-1ubuntu1~22.04)`.
Well, I'm lost here. How can the new kernel 5.19.0 have been built during the recent update with GCC-12 when I only have V11.3.0?
What do I need to do to fix this?

Below is the terminal output from /install-driver.sh and /var/lib/dkms/rtl88x2bu/5.13.1/build/make.log: 
  
#### # ./install-driver.sh
``` 
: install-driver.sh v20230109
: x86_64
: 5.19.0-28-generic
: gcc (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0
: dkms-2.8.7
: SecureBoot disabled
rtl88x2bu/5.13.1, 5.15.0-57-generic, x86_64: installed (WARNING! Diff between built and installed module!)
rtl88x2bu/5.13.1, 5.15.0-58-generic, x86_64: installed
rtl88x2bu/5 (copy).13.1, 5.15.0-57-generic, x86_64: built
rtl88x2bu/5 (copy).13.1, 5.15.0-58-generic, x86_64: built
Removing a dkms installation: rtl88x2bu
Module rtl88x2bu-5.13.1 for kernel 5.15.0-57-generic (x86_64).
Before uninstall, this module version was ACTIVE on this kernel.

88x2bu.ko:
 - Uninstallation
   - Deleting from: /lib/modules/5.15.0-57-generic/
rmdir: failed to remove '': No such file or directory
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

depmod.................
Module rtl88x2bu-5.13.1 for kernel 5.15.0-58-generic (x86_64).
Before uninstall, this module version was ACTIVE on this kernel.

88x2bu.ko:
 - Uninstallation
   - Deleting from: /lib/modules/5.15.0-58-generic/updates/dkms/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

depmod................
Deleting module rtl88x2bu-5.13.1 completely from the DKMS tree.
Removing 88x2bu.conf from /etc/modprobe.d
Removing source files from /usr/src/rtl88x2bu-5.13.1
Installing 88x2bu.conf to /etc/modprobe.d
The dkms installation routines are in use.
Copying source files to /usr/src/rtl88x2bu-5.13.1
Creating symlink /var/lib/dkms/rtl88x2bu/5.13.1/source -> /usr/src/rtl88x2bu-5.13.1
The driver was added to dkms successfully.

Kernel preparation unnecessary for this kernel. Skipping...

Building module:
cleaning build area...
'make' -j4 KVER=5.19.0-28-generic KSRC=/lib/modules/5.19.0-28-generic/build...(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.19.0-28-generic (x86_64)
Consult /var/lib/dkms/rtl88x2bu/5.13.1/build/make.log for more information.
Command exited with non-zero status 10
Compile time: 0.66 seconds
An error occurred. dkms build error:  10
Please report this error.
Please copy all screen output and paste it into the problem report.
Run the following before reattempting installation.
$ sudo ./remove-driver.sh
```
#### /var/lib/dkms/rtl88x2bu/5.13.1/build/make.log 
```
DKMS make.log for rtl88x2bu-5.13.1 for kernel 5.19.0-28-generic (x86_64)
Fri 20 Jan 2023 15:20:05 CST
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.19.0-28-generic/build M=/var/lib/dkms/rtl88x2bu/5.13.1/build  modules
make[1]: Entering directory '/usr/src/linux-headers-5.19.0-28-generic'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: x86_64-linux-gnu-gcc-12 (Ubuntu 12.1.0-2ubuntu1~22.04) 12.1.0
  You are using:           
/bin/sh: 1: gcc-12: not found
(standard_in) 1: syntax error
  CC [M]  /var/lib/dkms/rtl88x2bu/5.13.1/build/core/rtw_cmd.o
/bin/sh: 1: gcc-12: not found
make[2]: *** [scripts/Makefile.build:257: /var/lib/dkms/rtl88x2bu/5.13.1/build/core/rtw_cmd.o] Error 127
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [Makefile:1850: /var/lib/dkms/rtl88x2bu/5.13.1/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.19.0-28-generic'
make: *** [Makefile:2506: modules] Error 2
```

morrownr, 2023-01-21:
[How to install multiple GCC compiler versions.txt.tar.gz](https://github.com/morrownr/USB-WiFi/files/10473282/How.to.install.multiple.GCC.compiler.versions.txt.tar.gz)

See if the above file helps.

Nick

morrownr, 2023-01-22:
> For the last 3 kernel updates within 5.15.0-xx the DKMS driver update worked flawlessly :-)

As it should be.

> Now after also installing the first 5.19.0 variant, I just rebooted and tried to build the RTL8812bu driver for it as well.

When is the last time you updated the driver from the repo here? Here is why that is very important:

The Linux kernel is under constant revision. Patches that make big changes primarily happen in the 2 weeks before a new development cycle and mostly what we see during the 2 month (more or less) testing period is bug fixing. What this means is that every time a new kernel development cycle starts, I have to check for changes in the kernel and make appropriate changes in the driver. These types of changes usually do not happen with point updates to a kernel... such as 5.15.12 to 5.15.15. This means you usually do not have to worry much a new kernel flows into your system if it is a point update. However, if it is a new kernel, such as 5.15.17 to 5.19.3, is the upgrade you making, do not upgrade the kernel until after you pull the latest code from the repo here and compile and new driver. It is done like this:

```
$ sudo ./remove-driver.sh
$ git pull
$ sudo ./install-driver.sh
```

> How can I maintain drivers for both kernel lines, 5.15.0-x and now additionally 5.19.0-x, , so that I can fall back if needed, when install.sh always seems to remove everything for the older kernel line?

As long as your keep the driver in your ~/src/88x2bu-20210702, the clone directory, up to date, you should not have a problem... or rarely have a problem.

> From where is '$ sudo ./remove-driver.sh' supposed to be run?

The clone directory: ~/src/88x2bu-20210702

> How can the new kernel 5.19.0 have been built during the recent update with GCC-12 when I only have V11.3.0?

See the compressed file that I posted in the above message. 

> What do I need to do to fix this?

The first thing I would do it run: (all in the clone directory)

$ sudo ./remove-driver.sh

Hopefully that will clean things up. Then:

$ git pull

That should pull in all of the changes since the last time you updated the driver. Then follow the instructions in the compressed file install gcc 12 and set it as the default. Then: (as shown in the README, make sure you are in the clone directory)

$ sudo ./install-driver.sh

You make need to run that booted into each kernel you want supported but once done, dkms should keep them all current.

The key is knowing that a kernel upgrade from one release version to another is a BIG deal.

Final word: There is a reason that the menu item for the in-kernel driver supported adapters is near the top of the menu. Most Linux users would be happier if using usb wifi adapters that are supported by in-kernel drivers.

Hope this helps.

Nick

OS2User, 2023-01-23:
> See if the above file helps.

Nick, thank you very much for this great guide. It really helped me a lot. GCC-12 is now installed and active.
Trying next to rebuild the driver module for kernel 5.19.0-28.

morrownr, 2023-01-23:
@OS2User 

I have several guides that I think I need to add to the main menu.

Let me know if you are able to get things all the way to where you want them.

Good luck.

Nick

OS2User, 2023-01-24:
Hi Nick,

> > Now after also installing the first 5.19.0 variant, I just rebooted and tried to build the RTL8812bu driver for it as well.
> 
> When is the last time you updated the driver from the repo here? 

That was right before I tried to install the driver for the first time available 5.19.0 series kernel:  `/src/88x2bu-20210702# git pull` 
There where no updates/changes available.

> As long as your keep the driver in your ~/src/88x2bu-20210702, the clone directory, up to date, you should not have a problem... or rarely have a problem.
> 
> > From where is '$ sudo ./remove-driver.sh' supposed to be run?
> 
> The clone directory: ~/src/88x2bu-20210702
OK, here we seem to have a slight name convention mismatch, me thinks. 

When `# ./remove-driver.sh`  finishes, it says: 
`The driver was removed successfully.
You may now delete the driver directory if desired.`  

Therefore for me `/src/88x2bu-20210702`  was the driver's [source] directory it referred to, whereas the copy (which apparently shall not be named or mentioned) getting created during the installation run, `/src/rtl88x2bu-5.13.1` , is what I was referring to as its clone directory. 
And you seem to be referring to `/src/88x2bu-20210702`  as apparently the [git] clone directory.

Well, with that out of the way now, I think I am slowly starting to grasp how the driver installation and dkms is working and therefore need to modify my previous questions.

My understanding so far:
1.)  Since I didn't have gcc-12 available at the time of installing kernel 5.19.0-x-generic and shortly after, 6.1.0-10x-oem, dkms couldn't build the driver. Thanks to your provided great guide that is fixed now.
2.) Due to the previous 
``` 
$ sudo ./remove-driver.sh
$ git pull
$ sudo ./install-driver.sh
```   
cycle, all existing kernel driver structures in `/lib/modules/<kernel name>/updates/`  and `/var/lib/dkms/rtl88x2bu/`   remained missing for 5.15.0-5x-generic and 5.19.0-28-generic, after the driver for the in-use 6.1.0-100x-oem now got successfully built and installed.
(BTW, curious that in both directories are actual copies of the BU88x2bu.co driver and one is not a link from the other.)
Additionally, your install-driver.sh lists a potential third destination for the built driver: `MODDESTDIR="/lib/modules/${KVER}/kernel/drivers/net/wireless/"`  

3.) Anyhow, since I also wanted functional drivers ready for the two inactive kernels and not knowing any better, I resorted to have those removed by Mint's Update manager and then reinstall them again, to have dkms invoked at the end to do its magic.
I suspect, that there also would be a way to have dkms directly build the driver for those inactive kernels, rather than going the brutal way like I did.
How would this command need to be composed?

4.) If alternatively I would safekeep a driver copy for each kernel , in which location would I need to throw them, so that after rebooting with a different kernel, the corresponding driver would be found and automatically loaded, when needed?
Are there any additional configurations required somewhere?

Well, that should so far cover all of what is currently on my mind after doing my first steps into installing-drivers and dkms land ;-)

Your kind assistance is much appreciated :-)

 
> Most Linux users would be happier if using usb wifi adapters that are supported by in-kernel drivers.

I totally agree, but as I outlined [here](https://github.com/morrownr/USB-WiFi/issues/154#issuecomment-1399047202) there doesn't seem to exist any Mediatek based AC1200+ adapter unfortunately, in the 14.5x20.5x7mm nano size or the somewhat larger 19x40.3x9mm of the EDUP EP-1689 with activity-LED. By the way, both use the generic Realtek ID 0bda:b812. 
So there really was no Madiatek-based alternative to these particular Realtek based adapters for under $10.

morrownr, 2023-01-24:
> Therefore for me `/src/88x2bu-20210702` was the driver's [source] directory it referred to

The terminology takes some getting used to. The terminology is dictated by dkms, standard Linux C coding, git and other sources. git uses the word `clone` when you download the repo onto your system so the directory on your system where git places the source is called the clone directory. Then the source has to be copied to a new directory if dkms is being used for the installation because that is what dkms requires. If dkms is not being used for the installation then the directory does not need to be copied.

The installation script supports both dkms and non-dkms installations. Some distros do not support dkms so a non-dkms alternative is provided.

> , whereas the copy (which apparently shall not be named or mentioned) getting created during the installation run, `/src/rtl88x2bu-5.13.1` , is what I was referring to as its clone directory. And you seem to be referring to `/src/88x2bu-20210702` as apparently the [git] clone directory.

I have put a lot of effort into the documentation to include the installation steps. If those steps are followed closely, the result us usually a good installation. 
 
> Well, with that out of the way now, I think I am slowly starting to grasp how the driver installation and dkms is working and therefore need to modify my previous questions.

A lot can be learned from the install-driver.sh source code as I have tried to leave comments in needed places.

> Additionally, your install-driver.sh lists a potential third destination for the built driver: `MODDESTDIR="/lib/modules/${KVER}/kernel/drivers/net/wireless/"`

That is to tell the basic (non-dkms) installation where to put the driver. The driver will be in different locations depending on whether a dkms or basic installation is used.

> 3.) Anyhow, since I also wanted functional drivers ready for the two inactive kernels and not knowing any better, I resorted to have those removed by Mint's Update manager and then reinstall them [...]

Just boot into the kernel you want to add the driver to and run ./install-driver.sh.

> 4.) If alternatively I would safekeep a driver copy for each kernel , in which location would I need to throw them, so that after rebooting with a different kernel, the corresponding driver would be found and automatically loaded, when needed or manually per modprobe? Are there any additional configurations required somewhere?

My head hurts. 
 
> Well, that should so far cover all of what is currently on my mind after doing my first steps into installing-drivers and dkms land ;-)

I can see that you are looking to land the job as my Assistant in charge of helping maintain and improve the supporting scripts in the driver. So when can you start?

Nick