morrownr/USB-WiFi #192: CF-953AX in 6GHz mode

Aqua1ung, 2023-01-22:
So I purchased the CF-953AX based on the assumption that it supports the 6GHz band, though, so far, I have yet to figure out how to make it see my router's 6GHz WiFi.

So how do I do that? Currently on Ubuntu 22.10 w/6.1.7 kernel. Did all the updates recommended [here](https://github.com/morrownr/USB-WiFi/blob/main/home/How_to_Install_Firmware_for_Mediatek_based_USB_WiFi_adapters.md), to no avail. Is there an Ubuntu package that I need to install?

morrownr, 2023-01-22:
Hi @Aqua1ung 

> So I purchased the CF-953AX based on the assumption that it supports the 6GHz band

Several users of that adapter have reported that it supports the 6 GHz band. It is still relatively new and we are all trying to figure it out... let's see what we can do.

> I have yet to figure out how to make it see my router's 6GHz WiFi. So how do I do that?

You may want to skim a lot of it but #63 has a message on how we got a guy with a new Netgear A8000 adapter going today. Your adapter and his adapter both have the mt7921au chipset and both run with the in-kernel mt7921u driver but Netgear decided to use a company device ID instead of a Mediatek default ID so the thread runs for a few pages.

Once we got him to the point that you are at, what he did was change his wifi router from 6 GHz Auto to static channel and he set channel 37. He also had to set the country code:

$ sudo iw reg set US

If you are not the US, change accordingly. If you don't know you country code, tell me your country and I will look it up.

After doing those things, either toggle wifi or reboot and see what happens. I think the European Union has it together with 6 GHz better than the rest of the world right now. I don't think 6 GHz is legal in Taiwan yet. Can you imagine being a company trying to build a wifi router to sell and it is not legal yet to emit 6 GHz?

Let us know if this helps.

Nick

Aqua1ung, 2023-01-22:
Many thanks for the reply! That did help indeed! Switching to channel 37 did it! (197 does not, and I didn't try others, but I may, and will report back.)

The bad news is that the CF-953AX doesn't seem to support WPA3-Personal, which means that I have to set my router to "Enhanced Open," which I do not think it's a good idea. Anyway, if you know of any trick that I could employ to get my CF-953AX connect in an authenticated manner, I'm willing to listen. Unfortunately, my router (GT-AXE11000) only shows two options: "Enhanced Open" and "WPA3-Personal," so I may even be able to hack my router to allow WPA2 for the 6GHz band. (As a matter of fact, I see that "The Wi-Fi Alliance requires WPA3 security certification for Wi-Fi 6E devices that will operate in the 6 GHz band. However, there is no backward compatibility support for WPA2 security," so it looks as if there's not much to do in terms of router hacking.)

[image]

morrownr, 2023-01-22:
> The bad news is that the CF-953AX doesn't seem to support WPA3-Personal

WPA3-Personal is supported here. I have a CF-951AX, which for our purposes here, is the same as your CF-953AX in that it uses the same chipset and same driver. I've tested Ubuntu 22.10 and it supports WPA3. You might play around with some router settings as I'm sure there are still some minor incompatibilities to deal with.

What are you seeing that tells you WPA3 is not working?

> Switching to channel 37 did it! (197 does not, and I didn't try others, but I may, and will report back.)

If you could give a report about which channels are working for you, that would be great.

> I see that "The Wi-Fi Alliance requires WPA3 security certification for Wi-Fi 6E devices that will operate in the 6 GHz band.

Yes, WiFi 6 mandates WPA3.

Aqua1ung, 2023-01-22:
Alright, it's all good. I think I had to reboot to get WPA3 working. Now everything works just fine. Many thanks to you and the others for your efforts!

P.S. Interestingly, setting the country code via `$ sudo iw reg set US` does **not** survive reboot, so I had to add a `@reboot iw reg set US` line to my crontab.

morrownr, 2023-01-24:
> Alright, it's all good. I think I had to reboot to get WPA3 working. 

Good to hear.

> setting the country code via $ sudo iw reg set US does not survive reboot

This is curious. I wish I had an answer for you.

> Yes, WiFi 6 mandates WPA3.

That statement is not correct. WiFi 6e mandates WPA3 is the correct statement.

jiribrejcha, 2023-01-24:
@Aqua1ung, you can permanently save the reg domain by writing it to this file:
`/etc/default/crda
`
Here is the format:
`REGDOMAIN=US`

Aqua1ung, 2023-01-25:
> @Aqua1ung, you can permanently save the reg domain by writing it to this file: `/etc/default/crda `
> 
> Here is the format: `REGDOMAIN=US`

Thanks, but there's no "crda" file at that location on my system. Should I add it manually? Was it supposed to be there? And if yes, how come it isn't?

jiribrejcha, 2023-02-10:
@fs30000 https://github.com/WLAN-Pi/pi-gen/blob/bullseye64/wlanpi1/01-config-files/files/etc/wpa_supplicant/wpa_supplicant.conf

fs30000, 2023-02-25:
> @fs30000 https://github.com/WLAN-Pi/pi-gen/blob/bullseye64/wlanpi1/01-config-files/files/etc/wpa_supplicant/wpa_supplicant.conf

This file has nothing.

bjlockie, 2023-02-25:
I see the file.
Clear your cookies?

fs30000, 2023-02-27:
> I see the file. Clear your cookies?

I see the file too, but nothing relevant to this adapter.

laixinhk, 2024-04-11:
Just remind wireless regulatory domain database may need to be updated.
After modified regulatory domain, I still couldn't solve this issue until I updated  the wireless regulatory domain database.
For example on my PC:
```sh
$sudo apt install wireless-regdb
$reboot
```