morrownr/USB-WiFi #194: Warning LOTEKOO Wireless Dongle has Realtek rtl8188ftv chip in it now

Lordcorvin1, 2023-01-24:
On Canadian market, same listing https://www.amazon.ca/dp/B06Y2HKT75 has rtl8188ftv chip in it now
Even though the description still says Ralink 5370 Chipset

bjlockie, 2023-01-24:
Do an Amazon review and say that.
It is very annoying they would change the chipset.

morrownr, 2023-01-24:
@Lordcorvin1 

Thanks for the update. I have removed that link from the in-kernel adapters list.

Lordcorvin1, 2023-01-25:
> Do an Amazon review and say that. It is very annoying they would change the chipset.

Already did, also left a notice to correct product description as well.

> @Lordcorvin1
> 
> Thanks for the update. I have removed that link from the in-kernel adapters list.

No problem, thanks for quickly editing the link.

dubhater, 2023-03-03:
You don't have to throw it out if you bought it. The in-kernel rtl8xxxu supports the RTL8188FTV since 6.2.