morrownr/USB-WiFi #196: Unusual experience with a wifi adapter using RT3572 wifi adapter

STORMFIRE007, 2023-01-27:
Hi
I order a wifi adapter that claimed it had an RT5572 wifi chip in it, but turns out it was RT3572 instead, but i tried giving it a try anyway, but i couldn't enable Complete 5Ghz speeds on it, only the A band, which too seemed considerably slower than the N band of 2.4Ghz, i tried asking help around ubuntu forums and one guy said that apparently the wifi regulations regarding 5Ghz band has changed recently, which made RT3572 adapter not work properly with my raspberry pi, complete issue on this link on ubuntu forums:
https://askubuntu.com/questions/1446712/cant-create-5ghz-wifi-hotspot-on-ubuntu-22-04-1

bjlockie, 2023-01-27:
>RT3572. 802.11n 2T2R USB Single Chip. The RT3572 single chip is a highly integrated MAC/BBP and. 2.4/5GHz RF single chip with support for 300 Mbps PHY rate.

It's supposed to do 5ghz.
802.11a is 5ghz but it is only a 300mbps adapter I think.
I am skeptical about regulatory changes causing it not to work but I don't know.
I don't have that adapter.
I used OpenWrt on my raspberry pi 4 when I ran an AP.

STORMFIRE007, 2023-01-28:
yes, the 802.11a is 5Ghz, but it's theoretical max is only 54mbps, nowhere near to the supposed 300mbps speed, i am not exactly sure what caused that issue as i tried using the same adapter in 3 different OSes, ubuntu, raspberry pi OS and windows, maybe something wrong with the device itself perhaps? the 2.4Ghz was working perfectly fine