morrownr/USB-WiFi #197: Need help with AWUS1900 NetHunter support

cr4sh-me, 2023-01-27:
Im trying to get AWUS1900 get working with my OnePlus3T nethunter device. Kernel version is 3.18.x 
Kernel is patched with RTL88XXau patch, and have enabled support for it in kernel.
WIfi adapter is detected by lsusb and usb-devices, but cannot be rised up.

cr4sh-me, 2023-01-27:
> Are you trying to get a USB adapter working with Kali Linux on your phone? Is the patch/driver supposed to work with that really old kernel?

Yes, with kali nethunter.

Heres the patch https://gitlab.com/kalilinux/nethunter/build-scripts/kali-nethunter-kernel/-/tree/master/patches/3.18

bjlockie, 2023-01-27:
It sounds like it should work.
Does dmesg show anything?

cr4sh-me, 2023-01-27:
> It sounds like it should work.
> Does dmesg show anything?
> 

[image]

closed 2023-02-16 by cr4sh-me