morrownr/USB-WiFi #198: has anyone bought an Edup EP-AX1699 from alibaba?

bjlockie, 2023-01-28:
It says it is an mtk7921au:
https://www.alibaba.com/product-detail/EDUP-High-Speed-802-11AX-Dual_1600558720979.html

morrownr, 2023-01-29:
This is the first time I have seen this adapter.

Alibaba does not instill confidence in me. If I were ranking online retailers, they would be near the bottom. Examples of probable bad or questionable info in the ad:

- Security - WPA, WPA2 - I have WiFi 4 adapters that are over 10 years old that can do WPA3 and this only does WPA2?!?

- WIFI:11ac mode up to 866Mbps - it does not list AX transfer rate?!?

- LED - POWER - So, hard wired to on when power is on?!? That is okay if the LED is not bright.

- blue-tooth: -89dBm@1Mbps, -90dBm@2Mbps, -83dBm@3Mbps - so, if it is bt capable, is it limited to USB2?

- Chipset - MTK7921 - That chipset is used in PCIe cards.

- Bands - 2.4GHz and 5GHz ISM Band - as far as I know, Mediatek has not released a USB chipset that is only dual-band capable. There is the mt7921au and mt7922u (we think). Both are tri-band.

- Windows XP/Vista/7/8/10 compatible - XP ?!? Gotta be a lot of XP users out there. Bet they can't wait to get their hands on this adapter.

- Interface: USB2.0 - well there you go, bluetooth = USB2 unless some engineer has found a way so this may be the first thing I am posting from the ad that I think is correct.

. The adapter may be good but with that kind of an ad, I'll have to see it from another retailer before I consider it.

bjlockie, 2023-01-29:
Alibaba has ads for wifi6 devices that have wifi5 chipsets so they seem sketchy.

morrownr, 2023-01-29:
More info:

EDUP shows the following adapter which appears to use the same case:

https://www.szedup.com/product-item/wifi-6-adapter-edup-dual-band-ax1800-wifi-adapter-for-pc/

It goes by the part number EP-AX1696. It uses the rtl8832au chipset. Linux users should avoid adapters based on the rtl8852au and rtl8832au chipsets. I think Realtek may have stopped making that chipset and I suspect it had real problems. You can fix only so many hardware problems with software.

It could be that the EP-AX1696 is nearing end of life and they have switched to the mt7921au chipset and thus we see the ad for the EP-AX1699. Speculation on my part. EDUP has not posted a driver for the EP-AX1699 yet so this may be a very new product. Speculation on my part but probably pretty close to what the story is.

morrownr, 2023-02-05:
Hey @bjlockie 

Look what I found:

https://www.walmart.com/ip/USB-WiFi-6-Adapter-PC-TSV-AX1800Mbps-Dual-Band-2-4GHz-5GHz-Fast-USB3-0-High-Gain-5dBi-Antenna-802-11Ax-Dongle-Wireless-Network-Fit-OS-Windows-11-10-D/1065990864

I have no idea about the quality but at least the ad is closer to accurate than what we saw in the Alibaba ad. Also, I have bought from Walmart before and they will fix problems.

bjlockie, 2023-02-05:
I sent ***@***.*** an email asking about it.
I can't figure out how to order from Walmart to Canada but I think they 
will ship here from California.

morrownr, 2023-02-05:
Or you could search on Walmart Canada...

https://www.walmart.ca

hebeda, 2023-03-06:
> It says it is an mtk7921au: https://www.alibaba.com/product-detail/EDUP-High-Speed-802-11AX-Dual_1600558720979.html

i just signed up and i give you some quick insights on this device.

they are real  , i have 2x EP-AX1699 units on my desk - these are from the first production batch, these might not be the final product.

these where the only samples available in Stock in february 2023

i have ordered these for a costumer as sample devices 

the adapters are unmarked, no stickers,  no CE and FCC so far.

That why you will not see them anywhere soon, unless EDUP  received a big order , they wont do any certification.

Without FCC approval they not allowed to be sold in the United States.

Same applies for CE RED Declaration of Conformity, without CE declaration these are not allowed to be sold anywhere in the European Union.

the Chipset is indeed a Mediatek MTK7921 USB and it has support for 802.11N,  802.11ac and 802.11ax  and Bluetooth 5.2

the device got a USB 3.0 Interface (!)

so far i have tested them only on Windows 10 , the Windows Driver seems to be in very early ages, because you need to select manually (!) which Wireless Standard should be supported.

The Windows driver can be set to 802.11n, 802.11AC or 802.11AX , there is NO Autoselect option at all.

here comes the little strange part : 

the MAC ADRESS is a ALFA NETWORK Taiwan MAC Adress , the mac adresses are not the same, which leads to the conclusion, its not just cloned firmware from Alfa Networks , but they share probably the same OEM Manufacturer.

EDUP is a Company from China Mainland btw ..

Alfa Network has a similar Product, the AWUS036AXML , i am not sure about the certification status of this device:

https://www.alfa.com.tw/products/awus036axml?variant=39754360684616

if anyone want me to do anymore tests, feel free to ask me ...

morrownr, 2023-03-06:
@hebeda 

Thanks for the info. It is good to have more options.

> if anyone want me to do anymore tests, feel free to ask me ...

If you are running Linux and have the adapter working:

- Run and post the results of:

lsusb -t

lsusb

Is bluetooth working?

What do you think of the quality?

Thanks

hebeda, 2023-03-21:
i updated my last posting and posted the "iw list" output, if anyone is interested ....