morrownr/USB-WiFi #201: AR9271 USB 2.0 freezes multiple times in managed mode

turbodat, 2023-02-04:
AR9271 USB 2.0 freezes multiple times in managed(client) mode while  trying to download large files of about greater than 600MB AR9271 freezes with ssid in the network manager and becomes unresponsive until unplugged and repluged and later for some time it works and again it continues to freeze

Distro : Debian XFCE (testing)
Kernel version : 6.1.7
NetworkManager Applet version:1.30
driver package used: firmware-ath9k-htc (1.4.0-108-gd856466+dfsg1-1)
 

and the result is even worst in stable release of debian where it freezes multiple times even for downloading files of 100MB the card dies with no led light and the network manager completely becomes unresponsive and need to unplug the usb and replug to make it work again and the cycle of freezing continues after sometime.

Distro : Debian XFCE (stable)
Kernel version : 5.10
NetworkManager Applet version:1.20
driver package used: firmware-ath9k-htc (1.4.0-106-gc583009+dfsg1-1) 

The driver package used are from official repo.I thought usb is dying and I also tried with another ar9271 usb adapter and the result was same.But Adapter works great on windows.

The problem is same as this post https://github.com/qca/open-ath9k-htc-firmware/issues/173
and I don't think posting their could help since the issues are not solved.So posting it here
I read in one of the post that you have couple of ALFA AWUS036NHA adapters So can you test with your adapter 

Thank you for your patience for reading this post.

morrownr, 2023-02-05:
Hi @turbodat 

> AR9271 USB 2.0

I do have two adapters with that chipset so let me do a little testing to see what I can find out.

morrownr, 2023-02-05:
Tested on kernels 5.19 and 6.1.

```
[  5]   0.00-1.00   sec  5.63 MBytes  47.2 Mbits/sec    0    260 KBytes       
[  5]   1.00-2.00   sec  6.21 MBytes  52.1 Mbits/sec    0    272 KBytes       
[  5]   2.00-3.00   sec  6.28 MBytes  52.6 Mbits/sec    0    272 KBytes       
[  5]   3.00-4.00   sec  5.65 MBytes  47.4 Mbits/sec    0    272 KBytes       
[  5]   4.00-5.00   sec  6.28 MBytes  52.7 Mbits/sec    0    283 KBytes       
[  5]   5.00-6.00   sec  5.59 MBytes  46.9 Mbits/sec    0    310 KBytes       
[  5]   6.00-7.00   sec  5.16 MBytes  43.3 Mbits/sec    0    310 KBytes       
[  5]   7.00-8.00   sec  6.21 MBytes  52.1 Mbits/sec    0    310 KBytes       
[  5]   8.00-9.00   sec  5.65 MBytes  47.4 Mbits/sec    0    310 KBytes       
[  5]   9.00-10.00  sec  6.28 MBytes  52.7 Mbits/sec    0    310 KBytes       
[  5]  10.00-11.00  sec  4.60 MBytes  38.6 Mbits/sec    0    310 KBytes       
[  5]  11.00-12.00  sec  5.65 MBytes  47.4 Mbits/sec    0    310 KBytes       
[  5]  12.00-13.00  sec  5.72 MBytes  48.0 Mbits/sec    0    310 KBytes       
[  5]  13.00-14.00  sec  6.34 MBytes  53.2 Mbits/sec    0    310 KBytes       
[  5]  14.00-15.00  sec  5.78 MBytes  48.5 Mbits/sec    0    310 KBytes       
[  5]  15.00-16.00  sec  5.22 MBytes  43.8 Mbits/sec    0    465 KBytes       
[  5]  16.00-17.00  sec  5.65 MBytes  47.4 Mbits/sec    0    465 KBytes       
[  5]  17.00-18.00  sec  6.65 MBytes  55.8 Mbits/sec    0    465 KBytes       
[  5]  18.00-19.00  sec  5.59 MBytes  46.9 Mbits/sec    0    465 KBytes       
[  5]  19.00-20.00  sec  5.59 MBytes  46.9 Mbits/sec    0    465 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-20.00  sec   116 MBytes  48.5 Mbits/sec    0             sender
[  5]   0.00-20.01  sec   114 MBytes  47.8 Mbits/sec                  receiver

```

I just downloaded a 4.3 GB file. I did not see any problems.

```
[    7.390968] usb 1-1.1: ath9k_htc: Transferred FW: ath9k_htc/htc_9271-1.4.0.fw, size: 51008
[    7.644070] ath9k_htc 1-1.1:1.0: ath9k_htc: HTC initialized with 33 credits
[    7.923264] ath9k_htc 1-1.1:1.0: ath9k_htc: FW Version: 1.4
[    7.923271] ath9k_htc 1-1.1:1.0: FW RMW support: On

```

I don't have anything that I think will help. Maybe somebody has something.

bjlockie, 2023-02-06:
I would try a different cable.