morrownr/USB-WiFi #205: mt7612 not working with minimal kernel

desultory, 2023-02-13:
I have a pcie attached 7921 and usb attached 7612 on my system.  The following are all of my loaded kernel modules:

```
Linux router 6.1.0-gentoo-ansible #17 SMP Sun Feb 12 19:08:50 CST 2023 x86_64 AMD FX(tm)-8150 Eight-Core Processor AuthenticAMD GNU/Linux

Module                  Size  Used by
mt76x2u                24576  0
mt76x2_common          24576  1 mt76x2u
mt76x02_usb            12288  1 mt76x2u
mt76_usb               24576  2 mt76x02_usb,mt76x2u
mt7921e                20480  0
mt7921_common          57344  1 mt7921e
mt76x02_lib            53248  3 mt76x02_usb,mt76x2u,mt76x2_common
mt76_connac_lib        45056  2 mt7921e,mt7921_common
mt76                   61440  8 mt7921e,mt76_usb,mt76x02_lib,mt76x02_usb,mt7921_common,mt76x2u,mt76x2_common,mt76_connac_lib
```

I see the following if I unplug the adapter and plug it back in:
```
[  978.515902] usb 10-2: USB disconnect, device number 3
[  980.549242] usb 10-2: new high-speed USB device number 4 using xhci_hcd
[  980.741240] usb 10-2: New USB device found, idVendor=0e8d, idProduct=7612, bcdDevice= 1.00
[  980.741245] usb 10-2: New USB device strings: Mfr=2, Product=3, SerialNumber=4
[  980.741247] usb 10-2: Product: Wireless
[  980.741248] usb 10-2: Manufacturer: MediaTek Inc.
[  980.741250] usb 10-2: SerialNumber: 000000000
[  980.929329] usb 10-2: reset high-speed USB device number 4 using xhci_hcd
[  981.122469] mt76x2u 10-2:1.0: ASIC revision: 76120044
[  981.143968] Loading firmware: mt7662_rom_patch.bin
[  981.144026] mt76x2u 10-2:1.0: ROM patch build: 20141115060606a
[  981.307936] Loading firmware: mt7662.bin
[  981.307992] mt76x2u 10-2:1.0: Firmware Version: 0.0.00
[  981.307995] mt76x2u 10-2:1.0: Build: 1
[  981.307996] mt76x2u 10-2:1.0: Build Time: 201507311614____
[  981.983493] ieee80211 phy2: Failed to select rate control algorithm
[  981.983497] ieee80211 phy2: Failed to initialize rate control algorithm
```

I'm using the firmware from the linux-firmware package on gentoo which seems to contain the files you say need to copy.

morrownr, 2023-02-13:
Hi @desultory 

> 6.1.0-gentoo-ansible

It is not clear to me which Linux mainline kernel this is based on. The reason for concern: From kernel 6.0 rc1 until 6.1.7 or 8, the 7612u chipset was dead in the water. A patch by an Intel engineer that went in prior to 6.1 rc1 broke the driver. A Mediatek dev submitted a patch but the patch took some time to get merge approval.

So, you need either a kernel prior to 6.0 rc1 or later than 6.1.8. This is an issue that will fix itself over time but some distros may use bad kernels.

That is my best bet for the problem.

desultory, 2023-02-13:
> 

That should help, thanks. 
Interestingly, the kernel I had working was based on 6.0.8, and I believe it was me building modules into the kernel and removing thinks I thought I didn't need that caused it to stop working.  I also updated the linux-firmware package so I'm really not sure.

My kernel sources are based on:
https://gentoo.osuosl.org/distfiles/linux-6.1.tar.xz
https://gentoo.osuosl.org/distfiles/genpatches-6.1-13.base.tar.xz
https://gentoo.osuosl.org/distfiles/genpatches-6.1-13.extras.tar.xz