morrownr/USB-WiFi #206: MT7922 on ROG strix B650-A gaming wifi not working

Frozen-byte, 2023-02-13:
Hey, just wanna quickly report that the Wi-Fi/Bluetooth is currently not working on the Asus B650-A Motherboard.  
Already field a bug on [launchpad](https://bugs.launchpad.net/ubuntu/+source/linux/+bug/2006797)  

Since I don't know where to ask for help fixing this issue, maybe you know the right place to poke around.

Tried it with the preinstalled drivers from Ubuntu 20.10 5.19.0-35-generic,
and tried the copied files from [linux-firmware.git](https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/mediatek) (commit 91f4c6 from 2023-01-04)

morrownr, 2023-02-13:
Hi @Frozen-byte 

> Ubuntu 20.10 5.19.0-35-generic

If that is not a typo, I can tell you what the problem is... you need more recent distro with a very recent kernel. Run the following and post the results:

$ uname -r

Once we get the questions of what distro version and what kernel version worked out, we can proceed.

Frozen-byte, 2023-02-13:
Was a typo. It's 22.10 Just copied the Kernel version, thanks for correction, edited OP 😄

bjlockie, 2023-02-13:
You have a 7921au USB stick?
I don't think Bluetooth works on any of these because of a conflict with USB 3.

morrownr, 2023-02-13:
> You have a 7921au USB stick? I don't think Bluetooth works on any of these because of a conflict with USB 3.

If I am understanding correctly, this is built in so his wifi is probably not connected via USB. The BT is probably connected via USB.

morrownr, 2023-02-13:
@Frozen-byte 

> Was a typo. It's 22.10 

Recommend you start by updating the firmware. Go to:

https://github.com/morrownr/USB-WiFi

Select menu item 8. The first section is about the mt7922.

Frozen-byte, 2023-02-13:
@morrownr I have already tried that and double confirmed it, just jet. The files from linux-firmware git are in place, after reboot nothing changes.

> and tried the copied files from [linux-firmware.git](https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/mediatek) (commit 91f4c6 from 2023-01-04)

morrownr, 2023-02-13:
Run your kernel up to 6.1: (this guide is for 6.1.9 but I think 6.1.11 is available now)

2023-02-05

Purpose: To test drivers on or with kernels beyond what
is currently included in the release versions of Ubuntu.

Upgrading the Linux Kernel on Ubuntu based amd64 systems

Note: Turn secure boot off before installing. These kernels
do not support secure boot.

You will need to make a directory to hold some downloaded
files, Since we are downloading a 6.1 kernel I use:

~/src/6.1

... anywhere in your home directory should work fine.

Go to:

https://kernel.ubuntu.com/~kernel-ppa/mainline/

Note: This guide is an example for kernel 6.1.9 Please
make adjustments for other kernels.

Download into a clean directory: (no other .deb files)

```
linux-headers-6.1.9-060109_6.1.9-060109.202302010835_all.deb
linux-headers-6.1.9-060109-generic_6.1.9-060109.202302010835_amd64.deb
linux-image-unsigned-6.1.9-060109-generic_6.1.9-060109.202302010835_amd64.deb
linux-modules-6.1.9-060109-generic_6.1.9-060109.202302010835_amd64.deb
```

----

To install:

```
sudo dpkg -i *.deb
```

Once the installation is finished, reboot.

You can use grub during boot to decide which kernel to boot. The
default is the most modern kernel installed.

-----

To remove:

Note: To remove a kernel you installed this way, you will
need to boot the computer to a different kernel.

```
sudo apt remove linux-headers-6.1.9*
```

```
sudo apt remove linux-modules-6.1.9*
```

```
sudo apt remove linux-image-unsigned-6.1.9*
```

Note: Once complete, reboot and the default kernel will now be
the most modern kernel still on your system. To select another
kernel version to boot, use the grub menu.

-----

If that does not work then check to see if Airplane Mode is on.

Frozen-byte, 2023-02-16:
Hey, thank you for the guide.
I tested Kernel 6.1 and 6.2 both are stable for my system, but there is still no sign of the Chip, it keeps quiet:
```
#dmesg | grep mt7921e

[ 17.001057] mt7921e 0000:0b:00.0: enabling device (0000 -> 0002)
[ 17.001199] mt7921e 0000:0b:00.0: ASIC revision: 79220010
[ 20.247369] mt7921e 0000:0b:00.0: Message 0004008a (seq 1) timeout
[ 23.575534] mt7921e 0000:0b:00.0: Message 0004008a (seq 2) timeout
[ 26.907372] mt7921e 0000:0b:00.0: Message 0004008a (seq 3) timeout
[ 30.231376] mt7921e 0000:0b:00.0: Message 0004008a (seq 4) timeout
[ 33.559704] mt7921e 0000:0b:00.0: Message 0004008a (seq 5) timeout
[ 36.887365] mt7921e 0000:0b:00.0: Message 0004008a (seq 6) timeout
[ 40.215367] mt7921e 0000:0b:00.0: Message 0004008a (seq 7) timeout
[ 43.543670] mt7921e 0000:0b:00.0: Message 0004008a (seq 8) timeout
[ 46.871378] mt7921e 0000:0b:00.0: Message 0004008a (seq 9) timeout
[ 50.199714] mt7921e 0000:0b:00.0: Message 0004008a (seq 10) timeout
[ 50.280219] mt7921e 0000:0b:00.0: hardware init failed
```

bjlockie, 2023-02-16:
Do you have the firmware?

morrownr, 2023-02-16:
Let's make sure we are on the same sheet of music. The mt7922 support is in the mt7921 driver:

https://patchwork.kernel.org/project/linux-wireless/patch/27e39fd3c6d70837772e56f85bf9b01e8beeca47.1626370282.git.deren.wu@mediatek.com/

I'm beginning to wonder if the kernel is missing the pci device id for your chipset. What is the pci device id for your chipset?

diff --git a/drivers/net/wireless/mediatek/mt76/mt7921/pci.c b/drivers/net/wireless/mediatek/mt76/mt7921/pci.c
index 33782e1ee312..97ca8290881f 100644
--- a/drivers/net/wireless/mediatek/mt76/mt7921/pci.c
+++ b/drivers/net/wireless/mediatek/mt76/mt7921/pci.c
@@ -14,6 +14,7 @@ 
 
 static const struct pci_device_id mt7921_pci_device_table[] = {
 	{ PCI_DEVICE(PCI_VENDOR_ID_MEDIATEK, 0x7961) },
+	{ PCI_DEVICE(PCI_VENDOR_ID_MEDIATEK, 0x7922) },
 	{ },
 };
 
@@ -336,6 +337,8 @@  module_pci_driver(mt7921_pci_driver);
 MODULE_DEVICE_TABLE(pci, mt7921_pci_device_table);
 MODULE_FIRMWARE(MT7921_FIRMWARE_WM);
 MODULE_FIRMWARE(MT7921_ROM_PATCH);
+MODULE_FIRMWARE(MT7922_FIRMWARE_WM);
+MODULE_FIRMWARE(MT7922_ROM_PATCH);
 MODULE_AUTHOR("Sean Wang <sean.wang@mediatek.com>");
 MODULE_AUTHOR("Lorenzo Bianconi <lorenzo@kernel.org>");
 MODULE_LICENSE("Dual BSD/GPL");

Frozen-byte, 2023-02-16:
> Do you have the firmware?

What Firmware do you man? I tried the distributed firmware from Ubuntu and the new firmware from the linux-firmware git repository.

If you tell me how I check the firmware, I gladly answer you correctly.
Currently active firmware:
```
modinfo mt7921e
filename:       /lib/modules/6.2.0-060200rc8-generic/kernel/drivers/net/wireless/mediatek/mt76/mt7921/mt7921e.ko
license:        Dual BSD/GPL
author:         Lorenzo Bianconi <lorenzo@kernel.org>
author:         Sean Wang <sean.wang@mediatek.com>
firmware:       mediatek/WIFI_MT7922_patch_mcu_1_1_hdr.bin
firmware:       mediatek/WIFI_RAM_CODE_MT7922_1.bin
firmware:       mediatek/WIFI_MT7961_patch_mcu_1_2_hdr.bin
firmware:       mediatek/WIFI_RAM_CODE_MT7961_1.bin
srcversion:     2916AFA1218E56F98193D33
alias:          pci:v000014C3d00000616sv*sd*bc*sc*i*
alias:          pci:v000014C3d00000608sv*sd*bc*sc*i*
alias:          pci:v000014C3d00007922sv*sd*bc*sc*i*
alias:          pci:v000014C3d00007961sv*sd*bc*sc*i*
depends:        mt76-connac-lib,mt76,mt7921-common
retpoline:      Y
intree:         Y
name:           mt7921e
vermagic:       6.2.0-060200rc8-generic SMP preempt mod_unload modversions 
sig_id:         PKCS#7
signer:         Build time autogenerated kernel key
sig_key:        24:A5:5D:76:E8:25:BB:3B:49:4E:4C:13:0C:8E:11:1B:A6:23:43:0E
sig_hashalgo:   sha512
signature:      19:3D:BD:26:65:16:3B:71:E2:28:B1:1A:BA:E0:C2:A7:94:6E:38:32:
                33:09:41:03:5C:9F:D1:D3:55:D3:6D:2B:93:02:39:20:D6:9A:58:12:
                3C:40:55:08:0C:85:02:09:B9:76:8B:1F:6D:15:77:7C:EB:56:5E:E8:
                CF:ED:23:7A:E7:0B:1F:1F:AC:3D:6C:37:E6:52:F6:B8:F9:59:2D:A5:
                76:17:3F:BE:28:14:6E:2D:87:BC:82:0E:65:AF:C0:3B:F4:B2:A8:45:
                B3:D6:7C:94:0E:26:51:CD:87:34:AE:85:AE:45:82:67:DF:20:7E:CC:
                91:38:DF:BD:5E:57:1A:B9:38:28:F0:BA:F2:31:E0:5C:B9:E0:76:35:
                99:EB:8B:A4:07:2C:4E:15:E6:07:5C:F6:31:77:15:72:0F:13:37:05:
                3F:E4:CD:E6:D7:7D:5E:ED:D9:81:90:BB:CF:BD:EF:4B:58:A9:C0:81:
                32:94:23:27:37:17:48:4F:E6:3D:7B:F1:F4:E0:BE:71:4C:88:A4:A9:
                2D:E1:69:71:9A:51:8C:A9:04:A0:7A:57:02:77:91:E5:72:31:FC:3D:
                F5:E9:EC:AB:F1:57:75:FC:38:A6:6E:D0:73:D2:EF:02:0A:52:44:B9:
                AB:11:E2:B4:17:5C:43:83:34:2C:0F:2B:26:CE:A6:34:9B:C3:7E:EB:
                D8:53:5D:73:11:1C:82:3E:31:1D:B1:19:55:C6:31:EA:34:43:02:9D:
                48:10:1C:42:B7:F2:9F:C7:1E:12:2C:03:82:5A:1E:74:CE:02:D2:60:
                A0:97:C1:55:60:5B:9C:A5:0E:B2:3C:AD:87:9F:C2:8A:B8:9C:06:24:
                33:56:1C:BB:E5:7E:8E:67:B8:24:AC:60:CF:41:EB:F7:C1:F4:AF:23:
                0B:F1:08:CF:C2:D0:C0:98:08:14:9A:90:FF:47:48:23:68:65:30:8A:
                BD:1E:A8:2B:DD:14:7F:15:BB:67:F4:B6:D6:FB:9A:DA:9A:C4:E9:0B:
                77:74:DA:D5:7C:A9:3F:83:36:5C:A0:E2:DA:0A:4F:86:D6:47:3D:C3:
                D1:46:44:74:BD:FA:41:41:BD:81:65:19:21:AD:38:E3:DE:0E:29:26:
                DA:5C:3C:03:AF:83:C4:4D:3F:81:F6:CE:93:68:27:FB:60:D0:86:BE:
                76:B0:72:69:D3:68:94:6B:05:AC:02:9E:EB:15:4F:16:74:8C:8C:CB:
                CC:93:DA:6E:07:56:02:B9:C1:49:E2:19:F8:E3:8D:32:75:BE:16:F5:
                5A:AD:AD:63:9E:F0:54:E7:9D:5F:0C:A4:4E:A7:B6:A4:43:F1:EE:EB:
                B8:BE:95:B7:0B:E1:84:FC:8B:15:2B:78
parm:           disable_aspm:disable PCI ASPM support (bool)
```

Frozen-byte, 2023-02-16:
@morrownr  Thank you for the detailed help!
> I'm beginning to wonder if the kernel is missing the pci device id for your chipset. What is the pci device id for your chipset?

How do you ensure that? I do not know how to receive the pci device id and if it's correct:
```
# lspci | grep MEDIATEK
0b:00.0 Network controller: MEDIATEK Corp. MT7922 802.11ax PCI Express Wireless Network Adapter

# grep PCI_ID /sys/bus/pci/devices/*/uevent | grep 0b:00.0
/sys/bus/pci/devices/0000:0b:00.0/uevent:PCI_ID=14C3:0616
```

Is `14C3:0616` the ID you are looking for?

morrownr, 2023-02-16:
> I do not know how to receive the pci device id...

First, you grepped for MEDIATEK. Don't do that. Your chipset make show as an AMD chipset. AMD and Mediatek have a big contract where Mediatek supplies wifi chipsets and AMD rebrands them. Our in-kernel Linux drivers work on the rebranded chips but there is the possibility of a change in the USB or PCI device ID's which will keep the driver from knowing it is supposed to support the chip.

Do this:

$ lspci

...and post the entire output.

Then:

$ grep PCI_ID /sys/bus/pci/devices/*/uevent

...post that output as well.

Once we determine what you PCI_ID really is, if it is not in the driver yet, I'll show you how to put it in the driver and you can test it. If it works then you get to be a Linux kernel dev by submitting a patch.

Nick

morrownr, 2023-02-16:
> /sys/bus/pci/devices/0000:0b:00.0/uevent:PCI_ID=14C3:0616

> Is 14C3:0616 the ID you are looking for?

I should have said that this is probably what we are looking for but I wanted you to do an overall check before pressing on.

Frozen-byte, 2023-02-17:
Sure, for scrolls sake I've attached the whole logs as files.
[pci_id.txt](https://github.com/morrownr/USB-WiFi/files/10765686/pci_id.txt)
[lspci.txt](https://github.com/morrownr/USB-WiFi/files/10765687/lspci.txt)

morrownr, 2023-02-17:
> PCI_ID=14C3:0616

You were spot on. Now we need to look in the source where the PCI IDs are kept so as to see if this ID is there.

First a trivia question: Where is the ID for the bluetooth support? I did not  see it.

I keep a clone of of `https://github.com/torvalds/linux` on my system but you can look at it online as well.

Let's look in: /drivers/net/wireless/mediatek/mt76/mt7921/pci.c

```
static const struct pci_device_id mt7921_pci_device_table[] = {
	{ PCI_DEVICE(PCI_VENDOR_ID_MEDIATEK, 0x7961),
		.driver_data = (kernel_ulong_t)MT7921_FIRMWARE_WM },
	{ PCI_DEVICE(PCI_VENDOR_ID_MEDIATEK, 0x7922),
		.driver_data = (kernel_ulong_t)MT7922_FIRMWARE_WM },
	{ PCI_DEVICE(PCI_VENDOR_ID_MEDIATEK, 0x0608),
		.driver_data = (kernel_ulong_t)MT7921_FIRMWARE_WM },
	{ PCI_DEVICE(PCI_VENDOR_ID_MEDIATEK, 0x0616),
		.driver_data = (kernel_ulong_t)MT7921_FIRMWARE_WM },
	{ },
};
```

Also in: include/linux/pci_ids.h

include/linux/pci_ids.h:#define PCI_VENDOR_ID_MEDIATEK		0x14c3

Ah, ha! The current kernel has the ID so the question is, when did that happen?

I'm going over to: `https://www.kernel.org/`

5.19 is no longer supported. I'm going to download and search 6.1.12.

```
static const struct pci_device_id mt7921_pci_device_table[] = {
	{ PCI_DEVICE(PCI_VENDOR_ID_MEDIATEK, 0x7961) },
	{ PCI_DEVICE(PCI_VENDOR_ID_MEDIATEK, 0x7922) },
	{ PCI_DEVICE(PCI_VENDOR_ID_MEDIATEK, 0x0608) },
	{ PCI_DEVICE(PCI_VENDOR_ID_MEDIATEK, 0x0616) },
	{ },
};
```

Yup, 6.1.12 is good. Maybe this is not the problem or possibly the ID just went in and you haven't tested a version of the kernel that has the ID. You might think about using the guide I posted earlier to install 6.1.12. If you still don't get an interface after that, and given that the firmware file for 7922 should be the latest if you installed them, I am stumped.

Is Secure Boot on or off?

Frozen-byte, 2023-02-22:
Hello, sorry for the late answer. I needed time for experimenting and validating, my Issue is fixed (because I could not believe it was the solution)

I searched for the Internet and some Anons suggested resetting the Network-Adapter by disconnecting them from power.

So, I executed the following steps:
1. Removing the Motherboard from ANY power source (even took the battery off)
2. pressed power button several times to unload capacitors 
3. waited ~5 mins and connected power cable and inserted battery
- Then, after booting into the system, the network was not loaded at all.
4. opened the shell and `$ sudo modprobe  mt7921e`
- Everything was working now.

I am using/restarting my PC daily and there was only on occurrence where the WLAN was not working, a second `$ sudo modprobe  mt7921e` fixed it.

Strangely enough, I loaded the mod before I did the power surge, so I only can narrow it down to that step.

Thank you for the detailed help, that eliminated so many possible errors, I finally got the solution.

closed 2023-02-22 by Frozen-byte

morrownr, 2023-02-23:
What the hell!?!

Did you happen to see anything about what would cause that?

BTW, you might want to update your launchpad bug with this info.

Frozen-byte, 2023-02-23:
I have absolutely no clue. I just stumbled upon this "solution" here: https://unix.stackexchange.com/a/723829