morrownr/USB-WiFi #207: Having trouble connecting to 6 GHz band with mt7921au based adapter on Kali...

morrownr, 2023-02-13:
Help appreciated.

I am posting this for a friend and I don't have direct access to the computer. So, this is telephone tech support.

Kali is current and up to date.

He bought a new 6e (6 GHz) capable router. Can't seem to connect. Here is what he is seeing in dmesg:

```
[   33.796420] wlan1: authenticate with bc:0f:9a:49:05:fb
[   33.830284] wlan1: bad VHT capabilities, disabling VHT
[   33.830297] wlan1: Invalid HE elem, Disable HE
[   34.003014] wlan1: send auth to bc:0f:9a:49:05:fb (try 1/3)
[   34.017494] wlan1: authenticated
```

What the heck is that `Invalid HE elem, Disable HE` ?

It really does not look like an authentication problem. This is a new one one me so if anyone has any hints about how to connect a client to a new 6 GHz router on a 6 GHz channel, please speak up.

Thanks.

bjlockie, 2023-02-13:
What country is the router set to?

What channel is the wifi?

Disable any mesh.

Set the channel width to 20MHz wide on the router.

What is the result of $ sudo iw list ?

What is the result of $ uname -a ?

morrownr, 2023-02-15:
> What channel is the wifi?

Woring on getting these questions answered. I have already told him to run the router down to channel 37 to see if that helps.

gacragg-github, 2023-02-21:
Packet capture of the process would be nice, too.

morrownr, 2023-02-23:
Follow up: User switched to Ubuntu 22.10 as it works there.

Long distance tech support is a challenge. We were not able to determine what was up with his Kali installation but it looked to me like a wpa_supplicant issue. Ubuntu uses IWD these days instead of wpa_supplicant so that could be the difference.