morrownr/USB-WiFi #214: Comfast CF-952AX MT7921AU can't go above 3Mbit upload

ViRb3, 2023-02-18:
### System specs:
- Raspberry Pi 4B
- Raspberry Pi OS 11 64-bit, 6.11 kernel via `rpi-update`

### Tested setup:
- Directly to Raspberry Pi USB3 port
- Through USB3 powered hub
- Through USB3 powered hub + USB3 extension cable

### Consistent results:
```bash
root@raspberrypi:~# ./speedtest-go
Latency: 2.520892658s
Download Test: ...........
Upload Test: ...........

Download: 89.10 Mbit/s
Upload:  3.00 Mbit/s
```

I have to note, going through the hub the device is stuck in USB2 mode. When it was directly connected to Pi USB3 port, it could reach 220 Mbit download, though still only 3Mbit upload. Any way to force USB3 mode like in the old Realtek drivers?

@morrownr this is likely one for you :)

Thanks!

morrownr, 2023-02-18:
> @morrownr this is likely one for you :)

Okay, I'll give it a try but anyone else should feel free to jump in.

I may wear you out with questions.

> ./speedtest-go

I need more info about. Where does one get a copy?

I am assuming you are somehow testing with an internet server?

I ask because I always first test with iperf3 which is testing locally. There are so many variables once you get beyond you local lan that it is best to test yur local lan first if possible. Are you running OpenWRT on your router? Do you have an extra RasPi4B?

> Upload:  3.00 Mbit/s

What were you expecting?

>  I have to note, going through the hub the device is stuck in USB2 mode. When it was directly connected to Pi USB3 port, it could reach 220 Mbit download, though still only 3Mbit upload. Any way to force USB3 mode like in the old Realtek drivers?

The in-kernel drivers, such as the mt7612u and mt7921u, that support adapters that are capabile of USB3 do their work automatically. If everything that needs to be place for USB3 operation, then they turn on USB3 support. The logic in the drivers seems to be impeccable as I have never seen an error. There is likely something that the driver is seeing that is causing it to limit you to USB2.

When you upgraded to the 6.11 kernel via  `rpi-update`, did you also upgrade the rest of the driver (the firmware files) to the latest? There was a new release in January.

https://github.com/morrownr/USB-WiFi/blob/main/home/How_to_Install_Firmware_for_Mediatek_based_USB_WiFi_adapters.md

Be careful because the mt7921au firmware is now in section 2 on that page because mt7922 is now the latest.

Side note; Remember that upgrading the kernel only upgrades part of the driver, the in-kernel part, mt7921u. The firmware files are also part of the driver but are not part of the kernel so it is up to you or the distro maintainer to handle that and it is highly uplikely the RasPi folks are upgrading as fast as updates are coming.

> Through USB3 powered hub

Powered hubs on a RasPi... hmmm... good start to a scary novel. I recommend you leave the hubs out of this until we get this sorted out.

A solution that I use that works well with the CF-951AX (same chipset as yours) on a RasPi4B:

https://www.amazon.com/dp/B07S6B5X76

See https://github.com/morrownr/USB-WiFi/issues/139 for information on what the mt7921au chipset and driver can do. That test is with iperf3 in clean air. 

Would you mind sharing the following info with us as there has been a some conflicting info about the CF-952AX:

$ lsusb

@morrownr

ViRb3, 2023-02-18:
> I need more info about. Where does one get a copy?

https://github.com/showwin/speedtest-go
https://github.com/showwin/speedtest-go/releases/download/v1.4.0/speedtest-go_1.4.0_Linux_arm64.tar.gz

It's a CLI for https://speedtest.net. On my other SD Card with RTL8822BU USB card plugged in same port, I get:
```sh
root@raspberry:~/ # ./speedtest-go
Latency: 16.889302ms
Download Test: ...............
Upload Test: ....

Download: 209.81 Mbit/s
Upload: 119.48 Mbit/s
```

> When you upgraded to the 6.11 kernel via rpi-update, did you also upgrade the rest of the driver (the firmware files) to the latest? There was a new release in January.

I did not know this is a thing, will re-test now with updated firmware!

> Powered hubs on a RasPi... hmmm... good start to a scary novel. I recommend you leave the hubs out of this until we get this sorted out.

I'm aware, this is why I tested without powered hub too, and the upload is literally glued to 3Mbit. The RTL8822BU card does great, so I don't think it's a hub issue. FWIW, my ethernet is 1Gbit up/down, and I can get about 600Mbit up/down from my laptop over WiFi.

> Would you mind sharing the following info with us as there has been a some conflicting info about the CF-952AX:

```sh
$ lsusb
Bus 003 Device 004: ID 0e8d:7961 MediaTek Inc. Wireless_Device
```

ViRb3, 2023-02-19:
Well, this is awkward... Turns out that for whatever reason, the problem was in the speedtest app, as you properly suggested. Running either of these alternatives produce vastly different results:
- https://www.speedtest.net/apps/cli
- https://github.com/librespeed/speedtest-cli/

Sadly the results vary a lot from run to run, but seems like I'm getting about 200/200 on USB2 and 500/200 on USB3! The speeds should be symmetrical, but I suppose that upload will suffer with small devices like this. Is this correct?

Also turns out the problem was not in my USB3 hub, but rather the USB3 extension cable that follows it; directly attaching the Comfast to the hub properly joins USB3 mode.

All in all, very happy with the results so far! I think I'll wait for kernel 6.1 to hit stable in Raspberry Pi OS (should be very, very soon, perhaps even this month), but definitely switching to the Comfast after that.

ViRb3, 2023-02-19:
iperf3 results below:
```
root@raspberrypi:~# iperf3 -c 192.168.1.101
Connecting to host 192.168.1.101, port 5201
[  5] local 192.168.1.118 port 54250 connected to 192.168.1.101 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  29.7 MBytes   250 Mbits/sec    0   1.43 MBytes
[  5]   1.00-2.00   sec  33.8 MBytes   283 Mbits/sec    0   2.77 MBytes
[  5]   2.00-3.00   sec  35.0 MBytes   294 Mbits/sec    0   3.24 MBytes
[  5]   3.00-4.00   sec  33.8 MBytes   283 Mbits/sec    0   3.45 MBytes
[  5]   4.00-5.00   sec  32.5 MBytes   273 Mbits/sec    0   3.64 MBytes
[  5]   5.00-6.00   sec  33.8 MBytes   283 Mbits/sec    0   3.64 MBytes
[  5]   6.00-7.00   sec  32.5 MBytes   273 Mbits/sec    0   3.83 MBytes
[  5]   7.00-8.00   sec  33.8 MBytes   283 Mbits/sec    0   3.83 MBytes
[  5]   8.00-9.00   sec  35.0 MBytes   294 Mbits/sec    0   3.83 MBytes
[  5]   9.00-10.00  sec  31.2 MBytes   262 Mbits/sec    0   3.83 MBytes
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   331 MBytes   278 Mbits/sec    0             sender
[  5]   0.00-10.06  sec   330 MBytes   275 Mbits/sec                  receiver

iperf Done.
```

closed 2023-02-19 by ViRb3

morrownr, 2023-02-19:
It sounds like things are headed in the right direction. Good.

> Bus 003 Device 004: ID 0e8d:7961 MediaTek Inc. Wireless_Device

I appreciate you posting this. I am trying to sort out some inconsistent information about the Comfast CF-952AX. Another user that bought the 952AX said that it indeed does have the mt7921au chipset but that it used a device ID that was not one of the standard Mediatek provided device IDs so he submitted a patch to Linux wireless. In this case, I am seeing a standard Mediatek device ID so I am very puzzled. 

Can I get you to send me the link you used to buy the product?

Can you confirm that the adapter is indeed the CF-952AX and not the CF953AX?

I need to get the details right or it can cause confusion.

I wish I had a good solution for a cable but I don't. It seems that getting a good USB3 cable that will work with a specific adapter is trial and error. I will suggest you measure and buy the shortest cable that will do the job.

Thanks

ViRb3, 2023-02-19:
Whops, you're right, my model is the CF-953AX. Purchased from here:
- https://www.aliexpress.com/item/1005004469569274.html

morrownr, 2023-03-20:
@ViRb3 

Your message above contains good info and an interesting situation. Let me recommend that you copy it and start a new issue. Maybe some minor mods to clarify that some things in a new issue. The reason I say this is that we need more eyes on this than what will happen in this old issue. Let me ponder this and reply in the new issue.

@bjlockie 

>How do you test the mode?

lsusb -t

> I get 1201mbps link speeds for both upload and download.

That is good but it does not mean your adapter is in USB3 mode. You can see 1200 Mbps link speeds in USB2.

bjlockie, 2023-03-20:
>     I get 1201mbps link speeds for both upload and download.
>
> That is good but it does not mean your adapter is in USB3 mode. You 
> can see 1200 Mbps link speeds in USB2.
>
Should it show a faster link speed than the USB interface can do?
That doesn't seem very logical to me.

Class=Vendor Specific Class, Driver=mt7921u, 5000M