morrownr/USB-WiFi #216: Realtek wifi 8811CU, 8812CU, 8822BU - Not plug and play in kernel 6.2

userabcd0, 2023-02-21:
Above listed wifi adapters are not detected plug and play in linux kernel 6.2.

Manjaro and endeavouros.

Wifi adapters are comfast.

morrownr, 2023-02-21:
My bet is that either you did not install the firmware or the person that compiled the kernel has not turned on compiling for the new drivers yet.

I'll see about adding instructions for installing the firmware in main menu item about firmware tomorrow.

userabcd0, 2023-02-21:
Thanks, that will help.

Wifi adapters are working with out of kernel drivers.

morrownr, 2023-02-22:
@userabcd0 

Been busy but did add a short line with a link to the Realtek firmware:

https://github.com/morrownr/USB-WiFi/blob/main/home/How_to_Install_Firmware_for_Mediatek_based_USB_WiFi_adapters.md

A direct link is:

https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/rtw88

What will help get the driver turned on and the firmware installed/updated is for a bug to be filed with you distro. The bug should be specific and include the solution, which, in this case is point out the link to the Realtek firmware and showing exactly what needs to be turned on will kernel compilation. I just happen to have a guide that has a section that shows exactly what needs to be turned on:

https://github.com/morrownr/USB-WiFi/blob/main/home/Compiling_the_Linux_Mainline_Kernel.md

@morrownr

userabcd0, 2023-03-01:
Thanks for the replies and help everyone, this matter appears to have been solved.

On endeavouros the kernel updated this morning to 6.2.1.

Realtek 8811CU and 8822BU are working plug and play with in kernel drivers.

morrownr, 2023-03-01:
> 8822BU

I've been testing this one for two days now:

1. The driver appears limited to USB2 mode. *($ lsusb -t)
2. Performance and stability need work.

userabcd0, 2023-03-01:
Agree it has performance and stability issues.

The devices I tried disconnect after a short use time and then need to unlug and replug to work again.

morrownr, 2023-03-01:
Seeing the same thing here. Hopefully now that these drivers are in mainline we will see work increase on issues. I think the drivers will get there, it just be overnight.

I just a test with iperf3:

Our out-of-kernel driver located here at this site:

```
$ iperf3 -c 192.168.1.1
Connecting to host 192.168.1.1, port 5201
[  5] local 192.168.1.223 port 55274 connected to 192.168.1.1 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  60.8 MBytes   510 Mbits/sec    0   1.81 MBytes       
[  5]   1.00-2.00   sec  65.0 MBytes   545 Mbits/sec    0   1.81 MBytes       
[  5]   2.00-3.00   sec  63.8 MBytes   535 Mbits/sec    0   1.81 MBytes       
[  5]   3.00-4.00   sec  65.0 MBytes   545 Mbits/sec    0   1.81 MBytes       
[  5]   4.00-5.00   sec  65.0 MBytes   545 Mbits/sec    0   1.81 MBytes       
[  5]   5.00-6.00   sec  65.0 MBytes   545 Mbits/sec    0   1.81 MBytes       
[  5]   6.00-7.00   sec  65.0 MBytes   545 Mbits/sec    0   1.81 MBytes       
[  5]   7.00-8.00   sec  65.0 MBytes   545 Mbits/sec    0   1.81 MBytes       
[  5]   8.00-9.00   sec  65.0 MBytes   545 Mbits/sec    0   1.81 MBytes       
[  5]   9.00-10.00  sec  65.0 MBytes   545 Mbits/sec    0   1.81 MBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   645 MBytes   541 Mbits/sec    0             sender
[  5]   0.00-10.01  sec   642 MBytes   538 Mbits/sec                  receiver

iperf Done.
```

That is about 10 times faster than what I am seeing out of the in-kernel driver buy this will improve.