morrownr/USB-WiFi #218: For those that need to add a VID/PID to the mt7921u driver...

morrownr, 2023-02-23:
Just today a user posted about this and it is far easier than compiling a driver/kernel:

Below is an easy way to add the VID/PID to the Netgear A8000 adapter so that the mt7921u driver can recognize the adapter. 

From a terminal:
```
su
modprobe mt7921u
echo 0846 9060 > /sys/bus/usb/drivers/mt7921u/new_id
```

If you find that you mt7921au based adapter is not recognized due to the VIP/PID not being in the kernel yet, get the VIP/PID by running:

```
lsusb
```

Get the VID/PID and use it in the above example.

@morrownr

NathanRV, 2023-03-16:
Hi, 
I'm encountering the problem where the driver doesn't recognize my Netgear A8000 adapter and this solution works as a temporary fix but the adapter will stop working after a certain time and once I reboot I have to repeat these steps. I've got another usb wifi adapter (which is very slow now but it includes bluetooth so I'd like to keep it plugged in) and a PCI wifi adapter that I was using before and haven't gotten around to opening up the PC and getting it out. The weird part is that according to my understanding I shouldn't have to use this fix as the output of `uname -r` is `5.19.0-35-generic` and so my kernel is 5.19.0 and the adapter should be plug-and-play. The output of `lsusb` shows (among other usb devices) 
`Bus 004 Device 006: ID 0846:9060 NetGear, Inc. Wireless_Device` 
and `lsusb -t` shows (once again among other things)
```
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/3p, 5000M
        |__ Port 2: Dev 6, If 0, Class=Vendor Specific Class, Driver=, 5000M
```
I've tried plugging it in other usb ports without success (not every single one since since I've got a lot of them, 13). I'm using Ubuntu 22.04, any ideas?

bjlockie, 2023-03-16:
That adapter is not plug and play until kernel 6x I think.
Netgear didn't use the typical product id for that chipset.
You should be able to add those commands to a startup script.
You shouldn't need to reboot.
Maybe try a more recent kernel?
I use 6.2.5 with ubuntu.
https://kernel.ubuntu.com/~kernel-ppa/mainline/
Maybe upgrade the firmware?

morrownr, 2023-03-16:
Hi @NathanRV 
@russeree 

> I found that it should be PnP for my kernel on the USB WiFi adapters that are supported with Linux in-kernel drivers page...

As the author of the paragraph you quoted, I can now see that I need to edit the statement to include information indicating that if adapter makers use a company specific VID/PID, then the adapter will not be plug and play until such time as the company specific VID/PID is included in the in-kernel driver. This is how things have worked for many years and I failed to mention this simply because I did not think about it...

I have now edited the document you quoted to reflect this. Information about this was included in the section on the Netgear A8000 section but this could come up again so a more general warning is a good idea.

Another Netgear A8000 owner has already submitted a patch:

https://patchwork.kernel.org/project/linux-wireless/patch/20230123090555.21415-1-git@qrsnap.io/

It is not clear to me why the patch has not been acted on. I addressed this reply to @russeree so that he sees it. Reese, it is probably time for us to discuss what needs to happen to help this patch work its way into the kernel. Please reply.

@morrownr

russeree, 2023-03-16:
Absolutely, what steps need to be taken?

morrownr, 2023-03-16:
Thanks @russeree 

What to do? Well, I've been monitoring linix-wireless for enough years to understand what not to do for the part.

Your patch made it to patchworks and was assigned to Felix. Why is it not merged yet? I'll offet 2 theories:

1. Felix is really busy and just has not had time.
2. Felix sees something that he does like so it will require time to chase you down for a fix.

It could be a combination of both.

Options?

1. Send a reply to the original message asking if there is anything else you can do.
2. Look over the original message to see if it can be improve and send in a v2.
3. I could reply to the original message with an ack.

Let me ponder this and I'll get back to you with a recommendation. One thing I know for sure is that a v2 should include a request for it to go in 6.1 LTS and the dev kernel.

morrownr, 2023-03-17:
@russeree 

I posted the following to linux-wireless a few minutes ago:

```
I can confirm this VID/PID needs to go into 6.1 LTS and the current testing version of the kernel as I am getting an increasing amount of traffic from users that have purchased the Netgear A8000.

My site is github.com/morrownr/USB-WiFi

Helping Linux users with USB WiFi is what we do.

The OP could have added a comment to the patch showing the adapter that is causing this Patch to be submitted. Maybe he can submit a v2 that can be expedited?

Guidance?

Nick
​github.com/morrownr/USB-WiFi
```

morrownr, 2023-03-17:
Remember when posting to linux-wireless:

no top posting and your email client must be in text mode

russeree, 2023-03-17:
Thank you so much, this has been my issue. Submission was very easy with the git cli. 

I am having a bit of a harder time understating how to get a proper reply sent.

morrownr, 2023-03-17:
Usually the reply coming back is a message saying that the patch has been merged. It is not clear why the delay in this case. The patch definately made it as it is shown in Patchwork.

Sometimes those that are assigned to patches hesititate to see if additional confimation is available and there might not be much confirmation at this point. Hopefully my message helps confirm that your patch is correct.

Since part of the mission of this site is educational, could I get you to do and post a short checklist with how you made the patch with git cli? Others should see it and...

@russeree

morrownr, 2023-03-18:
@russeree 
@NathanRV 

Reese,

I'm sure you saw this if you are subscribed to linux-wireless but not everyone is subscribed so this is to document the reply for others:

```
I assigned this to me on patchwork, I'll queue this for v6.3 and change
the commit log to below. Felix&Lorenzo, ack?

wifi: mt76: mt7921: add Netgear AXE3000 (A8000)

Add support for the Netgear AXE3000 (A8000) based on the Mediatek
mt7921au chipset. A retail sample of the Netgear AXE3000 (A8000) yeilds
the following from lsusb D 0846:9060 NetGear, Inc. Wireless_Device. This
has been reported by other users on Github.
```

The above reply was from Kalle Valo. As best I can tell, he is boss at linux-wireless. Here are some take-aways:

- We should see the patch going into 6.3 rc? so we'll have to pay attention.
- The title that is used on messages is very important. He appears to be giving us a hint to follow protocol:

[PATCH] wifi: mt76: mt7921: add Netgear AXE3000 (A8000)

The need to add VID/PIDs for this mt7921au and mt7922u chipsets is likely to happen a lot as time passes. Would you mind working with me to provide a document that helps people submit patches? I can work on a good example patch message and round up the links to the official documentation if you could make a list of the steps to use git cli to make a patch. Interested?

Nathan,

[Checklist.tar.gz](https://github.com/morrownr/USB-WiFi/files/11009054/Checklist.tar.gz)

The above file is a checklist for installing the test compiles at Ubuntu mainline. It is for kernel 6.1.9 and you will be installing 6.3-rc4 or maybe 6.3-rc5 when they are available and depending on when the patch goes in but you can edit appropriately.

I could use a report on how well the checklist works as I probably need to get it up to speed and post it on the Main Menu here.

@morrownr

bjlockie, 2023-03-18:
Is there no way they can get it in 6.1?

russeree, 2023-03-18:
> Usually the reply coming back is a message saying that the patch has been merged. It is not clear why the delay in this case. The patch definately made it as it is shown in Patchwork.
>
> Sometimes [...]

Sorry for the delay and short responses, I absolutely would love to work on creating a checklist and some sort of script or tutorial that creates an easy method to add in the VID/PIDs for these wireless adapters. 

> [Checklist.tar.gz](https://github.com/morrownr/USB-WiFi/files/11009054/Checklist.tar.gz)
>
> The above file is a checklist for installing the test compiles at Ubuntu mainline. It is for kernel 6.1.9 [...]

I will check it out and test after clean up a few issues. 

On a second note, I wanted to say thanks so much for helping me though all of this. Even though it's a simple patch; contributing to the Linux kernel has always been my dream. At that something that is very useful and will give people access to a great wifi adapter on Linux.

morrownr, 2023-03-18:
> Is there no way they can get it in 6.1?

Sure it can go into 6.1. Let's see what happens.

morrownr, 2023-03-22:
@russeree 

A reply from Kalle Valo just came in here from linux-wireless:

```
Reese Russell <git@qrsnap.io> wrote:

> Add support for the Netgear AXE3000 (A8000) based on the Mediatek
> mt7921au chipset. A retail sample of the Netgear AXE3000 (A8000) yeilds
> the following from lsusb D 0846:9060 NetGear, Inc. Wireless_Device. This
> has been reported by other users on Github.
>
> Signed-off-by: Reese Russell <git@qrsnap.io>

There was a conflict, please rebase over wireless and resubmit as v2:

Recorded preimage for 'drivers/net/wireless/mediatek/mt76/mt7921/usb.c'
error: Failed to merge in the changes.
hint: Use 'git am --show-current-patch=diff' to see the failed patch
Applying: wifi: mt76: mt7921: add Netgear AXE3000 (A8000) support
Using index info to reconstruct a base tree...
M       drivers/net/wireless/mediatek/mt76/mt7921/usb.c
Falling back to patching base and 3-way merge...
Auto-merging drivers/net/wireless/mediatek/mt76/mt7921/usb.c
CONFLICT (content): Merge conflict in drivers/net/wireless/mediatek/mt76/mt7921/usb.c
Patch failed at 0001 wifi: mt76: mt7921: add Netgear AXE3000 (A8000) support

Patch set to Changes Requested.

--
https://patchwork.kernel.org/project/linux-wireless/patch/20230123090555.21415-1-git@qrsnap.io/

https://wireless.wiki.kernel.org/en/developers/documentation/submittingpatches
```

It appears there is a merge conflict and he is requesting a v2 of the patch. My recommendation for the title is:

[PATCH v2] wifi: mt76: mt7921: add Netgear AXE3000 (A8000) support

You are going to need to rebase. That is make the patch from a different kernel than previously used. Since he is trying to send this patch in the current testing kernel, using 6.3-rc3, I need to check what is current but I think that is it.

He also left a couple of links that you should read before reworking the patch and sending a reply.

My recommendation: When you have reworked this and have the message ready to send, post a new issue here are for patch reviews. We have several folks that stop in here at times that are very knowledgable about submitting patches.

@morrownr

russeree, 2023-03-29:
I will have some time to work on this tonight and will absolutely resubmit with the new title. Thanks so much and sorry for the delay.

morrownr, 2023-03-30:
@russeree 

It is better to get it right than be in a hurry. Recommend you write up a draft of v2 and post a new issue asking for it to be checked, There are folks that are experts doing this that stop by at times, This patch needs to go in kernels 6.2 and 6.1 in addition to 6.3 and I think there is an address that needs to be added for that to happen. I'd have to dig it up.

Thanks for working this. Many will appreciate it.

morrownr, 2023-04-03:
@russeree 
@981213

Reese,

@981213 recently submitted a patch to add the VIS/PID for a cf-952ax that used a company VID/PID. He might be able to work with you on [PATCH v2]. Getting this patch in 6.1 (LTS) and 6.2+ would really be appreciated by many people.

@morrownr

russeree, 2023-04-03:
Are you referencing this? 
https://lore.kernel.org/netdev/20230104063341.18863-1-gch981213@gmail.com/T/ 

This seems to be the exact same patch strategy that I employed. I did not include a comment as the fact is these are just USB PID/VID pointers to the chipset itself. As a note I have been running my kernel patch on a production machine for well over 2 months without any issues. 

None the less I am rebasing now over the newest pull from 

git://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git

russeree, 2023-04-04:
Okay here is the patch that I have written, it mimics my last patch closely but takes a queue from the now merged patch by @91213

```
From 3b5f77f62c01d26e065635ef61465625e04b11dc Mon Sep 17 00:00:00 2001
From: Reese Russell <git@qrsnap.io>
Date: Mon, 3 Apr 2023 16:56:39 -0700
Subject: [PATCH] [PATCH v2] wifi: mt76: mt7921: add Netgear AXE3000 (A8000)
 support

Issue: Though the Netgear AXE3000 (A8000) is based on the mt7921
chipset because of the unique USB VID:PID combination this device
does not initialize/register. Thus making it not plug and play.

Fix: Adds support for the Netgear AXE3000 (A8000) based on the Mediatek
mt7921au chipset. The method of action is adding the USD VID/PID
pair to the mt7921u_device_table[] array.

Notes: A retail sample of the Netgear AXE3000 (A8000) yeilds the following
from lsusb D 0846:9060 NetGear, Inc. Wireless_Device. This pair
0846:9060 VID:PID has been reported by other users on Github.
---
 drivers/net/wireless/mediatek/mt76/mt7921/usb.c | 3 +++
 1 file changed, 3 insertions(+)

diff --git a/drivers/net/wireless/mediatek/mt76/mt7921/usb.c b/drivers/net/wireless/mediatek/mt76/mt7921/usb.c
index 8fef09ed29c9..a08d20d5c556 100644
--- a/drivers/net/wireless/mediatek/mt76/mt7921/usb.c
+++ b/drivers/net/wireless/mediatek/mt76/mt7921/usb.c
@@ -18,6 +18,9 @@ static const struct usb_device_id mt7921u_device_table[] = {
 	/* Comfast CF-952AX */
 	{ USB_DEVICE_AND_INTERFACE_INFO(0x3574, 0x6211, 0xff, 0xff, 0xff),
 		.driver_info = (kernel_ulong_t)MT7921_FIRMWARE_WM },
+	/* Netgear A8000/AXE3000 */
+	{ USB_DEVICE_AND_INTERFACE_INFO(0x0846, 0x9060, 0xff, 0xff, 0xff),
+		.driver_info = (kernel_ulong_t)MT7921_FIRMWARE_WM },
 	{ },
 };
 
-- 
2.34.1
```

morrownr, 2023-04-04:
Recommend 3 tweaks to the subject:

[PATCH v2] wifi: mt76: mt7921u: add Netgear AXE3000 (A8000) VID/PID

> Are you referencing this?
https://lore.kernel.org/netdev/20230104063341.18863-1-gch981213@gmail.com/T/

Yes. Your patch went in about 2 weeks after his and his went right through whereas yours did not. It still is not obvious to me why that happened but sometimes things get bogged down and overlooked.

You might compare his To: and CC: list to make sure you are hitting the right spots. It seems to me that there is an address that needs to be included if you want this to go back to previous kernels and I cannot recall what the is but this really needs to be in 6.1 and 6.2+. I guess you could launch the patch and include a small question regarding how to get it in 6.1 and 6.2 if you can't find what I am talking about.

I think you did a good job.

981213, 2023-04-04:
The driver got a new driver_info line since 6.2 so it won't get automatically backported to 6.1 and earlier.
I think a manual backport patch without driver_info needs to be sent to linux-stable to get it included in older versions. You can follow option 3 in https://www.kernel.org/doc/html/latest/process/stable-kernel-rules.html after the patch is accepted in the latest linux version.

russeree, 2023-04-04:
@morrownr So other than matching lists my patch looks acceptable at first look?

russeree, 2023-04-04:
So I submitted to the linux-kernel@vger.kernel.org but I didn't realize the maintainers.pl file also included this. 
This means that my patch had a to:linux-kernel@vger.kernel.org and a cc:linux-kernel@vger.kernel.org and is not showing up on the list. 

Sorry for my mistake.

morrownr, 2023-04-04:
@russeree 

@981213 has some good words of wisdom and he pointed you to the link about how to get the patch into stable kernels like 6.1 and 6.2. I think he is correct is that you should wait until today's patch is merged, then you can look into getting it in 6.1 and 6.2. I really don't think it worth the effort for any kernel prior to 6.1.

I see that it is in patchwork. Hopefully it will flow rapidly in the kernel.

I happened to be thinking... it might be good if we have a little guide on the Main Menu here that gives the steps to submitting a patch. It could also include links to the officials docs and hints about various things to pay attention to and it could include a short example. Would you guys be interested in working on a guide?